iiab/iiab #3303: Call for testing: IIAB pre-installed on ubuntu-mate-22.04-desktop-arm64+raspi

jvonau, 2022-07-11:
2 New images for evaluation based on ubuntu-mate-22.04-desktop-arm64+raspi.img
[base](http://104.193.111.252/local_content/uploads/iiab-base-0711-ubuntu-mate-22.04-desktop-arm64+raspi.img.zip)
[small](http://104.193.111.252/local_content/uploads/iiab-min-0711-ubuntu-mate-22.04-desktop-arm64+raspi.img.zip)

Exact same user feel as the stock ubuntu-mate images, geared for Desktop therefore use of a monitor/keyboard/mouse is highly recommended. Feedback welcome under this issue, don't file new issues, keep it here.

https://github.com/iiab/iiab-factory/pull/231

jvonau, 2022-07-12:
Taken after providing internet access via tethered cell phone prior to reboot, seems stable with the screensaver turned off for now. http://sprunge.us/NXQi57

holta, 2022-07-12:
I've guessing it might be a bad microSD card (maybe, in any case overnight it froze [still hanging now] while writing to the card, 32% of the way through).  So I need to try again flashing the 3GB image in another way.

holta, 2022-07-12:
Another microSD card did not work.  Hopefully that was just extremely bad luck.  I'll try to a third one.

holta, 2022-07-12:
MATE completed its many updates and rebooted :heavy_check_mark: 

Very unfortunately however http://box and http://box.lan do not work (@georgejhunt very nicely made http://box.lan auto-launch with IIAB is install on RasPiOS with desktop!).

Background: this shortcoming is problematic, as teachers/students depend on this to get to know IIAB.  Long story short IIAB communities and IIAB documentation rely on this heavily, so the onboarding + learning experience is uniformly explainable + tangible across all devices &mdash; including IIAB's that happen to have a graphical desktop.

jvonau, 2022-07-12:
The P-400 and small RCA hdmi TV worked great for me, what are you using for a monitor? Like a said it's a full pristine setup, at least you don't have the IIAB install to wait through, just rename the builder,yml file to local_vars,yml and reboot as cmdserv needs to have the local_vars,yml file present or the service doesn't start up. The hotspot is disabled to allow client upstream wifi given the image was build in a container the alternate firmware was not installed, just a pass trough admin-console and the wifi will become active.

holta, 2022-07-12:
I'm not being dismissive.

I'm just saying I've been in dozens of schools where http://box and http://box.lan have to work (otherwise there's simply chaos).

holta, 2022-07-12:
> when the terminal opens banner "please run cd /opt/iiab/iiab and sudo ./iiab-network" like other the warnings do and have the terminal pop open when logging into the gui session?

Many people will open a browser but not a Terminal.

But as a short-term / intermediary solution something like the above (mandatory reminder/nudge in Linux terminal) is definitely pointing in the right direction.

jvonau, 2022-07-12:
On a side note with /etc/iiab/install-flags/iiab-complete running `sudo iiab` did ask to create iiab-admin set password then stopped, that should be enough to create the lessor restricted non-sudo account that could be used with admin-console or alternate GUI account but no sudo access.

jvonau, 2022-07-12:
> I'm just saying I've been in dozens of schools where http://box and http://box.lan have to work (otherwise there's simply chaos).

Sorry was evaluating every other aspect of the install, so that comes back to checking /etc/hosts prior to running iiab-network which I proceeded to do because that is the right thing to do to update ap0's mac address per installation or there will be mayhem with duplicate mac address in the air waves when more than 1 of the same image are booted close by. This goes for ALL iiab prepared images without exception so that part should already be documented with this 'issue' going away upon reboot.

holta, 2022-07-12:
> this 'issue' going away upon reboot

Currently a reboot does restore access to http://box and http://box.lan

As you probably well know; but just FYI I tried 3 reboots of this Ubuntu Mate [from 3GB image, on RPi 4] simply to verify that.

(The ideal would be that both names `box` and `box.lan` work for newbies exploring the desktop, if it's possible, _even prior_ to the 1st reboot.)

jvonau, 2022-07-12:
That is what the confirmation that I asked for is about, I'm check it on the next burn/boot, but I'm short of spare sdcards atm... 
This that you testing rpi-imager?
>8.239.59.22 - - [12/Jul/2022:08:42:28 -0500] "GET /local_content/uploads/iiab-base-0711-ubuntu-mate-22.04-desktop-arm64+raspi.img.zip HTTP/1.1" 206 90406354 "-" "Mozilla/5.0 rpi-imager/1.7.2"

holta, 2022-07-12:
Yes I used rpi-imager

> confirmation that I asked for

What confirmation?  I'm away most of the rest of the day, but will send if I can.

jvonau, 2022-07-12:
If /etc/hosts has the needed entries to resolve box box.lan just after firstboot login, I suspect the file was rewritten from what was originally on the image that I just confirmed is present using [mount-image.sh](https://github.com/jvonau/mk-image/blob/main/mount-image.sh)
```
127.0.0.1     box.lan localhost.localdomain localhost box box box.lan
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters
```

jvonau, 2022-07-12:
This is where I ask for iiab-diagnostics

holta, 2022-07-12:
> I have a nice blueman-applet in the top right corner.

Likewise, it's there now.  Not sure if it was there after the 4th boot.

> The top one looks like something the installer did, and emacs would be you, the other 2 are you or the system?

The top one happened automatically.  The other 4 paragraphs (at the bottom of /var/log/apt/history.log) are from my manual commands.

Sorry I haven't time today (to explore it more deeply) but iiab-diagnostics are here if it helps: http://sprunge.us/ZNeEG9?en

jvonau, 2022-07-12:
> The image shipped with /etc/iiab/local_vars.yml renamed to local_vars.yml.builder to encourage a review before using, can/should be copied local_vars.yml.
  
>177 =IIAB==========================================================================
 178 FILE DOES NOT EXIST: /etc/iiab/local_vars.yml
1249 =IIAB==========================================================================
1250 COMMAND: /usr/bin/journalctl -t IIAB-CMDSRV    # Admin Console CMDSRV log
1251 
1252 Jul 12 10:07:48 181-mate-iiab-3gb IIAB-CMDSRV[44617]: IIAB-CMDSRV : Starting Command Server
1253 Jul 12 10:07:48 181-mate-iiab-3gb IIAB-CMDSRV[44617]: IIAB-CMDSRV : Error in /etc/iiab/local_vars.yml
1254 Jul 12 10:07:48 181-mate-iiab-3gb IIAB-CMDSRV[44617]: IIAB-CMDSRV : Command Server Initialization Failed

In light of the above these would have to be 'True' in default_vars:
>  31 iiab-apps-to-be-installed :
  32 awstats
  33 bluetooth
  34 captiveportal
  35 kalite
  36 kiwix
  37 openvpn
  38 osm_vector_maps
  39 remoteit

jvonau, 2022-07-12:
> Sorry was evaluating every other aspect of the install, so that comes back to checking /etc/hosts prior to running iiab-network which I proceeded to do because that is the right thing to do to [...]

Hum, could tie in 'iiab' instead of exiting at iiab-complete offer to run iiab-network (with new network-complete flag as the final step) after suggesting raspi-config or other related actions (connect to upstream wifi) be completed first. Just ship images with iiab-complete and the in the wild provisioning step would be just `sudo iiab` `Y` to run iiab-network. You get the reduced iiab-admin user created at runtime as noted above, the network is adjusted for the current needs and the end user doesn't have to learn any new tricks. Does that sound plausible?

holta, 2022-07-12:
I don't understand all the details but Raspberry hotspots should (wherever possible!) start right up (on 1st boot of image, if feasible?)

Running [./iiab-network](https://github.com/iiab/iiab/blob/master/iiab-network) can and should be heavily promoted...on top of that?  Ideally in a targeted or even event-driven way...for those most likely to need it on non-Raspberry HW and any USB-smells-like-a-network scenarios?  Not sure if that's at all realistic in 2022, but it's an idea.)

jvonau, 2022-07-13:
> I don't understand all the details but Raspberry hotspots should (wherever possible!) start right up (on 1st boot of image, if feasible?)
>

Not even the stock image provided by RasPi Foundation functions as a 'client' out of the box without user intervention due to the legal requirements around wifi. When you create an image for redistribution you become a de-facto distro, based on, variant of one released but on your own and can be held liable for infractions under local laws, should really adhere to the local laws of where the equipment is deployed. I not going to play fast and loose when it comes down to international standards when we can strive for better, period. 
 > Running [./iiab-network](https://github.com/iiab/iiab/blob/master/iiab-network) can and should be heavily promoted...on top of that? Ideally in a targeted or even event-driven way...for those most likely to need it on non-Raspberry HW and USB-smells-like-a-network or Ethernet-link-detected scenarios?
> 
> (Not sure if that's at all realistic in 2022, but it's an idea.)

Write something up, in place of making a wish that I'm not going to grant because there is no clear objective.

jvonau, 2022-07-13:
Seems as if you are pushing the limits of the monitor's hardware beyond it's recommended settings. Without knowing what monitor you have connected all you are making is vague statements about not supporting some resolutions.

https://elinux.org/R-Pi_Troubleshooting#Green_LED_blinks_in_a_specific_pattern suggests your SD card might be an issue, any other storage usb devices connected? Official power supply used?

jvonau, 2022-07-13:
1366x768x60hz it what is set with the pics posted above with the monitor being a small rca tv using hdmi connections between the tv and my rpi-400. I have zero video issues here, but I'm not trying every setting under the sun looking to cut down/bash the distro. Back over to the soapbox owner.

jvonau, 2022-07-13:
Not loading the kernel and the GUI not loading are two very different issues, you seem to be confabulating the two.

holta, 2022-07-13:
> any other storage usb devices connected?

No

holta, 2022-07-13:
> Official power supply used?

Yes

jvonau, 2022-07-13:
FWIW: https://ubuntu-mate.org/raspberry-pi/install/ for a preview of the first boot routine
2Gb of ram might be pushing the lower limits of what is usable.
> 331 =IIAB==========================================================================
 332 COMMAND: /usr/bin/free    # RAM memory
 333 
 334                total        used        free      shared  buff/cache   available
 335 Mem:         1889064      863784      383520       86944      641760      790056
 336 Swap:        1048572       16384     1032188
 337 
 338 =IIAB=======================================================================

I've found good performance with 4Gb with the P-400 when that machine was my daily driver and I was running 20.10 and I build pre-updated images from that base image for 21.04 & 21.10 using [mk-image with examples](https://github.com/jvonau/mk-image) which worked fine. That is how I upgraded without risking the running sdcard. I just didn't push to share those images, now I am.

holta, 2022-07-13:
> 2Gb of ram might be pushing the lower limits of what is usable.

Just FYI RAM usage was right around 800 MB (i.e. 60% free and available, of 2 GB) immediately after logging in.

And about 1.2GB RAM usage after Firefox is launched.

But at other times it's entirely possible (similar to Ubuntu Desktop 22.04 LTS itself) that 4GB would be far preferable &mdash; especially for anybody using this as they full-time / regular computer.

jvonau, 2022-07-13:
> Again I'm just reporting the problem, in case others care:
> 
> The 2 problems (screen sync glitches unlike other OS's & repeated "7 flashes of RPi 4's green light", which it's alleged is related to kernel.img failing) would seem to occur together, despite these seeming to be very unrelated things.
> 
> I don't the reasons. Again I'm just mentioning the facts, in case others care.

You are mentioning observations without context (what monitor, resolution, sync rate) and presenting your opinion as facts. Have a better SDcard to try? Yours might be a bit worn out given the amount of writing they see, nothing lasts forever. I'd start with that given the 7 flashes, means the kernel didn't load at all.

jvonau, 2022-07-13:
side note: Seeing this across different installs:
>1639 Jul 12 16:30:18 181-mate-iiab-3gb systemd[1]: proc-sys-fs-binfmt_misc.automount: Got automount request for /proc/sys/fs/binfmt_misc, triggered by 4879 (df)
1640 Jul 12 16:30:18 181-mate-iiab-3gb systemd[1]: Mounting Arbitrary Executable File Formats File System...
1641 Jul 12 16:30:18 181-mate-iiab-3gb systemd[1]: Mounted Arbitrary Executable File Formats File System.

Seems related to using `df`, which is employed by the admin-console, there maybe a connection or not just noting it.

holta, 2022-07-13:
The OS boots about 90% of the time, so the 7 flashes after 5+ minutes of screen sync failing is probably not at all common nor worth worrying about.

Normally (during most boots) the repeated screen sync fails for about 30sec-to-2 min (before the 7 flashes also kick in) which may be ugly, but so be it, it then boots.

The particularities of which old Dell HDMI screen are not important (presumably brand particularities are buried on the back of the screen somewhere, but I'm going to worry about it).  It's a very generic old HDMI screen that works with all other OS's to date, and so be it.  I'm just mentioning this in passing, in case others later run into similar.

jvonau, 2022-07-13:
Can get the terminal to open when logged in, lets see if I can get a banner to display.
```
jerry@box:~$ cat ~/.config/autostart/mate-terminal.desktop 
[Desktop Entry]
Type=Application
Exec=mate-terminal
Hidden=false
Name[en_CA]=teminal
Name=teminal
Comment[en_CA]=test
Comment=test
X-MATE-Autostart-Delay=2
```

jvonau, 2022-07-14:
Actually I did one better, have the complete solution to the must run iiab-network post deployment but I'm not sure if this would be the right place to share it.

jvonau, 2022-07-14:
https://support.pishop.ca/article/33-raspberry-pi-act-led-error-patterns

georgejhunt, 2022-07-19:
I used rpi-imager to flash this image. It came up pretty well. I updated
/etc/hosts with "box.lan" not realizing that reboot was all that would be
required.
I didn't check all the iiab services. But admin-console worked, and it all
seems pretty snappy via remote/wired connection.
But the behavior of the desktop as a console, with usb keyboard and mouse
still leaves a lot to be desired. (hardware was rpi4-4GB). I had tried mint
on Ubuntu 6 months ago and I remember thinking that it wasn't ready for
prime time.

Observations:

   1. Click on the firefox icon at the top menu. It finishes painting the
   initial screen in 26 Seconds.
   2. There is keystroke latency in the search bar of 3-6 seconds which
   makes it irritating to correct a mistyped word.
   3. Overall the desktop seems to work, but it really feels sluggish. I
   tried toggling hardware acceleration in firefox.  Seemed better with
   hardware acceleration turned off.
   4. I tried starting over with a Ubuntu image, and putting lxqt on it
   (thinking that mate might be the culprit). lxqt (the newer lxde, which
   might not use xorg). It was slow too.
   5. Then I installed raspi OS desktop for comparison. Firefox painted the
   initial screen in 6 seconds (mate was 26). Chromium was a little slower --
   8 seconds.
   6. In general, it feels like there is internal misconfiguration, maybe
   xorg set wrong, where errors don't bubble up to anything understandable,
   but with stalls and timeouts.

jvonau, 2022-07-19:
1. first run configuration I bet being it is a snap installed app, the second launch should be faster.
2. I've seen this, drops entries on occasion too, when really loaded up.
3. Might just be firstboot related, as in not all the memory was recovered after cleanup, What's the impression after rebooting?
4. You could try the virgin mate install without iiab for comparison.
5. Now go back and install iiab over top of that image and re-time the test.
6. As above point 4 then file bugs upstream with mate, nothing video related was altered by myself.

jvonau, 2022-07-19:
I'd be interested to see iiab-diagnostics from the install.

jvonau, 2022-07-26:
As of today 97 packages can be upgraded from the upstream base image.
build time 49 mins for base on my hardware.
build time 57 mins for min on my hardware.

tim-moody, 2022-07-27:
interesting idea. how are upgrade_roles and upgrade_roles.txt maintained and by whom. Are upgrades cumulative; if I am one image back do I get the roles I need to upgrade? In the example at hand, the pylibs role should have been included. so will require a second upgrade.

To take it further why not compute the roles that have changed by a diff on the last commit on iiab against the commit of the image. For example

`git diff f374d05ea26e6c4e71c3d8d7eb47c77978ecc0c4 914285037a0393e04fa6ea35469d8c2013f8c701 --name-only |grep roles/`

jvonau, 2022-07-27:
> interesting idea. how are upgrade_roles and upgrade_roles.txt maintained and by whom. 

Initially I can, and for the most part this would be adding roles the would require re-installs that are safe to be added to upgrade_roles and the corresponding note in upgrade_roles.txt. Some roles updates originate when the version is bumped in "defaults" in iiab while others come from outside as when one of the apt installed roles that use 'latest' releases a later version. Ideally long term the person with the best situational awareness of all the moving pieces.
>Are upgrades cumulative; if I am one image back do I get the roles I need to upgrade? In the example at hand, the pylibs role should have been included. so will require a second upgrade.
>

Cumulative, and one could just list both roles within upgrade_roles and would be picked up when iiab-configure runs however because [pylibs is not optional](https://github.com/iiab/iiab/blob/master/roles/4-server-options/tasks/main.yml#L16-L20) the new files are installed with no need to explicitly list that role. Now iiab-configure has used by Adam in the past to move from small -> medium -> large while testing with great success.
 
> To take it further why not compute the roles that have changed by a diff on the last commit on iiab against the commit of the image. For example
> 
> `git diff f374d05ea26e6c4e71c3d8d7eb47c77978ecc0c4 914285037a0393e04fa6ea35469d8c2013f8c701 --name-only |grep roles/`

Results in:
```
roles/0-init/defaults/main.yml
roles/0-init/tasks/create_iiab_ini.yml
roles/0-init/tasks/hostname.yml
roles/0-init/tasks/main.yml
roles/0-init/tasks/network.yml
roles/0-init/tasks/validate_vars.yml
roles/1-prep/tasks/hardware.yml
roles/1-prep/tasks/main.yml
roles/2-common/tasks/fl.yml
roles/2-common/tasks/main.yml
roles/2-common/tasks/packages.yml
roles/4-server-options/tasks/main.yml
roles/6-generic-apps/tasks/main.yml
roles/awstats/tasks/main.yml
roles/awstats/templates/awstats.schoolserver.conf.j2
roles/azuracast/README.rst
roles/calibre-web/tasks/enable-or-disable.yml
roles/calibre-web/tasks/install.yml
roles/calibre-web/tasks/main.yml
roles/calibre/defaults/main.yml
roles/calibre/tasks/enable-or-disable.yml
roles/calibre/tasks/main.yml
roles/captiveportal/README.md
roles/captiveportal/tasks/install.yml
roles/captiveportal/tasks/main.yml
roles/captiveportal/templates/iiab-divert-to-nginx.j2
roles/cups/README.md
roles/cups/tasks/install.yml
roles/cups/templates/cups.conf.j2
roles/firmware/tasks/download.yml
roles/firmware/tasks/main.yml
roles/firmware/templates/iiab-check-firmware
roles/firmware/templates/iiab-firmware-warn.sh
roles/gitea/tasks/enable-or-disable.yml
roles/gitea/tasks/install.yml
roles/gitea/tasks/main.yml
roles/iiab-admin/README.rst
roles/iiab-admin/tasks/main.yml
roles/iiab-admin/tasks/pwd-warnings.yml
roles/iiab-admin/templates/sshpwd-lxde-iiab.sh.j2
roles/internetarchive/README.md
roles/internetarchive/tasks/enable-or-disable.yml
roles/internetarchive/tasks/main.yml
roles/jupyterhub/README.md
roles/jupyterhub/tasks/install.yml
roles/jupyterhub/tasks/main.yml
roles/jupyterhub/templates/getsite.py.j2.unused
roles/jupyterhub/templates/patch_FUA.sh.j2.unused
roles/kalite/tasks/enable-or-disable.yml
roles/kalite/tasks/install.yml
roles/kalite/tasks/main.yml
roles/kiwix/defaults/main.yml
roles/kiwix/tasks/install.yml
roles/kiwix/tasks/kiwix-apk.yml
roles/kiwix/tasks/main.yml
roles/kiwix/templates/iiab-make-kiwix-lib
[... 110 lines cut ...]
roles/postgresql/tasks/enable-or-disable.yml
roles/postgresql/tasks/main.yml
roles/pylibs/templates/iiab_lib.py
roles/remoteit/tasks/install.yml
roles/samba/README.rst
roles/samba/templates/smb.conf.j2
roles/sugarizer/tasks/install.yml
roles/sugarizer/tasks/main.yml
roles/sugarizer/tasks/main2.yml
roles/transmission/README.rst
roles/transmission/defaults/main.yml
roles/transmission/tasks/enable-or-disable.yml
roles/transmission/tasks/main.yml
roles/usb_lib/templates/mount.d/70-usb-library
roles/usb_lib/templates/umount.d/70-usb-library
roles/vnstat/tasks/install.yml
roles/vnstat/tasks/main.yml
roles/wordpress/tasks/main.yml
roles/www_base/files/html/html/credits.html
roles/www_base/tasks/php-stem.yml
roles/www_options/tasks/main.yml
roles/yarn/tasks/main.yml
runroles-base.yml
```
With the endless note editing that approach would have misleading results as above, the narrow focus would just involve the role's bump in the version used by the role. Perhaps when there is some relationship between git commits and 'iiab_revision' that will be possible.

tim-moody, 2022-07-27:
>With the endless note editing that approach would have misleading results as above

of course I intended to filter so a role only appears once. could further filter to only pick up roles of interest and installed roles. I agree some changes could be cosmetic and so immaterial. In the end there are not many roles that can be upgraded.

When creating the list manually how will you handle the following cases:

- I installed an image that I want to upgrade, but there have been several images since and upgrade_roles only knows about the changes since the most recent image.
- I installed the most recent image and some time ago I upgraded it. Now I want to upgrade again with new changes. Do I have to run roles that are the latest versions in my install?

jvonau, 2022-07-27:
> of course I intended to filter so a role only appears once. could further filter to only pick up roles of interest and installed roles. I agree some changes could be cosmetic and so immaterial. In [...]

In the end I don't care how you come up with your solution.
 
> When creating the list manually how will you handle the following cases:
> 
>     * I installed an image that I want to upgrade, but there have been several images since and upgrade_roles only knows about the changes since the most recent image.
>

I have had this code proposed in the past as #2500 if adopted then this would be moot point, we could of handled updates since that point in time till the latest code. For the record an older branch has [working examples](https://github.com/jvonau/iiab-upgrader/commit/c1ba82f86408101566a391e12ed49a5245ad432d) from that time frame. I'm not going to backfit the code to handle that given the past reception the code received and the treatment I received, just shortsightedness in play but now it a good idea. It was a great plan then and still is when you have a 1TB disk full of content and 2 lines changed in the code base for the fix you need.
 
>     * I installed the most recent image and some time ago I upgraded it. Now I want to upgrade again with new changes. Do I have to run roles that are the latest versions in my install?

The last revisions applied would be recorded in iiab.env and upgrade_roles carries the full listing, only the delta between the two would be applied. ie iiab.env has revision 5 applied while upgrade_roles lists 9 revisions 6-9 would be applied when run.

tim-moody, 2022-07-27:
>The last revisions applied would be recorded in iiab.env and upgrade_roles carries the full listing, only the delta between the two would be applied. ie iiab.env has revision 5 applied while upgrade_roles lists 9 revisions 6-9 would be applied when run.

sounds reasonable.

>in the end I don't care how you come up with your solution.

stops the conversation

jvonau, 2022-07-28:
Just trying to collaborate, in the end I'm just sharing how I intend to support my release of IIAB within the above images going forward is all. Keeping the basic images devoid of options keeps the delta small between image creation and installed time, that is why I chose 'base' and 'small' rather than the kitchen sink. Now with the presets working rather well just let the end user choose what to install and the content that is wanted once written to disk and booted. Having iiab-upgrade auto update itself gives me a small window into the user's environment that is used to update the basic iiab install and can offer new features going forward. Any support related about the images could be opened at [mk-image](https://github.com/jvonau/mk-image) and [iiab-upgrade](https://github.com/jvonau/iiab-upgrader/) respectively.

closed 2025-04-06 by jvonau