iiab/iiab #3307: Test IIAB on Linux Mint 21 Beta

holta, 2022-07-13:
Mint 21 Beta (based on Ubuntu 22.04) is downloadable here:

https://community.linuxmint.com/iso

Thanks everyone who can help!

holta, 2022-07-14:
FYI an "OEM install" of Mint 21 Beta was necessary for some reason (with VirtualBox).

Otherwise "Failed to start LSB: disk temperature monitoring daemon" flashed on screen during boot, and the VM's actual disk wasn't mounted, preventing install of the OS.  In short, you can explore Mint 21 (like a Live CD) that way, but not actually install it.

(Certainly that was the case using ISO linuxmint-21-cinnamon-64bit-beta.iso from 2022-07-11.)

holta, 2022-07-14:
A LARGE-sized IIAB install is currently underway...currently in Stage 6...so far so good:

```
root@box:~# iiab-summary
install_date = 2022-07-14T00:47:06Z   Current TZ: EDT

iiab: 767ee485, 29 PR's / 127 commits since tag 8.0-preview-2
    "Merge pull request #3302 from holta/wifi_net2"
    https://github.com/iiab/iiab   branch: master

iiab-admin-console: beeae89, 44 PR's / 125 commits since tag v0.5.5.9
    "Merge pull request #506 from tim-moody/0.6.0.4-enh2"
    https://github.com/iiab/iiab-admin-console   branch: master

Linux Mint 21 Vanessa   bookworm/sid
display-manager? active   Arch1: amd64   Arch2: i386
5.15.0-41-generic #44-Ubuntu SMP Wed Jun 22 14:20:53 UTC 2022 x86_64
Intel(R) Core(TM) i3-6100U CPU @ 2.30GHz   CPU(s): 2   Mem: 1977 Swap: 1162

Filesystem      Size  Used Avail Use% Mounted on
/dev/sda3        24G  9.5G   14G  42% /      ZIMs: 0 OER2Go: 0 Apps2B: 22

lo enp0s3 tun0
openvpn_enabled: True
openvpn_handle: 199-mint21beta-LARGE
192.168.0.199 10.8.0.50
```

holta, 2022-07-14:
A reminder that `sudo apt install nodejs npm` needs to be run manually prior to MInt 21's official release.  And also:

```
sudo echo 'nodejs_installed: True' >> /etc/iiab/iiab_state.yml
```

(This installed Node.js 12.22.9 and npm 8.5.1 which are generally good enough for testing!)

holta, 2022-07-14:
Node-RED 2.2.2 doesn't install with Node.js 12.x apparently?  In any case, opt out for now:

```
TASK [nodered : Copy /usr/lib/node_modules/node-red/settings.js to /home/nodered/.node-red/settings.js] ****************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Source /usr/lib/node_modules/node-red/settings.js not found"}
```

jvonau, 2022-07-16:
rxJ3B2?en
1211 =IIAB==========================================================================
1212 COMMAND: /usr/bin/sudo journalctl -b 0 -u networkd-dispatcher # networkd-dispatcher log
1213
1214 Jul 13 21:22:43 box systemd[1]: Starting Dispatcher daemon for systemd-networkd...
1215 Jul 13 21:22:45 box networkd-dispatcher[696]: WARNING: systemd-networkd is not running, output will be incomplete.
1216 Jul 13 21:22:45 box networkd-dispatcher[572]: ERROR:Unknown state for interface NetworkctlListState(idx=1, name='lo', type='loopback', operational='n/a', administrative='unmanaged'): n/a
1217 Jul 13 21:22:45 box networkd-dispatcher[572]: Traceback (most recent call last):
1218 Jul 13 21:22:45 box networkd-dispatcher[572]: File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
1219 Jul 13 21:22:45 box networkd-dispatcher[572]: self.handle_state(iface_name,
1220 Jul 13 21:22:45 box networkd-dispatcher[572]: File "/usr/bin/networkd-dispatcher", line 348, in handle_state
1221 Jul 13 21:22:45 box networkd-dispatcher[572]: raise UnknownState(operational_state)
1222 Jul 13 21:22:45 box networkd-dispatcher[572]: UnknownState: n/a
1223 Jul 13 21:22:45 box networkd-dispatcher[572]: ERROR:Unknown state for interface NetworkctlListState(idx=2, name='enp0s3', type='ether', operational='n/a', administrative='unmanaged'): n/a
1224 Jul 13 21:22:45 box networkd-dispatcher[572]: Traceback (most recent call last):
1225 Jul 13 21:22:45 box networkd-dispatcher[572]: File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
1226 Jul 13 21:22:45 box networkd-dispatcher[572]: self.handle_state(iface_name,
1227 Jul 13 21:22:45 box networkd-dispatcher[572]: File "/usr/bin/networkd-dispatcher", line 348, in handle_state
1228 Jul 13 21:22:45 box networkd-dispatcher[572]: raise UnknownState(operational_state)
1229 Jul 13 21:22:45 box networkd-dispatcher[572]: UnknownState: n/a
1230 Jul 13 21:22:45 box systemd[1]: Started Dispatcher daemon for systemd-networkd.

holta, 2022-07-16:
Above networkd-dispatcher concern discussed/mentioned here:

- #3264
- #3278

jvonau, 2022-10-05:
eb3c0a2684c81e6f916c8219c49918e8d304d1da Check if networkd-dispatcher is installed and if there is a traceback.

holta, 2022-10-06:
Mint 21 currently offers this `networkd-dispatcher` apt package: (which would seem to solve the earlier problem)

```
root@box:~# apt list networkd-dispatcher
Listing... Done
networkd-dispatcher/jammy-updates,jammy-updates,jammy-security,jammy-security,now 2.1-2ubuntu0.22.04.2 all [installed]
```

jvonau, 2022-10-07:
`sudo journalctl -u networkd-dispatcher --no-pager` on a current Mint installs and note if there are any tracebacks present would confirm there are no issues moving forward.

holta, 2022-10-07:
> > journalctl -u networkd-dispatcher
> 
> root@mint21:~# journalctl -u networkd-dispatcher | pastebinit -b sprunge.us http://sprunge.us/gfLsU1

I probably should have mentioned that the above is a vanilla/virgin install of Linux Mint 21.

(IIAB has not been installed.)

jvonau, 2022-10-07:
https://forums.linuxmint.com/viewtopic.php?t=372953
What if you `sudo systemctl enable systemd-networkd` then `sudo reboot` does the traceback go away?

holta, 2022-10-07:
> What if you `sudo systemctl enable systemd-networkd` then `sudo reboot` does the traceback go away?

Yes.

Traceback goes away, as you can see in lower 6 lines (resulting from reboot) here:

```
root@mint21:~# journalctl -u networkd-dispatcher | tail
Oct 07 10:03:59 mint21 networkd-dispatcher[534]:   File "/usr/bin/networkd-dispatcher", line 348, in handle_state
Oct 07 10:03:59 mint21 networkd-dispatcher[534]:     raise UnknownState(operational_state)
Oct 07 10:03:59 mint21 networkd-dispatcher[534]: UnknownState: n/a
Oct 07 10:03:59 mint21 systemd[1]: Started Dispatcher daemon for systemd-networkd.
Oct 07 12:58:39 mint21 systemd[1]: Stopping Dispatcher daemon for systemd-networkd...
Oct 07 12:58:39 mint21 systemd[1]: networkd-dispatcher.service: Deactivated successfully.
Oct 07 12:58:39 mint21 systemd[1]: Stopped Dispatcher daemon for systemd-networkd.
-- Boot b2aafab53cf64ae488d0cadaa383e718 --
Oct 07 12:58:57 mint21 systemd[1]: Starting Dispatcher daemon for systemd-networkd...
Oct 07 12:58:57 mint21 systemd[1]: Started Dispatcher daemon for systemd-networkd.
```

jvonau, 2022-10-07:
Good, think #3392 should be tested against Mint with WiFi so the 00-iiab-debug hooks get installed.

holta, 2022-10-07:
> Good, think #3392 should be tested against Mint with WiFi so the 00-iiab-debug hooks get installed.

Ok.

I don't have access to any laptops or similar WiFi devices where Mint 21 can be installed right now.

Let's try to find someone who does.

Finally, if PR #3392 is very safe, one option might be to merge it now (or soon) and then seek out this important Mint 21 test in the ensuing days.

jvonau, 2022-10-07:
How about I relax the WiFi conditional in 3392 and lets test that on the VM and all installs going forward? Some feedback is better than none..

holta, 2022-10-07:
> How about I relax the WiFi conditional in 3392 and lets test that on the VM and all installs going forward? Some feedback is better than none..

VM testing is easy & nearly instantaneous if that helps, great.

jvonau, 2022-10-07:
Done, will need a git pull to grab the latest commit and `sudo iiab-network` to install the hook

holta, 2022-10-07:
FYI the PR below is now merged &mdash; with testing explanation(s) on that ticket here:

- PR https://github.com/iiab/iiab/pull/3392#issuecomment-1271955699

holta, 2022-10-16:
Progress ongoing:

- #3398
- #3404

holta, 2022-10-23:
This should be close soon, as Mint 21 is heavily tested &mdash; and even Moodle 4.1 LTS pre-releases (on PHP 8.1) finally appear reliable (:

- #3182
- #3381
- PR #3382

holta, 2022-12-22:
@shanti-bhardwa the new Linux Mint 21.1 is now available:

https://news.itsfoss.com/linux-mint-21-1-release/

Any chance you might have time to run an IIAB install test in the coming week?!

holta, 2022-12-24:
Thank You @shanti-bhardwa &mdash; for having completed this IIAB on Mint 21.1 test:

http://sprunge.us/phftYl

holta, 2022-12-30:
Good enough!

Huge thanks to @shanti-bhardwa for his testing last week :heavy_check_mark:

closed 2022-12-30 by holta