iiab/iiab #3309: Hardware Failure Icons on top of IIAB main page (http://box.lan) e.g. Overheating, Disk Full/Malfunction, WiFi Dead, RTC Dead

holta, 2022-07-15:
IIAB sysadmins want RED or ORANGE icons somewhere near the very top of IIAB's home pages.

So teachers have a strong visual indicator that hardware fails months later, e.g. due to:

- CPU overheating
- Disk Full (e.g. > 99% Full to teacher/operator can take action)
- Disk Severe Malfunction
- Hotspot Died/Disconnected
- RTC Died

Obviously false alarms ([banner blindness](https://en.wikipedia.org/wiki/Banner_blindness)) are counterproductive as UX designers know all too well.  Emergency alerts should be exactly that (used only for that Actual Emergencies, when immediate preventive action is hopefully still possible).

Making these Emergency Alerts happen isn't exactly easy.  It need quite a bit of thinking and will be deferred to another IIAB Milestone (after IIAB 8.0 !) given the strong likelihood it cannot be completed until 2023 or so?

Related:

- #2157
- #2205
- #2528 
- #3106
- PR #3308

tim-moody, 2022-07-16:
>Related

the first 4 are only about the out of storage condition

I don't understand how the last is related.

holta, 2022-07-16:
If networking fails (e.g. like the WiFi silicon that failed in many XO-1.5 laptops, or WiFi hotspot hardware becomes unreachable) then @jvonau's research might become very relevant to sort this out.

(Hopefully not a common flaw but it does happen!)

tim-moody, 2022-07-16:
the icon for failure of the hotspot would be a blank page.

holta, 2022-07-16:
In most all cases yep.

Some of us pre-test IIAB's over Ethernet, and/or pre-test using an attached screen — but those tricks are not practical in many field communities.

(Speaking of OLPC, they developed a very handy and thorough pre-boot hardware diagnostic sequence in later years, that we might learn from... and Raspberry Pi should really emulate!)

jvonau, 2022-07-16:
"Disk Severe Malfunction" means what and by what metric?
'Wi-Fi Hotspot Died/Disconnected' means what and by what metric? Either hostapd works right or the firmware can get locked up for various reasons with the only fix is to reboot.
"RTC Died" gpio pins coming play here, can't get this wrong or bad things can happen. This would suggest that some third party has become good with the rtc hardware layout and is shipping hardware with a rtc installed and IIAB pre-installed on the hardware. Get them to write the code.

holta, 2022-07-16:
> "Disk Severe Malfunction"

S.M.A.R.T. (Self-Monitoring, Analysis and Reporting Technology) was designed exactly this — signaling impending disk failure.

Applicable to some SSD's too.  But probably not microSD cards unfortunately.

(USB content disks aren't the most common today, but mounting failures months could be signaled if such things become common in future.)

> "Wi-Fi Hotspot Died/Disconnected"

If a meaningful heuristic is too hard to come by, this one can be dropped.

Conversely: if Wi-Fi hardware is dying months later, and rebooting doesn't help, and if this is detectable with a few command-line tricks in the end, this should be mentioned in http://FAQ.IIAB.IO or similar.

> "RTC Died" gpio pins [come in] play here, can't get this wrong or bad things can happen. This would suggest that some third party has become good with the rtc hardware layout and is shipping hardware with a rtc installed and IIAB pre-installed on the hardware. Get them to write the code.

One would hope so more than 10 years after Raspberry Pi began!  Certainly this deserves careful investigation a decade later.

On other hardware (refurb'd old laptops and thin clients, old PC's) RTC's own Battery can and do also fail.  This was an all-too-common problem on OLPC laptops after a few short years — worst case after just a few short months in the pre-deployment warehouse — i.e. if the RTC Battery was no longer recharging properly after an "extended school summer vacation" of not being turned on at all.

So if in the end RTC battery failure (?) can be easily determined (i.e. used to work, but no longer now works?) organized school systems would Absolutely Love That.

holta, 2022-07-17:
> - CPU overheating

This is very much wanted (and needed) in the tropics.  Regardless how fast the planet cooks in coming years.

So here are 3 options to extract CPU temperature if we need them:

- `vcgencmd measure_temp` (RasPiOS)
- `landscape-sysinfo --sysinfo-plugins=Temperature` (Ubuntu Server)
- `cat /sys/class/thermal/thermal_zone*/temp` (Every Debian/Ubuntu OS? Generates multiple outputs so needs to be parsed...)

(Presumably the cross-platform option is best??)