iiab/iiab #3320: Issues after installing a lot of content

Bigun, 2022-07-24:
I know this probably isn't a bug, but I couldn't find a support forum to speak of, so here we go.

After setting up IIAB, I began by prepping an external 5 TB drive and moving the /library folder to the drive and perma mounting it on boot.  I then began to install about 1 TB of content, which I knew would take a while, but after 2 days I started to wonder if it was still functional.  I tried logging into the admin console, and it wouldn't respond.  It would reboot with an issued SSH reboot command either.  So I killed the power and restarted it.

Upon boot, the main page seems mostly responsive, save a few ZIM modules at the bottom of the page.  I then went into admin console and noticed an error.

`7/24/2022, 9:19:16 AM: Init Failed
7/24/2022, 9:19:16 AM: GET-SPACE-AVAIL SUCCEEDED (38 ms)
7/24/2022, 9:19:16 AM: GET-EXTDEV-INFO SUCCEEDED (48 ms)
7/24/2022, 9:19:16 AM: GET-OSM-VECT-STAT SUCCEEDED (64 ms)
7/24/2022, 9:19:16 AM: GET-VARS SUCCEEDED (210 ms)
7/24/2022, 9:19:16 AM: Command GET-EXTDEV-INFO sent to server
7/24/2022, 9:19:16 AM: GET-ZIM-STAT FAILED and returned Unexpected error in Command GET-ZIM-STAT
7/24/2022, 9:19:16 AM: GET-ANS SUCCEEDED (88 ms)
7/24/2022, 9:19:16 AM: Command GET-SPACE-AVAIL sent to server
7/24/2022, 9:19:16 AM: Command GET-OSM-VECT-STAT sent to server
7/24/2022, 9:19:16 AM: GET-IIAB-INI SUCCEEDED (32 ms)
7/24/2022, 9:19:15 AM: Command GET-OER2GO-STAT sent to server
7/24/2022, 9:19:15 AM: Command GET-ZIM-STAT sent to server
7/24/2022, 9:19:15 AM: Command GET-IIAB-INI sent to server
7/24/2022, 9:19:15 AM: Command GET-ANS sent to server
7/24/2022, 9:19:15 AM: Command GET-VARS sent to server
7/24/2022, 9:19:15 AM: GET-ADM-CONF SUCCEEDED (23 ms)
7/24/2022, 9:19:15 AM: Command GET-ADM-CONF sent to server
7/24/2022, 9:19:08 AM: Successfully connected to Server
7/24/2022, 9:19:08 AM: Checking Server Connection
7/24/2022, 9:19:08 AM: Starting init`

This is running on a Raspberry Pi 4 with an external 5 TB drive mounted in /library

http://sprunge.us/7G2yh8

holta, 2022-07-24:
Thanks @Bigun for posting.  We'll look into it!

> I couldn't find a support forum

Just FYI IIAB forums are here for now:
https://github.com/iiab/iiab/discussions

(But this GitHub issue is 100% fine!)

tim-moody, 2022-07-24:
The only thing I see in the diags is 

Jul 24 13:25:30 box kiwix-serve[1757]: The XML library file '/library/zims/library.xml' is empty (or has only remote books).

this is causing kiwix to fail and would cause GET-ZIM-STAT to fail

the first thing to try is from a terminal as root run

iiab-make-kiwix-lib to recreate /library/zims/library.xml

it's possible that the line cp $KIWIXLIB $KIWIXLIB.tmp will fail if /library/zims/library.xml doesn't exist

try touch /library/zims/library.xml to fix that

Bigun, 2022-07-24:
Ran iiab-make-kiwix-lib, got this:

root@box:/home/iiab-admin# iiab-make-kiwix-lib
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Traceback (most recent call last):
  File "/usr/bin/iiab-make-kiwix-lib.py", line 85, in <module>
    main()
  File "/usr/bin/iiab-make-kiwix-lib.py", line 50, in main
    zims_installed, path_to_id_map = iiab.read_library_xml(kiwix_library_xml)
  File "/usr/lib/python3/dist-packages/iiab/iiab_lib.py", line 76, in read_library_xml
    tree = ET.parse(lib_xml_file)
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 1229, in parse
    tree.parse(source, parser)
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 580, in parse
    self._root = parser._parse_whole(source)
xml.etree.ElementTree.ParseError: no element found: line 1, column 0
Finished making Kiwix library.xml

Tried touching that file and re-ran iiab-make-kiwix-lib with the same result.

tim-moody, 2022-07-25:
try /usr/bin/iiab-make-kiwix-lib.py -f

tim-moody, 2022-07-25:
Note: we should modify iiab.read_library_xml so returns empty results on any error. This will force complete rebuild of library.xml.

Or could be handled in iiab-make-kiwix-lib3.py with try/except in line 50.

Bigun, 2022-07-25:
root@box:/home/iiab-admin# /usr/bin/iiab-make-kiwix-lib.py -f
Cannot add zim /library/zims/content/blender.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/crashcourse_en_all_2022-07.zim to the library.
Cannot add zim /library/zims/content/dba.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/devops.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/drupal.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/gutenberg_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/stackoverflow.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/stackoverflow.com_en_all_nopic_2022-07.zim to the library.
Cannot add zim /library/zims/content/wikispecies_en_all_maxi_2022-06.zim to the library.
Writing zim_versions_idx
Zim files and library.xml are out of sync. Please run iiab-make-kiwix-lib.
root@box:/home/iiab-admin# iiab-make-kiwix-lib
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Traceback (most recent call last):
  File "/usr/bin/iiab-make-kiwix-lib.py", line 85, in <module>
    main()
  File "/usr/bin/iiab-make-kiwix-lib.py", line 50, in main
    zims_installed, path_to_id_map = iiab.read_library_xml(kiwix_library_xml)
  File "/usr/lib/python3/dist-packages/iiab/iiab_lib.py", line 76, in read_library_xml
    tree = ET.parse(lib_xml_file)
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 1229, in parse
    tree.parse(source, parser)
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 580, in parse
    self._root = parser._parse_whole(source)
xml.etree.ElementTree.ParseError: no element found: line 1, column 0
Finished making Kiwix library.xml

tim-moody, 2022-07-25:
none of this makes any sense.

can you paste in the results of 

```
ls -lav /library/zims
ls -lav /library/zims/content
cat /library/zims/library.xml
/opt/iiab/kiwix/bin/kiwix-manage --version
/opt/iiab/kiwix/bin/kiwix-manage /library/zims/library.xml show
```

Bigun, 2022-07-25:
root@box:/home/iiab-admin# ls -lav /library/zims
total 32
drwxr-xr-x 4 root root  4096 Jul 25 03:31 .
drwxr-xr-x 9 root root  4096 Jul 22 19:11 ..
drwxr-xr-x 2 root root 20480 Jul 25 03:16 content
drwxr-xr-x 2 root root  4096 Jun 16 21:29 index
-rw-r--r-- 1 root root     0 Jul 25 03:30 library.xml

root@box:/home/iiab-admin# ls -lav /library/zims/content
total 152310536
drwxr-xr-x 2 root root       20480 Jul 25 03:16 .
drwxr-xr-x 4 root root        4096 Jul 25 03:31 ..
-rw-r--r-- 1 root root    89649363 May  5 03:12 3dprinting.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   280383325 May  7 05:47 academia.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    85442263 May  5 11:33 ai.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    37239084 May  7  2020 alittlequestionaday_en_2020-05.zim
-rw-r--r-- 1 root root   172928717 Nov  9  2020 allthetropes_en_all_maxi_2020-11.zim
-rw-r--r-- 1 root root     3276727 Mar  6  2021 alpinelinux_en_all_maxi_2021-03.zim
-rw-r--r-- 1 root root     2804074 Mar  6  2021 alpinelinux_en_all_nopic_2021-03.zim
-rw-r--r-- 1 root root   394191746 May  7 13:09 android.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   212877195 May  7 08:57 anime.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   931174321 May  7 11:13 apple.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   438810444 Mar  2  2021 appropedia_en_all_maxi_2021-03.zim
-rw-r--r-- 1 root root   116863026 Mar  2  2021 appropedia_en_all_nopic_2021-03.zim
-rw-r--r-- 1 root root    36631585 Apr 16 15:47 archlinux_en_all_maxi_2022-04.zim
-rw-r--r-- 1 root root    37388405 May 17 14:25 archlinux_en_all_nopic_2022-05.zim
-rw-r--r-- 1 root root   247271266 May  5 22:54 arduino.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   205349447 Mar 14  2021 artofproblemsolving_en_all_maxi_2021-03.zim
-rw-r--r-- 1 root root   157515888 Mar 14  2021 artofproblemsolving_en_all_nopic_2021-03.zim
-rw-r--r-- 1 root root  2258149937 May  2 08:03 askubuntu.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   150220422 May  5 11:31 astronomy.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   372337321 May  5 11:30 aviation.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    75804994 May  5 06:45 avp.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    13144901 Dec 10  2020 basic_elecricity_en_all_min_2020-11.zim
-rw-r--r-- 1 root root    14849621 May  2 05:36 beer.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   343889422 May  6 15:02 bicycles.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    43349426 May  6 16:50 bioinformatics.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   357463056 May  7 08:29 biology.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   158006709 May  7 18:34 bitcoin.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    17851165 Mar 20  2021 bitcoin_en_all_maxi_2021-03.zim
-rw-r--r-- 1 root root     5989844 Mar 20  2021 bitcoin_en_all_nopic_2021-03.zim
-rw-r--r-- 1 root root   869363712 Jul 24 13:32 blender.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   100869299 May  7 14:54 boardgames.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   112933874 May  7 14:55 bricks.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    74277288 May  7 08:27 buddhism.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   958713960 Feb 12 23:17 bulbagarden_en_all_maxi_2022-02.zim
-rw-r--r-- 1 root root   278149648 Feb 11 07:24 bulbagarden_en_all_nopic_2022-02.zim
-rw-r--r-- 1 root root   350265988 Jul 24 13:28 chemistry.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    69845335 May  7 13:11 chess.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   260689045 Jul 24 13:30 chinese.stackexchange.com_en_all_2021-04.zim
-rw-r--r-- 1 root root   115387726 May  7 20:36 chinese.stackexchange.com_zh_all_2022-05.zim
-rw-r--r-- 1 root root   148214046 May  7 13:08 christianity.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    83497029 May  8 00:09 civicrm.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   352251644 May  8 05:42 codegolf.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   570575787 May  7 23:09 codereview.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    16673938 May  7 17:42 coffee.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    65959606 Jul 24 22:52 cogsci.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    43912459 May  8 05:44 computergraphics.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root     5431305 May  2 10:27 conlang.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   222641071 May  7 08:43 cooking.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    61204948 May  7 13:28 craftcms.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    44393059 May  8 05:39 crafts.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root 41506500608 Jul 24 13:33 crashcourse_en_all_2022-07.zim
-rw-r--r-- 1 root root   169054113 Jul 24 13:32 crypto.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root    17336674 May  8 18:08 cseducators.stackexchange.com_en_all_2022-05.zim
-rw-r--r-- 1 root root   270096892 May  8 17:59 cs.stackexchange.com_cs_all_2022-05.zim
-rw-r--r-- 1 root root   847159879 Jul 24 13:31 cs.stackexchange.com_en_all_2021-04.zim
[... cut ...]
root@box:/home/iiab-admin# cat /library/zims/library.xml
root@box:/home/iiab-admin#

root@box:/home/iiab-admin# /opt/iiab/kiwix/bin/kiwix-manage --version
kiwix-tools 3.3.0

libkiwix 11.0.0
+ libzim 7.2.2
+ libxapian 1.4.18
+ libcurl 7.67.0
+ libmicrohttpd 0.9.72
+ libz 1.2.12
+ libicu 58.2.0
+ libpugixml 0.12.0

libzim 7.2.2
+ libzstd 1.5.2
+ liblzma 5.2.4
+ libxapian 1.4.18
+ libicu 58.2.0

root@box:/home/iiab-admin# /opt/iiab/kiwix/bin/kiwix-manage /library/zims/library.xml show
root@box:/home/iiab-admin#

tim-moody, 2022-07-25:
I'd like to log into your server to see what's going on. if you run iiab-support it will allow me to log in.

holta, 2022-07-25:
> I'd like to log into your server to see what's going on. if you run iiab-support it will allow me to log in.

Good idea.

@Bigun is that ok with you?

(This could help a lot of people, figuring this out!)

tim-moody, 2022-07-25:
just doing some spot checking. you have
-rw-r--r-- 1 root root 267730944 May 7 13:54 drupal.stackexchange.com_en_all_2022-05.zim
I did a download today and got
-rw-r--r-- 1 root root 411731758 May  7 09:54 drupal.stackexchange.com_en_all_2022-05.zim

kiwix-manage can add mine to a library.xml file but not yours

tim-moody, 2022-07-26:
alternatively you can try

`iiab-make-kiwix-lib.py -f --no_tmp`

no matter what it says, don't run iiab-make-kiwix-lib afterwards. just run 

`/opt/iiab/kiwix/bin/kiwix-manage /library/zims/library.xml show`

and paste the results

Bigun, 2022-07-26:
https://dpaste.com/E9DRSGWNE

I'll work on getting iiab-support running momentarily.

Bigun, 2022-07-26:
Done and connected, look for my nick.  If you need the IP, let me know.

holta, 2022-07-26:
> Done and connected, look for my nick. If you need the IP, let me know.

1) All set, _huge thanks @Bigun:_

   Hopefully @tim-moody will be able to get to the bottom of  these issues today :heavy_check_mark:

2) QUICK SIDE QUESTIONS, such that others might do very similar &mdash; what kind of HDD or SSD physical drive & mounting/enclosure did you arrange to make this 5TB /library happen on `/dev/sda1` &mdash; if you're able to describe just a bit, towards helping others?

   - Can you share brand/model details?
   - What Linux mounting tricks did you use, e.g. in `/etc/fstab` ?
   - Did you arrange for additional electrical power, e.g. if the external 5TB drive needs that? (occasionally there are serious system stability problems here, if electrical power's not beefed up!)
     - And worse serious WiFi RF interference sometimes results from USB3 cabling, preventing use of IIAB's learning hotspot: #2638
   - Do you have a favorite enclosure &/or simple cabling for use in a school?
     - Did you also consider active &/or passive cooling?

tim-moody, 2022-07-26:
I logged in. looks like iiab-make-kiwix-lib.py -f --no_tmp got things back on track, though there are still the some zims that I think did not download completely (were listed above and also when I reran make kiwix lib):

Cannot add zim /library/zims/content/blender.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/crashcourse_en_all_2022-07.zim to the library.
Cannot add zim /library/zims/content/dba.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/devops.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/drupal.stackexchange.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/gutenberg_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/stackoverflow.com_en_all_2022-05.zim to the library.
Cannot add zim /library/zims/content/stackoverflow.com_en_all_nopic_2022-07.zim to the library.
Cannot add zim /library/zims/content/wikispecies_en_all_maxi_2022-06.zim to the library.

I don't know how they got out of /library/working/zims/ and into /library/zims/content

The complete download history is now in https://dpaste.com//H67H8YLNF Has to be a world record, but there are some failures, which could be tried again. You could try moving an incomplete zim back into /library/working/zims/ and selecting it again. Theoretically the download should try to continue from the point at which it stopped. Or just remove incomplete zims from /library/zims/content, select again, and download.

The iiab-admin password is not the default so I can't log into Admin Console, but running from the command line GET-ZIM-STAT now works, so you should be able to login.

There are still modules in /library/working/rachel. It may be possible to select these again and have have them continue their download.

Installed modules are in https://dpaste.com//8FR3HZK8J

Not sure where the home page menu stands; there could be missing items. Compare the list in /library/www/html/home/menu.json with the list of zims and modules.

That's about it for now, should allow you to surface the next set of issues if any.

Bigun, 2022-07-26:
@tim-moody - I've set the default iiab-admin password back to default if you want to have another look

@holta 
 - Seagate Backup Plus External 5TB drive
 - I mounted the drive @ /library and used the UUID of /dev/sda1 to mount the drive, here's the line from fstab:
    UUID=50b95ff2-53fb-46d0-a1e6-4acb2450e168 /library  ext4  defaults,nofail  0  0
 - The drive had it's own OEM power supply
  - The pi has passive cooling (some heatsinks), the external drive has no other cooling that I can see, I assume there may be some shielding in the case, but as I've never cracked it open, I'm not 100% sure.
    - This project was started because I had a spare pi, heard about the project from a TikTok, and got morbidly curious about how well a pi would perform with enough storage and loaded up with a obscene amount of information.  I was also curious about what other issues I would run into using offline information - for instance, it looks like the deprecated flash plugin may be a requirement to look at some of the math modules, also the "Try it yourself" module in the w3schools PHP section is inoperable (likely due to missing the PHP module not installed on the pi.  Again, was just curious.  I'm part of a Defcon local group, planning on presenting a talk on it fairly soon.

tim-moody, 2022-07-26:
>got morbidly curious about how well a pi would perform with enough storage and loaded up with an obscene amount of information

there's some rough testing code at https://github.com/iiab/iiab-factory/tree/master/testing. I did some load testing against kiwix.

>looks like the deprecated flash plugin may be a requirement to look at some of the math modules

phet has been rewritten with js, but the catalog may have old versions. not sure which other modules, but content rarely goes away.

>the "Try it yourself" module in the w3schools PHP section is inoperable (likely due to missing the PHP module not installed on the pi

php usually gets installed, but there could be a database or something that's missing. not sure who scraped w3schools; wasn't aware it was a module. Actually, just discovered it's on my server.

tim-moody, 2022-07-26:
I'll turn it back over to you now.

tim-moody, 2022-07-26:
#3321

Bigun, 2022-07-26:
Thank you for your help, going to present this to my DC group this Wednesday, I've already mentioned it and got a few head turns - maybe we can get a few contributors out of this.

tim-moody, 2022-07-26:
sounds good

holta, 2022-10-19:
@Bigun let us know if you have ongoing issues here, Thanks!

closed 2022-10-19 by holta