iiab/iiab #3324: Dell inspiron 1750 wlan issues

EMG70, 2022-07-28:
### Expected behavior and actual behavior:
Use teacher laptop as internal wlan for iian .The wlan is somehow being blocked by UFW .If ufw is disabled it comes back after every reboot.
[image]

### Steps to reproduce the problem:
Tested with assistance of Shanti and Adam Holt

### IIAB version, OS version, hardware details:
8 ,Linux Mint 20.3

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (http://sprunge.us/CZkc7E)
[image]
[image]

EMG70, 2022-07-28:
[image]
[image]

EMG70, 2022-07-28:
[iia_network.pdf](https://github.com/iiab/iiab/files/9214579/iia_network.pdf)

holta, 2022-07-28:
Thank you @EMG70!

Quick summary is that you've tried running `sudo iiab-network` many times (with USB WiFi dongle **Logitech C-U0010** inserted, and also with it removed) and the results seem to be the same regardless:

1. Your [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) are: http://sprunge.us/CZkc7E?en
2. Your `/opt/iiab/iiab/iiab-network.log` is: http://sprunge.us/cQh3Kn?en
3. Your `journalctl -u hostapd` output is: http://sprunge.us/XsJJYn?en

holta, 2022-07-28:
Thank you @EMG70 also for highlighting the red error (below, from your above PDF) &mdash; that's arising every time you run **sudo iiab-network** :

> TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway with wifi_up_down False] *** fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service hostapd: Job for hostapd.service failed because the control process exited with error code.\nSee \"systemctl status hostapd.service\" and \"journalctl -xe\" for details.\n"}

holta, 2022-07-28:
Thanks also @EMG70 for having confirmed that your Dell Inspiron 1750 appears to contain a "Broadcom BCM4315" WiFi chipset, according to `dmesg | grep 802` and `dmesg | grep wl`

Which doesn't look promising (``iw list` indicates it's not capable of AP Mode, but knowing what you face is a start) &mdash; so that @shanti-bhardwa and others can suggest possible options!

(e.g. if your **Logitech C-U0010** USB WiFi dongle is also not sufficient ?)

holta, 2022-07-29:
> The wlan is somehow being blocked by UFW

Clarifications & Correction:

- Yes **ufw** was somehow installed accidentally.

  - We're not sure how, likely manually as a result of an optional checkbox during the Mint 20.3 install process ??)

- ufw was blocking ssh (not WiFi)

- `sudo ufw disable` helped to restore ssh access, _but only as a temporary workaround until next boot_

- `sudo apt purge ufw` hopefully acts as a more permanent solution ?

  - @EMG70 please confirm, if this indeed works permanently, across reboots?

- Everyone should feel free to use `ufw` or any other firewall — if they first do their homework — _configuring it properly, in keeping with the ports they truly need._

  - This table can help:

    https://github.com/iiab/iiab/wiki/IIAB-Networking#list-of-ports--services

jvonau, 2022-07-31:
I'd like to see a fresh machine configured with network_install: False, network_enabled: False and vpn access, when available let me know.

EMG70, 2022-07-31:
HI,I am available now. Please bear with me as I am new to Linux

jvonau, 2022-07-31:
@EMG70 Do you have a second laptop with your OS installed but without iiab installed yet?

EMG70, 2022-07-31:
I only have one laptop with Linux but can quickly load on another old spare laptop

jvonau, 2022-07-31:
Once you have your mint install with no iiab on the laptop could you open a terminal within the GUI and run this command: `which iw` please, could show 'command not found' or something like '/usr/sbin/iw'. I'll guide you through the rest of the steps if needed.

holta, 2022-07-31:
FYI Linux Mint 21 is now official, as you can see here: https://linuxmint.com

Also downloadable at its mirror sites here: https://linuxmint.com/mirrors.php

(So do consider that, going forward, so that testing is uniform with what so many schools will be using over the coming 2 years!)

EMG70, 2022-07-31:
OK ,let me get second laptop ready and will install Mint 21,hopefully should be done pretty quick

jvonau, 2022-07-31:
I prefer one variable at a time... switching OS's might be apples to oranges, lets not make a linux newbie be your beta tester...

EMG70, 2022-07-31:
ok so we stick to same 20.3 for now.
Is it ok to use same old dell 1750 without iiab data,it might take me time to open other one and install mint

jvonau, 2022-07-31:
No there are mods made that would mask the problem once 'iiab' is ran even once as the apt package would be installed.

EMG70, 2022-07-31:
OK,am downloading mint 20.3 on other laptop now.Please give me 15-20mins

jvonau, 2022-07-31:
It will be time well spent, making iiab more bulletproof in the long run, thank you

EMG70, 2022-07-31:
Bit annoying,I have changed boot order to usb so i can load linux on lenovo laptop but not loading from usb as priority.I am still trying

EMG70, 2022-07-31:
I am not winning Guys,just wont load from selected usb mode

jvonau, 2022-07-31:
Sigh, guess that second laptop model whatever it is won't be useful if mint won't install on it.

EMG70, 2022-07-31:
Very frustrating.I am sorry,it seems its not changeable ,priority always from its ssd which i haven't got same type of ssd(small).
Any help is appreciated in making bios boot from usb(first option) is welcome as 
laptop model :Lenovo ideapad 330S 141KB

EMG70, 2022-07-31:
let me try your commands Jerry as soon as i get something running, will revert back

EMG70, 2022-07-31:
@jvonau Is it helpful to reload 20.3 only on that dell 1750 as I have struggled to get other laptop to load Linux mint,issues with its  bios which do not seem to be flexible.

jvonau, 2022-07-31:
Yes it would. Don't really want to force you to start completely over, to preserve what you have already have done with the first drive, can you swap a different drive into that machine?

EMG70, 2022-07-31:
Yes I have back up drives ,so should be ok.Will load mint only now

EMG70, 2022-07-31:
All loaded now,should I carry on with your earlier instructions?

jvonau, 2022-07-31:
Once you have captured the output of `which iw` make a note of that in this issue first.

jvonau, 2022-07-31:
Connect to the internet as you did before using the internal wifi of the laptop and optionally wired Ethernet but do not add the second usb wifi device at this time.
Proceed to install iiab with the curl script
post-install run iiab-diagnostics and post the url in this issue please.

jvonau, 2022-07-31:
I'll be back in an hour or so.

EMG70, 2022-07-31:
Ok Jerry

EMG70, 2022-07-31:
<pre>root@mint:/home/mint# which iw
/usr/sbin/iw
root@mint:/home/mint#</pre>

EMG70, 2022-07-31:
<pre>Installing Internet-in-a-Box requires /etc/iiab/local_vars.yml

PLEASE PICK (1) 🚵 SMALL (2) 🚢🚣 MEDIUM or (3) 🚂🚃🚃 LARGE (NEEDS 2+ GB RAM).
Or try (m) for 🚑 MEDICAL.

See &quot;What services (IIAB apps) are suggested during installation?&quot; and
&quot;What can I do with E-books and Internet-in-a-Box?&quot; in http://FAQ.IIAB.IO

WARNING: This Can Take More Than An Hour, depending on Internet speed, CPU,
temperature, and the speed of your microSD card.

Please type 1, 2, 3 or m: 1

--2022-07-31 16:30:16--  https://github.com/iiab/iiab/raw/master/vars/local_vars_small.yml
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/iiab/iiab/master/vars/local_vars_small.yml [following]
--2022-07-31 16:30:16--  https://raw.githubusercontent.com/iiab/iiab/master/vars/local_vars_small.yml
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8000::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17548 (17K) [text/plain]
Saving to: ‘/etc/iiab/local_vars.yml’

/etc/iiab/local_var 100%[===================&gt;]  17.14K  --.-KB/s    in 0.003s  

2022-07-31 16:30:16 (5.16 MB/s) - ‘/etc/iiab/local_vars.yml’ saved [17548/17548]

Edit /etc/iiab/local_vars.yml to customize your Internet-in-a-Box? [Y/n] n

 ██████████████████████████████████████████████████████████████████████████████
 ██                                                                          ██
 ██  RUN &apos;sudo iiab&apos; IF THIS INSTALL SCRIPT EVER FAILS, TO TRY TO CONTINUE!  ██
 ██                                                                          ██
 ██████████████████████████████████████████████████████████████████████████████

&apos;apt update&apos; is checking for OS updates, <b>which may force a reboot...</b>

&apos;apt update&apos; FAILED. VERIFY YOU&apos;RE ONLINE and resolve all errors below:

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

E: The repository &apos;cdrom://Linux Mint 20.3 _Una_ - Release amd64 20220104 focal Release&apos; does not have a Release file.
root@mint:/home/mint# 
</pre>

EMG70, 2022-07-31:
<pre>root@mint:/home/mint# sudo iiab-diagnostics
sudo: iiab-diagnostics: command not found
root@mint:/home/mint#</pre>

EMG70, 2022-07-31:
Please ignore below
> root@mint:/home/mint# sudo iiab-diagnostics
> sudo: iiab-diagnostics: command not found
> root@mint:/home/mint#

EMG70, 2022-07-31:
<pre>mint:/home/mint# iiab - diagnostics

  EXISTING /etc/iiab/local_vars.yml is being used to install Internet-in-a-Box

   🚂 🚃 🚄 🚅 🚆 🚇 🚈 🚉 🚊 🚋 🚌 🚍 🚎 🚏 🚐 🚑 🚒 🚚 🚛 🚜 🚞 🚟 🚠 🚡 🚲

                     Google &apos;local_vars.yml&apos; to learn more!

 ██████████████████████████████████████████████████████████████████████████████
 ██                                                                          ██
 ██  RUN &apos;sudo iiab&apos; IF THIS INSTALL SCRIPT EVER FAILS, TO TRY TO CONTINUE!  ██
 ██                                                                          ██
 ██████████████████████████████████████████████████████████████████████████████

&apos;apt update&apos; is checking for OS updates, <b>which may force a reboot...</b>

&apos;apt update&apos; FAILED. VERIFY YOU&apos;RE ONLINE and resolve all errors below:

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

E: The repository &apos;cdrom://Linux Mint 20.3 _Una_ - Release amd64 20220104 focal Release&apos; does not have a Release file.
root@mint:/home/mint#</pre>

EMG70, 2022-07-31:
root@mint:/home/mint# sudo iiab

  EXISTING /etc/iiab/local_vars.yml is being used to install Internet-in-a-Box

   🚂 🚃 🚄 🚅 🚆 🚇 🚈 🚉 🚊 🚋 🚌 🚍 🚎 🚏 🚐 🚑 🚒 🚚 🚛 🚜 🚞 🚟 🚠 🚡 🚲

                     Google 'local_vars.yml' to learn more!

 ██████████████████████████████████████████████████████████████████████████████
 ██                                                                          ██
 ██  RUN 'sudo iiab' IF THIS INSTALL SCRIPT EVER FAILS, TO TRY TO CONTINUE!  ██
 ██                                                                          ██
 ██████████████████████████████████████████████████████████████████████████████

'apt update' is checking for OS updates, which may force a reboot...

'apt update' FAILED. VERIFY YOU'RE ONLINE and resolve all errors below:

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

E: The repository 'cdrom://Linux Mint 20.3 _Una_ - Release amd64 20220104 focal Release' does not have a Release file.
root@mint:/home/mint#

EMG70, 2022-07-31:
<pre>root@mint:/home/mint# apt update
Ign:1 cdrom://Linux Mint 20.3 _Una_ - Release amd64 20220104 focal InRelease
Err:2 cdrom://Linux Mint 20.3 _Una_ - Release amd64 20220104 focal Release
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Hit:3 http://archive.canonical.com/ubuntu focal InRelease
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease                                                 
Ign:5 http://packages.linuxmint.com una InRelease                                          <font color="#C4A000">                      </font>
Hit:6 http://archive.ubuntu.com/ubuntu focal InRelease<font color="#C4A000">                                     </font>
Hit:7 http://packages.linuxmint.com una Release
Hit:8 http://archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:9 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Reading package lists... Done
<font color="#EF2929"><b>E: </b></font>The repository &apos;cdrom://Linux Mint 20.3 _Una_ - Release amd64 20220104 focal Release&apos; does not have a Release file.
<font color="#C4A000">N: </font>Updating from such a repository can&apos;t be done securely, and is therefore disabled by default.
<font color="#C4A000">N: </font>See apt-secure(8) manpage for repository creation and user configuration details.
root@mint:/home/mint# 
</pre>

EMG70, 2022-07-31:
[image]

EMG70, 2022-07-31:
wirelss adapter not working,I will wait for Jerry for instructions

jvonau, 2022-07-31:
How did you resolve the " cdrom://Linux Mint 20.3 _Una_ " error encountered above? How did you enable the device the first time? Did you have the ethernet active?

EMG70, 2022-07-31:
I did not resolve anything intentionally.I just followed your instructions then realised wlan wasn't showing and went to device manager which picked up that error .I have selected the Linux drivers and says still applying for past 20mins
[image]

EMG70, 2022-07-31:
I have been connected via Ethernet throughout

jvonau, 2022-07-31:
OK just let it continue for now. Not sure why you saw the `apt update` error noted above, bug somewhere in mint I would think..

EMG70, 2022-07-31:
Yes saw it and don't have an idea why

jvonau, 2022-07-31:
Me either, @holta @shanti-bhardwa Have you come across the 'cdrom' error noted above?

holta, 2022-07-31:
> Me either, @holta @shanti-bhardwa Have you come across the 'cdrom' error noted above?

I've never seen it before.

It'd be good to understand (if Mint 21 still has this error/issue ??) as Linux Mint seems to be getting more popular.

jvonau, 2022-07-31:
@EMG70 are you following somebody's cheat sheet/how to instructions from somewhere?

EMG70, 2022-07-31:
This laptop is too old Gents.I could not activate wifi via Fn + F2 on most of the time but would come up after hrs.It hasn't been responding since installing this 20.3 today .Hope i am not wasting your  time on old technology.

EMG70, 2022-07-31:
I have very limited knowledge on linux,last used it a few yrs back but mostly Virtual Box and ubuntu lab seeds

EMG70, 2022-07-31:
driver up[date is on 80% progress now,will hold on until its done

jvonau, 2022-07-31:
No not wasting my time but I think 'apt' might be messed up somehow as `apt update` is producing the 'cdrom' error. If you can get to a terminal on a hunch can you grab the output of `cat /etc/apt/sources.list` after the driver update is done.
Ref: https://forums.linuxmint.com/viewtopic.php?t=365938

EMG70, 2022-07-31:
Will do,Thanks.

jvonau, 2022-07-31:
If the installation media is still inserted in the usb port, I'd umount the drive and remove that usb stick before trying `apt update` again..

jvonau, 2022-07-31:
After the driver is updated not before that process finishes.

EMG70, 2022-07-31:
I had etched mint 20.3 onto the drive itself .OK will do apt update when its finished applying changes.

jvonau, 2022-07-31:
If I understand you correctly the starting point was an image file that was passed to you from somebody else and not a fresh install starting with booting from usb?

jvonau, 2022-07-31:
Re: cdrom error https://forums.linuxmint.com/viewtopic.php?t=357766

EMG70, 2022-07-31:
No,I saved from Linux website and used Uk server to download onto a stick and then copied it to hard drive which i inserted to boot

EMG70, 2022-07-31:
I have had to abandon that download as nothing seemed to be happening.I have seen dell drivers for same laptop but not sure which one is the executable ,after extraction there are many .exe files.please see following

[57 more replies]

closed 2022-08-16 by holta