iiab/iiab #3325: iiab-expand-rootfs.service often starts 2 simultaneous copies of /usr/sbin/iiab-expand-rootfs, causing it to crash

holta, 2022-07-29:
This happens to be on "RasPiOS with desktop" but likely occurs on many/most if not all OS's.

In the journalctl log below, you can see many examples of `/usr/sbin/iiab-expand-rootfs` _being launched twice simultaneously_.  The output of these simultaneous / duplicate processes is made even more clear by 32-digit hex 'Boot' identifier numbers at the top of each chunk of output, interweaved together, evidenced by each of the 2 running processes having 100% identical timestamps.

Unhealthy "conjoined twins" in a sense.

The 1st example (at the top of the log below) being the most onerous, as contention between those 2 processes caused the crash below (Line 48 of 117, made clear by: `/usr/sbin/iiab-expand-rootfs: line 59:   669 Segmentation fault      resize2fs $ROOT_PART`) &mdash; that prevented the drive from fully expanding to its actual physical capacity.

### `journalctl -u iiab-expand-rootfs` confirms the problem

<details>
  <summary>Click To Expand 'journalctl -u iiab-expand-rootfs' Log Output</summary>

```
root@box:~# journalctl -u iiab-expand-rootfs
-- Journal begins at Mon 2022-04-04 14:52:30 UTC, ends at Fri 2022-07-29 03:00:03 UTC. --
Jun 16 20:51:03 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jun 16 20:51:03 box iiab-expand-rootfs[434]: + '[' -f /.expand-rootfs ']'
Jun 16 20:51:03 box iiab-expand-rootfs[434]: + '[' -f /.resize-rootfs ']'
Jun 16 20:51:03 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jun 16 20:51:03 box systemd[1]: Finished Root Filesystem Auto-Expander.
-- Boot 84cc35b34186456ba2adf08f7b4d5399 --
Jun 16 20:52:17 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + '[' -f /.expand-rootfs ']'
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + echo '/usr/sbin/iiab-expand-rootfs: Expanding rootfs partition'
Jun 16 20:52:17 box iiab-expand-rootfs[453]: /usr/sbin/iiab-expand-rootfs: Expanding rootfs partition
Jun 16 20:52:17 box iiab-expand-rootfs[468]: ++ findmnt / -o SOURCE -n
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[505]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:52:17 box iiab-expand-rootfs[541]: ++ echo /dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[542]: ++ grep -o '[[:digit:]]*$'
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_PART_NUM=2
Jun 16 20:52:17 box iiab-expand-rootfs[551]: ++ cut -f 1 -d:
Jun 16 20:52:17 box iiab-expand-rootfs[550]: ++ tail -n 1
Jun 16 20:52:17 box iiab-expand-rootfs[549]: ++ parted /dev/mmcblk0 -ms unit s p
-- Boot 8ac77d02beb348ebbfb5fe4562f05b65 --
Jun 16 20:52:17 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + '[' -f /.expand-rootfs ']'
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + echo '/usr/sbin/iiab-expand-rootfs: Expanding rootfs partition'
Jun 16 20:52:17 box iiab-expand-rootfs[455]: /usr/sbin/iiab-expand-rootfs: Expanding rootfs partition
Jun 16 20:52:17 box iiab-expand-rootfs[473]: ++ findmnt / -o SOURCE -n
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[496]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:52:18 box iiab-expand-rootfs[455]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:52:18 box iiab-expand-rootfs[522]: ++ grep -o '[[:digit:]]*$'
Jun 16 20:52:18 box iiab-expand-rootfs[521]: ++ echo /dev/mmcblk0p2
Jun 16 20:52:18 box iiab-expand-rootfs[455]: + ROOT_PART_NUM=2
Jun 16 20:52:18 box iiab-expand-rootfs[535]: ++ parted /dev/mmcblk0 -ms unit s p
Jun 16 20:52:18 box iiab-expand-rootfs[537]: ++ cut -f 1 -d:
Jun 16 20:52:18 box iiab-expand-rootfs[536]: ++ tail -n 1
-- Boot 84cc35b34186456ba2adf08f7b4d5399 --
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + LAST_PART_NUM=2
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + '[' 2 -ne 2 ']'
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + growpart /dev/mmcblk0 2
-- Boot 8ac77d02beb348ebbfb5fe4562f05b65 --
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + LAST_PART_NUM=2
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + '[' 2 -ne 2 ']'
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + growpart /dev/mmcblk0 2
Jun 16 20:52:19 box iiab-expand-rootfs[631]: NOCHANGE: partition 2 is size 999210975. it cannot be grown
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:19 box iiab-expand-rootfs[669]: resize2fs 1.46.2 (28-Feb-2021)
Jun 16 20:52:20 box iiab-expand-rootfs[455]: /usr/sbin/iiab-expand-rootfs: line 59:   669 Segmentation fault      resize2fs $ROOT_PART
Jun 16 20:52:20 box iiab-expand-rootfs[455]: + rm -f /.expand-rootfs /.resize-rootfs
Jun 16 20:52:20 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jun 16 20:52:20 box systemd[1]: Finished Root Filesystem Auto-Expander.
-- Boot 84cc35b34186456ba2adf08f7b4d5399 --
Jun 16 20:52:20 box iiab-expand-rootfs[632]: CHANGED: partition=2 start=532480 old: size=23317568 end=23850048 new: size=999210975 end=999743455
Jun 16 20:52:20 box iiab-expand-rootfs[453]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:20 box iiab-expand-rootfs[735]: resize2fs 1.46.2 (28-Feb-2021)
[... 46 lines cut ...]
Jul 28 13:48:50 box iiab-expand-rootfs[470]: + '[' -f /.expand-rootfs ']'
Jul 28 13:48:50 box iiab-expand-rootfs[470]: + '[' -f /.resize-rootfs ']'
Jul 28 13:48:51 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jul 28 13:48:51 box systemd[1]: Finished Root Filesystem Auto-Expander.
-- Boot bc1ac6387d584b468ded38c9aee772c5 --
Jul 28 21:51:36 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jul 28 21:51:36 box iiab-expand-rootfs[456]: + '[' -f /.expand-rootfs ']'
Jul 28 21:51:36 box iiab-expand-rootfs[456]: + '[' -f /.resize-rootfs ']'
Jul 28 21:51:36 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jul 28 21:51:36 box systemd[1]: Finished Root Filesystem Auto-Expander.
-- Boot 1e391b0fe8a74fa0bb73ab9be6f051fd --
Jul 28 22:04:19 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jul 28 22:04:19 box iiab-expand-rootfs[451]: + '[' -f /.expand-rootfs ']'
Jul 28 22:04:19 box iiab-expand-rootfs[451]: + '[' -f /.resize-rootfs ']'
Jul 28 22:04:19 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jul 28 22:04:19 box systemd[1]: Finished Root Filesystem Auto-Expander.
```
  
</details>

### Remediation

1) Reverting to `raspi-config --expand-rootfs` (with manual reboot, already in our code but commented out for now) is a dead easy short-term fix on RasPiOS: https://github.com/iiab/iiab/blob/2efd7c01982ccecfb1297ae2bc93087172b38c2a/roles/1-prep/templates/iiab-expand-rootfs#L17

2) But it's hopefully also possible to fix [iiab-expand-rootfs.service](https://github.com/iiab/iiab/blob/master/roles/1-prep/templates/iiab-expand-rootfs.service) across all OS's &mdash; to put an end to this erroneous and quite common simultaneous-double-launching of [/usr/sbin/iiab-expand-rootfs](https://github.com/iiab/iiab/blob/master/roles/1-prep/templates/iiab-expand-rootfs) ?

tim-moody, 2022-07-29:
So to summarize, because one line of the script logs twice, you leapt to the conclusion that the script ran twice, ignoring the fact that the other three lines, including start and end messages only appear once. As a result you made changes that broke it and now think that because it's broken it should be eliminated altogether. As further evidence that it ran twice you took the fact that for one user it did nothing, not the same thing twice, without any idea as to why that user had a problem. Is that pretty much it?

holta, 2022-07-29:
Read the log and you will see that it started 2 simultaneous copies of /usr/sbin/iiab-expand-rootfs 2 times out of 10.

So it's almost definitely a timing issue during boot.  SystemD should be celebrated for helping us to clearly see this in its more modern logging system.

If you're accusing others of breaking things without evidence, that's not helpful.

holta, 2022-07-29:
> it started 2 simultaneous copies of /usr/sbin/iiab-expand-rootfs 2 times out of 10.

Clarification for people who don't read the logs:

The log shows that double-launching of iiab-expand-rootfs happens repeatedly &mdash; both during initial boot (when trying to expand the root filesystem) and also on subsequent boot(s) as well.

tim-moody, 2022-07-29:
Please cite the actual lines which you take to be evidence that it ran twice.

Does 'Starting Root Filesystem Auto-Expander' appear twice?

where does the log come from, what OS?

holta, 2022-07-29:
> where does the log come from, what OS?

Please do read this ticket (the top, especially).  To restate it: this occurred on 64-bit "RasPiOS with desktop" on RPi 400.

tim-moody, 2022-07-29:
>Please cite the actual lines which you take to be evidence that it ran twice.

tim-moody, 2022-07-29:
Jul 28 22:04:19 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jul 28 22:04:19 box iiab-expand-rootfs[451]: + '[' -f /.expand-rootfs ']'
Jul 28 22:04:19 box iiab-expand-rootfs[451]: + '[' -f /.resize-rootfs ']'
Jul 28 22:04:19 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jul 28 22:04:19 box systemd[1]: Finished Root Filesystem Auto-Expander.

holta, 2022-07-29:
> Please cite the actual lines which you take to be evidence that it ran twice.

Please analyze the log more patiently:

- Lines 7-55 show the 1st example of duplicate invocation of iiab-expand-rootfs
- Line 86-99 show another example

As explained at the top of this ticket, the interweaved chunks (sections, each with a `Boot` identifier 32-digit numbers) make clear how the 2 processes are happening simultaneously.

tim-moody, 2022-07-29:
If I run journalctl -u iiab-expand-rootfs I don't see the -- Boot 84cc35b34186456ba2adf08f7b4d5399 -- line, did you add that, is it a fresh boot? what does journalctl -b -u iiab-expand-rootfs show?

holta, 2022-07-29:
> If I run journalctl -u iiab-expand-rootfs I don't see the -- Boot 84cc35b34186456ba2adf08f7b4d5399 -- line, did you add that,

No, I assume Systemd adds those into journalctl logging.

> what does journalctl -b -u iiab-expand-rootfs show?

Indeed the `-b` flag gets rid of the interweaving / interleaving.

Making these 2 invocations (that have identical timestamps) more readable:

```
root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 -u iiab-expand-rootfs
-- Journal begins at Mon 2022-04-04 14:52:30 UTC, ends at Fri 2022-07-29 12:57:08 UTC. --
Jun 16 20:52:17 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + '[' -f /.expand-rootfs ']'
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + echo '/usr/sbin/iiab-expand-rootfs: Expanding rootfs partition'
Jun 16 20:52:17 box iiab-expand-rootfs[453]: /usr/sbin/iiab-expand-rootfs: Expanding rootfs partition
Jun 16 20:52:17 box iiab-expand-rootfs[468]: ++ findmnt / -o SOURCE -n
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[505]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:52:17 box iiab-expand-rootfs[541]: ++ echo /dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[542]: ++ grep -o '[[:digit:]]*$'
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_PART_NUM=2
Jun 16 20:52:17 box iiab-expand-rootfs[551]: ++ cut -f 1 -d:
Jun 16 20:52:17 box iiab-expand-rootfs[550]: ++ tail -n 1
Jun 16 20:52:17 box iiab-expand-rootfs[549]: ++ parted /dev/mmcblk0 -ms unit s p
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + LAST_PART_NUM=2
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + '[' 2 -ne 2 ']'
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + growpart /dev/mmcblk0 2
Jun 16 20:52:20 box iiab-expand-rootfs[632]: CHANGED: partition=2 start=532480 old: size=23317568 end=23850048 new: size=999210975 end=999743455
Jun 16 20:52:20 box iiab-expand-rootfs[453]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:20 box iiab-expand-rootfs[735]: resize2fs 1.46.2 (28-Feb-2021)
```

```
root@box:~# journalctl -b 8ac77d02beb348ebbfb5fe4562f05b65 -u iiab-expand-rootfs
-- Journal begins at Mon 2022-04-04 14:52:30 UTC, ends at Fri 2022-07-29 12:57:08 UTC. --
Jun 16 20:52:17 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + '[' -f /.expand-rootfs ']'
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + echo '/usr/sbin/iiab-expand-rootfs: Expanding rootfs partition'
Jun 16 20:52:17 box iiab-expand-rootfs[455]: /usr/sbin/iiab-expand-rootfs: Expanding rootfs partition
Jun 16 20:52:17 box iiab-expand-rootfs[473]: ++ findmnt / -o SOURCE -n
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[496]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:52:18 box iiab-expand-rootfs[455]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:52:18 box iiab-expand-rootfs[522]: ++ grep -o '[[:digit:]]*$'
Jun 16 20:52:18 box iiab-expand-rootfs[521]: ++ echo /dev/mmcblk0p2
Jun 16 20:52:18 box iiab-expand-rootfs[455]: + ROOT_PART_NUM=2
Jun 16 20:52:18 box iiab-expand-rootfs[535]: ++ parted /dev/mmcblk0 -ms unit s p
Jun 16 20:52:18 box iiab-expand-rootfs[537]: ++ cut -f 1 -d:
Jun 16 20:52:18 box iiab-expand-rootfs[536]: ++ tail -n 1
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + LAST_PART_NUM=2
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + '[' 2 -ne 2 ']'
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + growpart /dev/mmcblk0 2
Jun 16 20:52:19 box iiab-expand-rootfs[631]: NOCHANGE: partition 2 is size 999210975. it cannot be grown
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:19 box iiab-expand-rootfs[669]: resize2fs 1.46.2 (28-Feb-2021)
Jun 16 20:52:20 box iiab-expand-rootfs[455]: /usr/sbin/iiab-expand-rootfs: line 59:   669 Segmentation fault      resize2fs $ROOT_PART
Jun 16 20:52:20 box iiab-expand-rootfs[455]: + rm -f /.expand-rootfs /.resize-rootfs
Jun 16 20:52:20 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jun 16 20:52:20 box systemd[1]: Finished Root Filesystem Auto-Expander.
```

tim-moody, 2022-07-29:
so these are two separate boots. 

 Root Filesystem Auto-Expander didn't finish on the first one (? manual reboot) and so ran again on the second one and succeeded.

tim-moody, 2022-07-29:
OK. Now I see /usr/sbin/iiab-expand-rootfs: line 59:   669 Segmentation fault      resize2fs $ROOT_PART

tim-moody, 2022-07-29:
that could be run manually to test version

tim-moody, 2022-07-29:
but the segfault coudl also be from corrupt file system from previous reboot

tim-moody, 2022-07-29:
? never completed

tim-moody, 2022-07-29:
https://patchwork.ozlabs.org/project/linux-ext4/patch/20160810200113.GA10523@thunk.org/

this is old, but could be relevant

need to try to reproduce the resize2fs segmentation fault

holta, 2022-07-29:
> ? never completed

Possible.

These 3+3 pastes might shed more light on these "simultaneous boots" hopefully:

```
root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 | head -2
-- Journal begins at Mon 2022-04-04 14:52:30 UTC, ends at Fri 2022-07-29 13:35:30 UTC. --
Jun 16 20:52:13 box kernel: Booting Linux on physical CPU 0x0000000000 [0x410fd083]

root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 | grep -i 'ext4\|mmcblk'
Jun 16 20:52:13 box kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:D8:7F:ED vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=f91b8a44-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Jun 16 20:52:13 box kernel: mmcblk0: mmc0:aaaa SN512 477 GiB
Jun 16 20:52:13 box kernel:  mmcblk0: p1 p2
Jun 16 20:52:13 box kernel: mmcblk0: mmc0:aaaa SN512 477 GiB
Jun 16 20:52:13 box kernel: EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
Jun 16 20:52:13 box kernel: VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Jun 16 20:52:13 box kernel: EXT4-fs (mmcblk0p2): re-mounted. Opts: (null). Quota mode: none.
Jun 16 20:52:15 box systemd-fsck[287]: /dev/mmcblk0p1: 293 files, 61656/516190 clusters
Jun 16 20:52:17 box systemd[1]: Started Periodic ext4 Online Metadata Check for All Filesystems.
Jun 16 20:52:17 box systemd[1]: Starting Remove Stale Online ext4 Metadata Check Snapshots...
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[505]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[453]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:52:17 box iiab-expand-rootfs[541]: ++ echo /dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[549]: ++ parted /dev/mmcblk0 -ms unit s p
Jun 16 20:52:17 box systemd[1]: Finished Remove Stale Online ext4 Metadata Check Snapshots.
Jun 16 20:52:18 box iiab-expand-rootfs[453]: + growpart /dev/mmcblk0 2
Jun 16 20:52:20 box iiab-expand-rootfs[453]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:20 box kernel: EXT4-fs (mmcblk0p2): resizing filesystem from 2914696 to 124901371 blocks
Jun 16 20:52:31 box kernel: EXT4-fs (mmcblk0p2): resized to 6291456 blocks
Jun 16 20:52:41 box kernel: EXT4-fs (mmcblk0p2): resized to 11010048 blocks
Jun 16 20:52:53 box kernel: EXT4-fs (mmcblk0p2): resized to 16252928 blocks

root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 | tail -40
Jun 16 20:52:45 box systemd[1]: kalite-serve.service: Can't open PID file /library/ka-lite/kalite.pid (yet?) after start: Operation not permitted
Jun 16 20:52:45 box systemd[1]: Started Provides the kalite Server.
Jun 16 20:52:45 box lightdm[1408]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
Jun 16 20:52:45 box lightdm[1408]: pam_unix(lightdm-autologin:session): session opened for user iiab-admin(uid=1000) by (uid=0)
Jun 16 20:52:45 box systemd-logind[503]: New session 3 of user iiab-admin.
Jun 16 20:52:45 box systemd[1]: Started Session 3 of user iiab-admin.
Jun 16 20:52:45 box systemd[1]: systemd-fsckd.service: Succeeded.
Jun 16 20:52:47 box rtkit-daemon[1363]: Supervising 5 threads of 3 processes of 1 users.
Jun 16 20:52:47 box rtkit-daemon[1363]: Successfully made thread 1450 of process 1356 owned by '1000' RT at priority 5.
Jun 16 20:52:47 box rtkit-daemon[1363]: Supervising 6 threads of 3 processes of 1 users.
Jun 16 20:52:48 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:48 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:48 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:48 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:48 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:48 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:48 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:48 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:48 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:48 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:48 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:48 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:49 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:49 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:49 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:49 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:49 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
Jun 16 20:52:49 box kernel:  MAI: soc_pcm_open() failed (-19)
Jun 16 20:52:49 box kernel: hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[... cut ...]
Jun 16 21:15:13 box systemd[1]: Reached target Reboot.
Jun 16 21:15:13 box systemd[1]: Shutting down.
Jun 16 21:15:13 box systemd[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog
Jun 16 21:15:13 box systemd[1]: Failed to set timeout to 600s: Invalid argument
Jun 16 21:15:13 box systemd-shutdown[1]: Syncing filesystems and block devices.
Jun 16 21:15:14 box systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Jun 16 21:15:14 box systemd-journald[144]: Journal stopped
```

jvonau, 2022-07-29:
to 124901371 blocks
resized to 16252928 blocks

Didn't finish, rebooted/powered off too soon.

jvonau, 2022-07-29:
resizing filesystem from 15728640 to 124901371 blocks

tim-moody, 2022-07-29:
there's still the segfault on the second boot

holta, 2022-07-29:
Potentially more revealing details:

```
root@box:~# grep -in 'ext4\|mmcblk' /var/log/messages.1 | wc -l
92
root@box:~# grep -in 'ext4\|mmcblk' /var/log/messages.1 | pastebinit -b sprunge.us
http://sprunge.us/EEpAjD
```

http://sprunge.us/EEpAjD?en

```
root@box:~# grep -in 'ext4\|mmcblk' /var/log/syslog.1 | wc -l
149
root@box:~# grep -in 'ext4\|mmcblk' /var/log/syslog.1 | pastebinit -b sprunge.us
http://sprunge.us/9Au8Yx
```

http://sprunge.us/9Au8Yx?en

jvonau, 2022-07-29:
How about pastebin the entire journal?

jvonau, 2022-07-29:
Jun 16 19:37:24 raspberrypi kernel: [    1.586697] mmcblk0: mmc0:aaaa SC128 119 GiB
Jun 16 21:15:17 box kernel: [    1.675755] mmcblk0: mmc0:aaaa SN512 477 GiB

Looks like 2 different SDcards were involved.

jvonau, 2022-07-29:
Think there is a race condition between systemd-fsck and iiab-expand-rootfs

```
Jun 16 20:52:13 box kernel: EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
Jun 16 20:52:13 box kernel: VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Jun 16 20:52:13 box kernel: EXT4-fs (mmcblk0p2): re-mounted. Opts: (null). Quota mode: none.
Jun 16 20:52:16 box systemd-fsck[287]: /dev/mmcblk0p1: 293 files, 61656/516190 clusters
Jun 16 20:52:17 box systemd[1]: Started Periodic ext4 Online Metadata Check for All Filesystems.
Jun 16 20:52:17 box systemd[1]: Starting Remove Stale Online ext4 Metadata Check Snapshots...
Jun 16 20:52:17 box iiab-expand-rootfs[455]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:52:17 box iiab-expand-rootfs[496]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:52:18 box iiab-expand-rootfs[455]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:52:18 box iiab-expand-rootfs[521]: ++ echo /dev/mmcblk0p2
Jun 16 20:52:18 box iiab-expand-rootfs[535]: ++ parted /dev/mmcblk0 -ms unit s p
Jun 16 20:52:18 box systemd[1]: Finished Remove Stale Online ext4 Metadata Check Snapshots.
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + growpart /dev/mmcblk0 2
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:19 box kernel: EXT4-fs (mmcblk0p2): resizing filesystem from 15728640 to 124901371 blocks
Jun 16 20:52:19 box kernel: kernel BUG at fs/ext4/resize.c:528!
Jun 16 20:52:20 box kernel: pc : ext4_flex_group_add+0x9a0/0x1828
Jun 16 20:52:20 box kernel: lr : ext4_flex_group_add+0x98/0x1828
Jun 16 20:52:20 box kernel:  ext4_flex_group_add+0x9a0/0x1828
Jun 16 20:52:20 box kernel:  ext4_resize_fs+0xa98/0x1108
Jun 16 20:52:20 box kernel:  __ext4_ioctl+0x11dc/0x17b0
Jun 16 20:52:20 box kernel:  ext4_ioctl+0x40/0x68
Jun 16 21:15:04 box systemd[1]: Stopped Periodic ext4 Online Metadata Check for All Filesystems.
```

jvonau, 2022-07-29:
Might want to look at using 'After=' in the systemd unit file, for better staging of events.

tim-moody, 2022-07-29:
what is 

Jun 16 20:52:19 box kernel: kernel BUG at fs/ext4/resize.c:528! 

reminds of the segfault

tim-moody, 2022-07-29:
what is the fsck target to go after?

holta, 2022-07-29:
> what is
> 
> Jun 16 20:52:19 box kernel: kernel BUG at fs/ext4/resize.c:528!
> 
> reminds of the segfault

Good question.  Possibly it's the subsequent boot trying to recover from a failed rootfs expansion?

Here are the 100 lines surrounding the above line FWIW:

```
root@box:~# journalctl -b -7 | grep -C50 'kernel BUG' | pastebinit -b sprunge.us
http://sprunge.us/llESEp
```

http://sprunge.us/llESEp?en

holta, 2022-07-29:
> Think there is a race condition between systemd-fsck and iiab-expand-rootfs

Do these 2 shed any light?

```
root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 | grep -n fsck
27:Jun 16 20:52:13 box kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:D8:7F:ED vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=f91b8a44-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
292:Jun 16 20:52:13 box systemd[1]: Created slice system-systemd\x2dfsck.slice.
299:Jun 16 20:52:13 box systemd[1]: Listening on fsck to fsckd communication Socket.
362:Jun 16 20:52:13 box systemd-fsck[160]: e2fsck 1.46.2 (28-Feb-2021)
363:Jun 16 20:52:13 box systemd-fsck[160]: rootfs: clean, 319020/720544 files, 2504806/2914696 blocks
457:Jun 16 20:52:15 box systemd-fsck[287]: fsck.fat 4.2 (2021-01-31)
458:Jun 16 20:52:15 box systemd-fsck[287]: /dev/mmcblk0p1: 293 files, 61656/516190 clusters
944:Jun 16 20:52:45 box systemd[1]: systemd-fsckd.service: Succeeded.
```

```
root@box:~# journalctl -b 8ac77d02beb348ebbfb5fe4562f05b65 | grep -n fsck
27:Jun 16 20:52:13 box kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:D8:7F:ED vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=f91b8a44-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
297:Jun 16 20:52:13 box systemd[1]: Created slice system-systemd\x2dfsck.slice.
304:Jun 16 20:52:13 box systemd[1]: Listening on fsck to fsckd communication Socket.
368:Jun 16 20:52:13 box systemd-fsck[159]: e2fsck 1.46.2 (28-Feb-2021)
369:Jun 16 20:52:13 box systemd-fsck[159]: Setting free inodes count to 3567035 (was 401499)
370:Jun 16 20:52:13 box systemd-fsck[159]: Setting free blocks count to 13044906 (was 431960)
371:Jun 16 20:52:13 box systemd-fsck[159]: rootfs: clean, 319045/3886080 files, 2683734/15728640 blocks
472:Jun 16 20:52:16 box systemd-fsck[287]: fsck.fat 4.2 (2021-01-31)
473:Jun 16 20:52:16 box systemd-fsck[287]: There are differences between boot sector and its backup.
474:Jun 16 20:52:16 box systemd-fsck[287]: This is mostly harmless. Differences: (offset:original/backup)
475:Jun 16 20:52:16 box systemd-fsck[287]:   65:01/00
476:Jun 16 20:52:16 box systemd-fsck[287]:   Not automatically fixing this.
477:Jun 16 20:52:16 box systemd-fsck[287]: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
478:Jun 16 20:52:16 box systemd-fsck[287]:  Automatically removing dirty bit.
479:Jun 16 20:52:16 box systemd-fsck[287]: *** Filesystem was changed ***
480:Jun 16 20:52:16 box systemd-fsck[287]: Writing changes.
481:Jun 16 20:52:16 box systemd-fsck[287]: /dev/mmcblk0p1: 293 files, 61656/516190 clusters
1163:Jun 16 20:52:46 box systemd[1]: systemd-fsckd.service: Succeeded.
1711:Jun 16 21:15:13 box systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-f91b8a44\x2d01.service: Succeeded.
1713:Jun 16 21:15:13 box systemd[1]: Removed slice system-systemd\x2dfsck.slice.
1721:Jun 16 21:15:13 box systemd[1]: systemd-fsck-root.service: Succeeded.
```

holta, 2022-07-29:
And for comparison, here's (what appears to be!) the two subsequent boots: (following the above 2)

```
root@box:~# journalctl -b 02c8554a46b84a88938c27716049dcbb | grep -n fsck
27:Jun 16 21:15:13 box kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:D8:7F:ED vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=f91b8a44-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
295:Jun 16 21:15:13 box systemd[1]: Created slice system-systemd\x2dfsck.slice.
302:Jun 16 21:15:13 box systemd[1]: Listening on fsck to fsckd communication Socket.
359:Jun 16 21:15:13 box systemd-fsck[157]: e2fsck 1.46.2 (28-Feb-2021)
360:Jun 16 21:15:13 box systemd-fsck[157]: rootfs: clean, 319160/3886080 files, 2700858/15728640 blocks
472:Jun 16 21:15:16 box systemd-fsck[279]: fsck.fat 4.2 (2021-01-31)
473:Jun 16 21:15:16 box systemd-fsck[279]: /dev/mmcblk0p1: 293 files, 61656/516190 clusters
1050:Jun 16 21:15:46 box systemd[1]: systemd-fsckd.service: Succeeded.
1562:Jun 16 21:44:54 box systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-f91b8a44\x2d01.service: Succeeded.
1564:Jun 16 21:44:54 box systemd[1]: Removed slice system-systemd\x2dfsck.slice.
1572:Jun 16 21:44:54 box systemd[1]: systemd-fsck-root.service: Succeeded.
```

```
root@box:~# journalctl -b 565a5a9c7aae4b11b5f82343d3a62805 | grep -n fsck
27:Jun 16 21:44:54 box kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:D8:7F:ED vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=f91b8a44-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
292:Jun 16 21:44:54 box systemd[1]: Created slice system-systemd\x2dfsck.slice.
299:Jun 16 21:44:54 box systemd[1]: Listening on fsck to fsckd communication Socket.
356:Jun 16 21:44:54 box systemd-fsck[156]: e2fsck 1.46.2 (28-Feb-2021)
357:Jun 16 21:44:54 box systemd-fsck[156]: rootfs: clean, 319157/3886080 files, 2714611/15728640 blocks
464:Jun 16 21:44:57 box systemd-fsck[271]: fsck.fat 4.2 (2021-01-31)
465:Jun 16 21:44:57 box systemd-fsck[271]: /dev/mmcblk0p1: 293 files, 61656/516190 clusters
1049:Jun 16 21:45:27 box systemd[1]: systemd-fsckd.service: Succeeded.
```

holta, 2022-07-29:
These wordier logs might also be helpful??

```
root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 | grep -n 'ext4\|mmcblk\|GiB\|Raspberry Pi 4\|fsck' | wc -l
     29
root@box:~# journalctl -b 84cc35b34186456ba2adf08f7b4d5399 | grep -n 'ext4\|mmcblk\|GiB\|Raspberry Pi 4\|fsck' | pastebinit -b sprunge.us
http://sprunge.us/tOaqT6
```

http://sprunge.us/tOaqT6?en

```
root@box:~# journalctl -b 8ac77d02beb348ebbfb5fe4562f05b65 | grep -n 'ext4\|mmcblk\|GiB\|Raspberry Pi 4\|fsck' | wc -l
     54
root@box:~# journalctl -b 8ac77d02beb348ebbfb5fe4562f05b65 | grep -n 'ext4\|mmcblk\|GiB\|Raspberry Pi 4\|fsck' | pastebinit -b sprunge.us
http://sprunge.us/j8knTi
```

http://sprunge.us/j8knTi?en

holta, 2022-07-29:
> I don't know enough about Systemd to understand why 1 of the 2 "simultaneous boots" (specifically "84cc35b34186456ba2adf08f7b4d5399") does not appear here:
> 
> ```
> root@box:~# journalctl --list-boots
> -12 fd13670672ed4eb1ad9711a25c622cae Mon 2022-04-04 14:52:30 UTC—Mon 2022-04-04 14:52:47 UTC
> -11 051858ef96664f8b8fdfff4325e993af Mon 2022-04-04 14:52:47 UTC—Thu 2022-06-16 19:37:21 UTC
> -10 549ffca864984728b6a2a83b4da065cd Thu 2022-06-16 19:37:21 UTC—Thu 2022-06-16 20:51:00 UTC
>  -9 a06560a4507447f094afd6ff382231be Thu 2022-06-16 20:51:00 UTC—Thu 2022-06-16 20:52:14 UTC
>  -8 8ac77d02beb348ebbfb5fe4562f05b65 Thu 2022-06-16 20:52:14 UTC—Thu 2022-06-16 21:15:14 UTC
>  -7 02c8554a46b84a88938c27716049dcbb Thu 2022-06-16 21:15:14 UTC—Thu 2022-06-16 21:44:55 UTC
>  -6 565a5a9c7aae4b11b5f82343d3a62805 Thu 2022-06-16 21:44:55 UTC—Thu 2022-06-16 22:00:03 UTC
>  -5 ebc3de71ad43488cbea813b5409c4aff Wed 2022-07-27 23:32:46 UTC—Wed 2022-07-27 23:49:57 UTC
>  -4 9bd1b134a67845908742051d69d9474c Wed 2022-07-27 23:49:57 UTC—Thu 2022-07-28 00:19:36 UTC
>  -3 99cdf14cc3da416c8671a878bbf9d129 Thu 2022-07-28 00:19:36 UTC—Thu 2022-07-28 13:48:47 UTC
>  -2 ef13041c8a3c40a89e130149e8fc8fb1 Thu 2022-07-28 13:48:47 UTC—Thu 2022-07-28 21:51:33 UTC
>  -1 bc1ac6387d584b468ded38c9aee772c5 Thu 2022-07-28 21:51:33 UTC—Thu 2022-07-28 22:04:16 UTC
>   0 1e391b0fe8a74fa0bb73ab9be6f051fd Thu 2022-07-28 22:04:16 UTC—Fri 2022-07-29 12:40:02 UTC
> ```

Just FYI here's a bug in journalctl 247.3-7 on RasPiOS 2022-04-04:

- 'journalctl -b -8' and 'journalctl -b 8ac77d02beb348ebbfb5fe4562f05b65' are supposed to be 100% equivalent (that's the whole point of giving each _BOOT_ID a nickname, so you don't have to use the 32-char long form!)

- But they're most definitely NOT equivalent: journalctl (regularly) shows you info from the wrong boot.

SUMMARY: In situations like this when systemd (and journalctl) fail to record a boot (for whatever reason) then its internal BOOT ID numbering system gets screwed up across the board.  _So only use 32-char _BOOT_ID's if you need to avoid this mess !_

holta, 2022-07-29:
> How about pastebin the entire journal?

```
root@box:~# du -hs /var/log/journal
209M    /var/log/journal
```

(Also even if the size can be cut down to size, from the above 209MB, I'd need to speak to the machine owner about privacy issues in his journal.)

tim-moody, 2022-07-29:
After=local-fs.target for example. seems the fsck service can return two error targets, including a reboot, so should stay away from those.

holta, 2022-07-29:
> After=local-fs.target for example. seems the fsck service can return two error targets, including a reboot, so should stay away from those.

Sounds very reasonable.

Any suggestions on how to validate?

jvonau, 2022-08-01:
Related #2522 #3147

holta, 2022-08-01:
> Related #2522 #3147

Both tremendously useful context.

Thank you @jvonau for having linked to these:

- PR #2522
- #3147

holta, 2022-08-01:
I helped the owner of the 512GB microSD manually run `/usr/bin/iiab-expand-rootfs` to expand from 60GB to 470GB and it worked fine:

```
root@box:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       470G   11G  455G   3% /
devtmpfs        1.7G     0  1.7G   0% /dev
tmpfs           1.9G   11M  1.9G   1% /dev/shm
tmpfs           759M  1.4M  758M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   31M  222M  12% /boot
tmpfs           380M   20K  380M   1% /run/user/1000
tmpfs           380M   12K  380M   1% /run/user/0
```
 
Interestingly it took well over 3 min to complete despite no other activity on the RPi 400 machine &mdash; as journalctl confirmed:

```
Aug 01 14:47:02 box kernel: EXT4-fs (mmcblk0p2): resizing filesystem from 15728640 to 124901371 blocks
Aug 01 14:47:12 box kernel: EXT4-fs (mmcblk0p2): resized to 21495808 blocks
Aug 01 14:47:23 box kernel: EXT4-fs (mmcblk0p2): resized to 27262976 blocks
Aug 01 14:47:33 box kernel: EXT4-fs (mmcblk0p2): resized to 33030144 blocks
Aug 01 14:47:43 box kernel: EXT4-fs (mmcblk0p2): resized to 38797312 blocks
Aug 01 14:47:54 box kernel: EXT4-fs (mmcblk0p2): resized to 44564480 blocks
Aug 01 14:48:04 box kernel: EXT4-fs (mmcblk0p2): resized to 50331648 blocks
Aug 01 14:48:14 box kernel: EXT4-fs (mmcblk0p2): resized to 56098816 blocks
Aug 01 14:48:25 box kernel: EXT4-fs (mmcblk0p2): resized to 61865984 blocks
Aug 01 14:48:35 box kernel: EXT4-fs (mmcblk0p2): resized to 67633152 blocks
Aug 01 14:48:46 box kernel: EXT4-fs (mmcblk0p2): resized to 73400320 blocks
Aug 01 14:48:56 box kernel: EXT4-fs (mmcblk0p2): resized to 79167488 blocks
Aug 01 14:49:06 box kernel: EXT4-fs (mmcblk0p2): resized to 84934656 blocks
Aug 01 14:49:17 box kernel: EXT4-fs (mmcblk0p2): resized to 90701824 blocks
Aug 01 14:49:27 box kernel: EXT4-fs (mmcblk0p2): resized to 96468992 blocks
Aug 01 14:49:38 box kernel: EXT4-fs (mmcblk0p2): resized to 102236160 blocks
Aug 01 14:49:48 box kernel: EXT4-fs (mmcblk0p2): resized to 108003328 blocks
Aug 01 14:49:58 box kernel: EXT4-fs (mmcblk0p2): resized to 113770496 blocks
Aug 01 14:50:01 box CRON[41216]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Aug 01 14:50:01 box CRON[41217]: (www-data) CMD ([ -x /usr/share/awstats/tools/update.sh ] && /usr/share/awstats/tools/update.sh)
Aug 01 14:50:02 box CRON[41216]: pam_unix(cron:session): session closed for user www-data
Aug 01 14:50:09 box kernel: EXT4-fs (mmcblk0p2): resized to 119537664 blocks
Aug 01 14:50:19 box kernel: EXT4-fs (mmcblk0p2): resized filesystem to 124901371
```

PS Now that several people are in fact installing IIAB on 1TB microSD cards, those kinds of people need to be aware that `/usr/bin/iiab-expand-rootfs` can take 8 minutes (or more?) to do its job, on its boot.

_(Non-intuitive to people who don't realize what's happening, and could easily get frustrated by a very tedious 10+ minute boot sequence!)_

jvonau, 2022-08-02:
>PS Now that several people are in fact installing IIAB on 1TB microSD cards, these kinds of people need to be aware that /usr/bin/iiab-expand-rootfs can take 8 minutes (or more?) to do its job, on 1st boot.

>(Non-intuitive to people who don't realize what's happening, and could easily get frustrated by a very tedious 10+ minute boot sequence!)

How about a pop up a warning that the filesystem is being altered and not to reboot until complete? Tie the resize warning into the netwarn routine staged to display before the iiab-network question. Kind of makes the ssh version of netwarn a priority now. Still need to fix the staging to prevent the running of the resize routine before the fsck service is completed.

I've given more time to this project than I wished to in the past with the warnings unheeded, now the chickens are hatched and come home to roost. Good luck with whatever solution you choose for the home-brew'd images that are produced with the presently employed build routine.

holta, 2022-08-02:
> need to fix the staging to prevent the running of the resize routine before the fsck service is completed

1) Agreed.

2) Today I got permission from the owner of the machine to keep it online for a few more days, just in case there's further need to probe the logs of this live test case machine (512GB microSD card in a Raspberry Pi 400) &mdash; hoping we can learn from that.

3) A UX to educate operators of every different kind (in every different country) will never be easy, naturally, pop-ups and/or command-line notification reminders!  Information Overload (TMI, [banner blindness](https://en.wikipedia.org/wiki/Banner_blindness)) are very real issues, which is what makes UX design so hard, during boot-time especially.  Still, educating implementers/operators is necessary: so if there are very clean, low-maintenance and non-complex ways of doing this (for headless IIAB's as well, to keep Linux know-it-alls from shooting themselves in the foot too often!) then yes these should be definitely be investigated. :heavy_check_mark:

jvonau, 2022-08-08:
The opening [comment](https://github.com/iiab/iiab/pull/2522#issuecomment-692610563) clearly demonstrates that the resize operation can be launched before systemd-fsck-root.service, that in itself was the red flag warning that was glossed over becoming the primary cause of this issue. Bit of fiddling around, discovery pointed to a better approach back then, but in the face of resistance put forth why bother life is too short for the adversarial environment present. I figured it was just a matter of time before somebody ran across this known weakness to support my point of view, kind of sorry others have had their data put at risk using the service. Just remember I can't be the one that the finger points at, no commit access here.

jvonau, 2022-08-11:
> Jun 16 20:52:19 box iiab-expand-rootfs[631]: NOCHANGE: partition 2 is size 999210975. it cannot be grown
Jun 16 20:52:19 box iiab-expand-rootfs[455]: + resize2fs /dev/mmcblk0p2
Jun 16 20:52:19 box iiab-expand-rootfs[669]: resize2fs 1.46.2 (28-Feb-2021)
Jun 16 20:52:20 box iiab-expand-rootfs[455]: /usr/sbin/iiab-expand-rootfs: line 59:   669 Segmentation fault      resize2fs $ROOT_PART
Jun 16 20:52:20 box iiab-expand-rootfs[455]: + rm -f /.expand-rootfs /.resize-rootfs

To prevent the erroneous removal of /.expand-rootfs when there was a problem running the script perhaps bash should be called with -e to exit on first error for an "anti-snowballing" effect.

jvonau, 2022-09-13:
Used http://download.iiab.io/8.0/Old/iiab-8.0preview2-220616-MEDICAL-raspios32-lite-g2e2824c3.img.zip dated 2022-Jun-16 21:52
written to SDCard booted directly on a RPi0-W 
AP came up out of the box, ssh'd in over WiFi, ran some diagnostics, `sudo journalctl -b 0 -u iiab-expand-rootfs > ~/W-boot0.txt`
retrieved the file from an internet connected machine and to the WiFi LAN of the RPi0-W interesting results returned:

```
cat W-boot0.txt 
-- Journal begins at Mon 2022-04-04 12:05:58 UTC, ends at Tue 2022-09-13 12:47:16 UTC. --
Jun 16 20:54:07 box systemd[1]: Starting Root Filesystem Auto-Expander...
Jun 16 20:54:08 box iiab-expand-rootfs[286]: + '[' -f /.expand-rootfs ']'
Jun 16 20:54:08 box iiab-expand-rootfs[286]: + echo '/usr/sbin/iiab-expand-rootfs: Expanding rootfs partition'
Jun 16 20:54:08 box iiab-expand-rootfs[286]: /usr/sbin/iiab-expand-rootfs: Expanding rootfs partition
Jun 16 20:54:08 box iiab-expand-rootfs[305]: ++ findmnt / -o SOURCE -n
Jun 16 20:54:08 box iiab-expand-rootfs[286]: + ROOT_PART=/dev/mmcblk0p2
Jun 16 20:54:08 box iiab-expand-rootfs[331]: ++ lsblk -no pkname /dev/mmcblk0p2
Jun 16 20:54:09 box iiab-expand-rootfs[286]: + ROOT_DEV=/dev/mmcblk0
Jun 16 20:54:09 box iiab-expand-rootfs[348]: ++ echo /dev/mmcblk0p2
Jun 16 20:54:09 box iiab-expand-rootfs[349]: ++ grep -o '[[:digit:]]*$'
Jun 16 20:54:09 box iiab-expand-rootfs[286]: + ROOT_PART_NUM=2
Jun 16 20:54:09 box iiab-expand-rootfs[362]: ++ parted /dev/mmcblk0 -ms unit s p
Jun 16 20:54:09 box iiab-expand-rootfs[365]: ++ tail -n 1
Jun 16 20:54:09 box iiab-expand-rootfs[366]: ++ cut -f 1 -d:
Jun 16 20:54:23 box iiab-expand-rootfs[286]: + LAST_PART_NUM=2
Jun 16 20:54:23 box iiab-expand-rootfs[286]: + '[' 2 -ne 2 ']'
Jun 16 20:54:23 box iiab-expand-rootfs[286]: + growpart /dev/mmcblk0 2
Jun 16 20:54:24 box iiab-expand-rootfs[472]: /usr/bin/growpart: 522: arithmetic expression: expecting primary: "5*1000*1000+15*1000+32+"
Jun 16 20:54:24 box iiab-expand-rootfs[286]: + resize2fs /dev/mmcblk0p2
Jun 16 20:54:24 box iiab-expand-rootfs[484]: resize2fs 1.46.2 (28-Feb-2021)
Jun 16 20:54:25 box iiab-expand-rootfs[484]: The filesystem is already 1275303 (4k) blocks long.  Nothing to do!
Jun 16 20:54:25 box iiab-expand-rootfs[286]: + rm -f /.expand-rootfs /.resize-rootfs
Jun 16 20:54:25 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Jun 16 20:54:25 box systemd[1]: Finished Root Filesystem Auto-Expander.
```

The growpart line looks a bit strange, should just be the partition number and was not resized.

jvonau, 2022-09-13:
https://bugs.launchpad.net/cloud-utils/+bug/1881014 my guess is the patch mentioned has not made its way upstream to Debian yet.

holta, 2022-09-13:
> https://bugs.launchpad.net/cloud-utils/+bug/1881014 my guess is the patch mentioned has not made its way upstream to Debian yet.

That's really great progress in helping everyone understand these frustrating failures!

Can a more recent OS (possibly Ubuntu 22.10 daily build) be used to help confirm?

Or [22.04.1 on RPi](https://ubuntu.com/download/raspberry-pi) if that's faster to test?

jvonau, 2022-09-13:
Ubuntu is distributing the patched version, it's fixed over there. The issue is what `uname -r` returns for the RasPiOS kernels, some kernels the routine works because there are no non-arithmetic trailing characters present within the output. Would be better to know what is returned for the various kernels once booted.

holta, 2022-09-13:
> Ubuntu is distributing the patched version, it's fixed over there.

Indeed :+1:

Testing a recent Ubuntu-on-RPi could be extremely useful confirmation that both (1) intermittent failures and (2) deterministic failures are...both...increasingly tamed (with a few prayers included!)

- PR #3337

jvonau, 2022-09-13:
[Known slowness](https://github.com/iiab/iiab/issues/723#issuecomment-377297472) on W. Pic of [boot failure](https://github.com/iiab/iiab/issues/723#issuecomment-397437294) due to /boot not being clean, with that partition being a fat filesystem this could be caused by not 'ejecting' the media before removal on any given OS. TODO - check the version of cloud-guest-utils being used on RasPiOS, the patch mentioned above might be already included but is not enough to cover the /boot/kernel.img display of uname but otherwise works with the rest of the RasPiOS's kernels (kernel7.img kernel7l.img kernel8.img)

jvonau, 2022-09-14:
Info from booted W
```
iiab-admin@box:~ $ apt list cloud-guest-utils
Listing... Done
cloud-guest-utils/stable,now 0.31-2 all [installed]

iiab-admin@box:~ $ uname -r
5.15.32+

iiab-admin@box:~ $ ls /lib/modules
5.15.32+  5.15.32-v7+  5.15.32-v7l+  5.15.32-v8+
```
Looks like that patch went upstream as https://github.com/canonical/cloud-utils/pull/2 but doesn't really cover X.X.X+ as the error above notes 
> "5*1000*1000+15*1000+32+"

What is needed is a tweak to remove the trailing '+' from the 'mic' value [generated](https://github.com/canonical/cloud-utils/blame/main/bin/growpart#L671-L682) from uname -r running under sh
```
kver_to_num() {
	local kver="$1" maj min mic

	# Canonicalize the kernel version
	kver=${kver%%[!0-9.]*}.0.0

	maj=${kver%%[!0-9]*}
	kver=${kver#*.}
	min=${kver%%[!0-9]*}
	kver=${kver#*.}
	mic=${kver%%[!0-9]*}
	_RET=$((maj*1000*1000+min*1000+mic))
}
```
Any suggestions I can live test for a proposed patch?

holta, 2022-09-14:
> Debian is lagging in updating and is behind upstream's version

I'll assume this same lag also applies to the very latest [64-bit Raspberry Pi OS from 2022-09-06](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit).

i.e. it too has not yet included the https://bugs.launchpad.net/cloud-utils/+bug/1881014 patch.

(But if for some obscure this is wrong, please someone clarify if poss!)

jvonau, 2022-09-14:
> > Debian is lagging in updating and is behind upstream's version
> 
> I'll assume this same lag also applies to the very latest [64-bit Raspberry Pi OS from 2022-09-06](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit).
> 

`sudo apt list cloud-guest-utils` is your friend and can answer that question for you but from the looks of http://raspbian.raspberrypi.org/raspbian/pool/main/c/cloud-utils/ I'd say the newer version needed should be present now.

holta, 2022-09-15:
Even the very latest (from 2022-09-06) 64-bit "Raspberry Pi OS with desktop" unfortunately shows:

```
root@box:~# apt list cloud-guest-utils
Listing... Done
cloud-guest-utils/stable,stable,now 0.31-2 all [installed]
```

holta, 2022-09-15:
FWIW (also unfortunately) has the older version too:

```
root@deb11-srv:~# apt list cloud-guest-utils
Listing... Done
cloud-guest-utils/stable 0.31-2 all
```

More positively Debian 12 (Bookworm) that many Debian community people are already using (prior to its release around mid-2023) is all set &mdash; as confirmed by:

https://packages.debian.org/bookworm/cloud-guest-utils

holta, 2022-09-26:
Now that PR #3337 is merged (thanks to testing work by @jvonau and @tim-moody) this ticket should be closed in coming weeks &mdash; once reliable rootfs expansion is validated by others too.

(Preferably on both RasPiOS and Ubuntu!)

holta, 2022-12-22:
Appears solved:

Am moving this ticket to the IIAB 8.1 Milestone so these issues continue to be monitored / confirmed.