iiab/iiab #3330: ACCESS THROUGH LAN ISSUE

vinaybissa, 2022-08-01:
### Expected behavior and actual behavior:

### Steps to reproduce the problem:

### IIAB version, OS version, hardware details:

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

holta, 2022-08-01:
Hi @vinaybissa

Please run: **sudo iiab-diagnostics**

Then please reply with the **sprunge** URL that is generated above.

Thanks!

jvonau, 2022-08-03:
https://www.omserver.org/

vinaybissa, 2022-08-05:
Here is the output of iiab-diagnostics
http://sprunge.us/y62554?bash

holta, 2022-08-05:
> I was trying to access server through LAN in laptop from both of devices NUC & RPI 3 but not able to access. But i am able to access same through WiFi mode.

Can you explain specifically what you tried in detail?

With actual IP addresses and/or screenshots?

Thanks @vinaybissa !

vinaybissa, 2022-08-05:
@holta 
I HAVE A COPY OF IIAB INSTALLED IN RPI3 AND ALSO IN INTEL NUC, NOW ISSUE IS I WANT TO ACCESS THE SERVER ON MY LAPTOP  THROUGH LAN CABLE ONLY, SO I CONNECTED THE LAN CABLE BETWEEN (RPI3/NUC) AND LAPTOP, WHEN I TRY TO ACCESS (10.10.10.10 OR BOX.LAN) ON MY LAPTOP ITS NOT ACCESSIBLE SEEMS LAN GATEWAY IS NOT WORKING PROPERLY OR I MISSED ANY NETWORK PROTOCOL TO ESTABLISH CONNECTION BETWEEN MY LAPTOP AND PI

holta, 2022-08-05:
@vinaybissa

When using Ethernet from any client machine (e.g. your laptop) to the NUC, please try these URL's

- http://192.168.0.154
- *192.168.0.154/admin* (for Admin Console)

That is **if** the IP address shown in your http://sprunge.us/y62554?bash is still valid (it will change somewhat regularly, as this is likely a DHCP lease from your home router!)

Hopefully the above makes it clear, that IIAB is designed to be accessed over its WiFi hotspot primarily, e.g. for students nearby with smartphones and tablets (who all depend on WiFi).

PS If you really want IIAB to have a fixed IP address and fixed name on its Ethernet port, that is much more advanced, and you would need to configure that on your own.  These more advanced networking topics are mentioned here, if you have networking skills, and time to read these IIAB Docs carefully:

- https://github.com/iiab/iiab/tree/master/roles/network#readme
- https://github.com/iiab/iiab/wiki/IIAB-Networking

jvonau, 2022-08-05:
> I HAVE A COPY OF IIAB INSTALLED IN RPI3 AND ALSO IN INTEL NUC, NOW ISSUE IS I WANT TO ACCESS THE SERVER ON MY LAPTOP THROUGH LAN CABLE ONLY, SO I CONNECTED THE LAN CABLE BETWEEN (RPI3/NUC) AND LAPTOP, WHEN I TRY TO ACCESS (10.10.10.10 OR BOX.LAN) 

https://github.com/iiab/iiab/wiki/IIAB-Networking
@vinaybissa What you want is a 'LAN-Controller' but with Ubuntu and a desktop present using NetworkManager the auto creation of the wired slave under br0 is disabled to match the behavior of RasPiOS. You could use https://github.com/iiab/iiab/blob/master/roles/network/templates/network/systemd-br0-slave.j2 as a template saving the file as /etc/systemd/network/IIAB-Slave-enp0s25.network with the contents being:
```
# /etc/systemd/network/IIAB-Slave-enp0s25.network
[Match]
Name=enp0s25

[Link]
RequiredForOnline=no

[Network]
Bridge=br0
```
Next reboot. As long as /etc/systemd/network/IIAB-Slave-enp0s25.network is present the ethernet port will become part of br0 and 10.10.10.10 should be accessible across the wire, this will render the port unusable as a client for internet access. **_Under no circumstances plug the ethernet port into another network_** as this port will now have a dhcp server functional and will mess with the network. Don't do that, good way to anger the sysadmin of that network. Need to use the port for internet access, remove the file and reboot before connecting to another network.

holta, 2022-08-05:
Thanks @jvonau for the detailed response.

(That might help others as well, who Google/search for this "old school Ethernet" approach, and end up reading this ticket!?)

jvonau, 2022-08-05:
The above instructions do not work with RasPiOS as 'dhcpcd' is used in place of 'systemd-networkd' for networking chores.

holta, 2022-08-06:
@vinaybissa have you made progress, using yesterday's suggestions?

(And/or let us know if this ticket should can be closed in coming days!)

vinaybissa, 2022-08-08:
@jvonau - I have tried this approach, created a file  with these content in /etc/systemd/network/IIAB-Slave-enp0s25.network, and then i reboot the system.
But still the problem is same with the laptop its not accessible as earlier.

@holta - I tried your approach also with the router, router gives a dynamic ip to this server on that ip server is accessible. So somehow i was able to access same on LAN but for using this i will be needing that type of router which have capability to change the hostname so that instead of using Ip i can access server through the hostname

So the problem. statement is we are setting up a public community library in India where there is need of powerful machine for atleast 30 concurrent users, Raspberry Pi have lot of issues related to RTC and transmitting power so we want to switch to NUC, but max 11 user is able to connect through NUC for 12 the its showing Capacity Full, so we want to create a mesh which can support multiple user thats why LAN plays a important role in this.

jvonau, 2022-08-08:
> 19 display-manager? inactive

@vinaybissa From the url above suggests that you installed server version, I think the contents of /etc/netplan/00-installer-config.yaml might be overriding the IIAB-Slave-enp0s25.network file could you move 00-installer-config.yaml out of /etc/netplan as a test and reboot please.

holta, 2022-08-10:
How's it looking @vinaybissa ?

(@jvonau has provided you some quite valuable tips over the past week, so please let us know where things stand!)

vinaybissa, 2022-08-13:
@jvonau  Sorry i was not available since last days, so i have tested as per @jvonau  suggestion, now its accessible over lan on laptop on this IP address (10.10.10.10), assuming it will not impact the other things

holta, 2022-08-16:
@vinaybissa can this be closed now?

(Or do you have a related / follow-up question?)

vinaybissa, 2022-08-22:
@holta you can close this

closed 2022-08-22 by holta