iiab/iiab #3335: iiab.ini fails to parse

tim-moody, 2022-08-04:
iiab.ini can contain an illegal line that causes parsing to fail in the standard python parsing library

in the following fragment seen on a user machibne the second line is illegal

```
exclude_devices = none
wlp12s0
num_lan_interfaces = 1

```
probably a faulty script wrote this line

holta, 2022-08-04:
The root cause of yesterday's corrupted iiab.ini was resolved here:

- PR #3334 

But it's a good point that this will likely happen again &mdash; code reviews (of any code that is writing to iiab.ini) being essential to keep an eye on this :+1:

jvonau, 2022-08-04:
It's a user error, 3 default gateways would leave no devices to use with hostapd for the hot-spot support if none the devices don't support being used with ap0.

A better course of action might be to extend second_gateway_found to use 'stdout_lines' in place of second_gateway_found.'stdout' when putting together the 'exclude_devices' string. However, 3 gateway with no wifi device leftover to assign to hostapd could result in an Appliance being detected as there would be no interfaces to use for the iiab LAN's br0.

jvonau, 2022-08-04:
2022-08-01 07:22:01,742 p=3352 u=root n=ansible | TASK [network : Set the discovered wireless, if found] ************************************************
2022-08-01 07:22:01,823 p=3352 u=root n=ansible | ok: [127.0.0.1] => (item=wlx1ca7705e063c)
2022-08-01 07:22:01,843 p=3352 u=root n=ansible | ok: [127.0.0.1] => (item=wlp12s0)

2022-08-01 07:22:12,269 p=3352 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'wireless_list_1(wifi1)', 'value': 'wlp12s0'}

2022-08-01 07:22:14,809 p=3352 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'exclude_devices', 'value': 'wlx1ca7705e063c\nwlp12s0'})
2022-08-01 07:22:15,424 p=3352 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'num_lan_interfaces', 'value': '0'})

holta, 2022-09-04:
Thanks everyone who helped solve this (PR #3334) a bit more than a month ago.  

Please re-open if similar problems recur.

closed 2022-09-04 by holta