iiab/iiab #3343: Test IIAB on Ubuntu 22.10 "Kinetic Kudu"

holta, 2022-08-17:
The OS should be released 2022-10-20.

Thanks @pickypet if you & others can help here in coming weeks!

Somewhat related:

- #3182
- #3190
- #3289
- #3307

FYI Ubuntu 22.10 should be 2022-10-03 and contain PHP 8.1, PostgreSQL 14, and Python 3.10:

https://github.com/iiab/iiab/blob/master/vars/ubuntu-2210.yml

Coincidentally this is identical to [Ubuntu 22.04](https://github.com/iiab/iiab/blob/master/vars/ubuntu-2204.yml) so very few problems are expected!

holta, 2022-08-18:
To help @pickypete & others who want to help with **IIAB quality**, here's my recommendation for rapid testing on any Ubuntu machine:

```
multipass launch devel --bridged -n MY_VM -c 2 -m 2G -d 25G
```

[*] instead of `devel` you can equivalently write `kinetic` or `daily:22.10`

Then (optionally) if you want details of your new VM, run `multipass info MY_VM` to see:

```
Name:           MY_VM
State:          Running
IPv4:           10.88.67.47
Release:        Ubuntu Kinetic Kudu (development branch)
Image hash:     29bdfe284056 (Ubuntu 22.10)
CPU(s):         2
Load:           0.07 0.02 0.00
Disk usage:     2.3GiB out of 24.8GiB
Memory usage:   187.1MiB out of 1.9GiB
Mounts:         --
```

Then open a shell (command-line) on the VM:

```
multipass shell MY_VM
```

Finally, install IIAB within the Ubuntu 22.10 (pre-release) VM:

```
curl iiab.io/fast.txt | sudo bash
```

holta, 2022-08-18:
WARNING!  @pickypet (& all) please remember that pre-release OS's (like Ubuntu 22.10) almost always need you to first run the following prior (in the VM of course!) prior to starting IIAB's install:

```
apt install nodejs npm
echo 'nodejs_installed: True' >> /etc/iiab/iiab_state.yml
```

As a result of the issue we discussed yesterday and overnight:

- #3345

holta, 2022-08-18:
I went ahead and kicked off a LARGE-sized IIAB install test.  Here's the initial result:

```
TASK [nodered : Copy /usr/lib/node_modules/node-red/settings.js to /home/nodered/.node-red/settings.js] ****************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Source /usr/lib/node_modules/node-red/settings.js not found"}
```

So I'll set the following in [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?) (and re-run `sudo iiab` to continue installing IIAB...)

```
nodered_install: False
nodered_enabled: False
```

holta, 2022-08-18:
FYI Kolibri does not yet install on Ubuntu 22.10 &mdash; here are instructions to recover if you attempted that &mdash; so as to remove the `apt update` warning that will then block `sudo iiab` from proceeding:

- https://github.com/learningequality/kolibri/issues/9634

holta, 2022-08-18:
GOOD NEWS everything else worked (except for Node-RED and Kolibri mentioned above) and the LARGE-sized IIAB install with `INTERNET-IN-A-BOX (IIAB) SOFTWARE INSTALL IS COMPLETE` (on Ubuntu Server 22.10).

Of course Moodle and Sugarizer cannot run, just like with Ubuntu 22.04 &mdash; but hopefully upstream maintainers will fix that well before end-of-year:

- #3182
- #3190

holta, 2022-08-18:
> FYI Kolibri does not yet install on Ubuntu 22.10
> ...
> - learningequality/kolibri#9634

Great news &mdash; the Kolibri team is working on one-or-both of these issues:

- learningequality/kolibri-installer-debian#112
- PR learningequality/kolibri-installer-debian#113

jredrejo, 2022-08-30:
hello @holta , I've been testing kolibri in a Ubuntu 22.10 installation without any of the issues you found. I've done with a beta of the new package, where we added a conflict with python3-markdown, however that was not needed. In a fresh installation of Ubuntu there is no markdown package installed, and samba is running.
Also, installing the package as it's recommended in https://kolibri.readthedocs.io/en/latest/install/ubuntu-debian.html#install-from-ppa-repository there was none of the glitches you had with the repo keys.
In case you want to test it again you just need to ensure you're using the kolibri-proposed repo (it's a beta) instead of the normal kolibri repo. I.e.:
```
sudo add-apt-repository ppa:learningequality/kolibri-proposed
sudo apt-get update
sudo apt-get install kolibri
```

holta, 2022-08-30:
Thanks @jredrejo!

1) I confirm that `ppa:learningequality/kolibri-proposed` now works with Ansible[*] when installing on a fresh Ubuntu Server 22.10 daily build VM with a bare bones IIAB install _(no Samba yet for now!)_

   [*] IIAB uses [Ansible](https://github.com/iiab/iiab/wiki/Technical-Contributors-Guide#understanding-ansible) to automate the install of everything, i.e. using Ansible's [apt_repository](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/apt_repository_module.html) command in this case, resulting in:

   ```
   root@vm23:/etc/apt/sources.list.d# cat ppa_learningequality_kolibri_proposed_kinetic.list
   deb http://ppa.launchpad.net/learningequality/kolibri-proposed/ubuntu kinetic main
   ```

   Question: can `kolibri-proposed` be relied upon in general, as being the "latest pre-release" (or almost) during testting, with a good chance of working?  (Or is that a bad idea!)

2) Finally, do you know if the following message can be ignored?

   ```
   root@box:~# kolibri --version
   INFO: No C extensions are available for this platform
   /usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
     warnings.warn(
   kolibri, version 0.15.7b1
   ```

   As also mentioned here:

   - learningequality/kolibri-installer-debian#114

holta, 2022-08-30:
3) @jredrejo you're probably on top of this &mdash; but just FYI this DEPRECATION warning persists with Kolibri 0.15.7b1:

   > root@vm23:~# apt update
   > [...]
   > W: http://ppa.launchpad.net/learningequality/kolibri-proposed/ubuntu/dists/kinetic/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.

   As also mentioned here:

   - learningequality/kolibri-installer-debian#114

jredrejo, 2022-08-31:
Hello @holta , I'll try to answer all your questions here:

> * Question: can `kolibri-proposed` be relied upon in general, as being the "latest pre-release" (or almost) during testting, with a good chance of working?  (Or is that a bad idea!)

In general it's a bad idea because that's a repo used for internal tests, you can find there that the last version of kolibri is a beta or an alpha version that does not work. If you don't mind it because you're not going to pass it to the final users, you can use it, but beware of its unstable nature. Also, in case you need to sync some tests of the repos with LE, don't hesitate in contacting so we can upload to this unstable repo what you might need.

Btw, the latest stable version 0.15.7 is already in the official repository and it includes the conflict with python3-markdown, so you can use it without any problem.

> * Finally, do you know if the following message can be ignored?

Yes, I have investigated it and it's not caused by kolibri itself but by a bad version string formatting i ncluded in the file 
/usr/lib/python3/dist-packages/debian/_version.py
that belongs to the python3-debian package. I am sure they'll fix it in the future if this important package fails.

>you're probably on top of this — but just FYI this DEPRECATION warning persists with Kolibri 0.15.7b1:

Yes, this is not a problem with the package not even with our repository but with the way a signing key is added to your systems. Due to possible attacks to the repositories, apt-key is going to be deprecated and it won't be used in the future. So, for Debian > 12 or for Ubuntu > 22.04 the recommended way to add our key and install kolibri is:
```
sudo gpg --output /usr/share/keyrings/learningequality-kolibri.gpg --export  --keyserver hkp://keyserver.ubuntu.com:80 DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81

sudo su -c 'echo "deb [signed-by=/usr/share/keyrings/learningequality-kolibri.gpg] http://ppa.launchpad.net/learningequality/kolibri/ubuntu bionic main" > /etc/apt/sources.list.d/learningequality-ubuntu-kolibri.list'

sudo apt-get update
sudo apt install kolibri
```
instead of the methods explained at https://kolibri.readthedocs.io/en/latest/install/ubuntu-debian.html We will update the documentation to reflect it.
Thanks for all your feedback

jredrejo, 2022-08-31:
@holta just in case you're interested in tracking the docs changes, this is the issue to do it https://github.com/learningequality/kolibri-docs/issues/241

holta, 2022-08-31:
> @holta just in case you're interested in tracking the docs changes, this is the issue to do it [learningequality/kolibri-docs#241](https://github.com/learningequality/kolibri-docs/issues/241)

Thank you.  And let's hope this is simplified in future years :smile:

e.g. so hopefully Ansible (and any other higher-level approach) encapsulates these 3 lines clearly + succinctly:

```
sudo gpg --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81
sudo gpg --output /usr/share/keyrings/learningequality-kolibri.gpg --export  DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81

sudo su -c 'echo "deb [signed-by=/usr/share/keyrings/learningequality-kolibri.gpg] http://ppa.launchpad.net/learningequality/kolibri/ubuntu jammy main" > /etc/apt/sources.list.d/learningequality-ubuntu-kolibri.list'
```

jredrejo, 2022-08-31:
> e.g. so hopefully Ansible (and any other higher-level approach) encapsulates these 3 lines clearly + succinctly:

depending on how you create the ansible scripts, it could be easier if you previously get the `learningequality-kolibri.gpg` and `learningequality-ubuntu-kolibri.list` files.  Your ansible script would just copy these two files to the `/usr/share/keyrings/` and  `/etc/apt/sources.list.d/`

holta, 2022-08-31:
> it could be easier if you previously get the `learningequality-kolibri.gpg` and `learningequality-ubuntu-kolibri.list` files

Worth looking into, Thanks!

> Your ansible script would just copy these two files to the `/usr/share/keyrings/` and `/etc/apt/sources.list.d/`

1) First we'll try to use Ansible's [apt_key](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/apt_key_module.html) command to set up the keyring, as it appears designed for exactly this.

   ```
   apt_key:
     id: DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81
     file: /usr/share/keyrings/learningequality-kolibri.gpg
   ```

2) Then we'll try to keep using Ansible's [apt_repository](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/apt_repository_module.html) command to encode that into `/etc/apt/sources.list.d/learningequality-ubuntu-kolibri-CODENAME.list` where CODENAME is {jammy, kinetic, etc} as Ansible does automatically (really just 3 lines, works really well!) in our live code here: https://github.com/iiab/iiab/blob/624c4d8aaa0eb0b7725bd38cc401a792b7dbf7b0/roles/kolibri/tasks/install.yml#L48-L79

holta, 2022-09-01:
> 1. First we'll try to use Ansible's [apt_key](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/apt_key_module.html) command to set up the keyring, as it appears designed for exactly this:

Oops, while this revised Ansible code _might_ work...

```
apt_key:
  id: DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81 
  url: hkp://keyserver.ubuntu.com:80
  keyring: /usr/share/keyrings/learningequality-kolibri.gpg
```

...on the other hand, the `hkp` (OpenPGP HTTP Keyserver Protocol) might not be supported by Ansible?

_Either way, this Ansible warning is concerning:_

> The apt-key command has been deprecated and suggests to ‘manage keyring files in trusted.gpg.d instead’. See the Debian wiki for details. This [Ansible command] module is kept for backwards compatiblity for systems that still use apt-key as the main way to manage apt repository keys.

https://docs.ansible.com/ansible/latest/collections/ansible/builtin/apt_key_module.html#notes

holta, 2022-09-01:
> ...on the other hand, the `hkp` (OpenPGP HTTP Keyserver Protocol) might not be supported by Ansible?

Confirmed, the above Ansible snippet fails as follows.

> TASK [apt_key] *****************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failed to download key                                                                                         at hkp://keyserver.ubuntu.com:80: Request failed: <urlopen error unknown url ty                                                                                        pe: hkp>"}

CONCLUSION:   For better or for worse, Ansible has decided not to support `hkp` (OpenPGP HTTP Keyserver Protocol) !

holta, 2022-09-01:
```
sudo gpg --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81
```

The above line can be run repeatedly (no problem!)

```
sudo gpg --output /usr/share/keyrings/learningequality-kolibri.gpg --export DC5BAA93F9E4AE4F0411F97C74F88ADB3194DD81
```

Whereas if this 2nd line is run repeatedly, an interactive `Overwrite? (y/N)` prompt would block IIAB installation.

FYI adding gpg's `--yes` flag solves this, so long as clobbering the existing `learningequality-kolibri.gpg` is ok (should be fine!)

holta, 2022-10-23:
For Raspberry Pi-specific testing of Ubuntu 22.10, see:

- #3408

holta, 2022-12-22:
Good enough for Ubuntu 21.10 !

Let's follow up here towards fixing apt upgrades of Kolibri if possible:

- learningequality/kolibri-installer-debian#115 (apt upgrading Kolibri tricks operator into changing KOLIBRI_USER (accidentally) damaging Kolibri installations)

closed 2022-12-22 by holta