iiab/iiab #3348: iiab image contains wrong link to maps assets

tim-moody, 2022-08-20:
a user reported the following when installing maps onto an existing image (probably the medical one)

```
ok: [127.0.0.1] => (item=/library/www/osm-vector-maps/viewer/assets)
ok: [127.0.0.1] => (item=/library/www/osm-vector-maps/viewer/tiles)
ok: [127.0.0.1] => (item=/library/www/osm-vector-maps/installer)

TASK [osm-vector-maps : Download 26M https://raw.githubusercontent.com/iiab/maps/master/2020/cities1000.sqlite to /library/www/osm-vector-maps/viewer/] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/library/www/osm-vector-maps/viewer/", "elapsed": 0, "gid": 0, "group": "root", "mode": "0755", "msg": "Request failed", "owner": "root", "response": "HTTP Error 404: Not Found", "size": 4096, "state": "directory", "status_code": 404, "uid": 0, "url": ["https://raw.githubusercontent.com/iiab/maps/master/2020/cities1000.sqlite"](https://raw.githubusercontent.com/iiab/maps/master/2020/cities1000.sqlite)}

PLAY RECAP *********************************************************************
127.0.0.1 : ok=44 changed=12 unreachable=0 failed=1 skipped=9 rescued=0 ignored=0

```

Broken images should be removed.

holta, 2022-08-20:
> Broken images should be removed.

Agreed.

I had been waiting for Raspberry Pi OS's new images (likely they're posting those in a few short weeks).
But indeed that now seems like a bad idea (to keep waiting for those).

tim-moody, 2022-08-20:
after pulling iiab and iiab-admin-console

```
TASK [0-init : Also set 'FQDN_changed: True' -- if iiab_fqdn != ansible_fqdn (box.lan)] ***
skipping: [127.0.0.1]

TASK [0-init : Add 'runtime' variable values to /etc/iiab/iiab.ini] ************
fatal: [127.0.0.1]: FAILED! => {"msg": "'dict object' has no attribute 'iiab_remote_url'"}

PLAY RECAP *********************************************************************
127.0.0.1 : ok=24 changed=4 unreachable=0 failed=1 skipped=4 rescued=0 ignored=0
```

tim-moody, 2022-08-20:
newly added iiab.ini sections should not have to exist

holta, 2022-08-20:
> fatal: [127.0.0.1]: FAILED! => {"msg": "'dict object' has no attribute 'iiab_remote_url'"}

It's rare for [local_facts.fact](https://github.com/iiab/iiab/blob/master/scripts/local_facts.fact) to change.

But when this does happen, the workaround is likely:

```
cp /opt/iiab/iiab/scripts/local_facts.fact /etc/ansible/facts.d/local_facts.fact
```

tim-moody, 2022-08-20:
yes. I now figured that out. the cp fixes the problem.

holta, 2022-08-20:
This ticket can probably be closed now, as new IIAB images are now posted:
https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Summary

Prior images will be deleted.  For now they're intentionally hidden within:
https://iiab.io/8.0/Old/

jvonau, 2022-08-22:
Both iiab-install and iiab-configure both place local_facts.fact into place when called, what was used to generate the above errors given SKIP_ROLE_ON_ERROR is now in the picture?

holta, 2022-09-07:
This should now be solved, by the images published off of:

https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Summary

closed 2022-09-07 by holta