iiab/iiab #3352: hostapd tries to start too early on Zero W? (manually it works!)

holta, 2022-08-23:
`systemctl restart hostapd` works when run manually.

Whereas during boot, `hostapd.service` appears to fail every time, with the following error:

```
root@box:~# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-08-22 16:48:12 PDT; 3min 36s ago
    Process: 274 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
   Main PID: 274 (code=exited, status=1/FAILURE)
        CPU: 88ms

Aug 22 16:48:12 box hostapd[274]: nl80211: Driver does not support authentication/association or connect commands
Aug 22 16:48:12 box hostapd[274]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Aug 22 16:48:12 box hostapd[274]: Could not read interface wlan0 flags: No such device
Aug 22 16:48:12 box hostapd[274]: nl80211 driver initialization failed.
Aug 22 16:48:12 box hostapd[274]: wlan0: interface state UNINITIALIZED->DISABLED
Aug 22 16:48:12 box hostapd[274]: wlan0: AP-DISABLED
Aug 22 16:48:12 box hostapd[274]: wlan0: CTRL-EVENT-TERMINATING
Aug 22 16:48:12 box hostapd[274]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Aug 22 16:48:12 box systemd[1]: hostapd.service: Main process exited, code=exited, status=1/FAILURE
Aug 22 16:48:12 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
```

This suggests [hostapd.legacy.j2](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/hostapd.legacy.j2) needs timing adjustment(s) or similar?

FYI this is running on 32-bit RasPiOS Lite 2022-04-04 (with all apt updates applies) on an original Raspberry Pi Zero W.

iiab-diagnostics: http://sprunge.us/JuKIoy?bash

jvonau, 2022-08-23:
```
2355 Aug 22 16:48:10 box kernel: [   38.678168] cfg80211: Loading compiled-in X.509 certificates for regulatory database
2356 Aug 22 16:48:11 box kernel: [   40.258790] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
2357 Aug 22 16:48:11 box kernel: [   40.271693] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
```

holta, 2022-08-23:
One proven workaround is to added these 2 lines...

```
BindsTo=sys-subsystem-net-devices-wlan0.device
After=sys-subsystem-net-devices-wlan0.device
```

...into the `[Unit]` section of both these 2 files:

- [hostapd.legacy.j2](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/hostapd.legacy.j2) (if `wifi_up_down: False`)
- [iiab-clone-wifi.service.j2](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-clone-wifi.service.j2) (if `wifi_up_down: True` with ap0)

...as borrowed from:

https://unix.stackexchange.com/questions/257888/systemd-wait-for-network-interface-to-be-up-before-running-service/417839#417839

If the above belongs in a PR *(is this appropriate & reasonably safe long-term?*) then both each above `wlan0` sub-string should become `{{ discovered_wireless_iface }}`

holta, 2022-08-23:
@jvonau suspects (very reasonably) that the root cause may lie in last week's Raspberry Pi OS kernel update &mdash; whose breadcrumbs are here:

```
"Unpacking raspberrypi-kernel (1:1.20220811-1) over (1:1.20220331-1) ..."
```

holta, 2022-08-23:
Thanks to @jvonau who was *extremely helpful* last night, helping to work through this!

Code resiliency suggestions here from the wider Raspberry Pi ecosystem would also be great, if that too later becomes possible ?!

holta, 2022-09-03:
A new kernel for Raspberry Pi OS was pushed (via apt) early yesterday (Sept 1st):

```
raspberrypi-kernel/stable 1:1.20220830-1 arm64 [upgradable from: 1:1.20220811-1]
```

So let's restest (and fix as nec) Zero W hotspot functionality, getting it back on track (one way or another) during September 2022!

(If nec, the [above workaround](https://github.com/iiab/iiab/issues/3352#issuecomment-1223515602) &mdash; that was quite well tested 2 weeks ago &mdash; MIGHT be appropriate in the end?)

holta, 2022-09-15:
> @jvonau suspects (very reasonably) that the root cause may lie in last week's Raspberry Pi OS kernel update — whose breadcrumbs are here:
> 
> ```
> "Unpacking raspberrypi-kernel (1:1.20220811-1) over (1:1.20220331-1) ..."
> ```

The brand new Raspberry Pi OS (from 2022-09-06) introduces a new kernel:

```
root@box:~# apt list raspberrypi-kernel
Listing... Done
raspberrypi-kernel/stable,now 1:1.20220830-1 arm64 [installed]
raspberrypi-kernel/stable 1:1.20220830-1 armhf
```

This might (or might not!) also require the [fix that was tested 24 days ago](https://github.com/iiab/iiab/issues/3352#issuecomment-1223515602) !

_Thank you to anybody who can help test either:_

1. The latest IIAB installed using curl 1-install-script (see https://download.iiab.io) and then inserted into an **RPi Zero W.**
2. The latest IIAB installed using curl 1-install-script (see https://download.iiab.io) and then inserted into an **RPi Zero 2 W.**

(And if hostapd doesn't launch on boot, please try the [aforementioned fix/workaround](https://github.com/iiab/iiab/issues/3352#issuecomment-1223515602) !)

holta, 2022-09-27:
> The brand new Raspberry Pi OS (from 2022-09-06) introduces a new kernel:
> 
> ```
> root@box:~# apt list raspberrypi-kernel
> Listing... Done
> raspberrypi-kernel/stable,now 1:1.20220830-1 arm64 [installed]
> raspberrypi-kernel/stable 1:1.20220830-1 armhf
> ```

Please help test this PR if you can!

- PR #3379

Testing these scenarios would be useful:

- 32-bit RasPiOS Lite on Zero W
- 64-bit RasPiOS Lite on Zero 2 W

holta, 2022-09-29:
PR #3379 is now merged to address this issue that appears a result of (both) Raspberry Pi OS kernels released in Aug 2022 (20220811 and 20220830).  But let's keep this one for a few days or weeks to confirm?

Discussion ongoing here:

- PR #3179

holta, 2022-10-13:
> But let's keep this ticket open for a few days or weeks to confirm?

Let's close this ticket within the month if possible.

holta, 2022-10-23:
Good Enough!

Please re-open (or open a new ticket) if similar issues should arise.

closed 2022-10-23 by holta