iiab/iiab #3357: apt upgrading Kolibri is risky as settings accidentally get overridden

holta, 2022-09-01:
FYI here are some of the ~3 pink/blue screens that arise if one `apt` upgrades from Kolibri 0.15.6 to 0.15.7:

_(unfortunately these will lead to quite serious problems, as everyday IIAB operators will accidentally override pre-existing Kolibri systemd unit file & Linux username, something to think about!!)_

```
  ┌────────────────────────┤ Kolibri configuration ├────────────────────────┐
  │                                                                         │
  │ This will automatically call the system service to start Kolibri when   │
  │ your system boots.                                                      │
  │                                                                         │
  │ Do you wish to run Kolibri on start-up?                                 │
  │                                                                         │
  │                    <Yes>                       <No>                     │
  │                                                                         │
  └─────────────────────────────────────────────────────────────────────────┘
```

If one overrides the above default `<Yes>` to correctly select `<No>` (to preserve [IIAB's systemd file for Kolibri](https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri.service.j2)) the next screen is:

```
  ┌────────────────────────┤ Kolibri configuration ├────────────────────────┐
  │                                                                         │
  │ Running Kolibri manually:                                               │
  │                                                                         │
  │ You have chosen not to run Kolibri as a system service and that means   │
  │ you should run it manually. Starting Kolibri with your current user     │
  │ account:                                                                │
  │                                                                         │
  │   kolibri start                                                         │
  │                                                                         │
  │                                                                         │
  │ Starting Kolibri as a system-wide service:                              │
  │                                                                         │
  │   sudo service kolibri start                                            │
  │                                                                         │
  │                                                                         │
  │ Start Kolibri when the system boots:                                    │
  │                                                                         │
  │   sudo systemctl enable kolibri                                         │
  │                                                                         │
  │                                 <Ok>                                    │
  │                                                                         │
  └─────────────────────────────────────────────────────────────────────────┘
```

Finally this 3rd screen is generally pre-populated with an incorrect username: (IIAB mandates Linux username `kolibri` not `iiab-admin`, Ouch !)

```
  ┌────────────────────────┤ Kolibri configuration ├────────────────────────┐
  │ The default is to choose your preferred desktop user account, for       │
  │ instance to ensure access to importing data from external USB devices.  │
  │                                                                         │
  │ Entering a username that doesn't exist will create a new system user    │
  │ with home directory /var/<username>.                                    │
  │                                                                         │
  │ Which user account should own the Kolibri server?                       │
  │                                                                         │
  │ iiab-admin_____________________________________________________________ │
  │                                                                         │
  │                                 <Ok>                                    │
  │                                                                         │
  └─────────────────────────────────────────────────────────────────────────┘
```

I don't have answers; am simply posting the above (Linux Terminal) screenshots, so we can all think about it :thinking:

_Originally posted by @holta in https://github.com/iiab/iiab/issues/3356#issuecomment-1233694501_

holta, 2022-09-01:
_Also pasted in from #3356..._

> If one overrides the above default `<Yes>` to correctly select `<No>` (to preserve [IIAB's systemd file for Kolibri](https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri.service.j2)) the next screen is:

Strangely the above Kolibri prompt neither disables nor clobbers [`/etc/systemd/system/kolibri.service`](https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri.service.j2) no matter whether you answer `<Yes>` or `<No>`

'Yes' was tested on Ubuntu 20.04, 'No' was tested on Mint 21.  Again this is in response to "Do you wish to run Kolibri on start-up?"  _(So this definitely appears to be a Kolibri bug &mdash; a bug that ironically seems to prevent further problems for the moment &mdash; when apt upgrading Kolibri as part of IIAB !)_

In any case, if one goes ahead with the default `<Yes>`, the next screen (a bit different than above) is...just FYI:

```
    ┌──────────────────────┤ Kolibri configuration ├───────────────────────┐
    │                                                                      │
    │ The Kolibri system service                                           │
    │                                                                      │
    │ You have chosen to run Kolibri as a system service and it has been   │
    │ started automatically. You can start/stop/restart Kolibri with the   │
    │ following command                                                    │
    │                                                                      │
    │   sudo service kolibri [start, stop, restart]                        │
    │                                                                      │
    │                                <Ok>                                  │
    │                                                                      │
    └──────────────────────────────────────────────────────────────────────┘
```

jvonau, 2022-09-07:
What apt call did you use upgrade or dist-upgrade?

holta, 2022-09-07:
> What apt call did you use upgrade or dist-upgrade?

1) Likely I ran `apt dist-upgrade`
2) One can also run `apt install kolibri` presumably, to force the Kolibri upgrade to begin.
3) The 3rd Kolibri install screen seems to be pre-populated with a "main" Linux username (e.g. what's used by that distro's Graphical Desktop typically, if the distro isn't headless).  Which is almost always the wrong answer &mdash; with Internet-in-a-Box ( IIAB runs Kolibri with [username 'kolibri'](https://github.com/iiab/iiab/blob/master/roles/kolibri/defaults/main.yml#L41-L44) ) :sob:

cc: @jredrejo who might have recommendations!

jredrejo, 2022-09-08:
> > What apt call did you use upgrade or dist-upgrade?
> 
> 1. Likely I ran `apt dist-upgrade`
> 2. One can also run `apt install kolibri` presumably, to force the Kolibri upgrade to begin.
> 3. The 3rd Kolibri install screen seems to be pre-populated with a "main" Linux username (e.g. what's used by that distro's Graphical Desktop typically, if the distro isn't headless).  Which is almost always the wrong answer — with Internet-in-a-Box ( IIAB runs Kolibri with [username 'kolibri'](https://github.com/iiab/iiab/blob/master/roles/kolibri/defaults/main.yml#L41-L44) ) sob
> 
> cc: @jredrejo who might have recommendations!

hello, 

1. `upgrade` will have problems if the changes in the package involve adding/removing packages due to  new dependencies or conflict. If that's the case `dist-upgrade` will _upgrade_ the package but `upgrade` won't do it. 
2. just installing kolibri will do whatever is needed, so it'd have the same effect as dist-upgrade (but with the scope reduced to  kolibri)
3. yep, you're right,[ it does it here](https://github.com/learningequality/kolibri-installer-debian/blob/master/debian/kolibri.scripts-common#L352-L357) , If you tell me any flag I can use to find out if the device is running IIAB I'm happy to change it to use kolibri in such case

holta, 2022-09-08:
> 3\. yep, you're right,[ it does it here](https://github.com/learningequality/kolibri-installer-debian/blob/master/debian/kolibri.scripts-common#L352-L357) , If you tell me any flag I can use to find out if the device is running IIAB I'm happy to change it to use kolibri in such case

_Huge Thanks @jredrejo !_

We'll give this some thought so that this is done in a thoughtful + sustainable way &mdash; and get back to you :heavy_check_mark:

RECAP: https://github.com/learningequality/kolibri-installer-debian/blob/5f01725423f0bc5d93a8e57fb65e25c3edc46b13/debian/kolibri.scripts-common#L352-L357

holta, 2022-09-08:
@jredrejo in the past (@benjaoming I think!?) suggested we encode the Linux username (that Kolibri uses) into these 2 places:

1) `/etc/kolibri/username`
https://github.com/iiab/iiab/blob/ebdba360a38e1c18f2e4dfd9ba9980355d2a0202/roles/kolibri/tasks/install.yml#L24-L27
2) Linux Environment Variable `KOLIBRI_USER`
https://github.com/iiab/iiab/blob/ebdba360a38e1c18f2e4dfd9ba9980355d2a0202/roles/kolibri/tasks/install.yml#L100-L105

QUESTION: Might it be possible for Kolibri's installer to auto-extract either above (programmatically) to pre-populate the [aforementioned](https://github.com/iiab/iiab/issues/3357#issue-1358271391) user prompt during Kolibri upgrade ?

jredrejo, 2022-09-08:
kolibri installer is a normal deb package, you can always [preseed it](https://wiki.debian.org/PackageManagement/Preseed). 

As an example, [this is the preseed](https://github.com/learningequality/pi-gen/blob/master/stage2/03-kolibri-server/00-debconf) used to build our own RPi image.

You can use the same, replacing `pi` by `kolibri` if that's your case. Installing the package after using this preseed will make it use your configs without questions.

holta, 2022-09-08:
> Installing the package after using this preseed will make it use your configs without questions.

Interesting!

Is this in fact the most robust + resilient way long-term ??

(I'd been hoping Kolibri could retain its own state, so as not to accidentally damage itself during upgrades &mdash; even if the deb/apt upgrade screens happen to change year by year &mdash; in coming years?  Or is that completely unrealistic?)

> kolibri installer is a normal deb package, you can always [preseed it](https://wiki.debian.org/PackageManagement/Preseed).

Just FYI the above doc contains many translation and technical errors :shushing_face:

Even the original French doc contains errors, e.g. `debconf-get-selections | grep -B1 '^kolibri '` does not work, as TABS are the relevant separators, not spaces!

So it turns out `\s` or `[[:space:]]` or `[[:blank]]` thankfully work, as in this quick example to explore this avenue:

```
root@box:~# debconf-get-selections | grep -B1 '^kolibri\s'
# Do you wish to run Kolibri on start-up?
kolibri kolibri/init    boolean true
--
# The Kolibri system service
kolibri kolibri/init-instructions       note
--
# Changing system service owner:
kolibri kolibri/pre-010-upgrade-system-user     note
--
# Running Kolibri manually:
kolibri kolibri/manual-instructions     note
--
# Which user account should own the Kolibri server?
kolibri kolibri/user    string  ubuntu
```

holta, 2022-09-08:
Just FYI if one runs `dpkg-reconfigure kolibri` then a 4th screen also appears:

```
 ┌─────────────────────────┤ Kolibri configuration ├─────────────────────────┐
 │                                                                           │
 │ Changing system service owner:                                            │
 │                                                                           │
 │ You are updating from an older version of Kolibri and we recommend that   │
 │ you switch the user account that runs the system service to your normal   │
 │ desktop user. This allows Kolibri to access your USB devices for          │
 │ importing and exporting content channels.                                 │
 │                                                                           │
 │ After finishing the installation/upgrade, follow these instructions to    │
 │ change the user setting and move your previous content channels and       │
 │ database:                                                                 │
 │                                                                           │
 │    https://learningequality.org/r/docs-debian-change-user                 │
 │                                                                           │
 │                                  <Ok>                                     │
 │                                                                           │
 └───────────────────────────────────────────────────────────────────────────┘
```

When trying to visit the above URL, it redirects to this broken link:

https://kolibri.readthedocs.io/en/latest/install.html#changing-the-owner-of-kolibri-system-service

Finally, the final output of the above `dpkg-reconfigure kolibri` command (after it returns from the pink/blue interactive screens) is...

```
root@box:~# dpkg-reconfigure kolibri
Synchronizing state of kolibri.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable kolibri
```

jvonau, 2022-09-08:
Would to not be easier just to create /etc/default/kolibri with the alternate iiab info before installing the deb file? That should be referenced first by the deb install routine as it's listed in conffiles?

holta, 2022-09-08:
> easier just to create /etc/default/kolibri with the alternate iiab info

Great question!

Does Kolibri support setting `KOLIBRI_USER` etc within `/etc/default/kolibri` as possibly suggested below?

- https://github.com/learningequality/kolibri-installer-debian/issues/62

holta, 2022-09-08:
@jredrejo

1) Short Term Question: Should IIAB run the following command immediately after installing Kolibri?

   ```
   echo 'kolibri kolibri/user string kolibri' | debconf-set-selections
   ```

   The above would seem to help &mdash; at least nudging the operator in the right direction, so they're prompted with a more meaningful KOLIBRI_USER default &mdash; to (try!) to avoid shooting themselves in the foot during apt updates :sweat_smile:

2) Longer Term Question: Can Kolibri consider relying on KOLIBRI_USER and KOLIBRI_HOME in some standardized/canonical place ( `/etc/defaults/kolibri` or any other place! ) so future Kolibri upgrades a far more _"foolproof"_ and don't accidentally damage Kolibri's installation?

holta, 2022-09-08:
KOLIBRI_USER or KOLIBRI_HOME should almost always be preserved from the prior Kolibri installation (during Kolibri upgrades).

But FWIW https://askubuntu.com/questions/429592/what-is-the-purpose-of-etc-default/429604#429604 does mention that `/etc/default` is almost perfect for this &mdash; e.g. if post-installation brain/heart surgery (very rarely, but occasionally) is desired:

> Basically, `/etc/default` contains some parameters that the end user or administrator is likely to change, rather than embedding the values in the actual boot scripts. **In this way, changes will persist even if you upgrade the package and the boot script is replaced.**

holta, 2022-09-09:
> > Does Kolibri support setting `KOLIBRI_USER` etc within `/etc/default/kolibri` as possibly suggested below?
> > 
> > * [Incorrect value of KOLIBRI_HOME in /etc/default/kolibri learningequality/kolibri-installer-debian#62](https://github.com/learningequality/kolibri-installer-debian/issues/62)
> 
> Yes

Interesting!  @jredrejo does this mean that IIAB can safely now...

- stop recording `KOLIBRI_USER` in `/etc/kolibri/username` ?

- stop recording `KOLIBRI_USER` and `KOLIBRI_HOME` into Linux Environment Variables ?

([As mentioned above.](https://github.com/iiab/iiab/issues/3357#issuecomment-1240946532))

jredrejo, 2022-09-09:
> Interesting! @jredrejo does this mean that IIAB can safely now...
> 
> * stop recording `KOLIBRI_USER` in `/etc/kolibri/username` ?
> * stop recording `KOLIBRI_USER` and `KOLIBRI_HOME` into Linux Environment Variables ?

you either continue doing it or you preseed these values, whatever you prefer

jredrejo, 2022-09-09:
> Just FYI if one runs `dpkg-reconfigure kolibri` then a 4th screen can also appear:
> 
> ```
>  ┌─────────────────────────┤ Kolibri configuration ├─────────────────────────┐
>  │                                                                           │
>  │ Changing system service owner:                                            │
>  │                                                                           │
>  │ You are updating from an older version of Kolibri and we recommend that   │
>  │ you switch the user account that runs the system service to your normal   │
>  │ desktop user. This allows Kolibri to access your USB devices for          │
>  │ importing and exporting content channels.                                 │
>  │                                                                           │
>  │ After finishing the installation/upgrade, follow these instructions to    │
>  │ change the user setting and move your previous content channels and       │
>  │ database:                                                                 │
>  │                                                                           │
>  │    https://learningequality.org/r/docs-debian-change-user                 │
>  │                                                                           │
>  │                                  <Ok>                                     │
>  │                                                                           │
>  └───────────────────────────────────────────────────────────────────────────┘
> ```
> 
> When trying to visit the above URL, it redirects to this broken link:
> 
> https://kolibri.readthedocs.io/en/latest/install.html#changing-the-owner-of-kolibri-system-service

Thanks for the warning, that's an alias that was not updated after the documentation was updated. We'll fix it

> Finally, the final output of the above `dpkg-reconfigure kolibri` command (after it returns from the pink/blue interactive screens) is...
>
> ```
> root@box:~# dpkg-reconfigure kolibri
> Synchronizing [...]

yes, I know, reconfigure means executing the postinst script of the package so it shows the script messages , you can set the non-interactive envvar if you want to hide them: https://linuxhint.com/debian_frontend_noninteractive/

jredrejo, 2022-09-09:
> > Installing the package after using this preseed will make it use your configs without questions.
> 
> Interesting!
> 
> Is this in fact the most robust + resilient way long-term ??

Well, that's what the Debian installer has been using for more than 20 years and Ubuntu and other derivatives use, so I'd say yes ;) 

> 
> (I'd been hoping Kolibri could retain its own state, so as not to accidentally damage itself during upgrades — even if the deb/apt upgrade screens happen to change year by year — in coming years? Or is that completely unrealistic?)

it's totally realistic, even if kolibri is not uploaded to the official Debian repositories, its package is treated as a standard package so being able to upgrade without errors is a must. That's why I think that relying in the standard Debian packages mechanisms is far better than overriting config files or similar strategies. Debian policies are quite strict and have proved to be bulletproof.

> 
> > kolibri installer is a normal deb package, you can always [preseed it](https://wiki.debian.org/PackageManagement/Preseed).

holta, 2022-09-09:
@jredrejo Quick Question: is Line 35 of `/etc/default/kolibri` ignoring any pre-existing KOLIBRI_HOME environment variable on purpose?

```
KOLIBRI_HOME="$(getent passwd $KOLIBRI_USER | awk -F ':' '{print $6}')/.kolibri"
```

( That's https://github.com/learningequality/kolibri-installer-debian/blob/5f01725423f0bc5d93a8e57fb65e25c3edc46b13/debian/startup/kolibri.default#L35 )

FWIW in the case of IIAB, the above code overwrites IIAB's current environment variable `KOLIBRI_HOME=/libary/kolibri` and changes it to `KOLIBRI_HOME=/home/kolibri/.kolibri` (which seems very wrong?!)

Ref:

- PR learningequality/kolibri-installer-debian#63

jredrejo, 2022-09-09:
> @jredrejo Quick Question: is Line 35 of `/etc/default/kolibri` ignoring any pre-existing KOLIBRI_HOME environment variable on purpose?
>
> ```
> KOLIBRI_HOME="$(getent passwd $KOLIBRI_USER | awk -F ':' [...]

`kolibri.default` is just for the kolibri service (or daemon, if you prefer to call it).  Setting `KOLIBRI_HOME` to a different value was never thought to be done within the service. If you start kolibri normally (i.e. `kolibri start`) , your env var is not overwritten. If you want to have this as a new feature you can create a github issue and I'll do it when I got some bandwidth.

holta, 2022-09-09:
> you can create a github issue and I'll do it when I got some bandwidth

Thanks @jredrejo for all your help explaining.

This is starting to make a lot more sense now, thanks to your explanations.

So I wrote up a suggestion (trying!) to get at the root cause &mdash; and hopefully solve that:

- learningequality/kolibri-installer-debian#115

But if I got any details wrong, I apologize (if so just LMK, Thanks!!)

holta, 2023-06-29:
Should be largely fixed now!

(If issues recur, this ticket can be re-opened.)

closed 2023-06-29 by holta