iiab/iiab #3362: Admin Console > Install Content > Clone IIAB fails w/ RasPiOS with desktop (1) USB Stick starts then fails (2) USB HDD not visible at all

holta, 2022-09-07:
Thanks to @georgejhunt helping me investigate what's happening!  Hopefully we'll have more info within ~24h.

(1) The external USB Stick can be made to work by manually dropping the secondary mount, e.g.:

```
umount /media/wce/IIAB
```

The errors (below) suddenly then go away, and **Admin Console > Install Content > Clone IIAB** appears to actually work, Amazing!  FYI the OS is 64-bit Raspberry Pi OS with desktop, running on a Raspberry Pi 400.

(2) It will be interesting to see if a similar manual workaround might (or might not) help with external USB HDD (a spinning disk) e.g.:

```
umount /media/wce/ABFB-745A
```

So far **Admin Console > Install Content > Clone IIAB > Find USB Device** (BLUE BUTTON) does not work at all with the (spinning) USB HDD.  No writable devices are found.  Which is odd as both external devices (USB Stick and USB HDD, one at a time) auto-mount to `/media/usb0` and yet behave very differently !

(3) Original failure of **Admin Console > Install Content > Clone IIAB** attempting to write to the USB Stick is here, is immediate, happening nearly instantly after you click Clone IIAB.  Here are the resulting errors:

[image]

```
Error: Partition(s) 1 on /dev/sda have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use. As a result, the old partition(s) will remain in use. You should reboot now before making further changes.
Preparing target...
Could not create FAT.                    
```

```
cp: cannot create directory '/tmp/tmp.8U7FSSJ2T4/././home/wce/Pictures': No such file or directory
cp: cannot create directory '/tmp/tmp.8U7FSSJ2T4/././home/wce/Videos': No such file or directory
cp: cannot create directory '/tmp/tmp.8U7FSSJ2T4/././home/wce/.pki': No such file or directory
cp: preserving times for '/tmp/tmp.8U7FSSJ2T4/././home/wce': No such file or directory
cp: cannot create directory '/tmp/tmp.8U7FSSJ2T4/././home/iiab-admin': No such file or directory
cp: preserving times for '/tmp/tmp.8U7FSSJ2T4/././home': No such file or directory

[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)
[ ] (0%)cp: cannot copy a directory, '/tmp/tmp.hqsp6iCjc9/.', into itself, '/tmp/tmp.8U7FSSJ2T4/./.'

[##################################################] (100%)umount: /tmp/tmp.8U7FSSJ2T4: not mounted.

Could not unmount partition.
```

jvonau, 2022-09-07:
Turn the auto-mount feature off within the Desktop?

holta, 2022-09-07:
> Turn the auto-mount feature off within the Desktop?

Indeed that must be the core of the problem (certainly with the USB Stick).

Hopefully the coming 24h will shed even more light... e.g. why the USB HDD is behaving so differently from the USB Stick / Flash Drive (etc).

holta, 2022-09-08:
> (2) It will be interesting to see if a similar manual workaround might (or might not) help with external USB HDD (a spinning disk) e.g.:
> 
> ```
> umount /media/wce/ABFB-745A
> ```

Unfortunately the above did not work:

(A) No matter how this external USB HDD's /dev/sda1 is mounted or umount'ed (the **Find USB Device** blue button is clicked every time) this Admin Console interface can't seem to ever find /dev/sda :

[image]

(B) RECAP: In comparison, this Admin Console interface never seems to have a problem finding the external USB STICK's /dev/sda (i.e. when trying a thumb drive INSTEAD OF a spinning disk)  as seen in this prior screenshot:

[image]

(C) @georgejhunt can you tell why manually attempting `piclone_cmd /dev/sda` (on the spinning HDD) failed as follows?

```
root@box:~# piclone_cmd /dev/sda
Preparing target...
Reading partitions...
Preparing partitions...
mke2fs 1.46.2 (28-Feb-2021)

Copying partition 1 of 2...
[##################################################] (100%)umount: /tmp/tmp.akTIYioQnd: not mounted.

Could not unmount partition.
```

(D) Just FYI this is how far it got &mdash; repartitioning the external USB HDD &mdash; but not much further:

```
root@box:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk
├─sda1        8:1    0   256M  0 part
└─sda2        8:2    0 931.3G  0 part
mmcblk0     179:0    0 476.7G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /tmp/tmp.405VwE0710
└─mmcblk0p2 179:2    0 476.5G  0 part /

root@box:~# umount /tmp/tmp.405VwE0710

root@box:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk
├─sda1        8:1    0   256M  0 part
└─sda2        8:2    0 931.3G  0 part
mmcblk0     179:0    0 476.7G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0 476.5G  0 part /

root@box:~# blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="5EA1-EC88" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="31e9c350-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="f0702b0a-e54b-4c34-bd2d-6e52a561898c" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="31e9c350-02"
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="boot" LABEL="boot" UUID="5EA1-EC88" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="969bfef6-01"
/dev/sda2: LABEL="rootfs" UUID="f0702b0a-e54b-4c34-bd2d-6e52a561898c" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="969bfef6-02"
```

jvonau, 2022-09-08:
Is the desktop still the same as [older versions](https://raspberrypi.stackexchange.com/questions/65739/stock-pi3-what-process-is-auto-mounting-removable-media)?

jvonau, 2022-09-08:
> SEC_TYPE="msdos" 

That is something different.

holta, 2022-09-08:
Lots of good questions during today's call (http://minutes.iiab.io) but no clear/final answer just yet!

Needs more investigation in coming weeks :thinking:

Meantime, those using 'Raspberry Pi OS with desktop' can click the **Raspberry start icon** (top-left of screen) **> Accessories > SD Card Copier** which has been confirmed to solves all problems (when backing up to an external USB HDD).

holta, 2022-09-09:
> Is the desktop still the same as [older versions](https://raspberrypi.stackexchange.com/questions/65739/stock-pi3-what-process-is-auto-mounting-removable-media)?

1. Disabling automount within "Raspberry Pi OS with desktop" (by clicking **File Manager > Edit > Preference > Volume Management > Mount removable media automatically when they are inserted**) does not help.  I ran about 20 tests (repeatedly re-attaching a real/spinning USD HDD, and also inserting a solid-state USB Stick) and disabling automount appears to make no difference at all in the end.
   a. ASIDE/FYI: This [5-year old GUI](https://raspberrypi.stackexchange.com/questions/65739/stock-pi3-what-process-is-auto-mounting-removable-media/65945#65945) looks completely different now.  But the idea is the same (and the verbiage remains essentially the same, as you click, for people who want to turn off RasPiOS automount).
2. **Clone IIAB** failed to complete about ~75% of the time (regardless whether OS automount was on or off).
3. It always failed within the 1st minute or so.  So re-trying (while very ugly, and extremely confusing) often works if you re-attempt about 4 times.  (Not something that new users will accept, but FYI!)
4. _Even if the USB HDD has no partitions that are double-mounted_ (this ticket's original issue, or so it appeared!), the **Clone IIAB** and what it invokes (`piclone_cmd /dev/sda`) still fail very regularly during initial stages.  The 4 most common failures were:
   a. Admin Console pops up a message saying that some device is busy.  Clicking the **Clone IIAB** RED BUTTON a 2nd time usually works to get past this.  Not a big deal (many people give up completely when they see an error message, but people who actuall read it will likely try clicking the **Clone IIAB** RED BUTTON again, which seems to usually work).
   b. `piclone_cmd` struggles to re-partition many pre-existing partition schemes:
[image]
Likewise:
   ```
   Error: Partition(s) 4 on /dev/sda have been written, but we have been unable to inform the kernel of the change, probably 
   because it/they are in use. As a result, the old partition(s) will remain in use. You should reboot now before making further 
   changes.
   Preparing target...
   Could not create FAT.
   ```
   c. **"Re-reading the device or partition table failed.: Device or resource busy"**
[image]
   d. **PARTITION 1 COPY CLAIMS SUCCESS BUT... "Could not unmount partition."**
   ```
   The device contains 'iso9660' signature and it may remain on the device. It is recommended to wipe the device with wipefs(8) 
   or fdisk --wipe, in order to avoid possible collisions.
   Re-reading the partition table failed.: Device or resource busy
   Preparing target...
   Reading partitions...
   Preparing partitions...

   Copying partition 1 of 2...

   [ ] (0%)
   [ ] (0%)
   [############################################ ] (88%)
   [##################################################] (100%)umount: /tmp/tmp.Z83pdkPntq: not mounted.

   Could not unmount partition.
   ```

ASIDE: All ~20 above tests were done using 64-bit "Raspberry Pi OS with desktop" on Raspberry Pi 4.  Which reconfirms what was reported earlier (top of this ticket) with the same OS but on Raspberry Pi 400.

I did not test **Clone IIAB** (and its underlying `/usr/sbin/piclone_cmd` commands that Admin Console installs from either https://download.iiab.io/packages/piclone_cmd_aarch64 or https://download.iiab.io/packages/piclone_cmd_armv7l) with any 32-bit or "Lite" (headless) versions of Raspberry Pi SO.  I would assume many of the same failures occur on those OS variants as well &mdash; but I might be wrong, and if someone has time to test any of these other versions of the OS to corroborate &mdash; that might be incredibly helpful!

holta, 2022-09-09:
I posted a public warning in http://FAQ.IIAB.IO -> **"43 How do I back up, shrink & copy IIAB microSD cards?"** until we sort this out:

https://wiki.iiab.io/index.php?title=FAQ&type=revision&diff=418&oldid=417

holta, 2022-09-18:
Just FYI `rpi-clone` look Potentially Great... _but:_

1. @jvonau mentions there's a risk with `rpi-clone` if IIAB is still using **fuse** :

   https://github.com/billw2/rpi-clone excerpt: "FUSE mounts (ssh mounts) should be unmounted before cloning or else the directory mounted on will not stat and the directory will not be made on the clone. You will get a readlink stat error from rsync because root can't access a users FUSE mount - only the user can."

1. Bill Wilson's `rpi-clone` can freeze on Ubuntu & Ubuntu Mate:

   https://github.com/billw2/rpi-clone/issues/125

jvonau, 2022-09-18:
(A) Picture I can't reproduce if /dev/sdX is still present in /dev/ so I suspect there might be an intermittent power issue (does this device have it's own source of power or a powered usb hub in use?) that might of removed the spinning usb storage device from /dev/.

holta, 2022-09-18:
> does this device have it's own source of power or a powered usb hub in use?

Yes.  The external USB HDD had its own independent power supply.  (Electricity interruptions are not an issue.)

USB HDD versus USB Sticks would appear to mount to identical (or nearly identical??) mount points...no matter which kind is spontaneously inserted into a Raspberry Pi running Raspberry Pi OS with desktop.

Somehow Admin Console treats USB HDD very differently from USB Sticks.

I don't know anybody who's tried this same experiment with an external USB SSD.  That data point would be great have.  (Do we know anybody who owns an external USB SSD?)

jvonau, 2022-09-18:
Side note https://www.pragmaticlinux.com/2021/03/fix-for-getting-your-ssd-working-via-usb-3-on-your-raspberry-pi/

jvonau, 2022-09-18:
We have a reproducer this is displayed 

[image]

While mount show this 
/dev/sda1 on /media/usb0 type fuseblk (rw,nodev,noexec,noatime,sync,user_id=0,group_id=0,allow_other,blksize=4096)

blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="29F5-65C4" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="4083909b-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="cbe4d267-24de-4402-9a4b-1413a1da5eb8" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="4083909b-02"

df
Filesystem      1K-blocks     Used  Available Use% Mounted on
/dev/root       122578588 67655312   52414848  57% /
devtmpfs          1776952        0    1776952   0% /dev
tmpfs             1942712    49752    1892960   3% /dev/shm
tmpfs              777088     1456     775632   1% /run
tmpfs                5120        4       5116   1% /run/lock
/dev/mmcblk0p1     261108    31222     229886  12% /boot
tmpfs              388540       24     388516   1% /run/user/1000
/dev/loop0      111702920 59427154   46505008  57% /library-ext
/dev/sda1      1953479676   151220 1953328456   1% /media/usb0

jvonau, 2022-09-19:
Non Reproducer of (A) Can find the device prior use was bootable iso9660 : 

mount
/dev/sda1 on /media/jerry/Ubuntu 21.10 amd64 type iso9660 (ro,nosuid,nodev,relatime,nojoliet,check=s,map=n,blocksize=2048,uid=1000,gid=1000,dmode=500,fmode=400,iocharset=utf8,uhelper=udisks2)
/dev/sda2 on /media/usb0 type vfat (rw,nodev,noexec,noatime,nodiratime,sync,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)

blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="29F5-65C4" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="4083909b-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="cbe4d267-24de-4402-9a4b-1413a1da5eb8" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="4083909b-02"

df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122578588 67666936  52403224  57% /
devtmpfs         1776952        0   1776952   0% /dev
tmpfs            1942712    49752   1892960   3% /dev/shm
tmpfs             777088     1472    775616   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261108    31222    229886  12% /boot
tmpfs             388540       24    388516   1% /run/user/1000
/dev/loop0     111702920 59427154  46505008  57% /library-ext
/dev/sda1        3038844  3038844         0 100% /media/jerry/Ubuntu 21.10 amd64
/dev/sda2           4198     3206       992  77% /media/usb0

jvonau, 2022-09-19:
@holta can you grab the output of `udisksctl info -b /dev/sda` of the problem spinning HDD prior to trying to clone?

holta, 2022-09-19:
> output of udisksctl info -b /dev/sda and lsblk of the problem spinning HDD prior to trying to clone

1) Pasted in below.

2) This is not the original powered USB HDD (I no longer have access to that) but it's the same situation here &mdash; **Admin Console > Install Content > Clone IIAB > Find USB Device** (BLUE BUTTON) cannot find it.

3) "RasPiOS with desktop" **> File Manager > Edit > Preference > Mount removable media automatically when they are inserted** is ON at the moment.  This neither helps not hinders (in all my experiments so far) but just FYI.

```
root@box:~# udisksctl info -b /dev/sda
/org/freedesktop/UDisks2/block_devices/sda:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/sda
    DeviceNumber:               2048
    Drive:                      '/org/freedesktop/UDisks2/drives/WD_My_Passport_0748_575842314131333237323933'
    HintAuto:                   true
    HintIconName:
    HintIgnore:                 false
    HintName:
    HintPartitionable:          true
    HintSymbolicIconName:
    HintSystem:                 false
    Id:                         by-id-usb-WD_My_Passport_0748_575842314131333237323933-0:0
    IdLabel:
    IdType:
    IdUUID:
    IdUsage:
    IdVersion:
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/sda
    ReadOnly:                   false
    Size:                       1000170586112
    Symlinks:                   /dev/disk/by-id/usb-WD_My_Passport_0748_575842314131333237323933-0:0
                                /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1:1.0-scsi-0:0:0:0
    UserspaceMountOptions:
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/sda2
                        /org/freedesktop/UDisks2/block_devices/sda1
    Type:               dos
```

```
root@box:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk
├─sda1        8:1    0   256M  0 part /media/usb1
└─sda2        8:2    0 931.2G  0 part /media/usb0
mmcblk0     179:0    0 119.1G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0 118.8G  0 part /
```

```
root@box:~# mount | grep sda
/dev/sda2 on /media/usb0 type ext4 (rw,nodev,noexec,noatime,nodiratime,sync)
/dev/sda1 on /media/usb1 type vfat (rw,nodev,noexec,noatime,nodiratime,sync,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

jvonau, 2022-09-20:
Continuation of reproducer of (A) above
```
jerry@NM-64-desktop-RasPiOS:~ $ udisksctl info -b /dev/sdb
/org/freedesktop/UDisks2/block_devices/sdb:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/sdb
    DeviceNumber:               2064
    Drive:                      '/org/freedesktop/UDisks2/drives/WD_My_Passport_2626_57584E324133314337463248'
    HintAuto:                   true
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 false
    Id:                         by-id-usb-WD_My_Passport_2626_57584E324133314337463248-0:0
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/sdb
    ReadOnly:                   false
    Size:                       2000365289472
    Symlinks:                   /dev/disk/by-id/usb-WD_My_Passport_2626_57584E324133314337463248-0:0
                                /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:2:1.0-scsi-0:0:0:0
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/sdb1
    Type:               gpt
```
Shows up on the desktop as "My Passport", I bet the space in the name or lack of IdLabel: IdType: IdUUID: is tripping up blkid, not being displayed there then not found by admin-console.
Side note /sys/fs/fuse/connections would have entries when used to mount stuff and need to be 'ejected' to fully close the device after unmounting.

jvonau, 2022-09-23:
observation on U22.04
```
cat /etc/fstab
/mnt/spin /library none defaults,bind,nofail 0 0
mount
/dev/sda1 on /mnt/spin type ext4 (rw,relatime,errors=remount-ro)
/dev/sda1 on /library type ext4 (rw,relatime,errors=remount-ro)

lsblk
sda           8:0    0   3.6T  0 disk 
└─sda1        8:1    0   3.6T  0 part /library
                                      /mnt/spin
lsblk -a
sda           8:0    0   3.6T  0 disk 
└─sda1        8:1    0   3.6T  0 part /library
                                      /mnt/spin
lsblk -a -o name,fstype,mountpoint
sda                  
└─sda1      ext4     /library
```
The use of -o returns only the last mountpoint

observation on RaspiOS without Desktop
```
sudo mount /dev/sdb2 /mnt
mount
/dev/sdb2 on /media/usb2 type exfat (rw,nodev,noexec,noatime,nodiratime,sync,fmask=0000,dmask=0000,allow_utime=0022,iocharset=utf8,errors=remount-ro)
/dev/sdb2 on /mnt type exfat (rw,relatime,sync,fmask=0000,dmask=0000,allow_utime=0022,iocharset=utf8,errors=remount-ro)

lsblk
sdb           8:16   0 465.8G  0 disk 
├─sdb1        8:17   0   200M  0 part /media/usb1
└─sdb2        8:18   0 465.6G  0 part /mnt
lsblk -a
sdb           8:16   0 465.8G  0 disk 
├─sdb1        8:17   0   200M  0 part /media/usb1
└─sdb2        8:18   0 465.6G  0 part /mnt
```
last mount point is only ever shown.

Another reproducer of (A) above without Desktop
```
udisksctl info -b /dev/sdb
/org/freedesktop/UDisks2/block_devices/sdb:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/sdb
    DeviceNumber:               2064
    Drive:                      '/org/freedesktop/UDisks2/drives/Seagate_Expansion_SSD_00000000NAAL02RS'
    HintAuto:                   true
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 false
    Id:                         by-id-usb-Seagate_Expansion_SSD_00000000NAAL02RS-0:0
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/sdb
    ReadOnly:                   false
    Size:                       500107861504
    Symlinks:                   /dev/disk/by-id/usb-Seagate_Expansion_SSD_00000000NAAL02RS-0:0
                                /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:2:1.0-scsi-0:0:0:0
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/sdb2
                        /org/freedesktop/UDisks2/block_devices/sdb1
    Type:               gpt

mount | grep sd
/dev/sda1 on /media/usb0 type fuseblk (rw,nodev,noexec,noatime,sync,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096)
/dev/sdb1 on /media/usb1 type vfat (rw,nodev,noexec,noatime,nodiratime,sync,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdb2 on /media/usb2 type exfat (rw,nodev,noexec,noatime,nodiratime,sync,fmask=0000,dmask=0000,allow_utime=0022,iocharset=utf8,errors=remount-ro)
/dev/sdb2 on /mnt type exfat (rw,relatime,sync,fmask=0000,dmask=0000,allow_utime=0022,iocharset=utf8,errors=remount-ro)

blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="37E2-62C3" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="cd9edb0a-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="6a932c1f-7335-42d9-9351-1b1b2ca538d4" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="cd9edb0a-02"
/dev/sda1: LABEL="My Passport" BLOCK_SIZE="512" UUID="0E3EA2993EA2797B" TYPE="ntfs" PARTLABEL="My Passport" PARTUUID="5d37b6ae-c521-45a6-962b-5ec75fc4811e"
/dev/sdb1: LABEL_FATBOOT="EFI" LABEL="EFI" UUID="67E3-17ED" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="EFI System Partition" PARTUUID="64091b9a-5d22-47d1-9d1b-6d6756962783"
/dev/sdb2: LABEL="Seagate" UUID="5D89-9C91" BLOCK_SIZE="512" TYPE="exfat" PARTLABEL="Seagate" PARTUUID="14b6c811-a98b-48bd-88d7-8f4f1043fab1"

lsblk -aP -o NAME,FSTYPE,TYPE,SIZE,MOUNTPOINT,LABEL,UUID,PARTLABEL,PARTUUID,MODEL
NAME="sda" FSTYPE="" TYPE="disk" SIZE="1.8T" MOUNTPOINT="" LABEL="" UUID="" PARTLABEL="" PARTUUID="" MODEL="My_Passport_2626"
NAME="sda1" FSTYPE="ntfs" TYPE="part" SIZE="1.8T" MOUNTPOINT="/media/usb0" LABEL="My Passport" UUID="0E3EA2993EA2797B" PARTLABEL="My Passport" PARTUUID="5d37b6ae-c521-45a6-962b-5ec75fc4811e" MODEL=""
NAME="sdb" FSTYPE="" TYPE="disk" SIZE="465.8G" MOUNTPOINT="" LABEL="" UUID="" PARTLABEL="" PARTUUID="" MODEL="Expansion_SSD"
NAME="sdb2" FSTYPE="exfat" TYPE="part" SIZE="465.6G" MOUNTPOINT="/mnt" LABEL="Seagate" UUID="5D89-9C91" PARTLABEL="Seagate" PARTUUID="14b6c811-a98b-48bd-88d7-8f4f1043fab1" MODEL=""
NAME="sdb1" FSTYPE="vfat" TYPE="part" SIZE="200M" MOUNTPOINT="/media/usb1" LABEL="EFI" UUID="67E3-17ED" PARTLABEL="EFI System Partition" PARTUUID="64091b9a-5d22-47d1-9d1b-6d6756962783" MODEL=""
NAME="mmcblk0" FSTYPE="" TYPE="disk" SIZE="59.5G" MOUNTPOINT="" LABEL="" UUID="" PARTLABEL="" PARTUUID="" MODEL=""
NAME="mmcblk0p1" FSTYPE="vfat" TYPE="part" SIZE="256M" MOUNTPOINT="/boot" LABEL="boot" UUID="37E2-62C3" PARTLABEL="" PARTUUID="cd9edb0a-01" MODEL=""
NAME="mmcblk0p2" FSTYPE="ext4" TYPE="part" SIZE="59.3G" MOUNTPOINT="/" LABEL="rootfs" UUID="6a932c1f-7335-42d9-9351-1b1b2ca538d4" PARTLABEL="" PARTUUID="cd9edb0a-02" MODEL=""
```
Both drives have factory formatting neither is found with 'Find USB Device' with [84ed438bdd9ce21eec212c33e241c2c32621d378](https://github.com/iiab/iiab-admin-console/commit/84ed438bdd9ce21eec212c33e241c2c32621d378)

drszxn, 2022-09-23:
I will open a new issue "drag and drop not working". :-/

holta, 2022-09-23:
> I will open a new issue "drag and drop not working". :-/

Thanks @drszxn (I'm assuming this relates to #3375 ?)

drszxn, 2022-09-24:
@holta Yes. It looks like now the issue can be broken into three parts that are maybe bugs and feature requests:

1) there are no clear instructions on how to delete or remove menu items
2) menu items for installed content do not go away when you delete content
3) I am unable to drag/drop

holta, 2022-09-26:
@jvonau confirmed:

> Both drives have factory formatting neither is found with 'Find USB Device' with [84ed438bdd9ce21eec212c33e241c2c32621d378](https://github.com/iiab/iiab-admin-console/commit/84ed438bdd9ce21eec212c33e241c2c32621d378)

1) Can you suggest any targeted (very compact) patch to address the above, ideally?

2) Even if [rpi-clone](https://github.com/billw2/rpi-clone) isn't quite a drop-in replacement for [piclone_cmd](https://github.com/georgejhunt/piclone_cmd) (which we should investigate later, even if quite naturally neither are perfect!)

3) Let's give newbies/newcomers a fighting chance by further clarifying http://FAQ.IIAB.IO -> "43 How do I back up, shrink & copy IIAB microSD cards?" By/around early October if at all possible! 💾 -> 💾

jvonau, 2022-09-29:
Powered off above device, removed _**internal SDCard**_ and usb booted the cloned external SDCard just fine.
Summation:

- Some 'spinning' storage devices are labeled in a way that some linux tools don't like resulting in the not found issue A above
- Failure to unmount is likely desktop related.

holta, 2022-10-23:
FYI here are 2 other options mentioned by https://github.com/moodlebox/moodlebox/issues/259:

- [Image File Utilities](https://forums.raspberrypi.com/viewtopic.php?t=332000) like [image-backup](https://github.com/scruss/RonR-RaspberryPi-image-utils) (June 2020 repo archived by the owner)
- https://github.com/gitbls/sdm (Raspberry Pi SD Card Image Manager) appears actively maintained, well documented ([wiki](https://github.com/gitbls/sdm/wiki), 12m25s [YouTube](https://youtu.be/CpntmXK2wpA)), including many features similar to Raspberry Pi Imager or cloud-init

RECAP: The [rpi-clone](https://github.com/billw2/rpi-clone) bash script is the old standby (despite it being untouched since May 2020, it's still very popular after almost a decade of use!)

Whereas [piclone_cmd](https://github.com/nwright-mcc/piclone_cmd) appears unmaintained since Feb 2019.  Noting that @georgejhunt recompiled it for other platforms in March 2020: https://github.com/georgejhunt/piclone_cmd

holta, 2022-12-29:
Good enough for now.

If further refinements are required, they can be added during the IIAB 8.1 release cycle.

jvonau, 2022-12-30:
This is really a dead issue, should be closed.