iiab/iiab #3366: missing /etc/iiab/install-flags directory

tim-moody, 2022-09-16:
initial install with iiab-install
tried to add osm service with admin console

```
TASK [network : Create /etc/iiab/install-flags/iiab-network-complete on second pass of network role.] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Error, could not touch target: [Errno 2] No such file or directory: b'/etc/iiab/install-flags/iiab-network-complete'", "path": "/etc/iiab/install-flags/iiab-network-complete"}

```

holta, 2022-09-16:
@tim-moody is this regardless of your use of [local_vars_none.yml](https://github.com/iiab/iiab/blob/master/vars/local_vars_none.yml) that you mentioned you used when first discovering this?

Did you or did you not run `curl iiab.io/install.txt | bash` (or similar) ?

tim-moody, 2022-09-16:
As mentioned above I ran iiab-install

jvonau, 2022-09-16:
Suppose the issue is the non use of the curl routine where ['iiab'](https://github.com/iiab/iiab-factory/blob/master/iiab#L75) creates the directory.

jvonau, 2022-09-17:
```
Moving downloaded health.stackexchange.com_en_all_2022-05.zim to production
Re-indexing Kiwix Library
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Writing zim_versions_idx
Finished making Kiwix library.xml
Updating home page menues
Updating Home Page Menu
Updating kiwix menu items
Zim files and library.xml are out of sync. Please run iiab-make-kiwix-lib.
```

holta, 2022-09-22:
Resolved by:

- PR #3369

closed 2022-09-22 by holta