iiab/iiab #3367: OSM links incorrect on IIAB main page

holta, 2022-09-17:
@georgejhunt there's a sequencing/timing issue that occasionally leads to IIAB Maps being unreachable from IIAB's Main Page.

These broken links arose for me when:

- I did a fresh install of IIAB using the 1-line installer seen on https://download.iiab.io
- I did not reboot
- I used IIAB's Admin Console to install the entire ~81 GB OpenStreetMap vector map data for the entire world.
- I used Admin Console to install satellite photos (a 1000 x 1000 km square containing Senegal).

Every map link on IIAB's main page (French or English, I tried more than 6 different OSM menu-defs) then contained this broken link:

- http://box.lan/osm-vector-maps/satellite_z0-z9_2020.mbtiles/viewer

Rebooting did not help!  Changing browser did not help!  *(Obviously these map links really need to be...)*

- http://box.lan/osm-vector-maps/viewer

**What might be causing this ?**

ASIDE: A kludgy workaround (very ugly but it works until this bug is fixed!) is to tweak 2 lines in each OSM menu-def ([French one](https://github.com/iiab-share/js-menu-files/commit/647b09c48c4003a4e06d62e1bcf5fe6cdda34630) and [English one](https://github.com/iiab-share/js-menu-files/commit/6a7b191c81ee613ffed4d554f08df7ab96c72d66)).  Specifically here are the lines that can be hacked &mdash; until a legit fix is found:

- From `"intended_use": "map",` to `"intended_use": "webroot",`
- From `"start_url": "viewer",` to `"start_url": "osm-vector-maps/viewer/",`

CONTEXT: This was a 2022-09-15 fresh install of MEDIUM-sized install (minus KA Lite) on 64-bit Raspberry Pi OS with desktop on RPi 4.

RECAP:: All content was installed prior to rebooting IIAB.  Might this be causing OSM to accidentally generate the above broken links?

tim-moody, 2022-09-17:
This must be happening late in the process. I did the following and the link is http://192.168.3.95/osm-vector-maps/viewer :

On Ub22 VM install iiab with none local vars.
In Adm Cons config check maps, save and ICO

There are two links on the home page

Install IIAB Maps
World Maps & Satellite Photos - this is en-osm_viewer_v2 menu def

Note that no maps or satellite photos have been installed yet, only the base install

holta, 2022-09-17:
Huge thanks Tim for the data points yesterday & today.

I'll try to add yet more data points so we can isolate this bug.

tim-moody, 2022-09-17:
My VM does not have room for the entire world.
In Adm Cons select Central AM and install

The home page is unchanged

holta, 2022-09-17:
Many Thanks @tim-moody

That helps me further try to isolate it.

tim-moody, 2022-09-17:
In Adm Cons install satellite photo of 1000^2 over Central Am

Note that prior to refreshing Adm Cons this option said maps were not installed

After 15 minutes still running. (screen says can take 5 hours)

tim-moody, 2022-09-17:
2 1/2 hours and still on level 13

tim-moody, 2022-09-17:
finished in 2:35. home page is unchanged.

holta, 2022-09-17:
1) Very hard to reproduce what happened Thur/Friday.

2) Likewise I just tried a "ZERO-sized" IIAB install + Admin Console + OSM:

   - Ubuntu 22.10 (Kinetic Kudu pre-release)
   - "San Jose Small File for Testing" (OSM test vector data)
   - 100 x 100km satellite region over Quetico Provincial Park

3) So I'll try to buy a 1TB SSD, to clone the failing IIAB Senegal image from Thursday/Friday, to diagnose directly what appears to have gone wrong there.

tim-moody, 2022-09-17:
I think the key is that your link looks like a v1 link, not v2, which makes me think it had the wrong menu def. I wonder if doing the world map download either tried to generate a menu def or somehow added the wrong one to menu.json.

holta, 2022-09-18:
You're right.  Ugh this was entirely fault.  Apologies I forgot I hand copied a couple items from their earlier Cameroon machine, in the mad rush to build up this Terabyte IIAB in a single day before their flight.

closed 2022-09-18 by holta