iiab/iiab #3375: Unable to delete Menu Items

drszxn, 2022-09-23:
### Expected behavior and actual behavior:
Expected: when you remove content, the menu item is deleted
Actual Behavior: When you remove the content, the menu items does not delete

### Steps to reproduce the problem:
1. Log into /admin
2. Install Content Tab
3. Under the "Installation Tasks" menu on left in the green, click on "Manage Content".
4. Under "Installed ____ Files, check the item to remove, and then press the darker blue right side menu "Remove Selected Content".
5. Job finishes and the item disappears from the list of the installed files.
6. Go to "Content Menus" and under "Actions" press "Refresh Lists", "Sync Menu Defs", and "Update Home Menu" 

Expectation: delete items disappear from the list. 

Actual result: delete items remain

### IIAB version, OS version, hardware details:
Fresh install of iiab-8.0preview3-220820-SMALL-raspios64-lite-g2543831e.img using Raspberry Pi Imager
Windows 10
Raspberry Pi 4b 8G with a 64GB microSD
Chrome Version 105.0.5195.126 (Official Build) (64-bit)

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/TssobM

If there is something I am missing, please let me know. I wasn't able to find a strait forward how-to and I am I have reinstalled a few times, just to make sure.

drszxn, 2022-09-23:
I deleted it...
[image]

It won't go away and there is no "delete" option.
[image]

jvonau, 2022-09-23:
Just drag the menu item from the "current menu" on the left to "available menu" on the right and click 'update home menu' on the left side, then check the /home page.

jvonau, 2022-09-23:
@holta Filesystem failed to expand
=IIAB==========================================================================
COMMAND: /usr/bin/df -h    # Disk usage
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       6.7G  5.7G  877M  87% /
=IIAB==========================================================================
COMMAND: /usr/bin/df -ah    # Disk usage detail
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       6.7G  5.7G  877M  87% /
=IIAB==========================================================================
COMMAND: /usr/bin/lsblk    # Partition mount points
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0   58G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 57.7G  0 part /

holta, 2022-09-23:
> If there is something I am missing, please let me know. I wasn't able to find a strait forward how-to and I have reinstalled a few times, just to make sure.

@drszxn thanks for writing!

1) You aren't the first to encounter this confusing UX:

- #1873

   RECAP: We definitely need to better explain deletion 👍 

   And possibly in future change the graphical interface to make this more intuitive.

2) We're working hard on another serious bug you also encountered:

- #3362

_Keep in touch please, as we work on trying to solve both above issues hopefully soon_ 👍

tim-moody, 2022-09-23:
From the help:

Current Item List
Click this tab to change the Menu Items included in the menu. You will see two lists of links.

On the left are the links in the menu you are editing and on the right are all of the available Menu Item Definitions.

To change the order of links drag a link up or down the list on the left.
To add a link drag it from the right to the left.
**To remove a link drag it from the left to the right.**

tim-moody, 2022-09-23:
agree a nice enhancement would be to remove a menu item when the content is removed

drszxn, 2022-09-23:
I tried to drag/drop and that function was not working through the same browser. I will file another issue case for that. 

Thank you all for the very quick response!

jvonau, 2022-09-23:
> > If there is something I am missing, please let me know. I wasn't able to find a strait forward how-to and I have reinstalled a few times, just to make sure.
>     2. We're working hard on another serious bug you also encountered:
> 
> 
>     * [Admin Console > Install Content > Clone IIAB (1) does not find USB HDD's (2) often fails to repartition/mount/unmount in the 1st minute [should rpi-clone be considered instead of piclone_cmd?] #3362](https://github.com/iiab/iiab/issues/3362)
> 

@holta Triage is wrong, issue S/B
 * rootfs can fail to fully expand: race condition betw iiab-expand-rootfs & systemd-fsck ? (journalctl confusingly/regularly portrays 2 simultaneous boots due to lack of RTC?) #3325 

> 
>└─mmcblk0p2 179:2 0 57.7G 0 part /
>/dev/root 6.7G 5.7G 877M 87% /

Disk is showing full usage of 57.7G while the filesystem is only showing 6.7 G

@drszxn Could you double check and confirm the available space on the device by going to "Utilities" -> "Display System Storage", should be able to view the available space. Should have over 50G of 'Size'

[image]

holta, 2022-09-23:
> Disk is showing full usage of 57.7G while the filesystem is only showing 6.7 G
>
> @drszxn Could you double check and confirm the available space on the device by going to "Utilities" -> "Display System Storage", should be able to view the available space. Should have over 50G of 'Size'

Thanks @drszxn and @jvonau, incredibly useful if we can shed light here together!

jvonau, 2022-09-23:
> I tried to drag/drop and that function was not working through the same browser. I will file another issue case for that.
> 
> Thank you all for the very quick response!

Be good to know which browsers are tried.

drszxn, 2022-09-24:
@jvonau 

[image]

drszxn, 2022-09-24:
It's a 64GB microSD but the image I used doesn't take up the whole card.

holta, 2022-09-24:
In response to @drszxn's [summary](https://github.com/iiab/iiab/issues/3362#issuecomment-1257066955) :

> * there are no clear instructions on how to delete or remove menu items

Certainly there's a [help screen](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/ContentMenus.rst#edit-content-menus) that says (above halfway down the page) "To remove a link drag it from the left to the right."

My opinion is that the struggle to make this UX/UI more discoverable + intuitive (1) won't be easy, (2) but is indeed critically important!

> * menu items for installed content do not go away when you delete content

Getting this right will be (wicked) hard, but we're trying!  Greatly simplifying these UI/UX concepts + affordances is probably part of the secret.  Alongside a growing collection of YouTube-like videos that almost anyone can follow (that's the goal!)

If you or anyone have specific + concrete suggestions, please see:

- iiab/iiab-admin-console#510

>   --> 3) I am unable to drag/drop  <-- This seemed to all the sudden started working so never mind

Thanks for persisting, Great!

holta, 2022-10-06:
@drszxn FYI: if you do a fresh install of Internet-in-a-Box (onto a fresh OS) using https://download.iiab.io — the ability to delete content (via Admin Console) should now be much improved:

- PR iiab/iiab-admin-console#513

jvonau, 2022-10-07:
Shame one has to completely start over to pickup a single incremental change. @drszxn what is the intended purpose of the second WiFi device?

tim-moody, 2022-10-07:
>Shame one has to completely start over to pickup a single incremental change

Not sure what you mean. Git pull and install takes a few minutes. Only things that have changed will get installed.

jvonau, 2022-10-07:
> > Shame one has to completely start over to pickup a single incremental change
> 
> Not sure what you mean. Git pull and install takes a few minutes. Only things that have changed will get installed.

Sorry @tim-moody I was referring to @holta''s reply:
> @drszxn FYI: if you do a fresh install of Internet-in-a-Box (onto a fresh OS) using https://download.iiab.io — the ability to delete content (via Admin Console) should now be much improved: