iiab/iiab #3377: iiab working on WSL 0.67.6

turingtrust, 2022-09-23:
Microsoft recently added Systemd support to WSL:
[https://devblogs.microsoft.com/commandline/systemd-support-is-now-available-in-wsl/](url)

I have successfully installed iiab in WSL with systemd enabled on Ubuntu 20.04 LTS on a Windows 11 host.

The install failed when trying to get the plocate package, but I fixed it by editing /opt/iiab/iiab/roles/2-common/tasks/packages.yaml and renaming plocate to mlocate, the install then succeeded.

I can now access the iiab webpage from the host with the url: **localhost/**

Could other people test this please? I have only tried this with the minimal install option.

_This is my first time writing an issue on github so forgive me if the formatting is incorrect._

holta, 2022-09-23:
Hi @turingtrust, thanks for the report!

1) Can you run `sudo iiab-diagnostics` ?

   After you've run that, it should generate a `sprunge.us` URL.

   Please post that URL here if possible, Thank You.

2) Please very strongly consider Ubuntu 22.04 instead of 20.04 if at all possible?

   So that you're on a path for much better IIAB support in coming years.  In this example, Ubuntu 20.04 makes certain assumptions from the 2019 era.  To keep focused, we do now need to start looking to the future e.g. OS's like Debian 11 Bullseye, Raspberry Pi OS "Bullseye", Mint 21, Ubuntu 22.04.1

turingtrust, 2022-09-23:
> Hi @turingtrust, thanks for the report!
>
>     1. Can you run `sudo iiab-diagnostics` ?
>        After you've run that, it should generate a `sprunge.us` URL.
>        Please post that URL here if [...]

[http://sprunge.us/i12evA](url)

I switched to 20.04 because apt upgrade was breaking when running the iiab setup. Upgrading fully before enabling systemd support might fix that though.

andrewclark5, 2022-09-23:
Here is the Author's full install test procedure:-

https://docs.google.com/document/d/1IV9vNqzMuqvLwtY78O2lUkYnwt5hnNHOassQF2XCJUQ/edit?usp=sharing

holta, 2022-09-23:
> Here is the Author's full install test procedure:-
> 
> https://docs.google.com/document/d/1IV9vNqzMuqvLwtY78O2lUkYnwt5hnNHOassQF2XCJUQ/edit?usp=sharing

Amazing.  Many thanks @andrewclark5 & @turingtrust !

turingtrust, 2022-09-23:
> > I switched to 20.04 because apt upgrade was breaking when running the iiab setup.
> 
> Share details or screenshots if at all possible? Many Thanks !
> 
> ( Ubuntu 22.04 should be very solid now, more than 5 months after its release. With Moodle support coming in just over a month from now, should you later need that. Sugarizer requires a manual install of MongoDB for now, should you need that for younger children. We hope that is cleanly automated well before end of year. Background notes here: https://github.com/iiab/iiab/wiki/IIAB-Platforms )
> 
>     * [Test Moodle 4.0 ETA 2022-04-19, including on Ubuntu 22.04 [with PHP 8.1] #3182](https://github.com/iiab/iiab/issues/3182)
> 
>     * [MongoDB 4.4, 5.0 & 6.0RC1 don't run on Ubuntu 22.04 (preventing Sugarizer install) [missing apt package libssl1.1 -> libssl3 ?] #3190](https://github.com/iiab/iiab/issues/3190)

It turns out I wasn't actually running 22.04 for the first attempt. the error I got during apt upgrade can be fixed by upgrading before enabling systemd support.

I've just tried with 22.04 now and get this error:
[image]

holta, 2022-09-23:
Strange!

Thanks @turingtrust can you re-run `sudo iiab` (repeatedly if nec) to confirm the `Unable to start service ssh` error is repeatable?

If so, can you post the sprunge.us URL output from `iiab-diagnostics` and any other relevant diagnostics?  Specifically, what is the output from...?

```
sudo systemctl status ssh
```

turingtrust, 2022-09-23:
> Strange!
>
> Thanks @turingtrust can you re-run `sudo iiab` (repeatedly if nec) to confirm the `Unable to start service ssh` error is repeatable?
>
> If so, can you post the sprunge.us URL output from [...]

running iiab repeatedly doesn't change anything.

http://sprunge.us/1rysmr

[image]

turingtrust, 2022-09-23:
The install worked after generating ssh keys.

http://sprunge.us/kQXmE1

[image]

holta, 2022-09-23:
> I think I fixed it by running ssh-keygen -A

Amazing!

Any idea why this was necessary?

Any chance Microsoft's WSL VM (0.67.6) is still teething/converging ?

turingtrust, 2022-09-23:
> 

I just tried the solution from here: [https://www.garron.me/en/linux/sshd-no-hostkeys-available-exiting.html](url)

The admin console is broken now:
[image]

It worked fine on 20.04

holta, 2022-09-23:
Pop-up shows:

> Connection to Uwsgi Server failed.

http://sprunge.us/kQXmE1?bash shows:

> 1629 Sep 23 18:42:59 DESKTOP-9687BEA systemd[1]: iiab-cmdsrv.service: Can't open PID file /run/iiab-cmdsrv.pid (yet?) after start: Operation not permitted
1630 Sep 23 18:42:59 DESKTOP-9687BEA systemd[1]: iiab-cmdsrv.service: Supervising process 527 which is not our child. We'll most likely not notice when it exits.
...
1674 Sep 23 18:43:03 DESKTOP-9687BEA IIAB-CMDSRV[527]: IIAB-CMDSRV : Job Minder Thread Started
1675 Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.313860] WSL (2) ERROR: InitCreateSessionLeader:1072: systemd: failed to start within 10000
1676 Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.313865] ms
1677 Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.321701] WSL (2): systemd: creating login session for turing
1678 Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.348291] systemd-journald[59]: File /var/log/journal/05a72943a58b49bf96254d5b27d28666/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
1679 Sep 23 18:43:16 DESKTOP-9687BEA systemd[1]: dmesg.service: Deactivated successfully.

@turingtrust I assume that `systemctl status iiab-cmdsrv` and `systemctl restart iiab-cmdsrv` don't quite solve this?

@tim-moody are you perhaps able to reproduce this Admin Console problem on Windows 11 + very latest systemd-enabled WSL (i.e. Ubuntu 22.04) if you have time this weekend?

I do not have access to Windows 11 unfortunately.  (Any chance it's a WSL teething issue with its brand new systemd release?)

turingtrust, 2022-09-23:
> @turingtrust I assume that `sudo systemctl status iiab-cmdsrv` and `sudo systemctl restart iiab-cmdsrv` don't quite solve this?

`turing@DESKTOP-9687BEA:~$ sudo systemctl status iiab-cmdsrv
Failed to get properties: Transport endpoint is not connected`

jvonau, 2022-09-23:
> Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.313860] WSL (2) ERROR: InitCreateSessionLeader:1072: systemd: failed to start within 10000
Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.313865] ms
Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.321701] WSL (2): systemd: creating login session for turing
Sep 23 18:43:08 DESKTOP-9687BEA kernel: [   13.348291] systemd-journald[59]: File /var/log/journal/05a72943a58b49bf96254d5b27d28666/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.

Seems like WSL teething pain, along with the need to generating ssh keys, is this instance is a clone of another configured VM by any chance?

turingtrust, 2022-09-27:
It might be best to leave this until microsoft moves systemd support out of beta.

I also tried debian which didn't work.

Note: if you want to test this, you should be able to install windows 11 on any machine that supports windows 10.

If you make a bootable usb with the windows 11 iso using [Rufus](https://rufus.ie/en/) it will give you the option to bypass the TPM requirement.
[image]

jvonau, 2022-10-04:
Notes:
1403 2022-09-23 18:32:53,167 p=11326 u=root n=ansible | TASK [network : Wired enslaving - Assigns lan_list_results to br0 as wired slaves if present] **************************
1404 2022-09-23 18:32:53,434 p=11326 u=root n=ansible | changed: [127.0.0.1] => (item=bond0)
1405 2022-09-23 18:32:53,678 p=11326 u=root n=ansible | changed: [127.0.0.1] => (item=bonding_masters)
1406 2022-09-23 18:32:53,925 p=11326 u=root n=ansible | changed: [127.0.0.1] => (item=dummy0)
1407 2022-09-23 18:32:54,167 p=11326 u=root n=ansible | changed: [127.0.0.1] => (item=sit0)

holta, 2022-10-11:
Anyone have time to test IIAB on WSL 0.70.0 in coming days?

https://github.com/microsoft/WSL/releases/tag/0.70.0

holta, 2022-10-19:
WSL 0.70.4 released:

https://github.com/microsoft/WSL/releases/tag/0.70.4

holta, 2022-11-15:
WSL 1.0.0 released:
https://github.com/microsoft/WSL/releases/tag/1.0.0

Nearly equivalent to the 11-day-old:
https://github.com/microsoft/WSL/releases/tag/0.70.8

turingtrust, 2022-11-21:
> WSL 1.0.0 released: https://github.com/microsoft/WSL/releases/tag/1.0.0
> 
> Nearly equivalent to the 11-day-old: https://github.com/microsoft/WSL/releases/tag/0.70.8
> 
> Related:
> 
>     * PR [Wsl [Windows Subsystem for Linux] #3004](https://github.com/iiab/iiab/pull/3004)

Just tested wsl 1.0.0 on Ubuntu 22.04.1

Had to create /etc/network/interfaces.d for install to compete.

Everything appears to work now.

holta, 2022-11-21:
@turingtrust if possible can you post the pastebin URL, after you run `sudo iiab-diagnostics` ?

turingtrust, 2022-11-21:
> @turingtrust if possible can you post the pastebin URL, after you run `sudo iiab-diagnostics` ?

http://sprunge.us/NDwCnZ

holta, 2022-11-21:
Thanks!

> "Destination directory /etc/network/interfaces.d does not exist"

...excerpted from Line 1503 at the bottom here:

```
TASK [network : Supply resolvconf.conf] ********************************************************************************
changed: [127.0.0.1]
TASK [network : Supply dhcpcd.conf] ************************************************************************************
skipping: [127.0.0.1]
TASK [network : Copy the bridge script] ********************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "checksum": "6d8982e788d2bf2785429e2067fe281c9a77947b", "msg": "Destination directory /etc/network/interfaces.d does not exist"}
```

jvonau, 2022-11-21:
>Had to create /etc/network/interfaces.d for install to compete.

I doubt it, the extract below show the directory as not being created.. I'll give you `sudo iiab` would of completed without completing the network, that is part of the install process failsafe to continue on, to be able to install admin_console.
>=IIAB==========================================================================
FILE DOES NOT EXIST: /etc/network/interfaces
=IIAB==========================================================================
FILE DOES NOT EXIST: /etc/hostapd/hostapd.conf
=IIAB==========================================================================
FILE DOES NOT EXIST: /etc/wpa_supplicant/wpa_supplicant.conf

>3. CONTENT OF DIRECTORIES (1-LEVEL DEEP)
=IIAB==========================================================================
DIRECTORY DOES NOT EXIST: /etc/network/interfaces.d
=IIAB==========================================================================
DIRECTORY /etc/systemd/network FILES WILL FOLLOW...IF THEY EXIST
=IIAB==========================================================================
DIRECTORY DOES NOT EXIST: /etc/NetworkManager/system-connections
=IIAB==========================================================================
DIRECTORY /etc/netplan FILES WILL FOLLOW...IF THEY EXIST

The below are only present with WSL and previous noted above as the trigger to install br0
>=IIAB==========================================================================
COMMAND: /usr/sbin/ip addr    # Network interfaces
2: bond0: <BROADCAST,MULTICAST,MASTER> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether ee:62:a6:bc:5c:e1 brd ff:ff:ff:ff:ff:ff
3: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 96:b8:4f:56:6f:30 brd ff:ff:ff:ff:ff:ff
5: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0

jvonau, 2022-11-21:
I suggest that 'network_enable: False' be added to local_vars.yml to avoid networking additions that wsl doesn't support.

holta, 2022-11-22:
> I suggest that 'network_enable: False' be added to local_vars.yml to avoid networking additions that wsl doesn't support.

Thanks @jvonau.  I think you mean "enabled" with a "d" — as follows:

```
network_enabled: False
```

holta, 2022-11-22:
Today's WSL 1.0.1 offers a few bug fixes:

https://github.com/microsoft/WSL/releases/tag/1.0.1

jvonau, 2022-11-22:
> > I suggest that 'network_enable: False' be added to local_vars.yml to avoid networking additions that wsl doesn't support.
> 
> Thanks @jvonau. I think you mean "enabled" with a "d" — as follows:
> 
> ```
> network_enabled: False
> ```

I stand corrected, should include 'd'

holta, 2022-12-01:
This pre-release adds a `--pre-release` option to `wsl.exe --update` among other things:

https://github.com/microsoft/WSL/releases/tag/1.0.3

holta, 2023-01-18:
Just FYI a new WSL pre-release:

https://github.com/microsoft/WSL/releases/tag/1.1.0

holta, 2023-04-04:
WSL 1.2.0 just released:

https://github.com/microsoft/WSL/releases/tag/1.2.0

holta, 2023-09-18:
WSL is right now jumping from 1.3.x to pre-release 2.0.0 :

- https://github.com/microsoft/WSL/releases/tag/2.0.0