iiab/iiab #3378: Filesystem failed to expand

jvonau, 2022-09-24:
@holta Filesystem failed to expand
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       6.7G  5.7G  877M  87% /      ZIMs: 3 OER2Go: 1 Apps2B: 0

=IIAB==========================================================================
COMMAND: /usr/bin/df -ah    # Disk usage detail
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       6.7G  5.7G  877M  87% /
=IIAB==========================================================================
COMMAND: /usr/bin/lsblk    # Partition mount points
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0   58G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 57.7G  0 part /

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3375#issuecomment-1256016243_

holta, 2022-09-24:
Perhaps this is to be expected — if @drszxn had not yet rebooted?

CLARIF: After PR #3337 is merged, a manual reboot will no longer be required to expand the root filesystem.

Related:

- PR #3336
- PR #3337

jvonau, 2022-09-24:
Post writing http://download.iiab.io/8.0/iiab-8.0preview3-220820-SMALL-raspios64-lite-g2543831e.img.zip in host machine:
```
lsblk
sdb           8:16   1  59.5G  0 disk 
├─sdb1        8:17   1   256M  0 part /media/jvonau/boot
└─sdb2        8:18   1   6.8G  0 part /media/jvonau/rootfs

df
/dev/sdb2         6969504   5054960    1755796  75% /media/jvonau/rootfs
/dev/sdb1          258095     30943     227153  12% /media/jvonau/boot
```
Firstboot in device: 
```
iiab-admin@box:~ $ journalctl -u iiab-expand-rootfs.service --no-pager
-- Journal begins at Mon 2022-04-04 10:41:41 EDT, ends at Sat 2022-09-24 01:09:09 EDT. --
Aug 20 10:20:45 box systemd[1]: Starting Root Filesystem Auto-Expander...
Aug 20 10:20:45 box iiab-expand-rootfs[403]: + '[' -f /.expand-rootfs ']'
Aug 20 10:20:45 box iiab-expand-rootfs[403]: + '[' -f /.resize-rootfs ']'
Aug 20 10:20:45 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Aug 20 10:20:45 box systemd[1]: Finished Root Filesystem Auto-Expander.
-- Boot 80e8ca626f6e485681009957f0bb7830 --
Aug 20 10:24:24 box systemd[1]: Starting Root Filesystem Auto-Expander...
Aug 20 10:24:24 box iiab-expand-rootfs[447]: + '[' -f /.expand-rootfs ']'
Aug 20 10:24:24 box iiab-expand-rootfs[447]: + echo '/usr/sbin/iiab-expand-rootfs: Expanding rootfs partition'
Aug 20 10:24:24 box iiab-expand-rootfs[447]: /usr/sbin/iiab-expand-rootfs: Expanding rootfs partition
Aug 20 10:24:24 box iiab-expand-rootfs[447]: + '[' -x /usr/bin/raspi-config ']'
Aug 20 10:24:24 box iiab-expand-rootfs[447]: + raspi-config --expand-rootfs
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Welcome to fdisk (util-linux 2.36.1).
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Changes will remain in memory only, until you decide to write them.
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Be careful before using the write command.
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Command (m for help): Disk /dev/mmcblk0: 59.52 GiB, 63910707200 bytes, 124825600 sectors
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Units: sectors of 1 * 512 = 512 bytes
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Sector size (logical/physical): 512 bytes / 512 bytes
Aug 20 10:24:26 box iiab-expand-rootfs[613]: I/O size (minimum/optimal): 512 bytes / 512 bytes
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Disklabel type: dos
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Disk identifier: 0x7790a859
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Device         Boot  Start      End  Sectors  Size Id Type
Aug 20 10:24:26 box iiab-expand-rootfs[613]: /dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
Aug 20 10:24:26 box iiab-expand-rootfs[613]: /dev/mmcblk0p2      532480 14768911 14236432  6.8G 83 Linux
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Command (m for help): Partition number (1,2, default 2):
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Partition 2 has been deleted.
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Command (m for help): Partition type
Aug 20 10:24:26 box iiab-expand-rootfs[613]:    p   primary (1 primary, 0 extended, 3 free)
Aug 20 10:24:26 box iiab-expand-rootfs[613]:    e   extended (container for logical partitions)
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Select (default p): Partition number (2-4, default 2): First sector (2048-124825599, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (532480-124825599, default 124825599):
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Created a new partition 2 of type 'Linux' and of size 59.3 GiB.
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Partition #2 contains a ext4 signature.
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Command (m for help):
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Disk /dev/mmcblk0: 59.52 GiB, 63910707200 bytes, 124825600 sectors
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Units: sectors of 1 * 512 = 512 bytes
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Sector size (logical/physical): 512 bytes / 512 bytes
Aug 20 10:24:26 box iiab-expand-rootfs[613]: I/O size (minimum/optimal): 512 bytes / 512 bytes
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Disklabel type: dos
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Disk identifier: 0x7790a859
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Device         Boot  Start       End   Sectors  Size Id Type
Aug 20 10:24:26 box iiab-expand-rootfs[613]: /dev/mmcblk0p1        8192    532479    524288  256M  c W95 FAT32 (LBA)
Aug 20 10:24:26 box iiab-expand-rootfs[613]: /dev/mmcblk0p2      532480 124825599 124293120 59.3G 83 Linux
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Command (m for help): The partition table has been altered.
Aug 20 10:24:26 box iiab-expand-rootfs[613]: Syncing disks.
Aug 20 10:24:28 box iiab-expand-rootfs[475]: Please reboot
Aug 20 10:24:28 box iiab-expand-rootfs[447]: + rm -f /.expand-rootfs /.resize-rootfs
Aug 20 10:24:28 box systemd[1]: iiab-expand-rootfs.service: Succeeded.
Aug 20 10:24:28 box systemd[1]: Finished Root Filesystem Auto-Expander.

lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 59.5G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 59.3G  0 part /

[... 77 lines cut ...]
Aug 20 09:33:00 raspberrypi resize2fs_once[501]: Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
Aug 20 09:33:00 raspberrypi resize2fs_once[501]: old_desc_blocks = 1, new_desc_blocks = 4
Aug 20 09:33:00 raspberrypi resize2fs_once[501]: The filesystem on /dev/mmcblk0p2 is now 15563776 (4k) blocks long.
[... cut ...]
Sep 24 01:27:52 box systemd[1]: Started LSB: Resize the root filesystem to fill partition.

iiab-admin@box:~ $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       61139384 5098000  55333352   9% /
devtmpfs         1777020       0   1777020   0% /dev
tmpfs            1942748       0   1942748   0% /dev/shm
tmpfs             777100     944    776156   1% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/mmcblk0p1    258095   30943    227153  12% /boot
tmpfs             388548       0    388548   0% /run/user/1000
```

jvonau, 2022-09-24:
Think the reboot could be avoided

holta, 2022-09-24:
> Think the reboot could be avoided just calling
> 
> resize2fs "replace with variable" update-rc.d resize2fs_once remove rm /etc/init.d/resize2fs_once
> 
> or borrow parts (fdisk) of the do_expand_rootfs() function from raspi-config

Apologies I'm confused by the above:

Isn't this ticket mainly discussing RasPiOS (where PR #3337 as it stands should handle rapid auto-reboot very cleanly) ?

Or other OS's with resize2fs (where reboot is apparently not required) ?

jvonau, 2022-09-24:
Neither, I spotted that the created image in the report didn't have a resized file system but jumped to just running the script in an effort to suppress the reboot.. I'll circle back with just reboot after the initial login on firstboot cause I'm wondering if using any of the pre-seed values in Rpi-imager changes the sequence of events given Rpi-imager creates 'systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target' in the /boot/cmdline.txt file and /boot/firstrun.sh is created when you do on the newly written iiab image.

holta, 2022-09-24:
To reduce complexity, PR #3337 (or something close to it) needs be merged and broadly tested shortly, after much progress over 7 weeks now:

Certainly #3375 (and this ticket) are about RasPiOS.  As such, is there solid evidence of a bug in `raspi-config --expand-rootfs` ?

How do we best identify conditions to reproduce that bug if so, as a public service for others?

(Or conversely, if there's solid but complicated evidence &mdash; let's report that upstream as clearly as we can in coming days &mdash; so Raspberry Pi Foundation people can hopefully investigate!)

jvonau, 2022-09-24:
No a simple reboot works on the produced images but doesn't give any user feedback that it is running. I'm looking towards the future with what was released on the current upstream images where there is feedback that the resize is in progress.

jvonau, 2025-04-06:
There are many examples in other issues of users not have a resized filesystem for a number of different reasons.

closed 2025-04-06 by jvonau