iiab/iiab #3385: iiab-network "runs but does not add Ethernet-connected WiFi AP hotspot" on Ubuntu 22.04 (Server not Desktop) NUC

holta, 2022-09-29:
@rickgurmet reports that he's tried running `sudo iiab-network` in many different ways.  It is not working as expected.  I don't know if this is a bug or a question so for now this ticket is tagged as both, until we figure it out.

He wants the WiFi AP (Access Point) to handle more WiFi client devices than just the very meager (maximum of 12?) offered by the NUC's internal WiFi AP.

The WiFi AP device (not much of a router) has only 1 Ethernet port, as does the IIAB/NUC &mdash; so there's only 1 way to connect the Ethernet cable between the 2 devices &mdash; which greatly simplifies things thankfully.

He's tried running `sudo iiab-network` both with the Ethernet cable plugged into the NUC and also with the NUC's Ethernet port completely empty.

Then he runs `brctl show` and the Ethernet interface (`eno1`) is unfortunately not added to the "bridge" (for LAN-side WiFi AP hotspots).

On the side he discovered that he can temporarily add `eno1` under the bridge by running `brctl addif br0 eno1` &mdash; but unforunately this setting does not survive reboots:

[image]

/opt/iiab/iiab/iiab-network.log (3.2 MB, 53108 lines) is attached here, hoping this might help us all understand better: 
[iiab-network.log](https://github.com/iiab/iiab/files/9678258/iiab-network.log)

BACKGROUND: He installed the latest IIAB onto this NUC on Sept ~16, and more recently `git pull` was run to update `/opt/iiab/iiab` to the latest code there just in case.

iiab-diagnostics: http://sprunge.us/3fyNhH?bash

Quick Questions for @rickgurmet ...that might possibly shed more light:

- Is the WiFi AP device running DHCP yes or no?
- What's the make and model of this WiFi AP device?

_Thanks @rickgurmet if you can explain more detail other things you've tried, to help everyone try to resolve this efficiently in coming days._ :+1:

jvonau, 2022-09-29:
same issue as https://github.com/iiab/iiab/issues/3330#issuecomment-1208130277

> 1123 -IIAB--------------------------------------------------------------------------
1124 -rw-r--r-- 1 root root 154 Sep 16 10:57 /etc/netplan/00-installer-config.yaml
1125                         ...ITS LAST 100 LINES FOLLOW...
1126 
1127 # This is the network config written by 'subiquity'
1128 network:
1129   ethernets:
1130     eno1:
1131       dhcp4: true
1132     enx0c37961e7ce1:
1133       dhcp4: true
1134   version: 2

You can \# out lines eno1:  dhcp4: true or move 00-installer-config.yaml out of the way and reboot.

rickgurmet, 2022-09-30:
@holta @jvonau Thank you, it worked. 
Just checked /etc/systemd/network/IIAB-Slave-eno1.network and the bridge br0 had interface eno1 (the ethernet port) but still, it was not working. After referring to issue [#3330](https://github.com/iiab/iiab/issues/3330#issuecomment-1208130277) , I moved the file /etc/netplan/00-installer-config.yaml to some other location as stated [here](https://github.com/iiab/iiab/issues/3330#issuecomment-1208130277) and it worked. 
I hope @vinaybissa didn't face any issue after moving the file to some other location.

holta, 2022-09-30:
1. Thanks @rickgurmet who explained that he got this working _without running any (additional) commands:_

   > I just did a reboot and checked if br0 has an interface eno1 and it did, and I just plugged in to the ethernet port and it worked.

   > I even rebooted several times to check if it works and it still did

2. Does that mean that his yesterday running of `sudo iiab-network` and/or `brctl addif br0 eno1` were... _also necessary ?_

rickgurmet, 2022-09-30:
An alternative and a good way that @jvonau suggested [here](https://github.com/iiab/iiab/issues/3385#issuecomment-1262734247) was to comment out the following in my case, where eno1 was my ethernet port name in 00-installer-config.yaml.

 `eno1: 

 dhcp4: true`

Now the advantage of this was, if you have an external ethernet adapter plugged in to your usb port, then you can use the external ethernet adapter as dhcp (to connect to the internet) while using the ethernet port to access the IIAB.
In my case it was easy for me to update the server or download contents from the web.

holta, 2022-09-30:
Thanks @rickgurmet for having confirmed + explained.

> external ethernet adapter plugged in to your usb

I see you've added new hardware (a USB-to-Ethernet dongle) to your IIAB/NUC.

1. So a key question is "encoding your intentions" as to which of the 2 Ethernets should be...
   - WAN-side (Internet)

   ...and which of the 2 Ethernets should be...
   - LAN-side (WiFi AP/hotspot)

   IIAB software cannot possibly guess your intentions, obviously :smiley:

2. One way to "encode your intentions" is specified in **Common Customizations > (4)** here:

   https://github.com/iiab/iiab/wiki/IIAB-Networking#common-customizations

   i.e. the "official" way is to put a line like the following in your [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?) :

   ```
   reserved_device: eth0
   ```
   _Above is just an example if you wanted Ethernet `eth0` to be WAN-side [Internet] instead of LAN-side [WiFi AP/hotspot]._ 

   _Then as usual, you'd to run `sudo iiab-network` to make it all happen (i.e. put your networking into place!)_

jvonau, 2022-10-02:
What is the 'bug' notation in relation to?

holta, 2022-10-02:
At the moment it's too confusing for new people to figure out whether they need one-or-both approaches:

A) Adding `reserved_device: XXXX` in local_vars.yml and then running `sudo iiab-network`
B) Hand editing `/etc/netplan/00-installer-config.yaml`

Should both be recommended and in which precise circumstances?

PS Possibly this does not currently affect Ubuntu on Raspberry Pi hardware ?

jvonau, 2022-10-02:
Gleaned from http://sprunge.us/5oa3JZ, seems like documentation fodder
```
-IIAB--------------------------------------------------------------------------
-rw-r--r-- 1 root root 116 Sep 27 13:44 /etc/systemd/network/IIAB-Slave-eno1.network
                        ...ITS LAST 100 LINES FOLLOW...

# /etc/systemd/network/IIAB-Slave-eno1.network
[Match]
Name=eno1

[Link]
RequiredForOnline=no

[Network]
Bridge=br0

-IIAB--------------------------------------------------------------------------
-rw-r--r-- 1 root root 138 Sep 17 16:52 /etc/systemd/network/IIAB-Slave-enx0c37961e7afc.network
                        ...ITS LAST 100 LINES FOLLOW...

# /etc/systemd/network/IIAB-Slave-enx0c37961e7afc.network
[Match]
Name=enx0c37961e7afc

[Link]
RequiredForOnline=no

[Network]
Bridge=br0

-IIAB--------------------------------------------------------------------------
-rw-r--r-- 1 root root 138 Sep 25 05:58 /etc/systemd/network/IIAB-Slave-enx0c37961e7ce1.network
                        ...ITS LAST 100 LINES FOLLOW...

# /etc/systemd/network/IIAB-Slave-enx0c37961e7ce1.network
[Match]
Name=enx0c37961e7ce1

[Link]
RequiredForOnline=no

[Network]
Bridge=br0

-IIAB--------------------------------------------------------------------------
-rw-r--r-- 1 root root 116 Sep 27 13:44 /etc/systemd/network/IIAB-Slave-wlp0s20f3.network
                        ...ITS LAST 100 LINES FOLLOW...

# /etc/systemd/network/IIAB-Slave-eno1.network
[Match]
Name=eno1

[Link]
RequiredForOnline=no

[Network]
Bridge=br0

=IIAB==========================================================================
DIRECTORY DOES NOT EXIST: /etc/NetworkManager/system-connections
=IIAB==========================================================================
DIRECTORY /etc/netplan FILES WILL FOLLOW...IF THEY EXIST
-IIAB--------------------------------------------------------------------------
-rw------- 1 root root 86 Sep 16 10:57 /etc/netplan/00-installer-config-wifi.yaml
                        ...ITS LAST 100 LINES FOLLOW...

# This is the network config written by 'subiquity'
network:
  version: 2
  wifis: {}

-IIAB--------------------------------------------------------------------------
-rw-r--r-- 1 root root 156 Sep 30 16:23 /etc/netplan/00-installer-config.yaml
                        ...ITS LAST 100 LINES FOLLOW...

# This is the network config written by 'subiquity'
network:
  ethernets:
#    eno1:
#     dhcp4: false
    enx0c37961e7ce1:
      dhcp4: true
  version: 2
```
For A) I'll go with when there is more than 2 wired interfaces interfaces to avoid:
```
-rw-r--r-- 1 root root 116 Sep 27 13:44 /etc/systemd/network/IIAB-Slave-wlp0s20f3.network
                        ...ITS LAST 100 LINES FOLLOW...

# /etc/systemd/network/IIAB-Slave-eno1.network
[Match]
Name=eno1
```
For B) if the initial install of Ubuntu had both interfaces present: Always but if new hardware was added double check anyway.

> wireless_list_1(wifi1) = not found-1
wireless_list_2(wifi2) = wlp0s20f3
num_wifi_interfaces = 1
discovered_wireless_iface = wlp0s20f3
discovered_wired_iface = eno1
exclude_devices = none
num_lan_interfaces = 2

> 2022-09-27 13:44:08,721 p=20120 u=root n=ansible | TASK [network : Wired enslaving - Assigns lan_list_results to br0 as wired slaves if present] ****************************************************************************
2022-09-27 13:44:09,064 p=20120 u=root n=ansible | changed: [127.0.0.1] => (item=eno1)
2022-09-27 13:44:09,369 p=20120 u=root n=ansible | changed: [127.0.0.1] => (item=wlp0s20f3)

The above is a potential bug, that is the unused wifi device, the parent of ap0, the systemd bridging code predates ap0 usage by a year or so.

holta, 2022-10-02:
This ticket is a big step forward to help everyone understand.

Let's get the most common scenarios cleanly communicated in coming weeks.

Thanks @jvonau @rickgurmet and everyone who's helped.  Networking's Never Simple.  Streamlining will not happen overnight (there are too many corner cases!) and that's just fine.

_I'd rather this gets done professionally (where we can) one week at a time &mdash; so newcomers gain crystal clear instructions &mdash; for the most supported networking layouts_ :chart_with_upwards_trend:

jvonau, 2022-10-03:
Think the summary would be ensure your default gateway device is defined in /etc/netplan/ somewhere(the file names vary) with the systemd-networkd backend in use and # out members of the br0 device. Network-Manager backend doesn't need to have the default gateway device defined by the installer but might have something present.

jvonau, 2022-10-23:
Note #2751 #2752 `brctl` hasn't been used with 'iiab' since https://github.com/iiab/iiab/pull/2759

holta, 2024-03-29:
Thank you to @mbonett who's added an explanatory diagram here:

- https://github.com/iiab/iiab/discussions/3724#discussioncomment-8947062

jvonau, 2025-01-27:
status?

holta, 2025-01-27:
This ticket seems like great "living documentation".

To help people who run into similar, as new OS's like Ubuntu 25.04 go into final test soon in coming months?

(Or we can close this ticket, if that's strongly preferred!)