iiab/iiab #3393: Ansible conditional acting weird.

jvonau, 2022-10-05:
Related: #3392

pi@0906gui:/opt/iiab/iiab $ `cat /etc/rpi-issue`
Raspberry Pi reference 2022-09-06
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 827affcc11aaf7aa577d15daf02fb40b64392380, stage4
```
2022-10-04 02:01:12,991 p=7453 u=root n=ansible | included: /opt/iiab/iiab/roles/network/tasks/dnsmasq.yml for 127.0.0.1
2022-10-04 02:01:13,110 p=7453 u=root n=ansible | TASK [1-prep : Install dnsmasq] **************************************************************>
2022-10-04 02:01:24,154 p=7453 u=root n=ansible | changed: [127.0.0.1]
2022-10-04 02:01:24,233 p=7453 u=root n=ansible | TASK [1-prep : Install /etc/dnsmasq.d/dnsmasq-iiab, allowing systemd-resolved AND dnsmasq to w>
2022-10-04 02:01:25,440 p=7453 u=root n=ansible | changed: [127.0.0.1] => (item={'src': 'roles/network/templates/network/dnsmasq.service.u18', '>
2022-10-04 02:01:26,587 p=7453 u=root n=ansible | changed: [127.0.0.1] => (item={'src': 'roles/network/templates/network/dnsmasq-iiab', 'dest': >
2022-10-04 02:01:26,670 p=7453 u=root n=ansible | TASK [1-prep : Don't use stock dnsmasq systemd unit file during boot but start now] **********>
2022-10-04 02:01:30,466 p=7453 u=root n=ansible | changed: [127.0.0.1]
2022-10-04 02:01:30,544 p=7453 u=root n=ansible | TASK [1-prep : Install package networkd-dispatcher (OS's other than RasPiOS and Linux Mint)] *>
2022-10-04 02:01:40,592 p=7453 u=root n=ansible | changed: [127.0.0.1]
2022-10-04 02:01:40,667 p=7453 u=root n=ansible | TASK [1-prep : Install 11 network packages: avahi-daemon, hostapd, iproute2, iptables-persiste>
2022-10-04 02:01:58,699 p=7453 u=root n=ansible | changed: [127.0.0.1]
2022-10-04 02:01:58,780 p=7453 u=root n=ansible | TASK [1-prep : Unmask the Access Point 'hostapd' service] ************************************>
2022-10-04 02:02:02,218 p=7453 u=root n=ansible | changed: [127.0.0.1]
```
```
Start-Date: 2022-10-04  02:01:18
Commandline: /usr/bin/apt-get -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold install dnsmasq=2.85-1
Requested-By: pi (1000)
Install: dnsmasq:arm64 (2.85-1)
End-Date: 2022-10-04  02:01:22

Start-Date: 2022-10-04  02:01:36
Commandline: /usr/bin/apt-get -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold install networkd-dispatcher=2.1-2
Requested-By: pi (1000)
Install: networkd-dispatcher:arm64 (2.1-2)
End-Date: 2022-10-04  02:01:39

Start-Date: 2022-10-04  02:01:46
Commandline: /usr/bin/apt-get -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold install hostapd=2:2.9.0-21 iptables-persi>
Requested-By: pi (1000)
Install: netmask:arm64 (2.4.4-3), iptables-persistent:arm64 (1.0.15), netfilter-persistent:arm64 (1.0.15, automatic), hostapd:arm64 (2:2.9.0-21)
End-Date: 2022-10-04  02:01:56
```
>TASK [1-prep : Install package networkd-dispatcher (OS's other than RasPiOS and Linux Mint)]
 changed: [127.0.0.1]

https://github.com/iiab/iiab/blame/master/roles/network/tasks/install.yml#L10

The intent is to skip the package but was not skipped, just wondering if there is a bigger issue at play.

holta, 2022-10-05:
Line 10 looks suspicious:

```
when: not is_raspbian or not is_linuxmint
```

That's equivalent to:

```
when: true
```

Which is equivalent to the line not existing at all.

_The reason is all OS's are (not is_raspbian) or (not is_linuxmint)._

Possibly Line 10 was meant to be...?

```
when: not is_raspbian and not is_linuxmint
```

holta, 2022-10-05:
If Line 10 should contain `and` instead of `or`, should the PR below fix it...?

- PR #3392

(Or do you prefer another PR?)

jvonau, 2022-10-05:
I prefer more info because in the end I'm inclined to remove the restriction on RasPiOS given the recent addition of NetworkManager and optional use into the OS. I have to consider how dnsmasq is being [started ](https://github.com/iiab/iiab/blob/master/roles/network/tasks/enable_services.yml#L90-L110) when systemd-networkd has control of br0 and if the br0 is [available](https://github.com/iiab/iiab/blob/master/roles/network/templates/network/dnsmasq-iiab) when starting dnsmasq along with the general dnsmasq failure to install cleanly without iiab in the picture.

jvonau, 2022-10-06:
RasPiOS doesn't have the traceback as it doesn't have the problem version of networkd-dispatcher when using NM backend 

apt list networkd-dispatcher
Listing... Done
networkd-dispatcher/stable,stable,now 2.1-2 all [installed]

jvonau, 2022-10-06:
Removed restriction in 3392

closed 2022-10-06 by jvonau