iiab/iiab #3398: Should IIAB run 'rfkill unblock wlan' ?  This worked to get IIAB running on Mint 21 on Acer TravelMate P643-M

holta, 2022-10-12:
@jvonau & All:

I don't know all the ramifications but `rfkill unblock wlan` saved the day installing the latest IIAB onto Mint 21 on an Acer laptop (TravelMate P643-M).

What might be the best way to make this happen for new implementers in general?

ASIDE: iiab-diagnostics appear a bit odd as this is a continuation of an IIAB install someone apparently began 2 months ago: http://sprunge.us/rjTcse?bash

Tangentially Related:

- #3397

holta, 2022-10-12:
Just FYI rfkill claimed something was "HARD BLOCKED" but apparently that's an exaggeration?

As the fix was in the end just 1 command (`rfkill unblock wlan`) as seen here:

```
root@box:/opt/iiab/iiab# systemctl status hostapd
○ hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Wed 2022-10-12 18:13:46 EDT; 7min ago
   Main PID: 64719 (code=exited, status=0/SUCCESS)
        CPU: 16ms

Oct 12 18:13:46 box systemd[1]: Starting Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Oct 12 18:13:46 box hostapd[64718]: rfkill: WLAN hard blocked
Oct 12 18:13:46 box hostapd[64718]: ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Oct 12 18:13:46 box ip[64720]: RTNETLINK answers: Operation not possible due to RF-kill
Oct 12 18:13:46 box systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Oct 12 18:13:46 box systemd[1]: hostapd.service: Deactivated successfully.
```

```
root@box:~# rfkill
ID TYPE      DEVICE             SOFT      HARD
 0 wlan      acer-wireless   blocked unblocked
 1 bluetooth hci0          unblocked unblocked
 2 wlan      phy0            blocked   blocked
```

```
root@box:~# rfkill unblock wlan
```

```
root@box:~# rfkill
ID TYPE      DEVICE             SOFT      HARD
 0 wlan      acer-wireless unblocked unblocked
 1 bluetooth hci0          unblocked unblocked
 2 wlan      phy0          unblocked unblocked
```

```
root@box:~# systemctl restart hostapd
```

```
root@box:~# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-10-12 18:25:36 EDT; 7min ago
    Process: 74850 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=0/SUCCESS)
    Process: 74855 ExecStartPost=/sbin/ip link set ap0 up (code=exited, status=0/SUCCESS)
   Main PID: 74854 (hostapd)
      Tasks: 1 (limit: 4296)
     Memory: 3.1M
        CPU: 5.878s
     CGroup: /system.slice/hostapd.service
             └─74854 /usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf

Oct 12 18:25:35 box systemd[1]: Starting Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Oct 12 18:25:35 box hostapd[74850]: ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Oct 12 18:25:35 box hostapd[74850]: ap0: interface state COUNTRY_UPDATE->ENABLED
Oct 12 18:25:35 box hostapd[74850]: ap0: AP-ENABLED
Oct 12 18:25:36 box systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
```

holta, 2022-10-12:
I should have mentioned:

The fix really works, in the sense that it's "permanent" (i.e. hostapd starts properly after reboot).

holta, 2022-10-12:
Possibly both relevant?

https://github.com/iiab/iiab/blob/e46e34679ec98eeae0a3ed95b85b1f63846469c4/roles/network/tasks/rpi_debian.yml#L55-L57

https://github.com/iiab/iiab/blob/e46e34679ec98eeae0a3ed95b85b1f63846469c4/roles/network/templates/hostapd/iiab-test-wifi.j2#L15-L23

jvonau, 2022-10-13:
Incomplete information, information being withheld, or games being played, this is what I see:
```
 341 wireless_list_1(wifi1) = not found-1
 342 wireless_list_2(wifi2) = wlp3s0
 343 num_wifi_interfaces = 1
 344 discovered_wireless_iface = wlp3s0
 345 discovered_wired_iface = none
 346 exclude_devices = none
 347 num_lan_interfaces = 0
 348 gui_static_wan = False
 349 iiab_lan_iface = br0
 350 iiab_wan_iface = enp4s0
 351 can_be_ap = True

1203 3: wlp3s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
1204     link/ether 74:e5:43:2c:9e:50 brd ff:ff:ff:ff:ff:ff
```
https://maclookup.app/macaddress/74E5F9 That should be an intel chipset
```
1492 COMMAND: /usr/sbin/rfkill list    # Are WiFi and Bluetooth interfaces blocked?
1493 
1494 0: acer-wireless: Wireless LAN
1495 	Soft blocked: no
1496 	Hard blocked: no
1497 1: hci0: Bluetooth
1498 	Soft blocked: no
1499 	Hard blocked: no
1500 2: phy0: Wireless LAN
1501 	Soft blocked: no
1502 	Hard blocked: no
```
Strange, the above would suggest 2 wifi devices present
```
2088 [   23.686698] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
2089 [   23.689333] ieee80211 phy0: Atheros AR9462 Rev:2 mem=0xffffb83440a00000, irq=19
2090 [   23.823702] ath9k 0000:03:00.0 wlp3s0: renamed from wlan0

2111 03:00.0 Network controller [0280]: Qualcomm Atheros AR9462 Wireless Network Adapter [168c:0034] (rev 01)
```
Is it possible that there was/are 2 devices present but only one of the devices(AR9462) has the correct firmware loaded and the other (acer-wireless) is just along for the ride?

jvonau, 2022-10-13:
https://www.linux.org/threads/wireless-lan-hard-blocked-yes.37703/

holta, 2022-10-13:
> Is it possible that there was/are 2 WiFi devices present but only one of the devices(AR9462) has the correct firmware loaded and the other (acer-wireless) is just along for the ride?

I don't understand it.  Certainly there are no WiFi peripherals/devices attached externally to the laptop.  But possibly there might be 2 WiFi devices inside it ???

I'll ask him to completely wipe the laptop with a fresh copy of Mint 21, and then reinstall a (hopefully MINIMUM-sized) IIAB using: **curl iiab.io/install.txt | bash**

I (hope) he has time to do that in the coming days.

So that the install process & diagnostics are as cleanly understandable as possible.
And at least we'll then be able to see what's repeatable (or not).

holta, 2022-10-16:
> I should have mentioned:
> 
> The fix really works, in the sense that it's "permanent" (i.e. hostapd starts properly on its own after reboot).

CORRECTION:

The above worked across a quick reboot, _but did not work_ after cold starting the laptop (Acer TravelMate P643-M) after it had been powered off.  (The output of `rfkill` reverted to its original problematic state, as follows...)

```
root@box:~# rfkill
ID TYPE      DEVICE             SOFT      HARD
 0 wlan      acer-wireless   blocked unblocked
 1 bluetooth hci0          unblocked unblocked
 2 wlan      phy0            blocked   blocked
```

PS He (Peter) found `Fn`+`F3` hoping that might possibly help UNBLOCK WiFi...somewhat more permanently...ideally ??

PPS Regardless, he hopes to wipe the laptop's HDD and install a fresh copy of Mint 21 to try again in coming hours or days.

holta, 2022-10-16:
It would seem (?) miraculously that wiping and installing Mint 21 + IIAB afresh worked nearly flawlessly:

http://sprunge.us/FMrBYY?bash

Tangential minor bug:

- #3404

holta, 2022-10-23:
1) In the case of Acer TravelMate P643-M it appears WiFi was blocked by the keyboard sequence `Fn` + `F3` &mdash; so automating `rfkill unblock wlan` (which works, but until reboot it appears!) would perhaps do more harm than good &mdash; by papering over the operator's understanding of their own computer.

2) In other cases, I'd like to understand better why IIAB's code sometimes uses `rfkill unblock 0` &mdash; and other times uses `rfkill unblock wifi`
   - The 2 code snippets are shown at: https://github.com/iiab/iiab/issues/3398#issuecomment-1276814881

holta, 2022-12-22:
In conclusion, the answer to the question in the subject line is probably "No" &mdash; even though it would have brought WiFi to life in this particular scenario.

(The main reason being that `rfkill unblock wlan` would have to be forced on every boot, so as to override the fact that their hardware/keyboard toggle was set to boot every time with WiFi OFF.  And it was felt that educating the operator was more important than a hacky/kludgy workaround.)

closed 2022-12-22 by holta

holta, 2022-12-26:
Thanks @martignoni for pointing out the underlying mechanism here :heavy_check_mark: 

- https://github.com/moodlebox/moodlebox/issues/279
- https://github.com/RPi-Distro/raspi-config/blob/5b11405bf839083557ec64c263fc79165e01c3cd/raspi-config#L576