iiab/iiab #3399: IIAB installs on Debian 12 Bookworm pre-releases IF (1) networkd-dispatcher bypassed (2) crontab bypassed for Kiwix/Matomo (3) MongoDB 4.4 forced to 6.0 w/ libssl1.1

holta, 2022-10-13:
Testing on Debian 12 Bookworm pre-releases is easy if you create an instant VM:

```
multipass launch -n deb12 -c 2 -d 25G -m 2G https://cloud.debian.org/images/cloud/bookworm/daily/latest/debian-12-generic-amd64-daily.qcow2
```

Shell into the VM using:

```
multipass shell deb12
```

Then within the VM, run:

```
echo 'VERSION_ID="12"' >> /etc/os-release
curl iiab.io/install.txt | bash
```

A LARGE-sized IIAB install just about worked &mdash; here are the quick+dirty workarounds I used for now:

1) The 1st error was:

   > TASK [1-prep : Install 12 network packages: avahi-daemon, hostapd, iproute2, iptables-persistent, iw, libnss-mdns, netmask, net-tools, networkd-dispatcher, rfkill, wireless-tools, wpasupplicant -- later used by https://github.com/iiab/iiab/tree/master/roles/network] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'networkd-dispatcher' is available"}

   Any suggested workarounds, to explore how IIAB will work on Debian 12 Bookworm (the basis for Raspberry Pi OS 12) in 2023?

   For now, I commented out [Line 29 of network/tasks/install.yml](https://github.com/iiab/iiab/blob/e46e34679ec98eeae0a3ed95b85b1f63846469c4/roles/network/tasks/install.yml#L29) to continue.

2) The 2nd error was:

   > TASK [kiwix : Make a crontab entry to restart kiwix-serve at 4AM (debuntu)] ********************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /etc/crontab does not exist !", "rc": 257}

   I commented out [Lines 26-31 of kiwix/tasks/enable-or-disable.yml](https://github.com/iiab/iiab/blob/e46e34679ec98eeae0a3ed95b85b1f63846469c4/roles/kiwix/tasks/enable-or-disable.yml#L26-L31) to continue.

3) The 3d error was entirely expected, as Moodle's 8.1 support is converging right now week-to-week:

   > TASK [moodle : Execute /opt/iiab/moodle/moodle_installer IF /opt/iiab/moodle/config.php doesn't yet exist -- REQUIRES 'max_input_vars = 5000' (or higher) in /etc/php/8.1/cli/php.ini IF PHP 8+ (as set up by www_options/tasks/main.yml) -- WHEREAS LATER ON Moodle uses /etc/php/8.1/fpm/php.ini during regular operation] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/opt/iiab/moodle/moodle_installer", "delta": "0:00:00.121594", "end": "2022-10-13 00:32:12.411611", "msg": "non-zero return code", "rc": 255, "start": "2022-10-13 00:32:12.290017", "stderr": "+ sudo -u www-data /usr/bin/php /opt/iiab/moodle/admin/cli/install.php --wwwroot=http://box.lan/moodle --dataroot=/library/moodle --dbtype=pgsql --dbname=moodle --dbuser=Admin --dbpass=changeme --fullname=Your_School --shortname=School --adminuser=admin --adminpass=changeme --non-interactive --agree-license --allow-unstable\nPHP Deprecated:  Return type of moodle_recordset::rewind() should either be compatible with Iterator::rewind(): void, or the #[\\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.php on line 61\nPHP Fatal error:  Type of xml_format_exception::$line must be int (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line 42", "stderr_lines": ["+ sudo -u www-data /usr/bin/php /opt/iiab/moodle/admin/cli/install.php --wwwroot=http://box.lan/moodle --dataroot=/library/moodle --dbtype=pgsql --dbname=moodle --dbuser=Admin --dbpass=changeme --fullname=Your_School --shortname=School --adminuser=admin --adminpass=changeme --non-interactive --agree-license --allow-unstable", "PHP Deprecated:  Return type of moodle_recordset::rewind() should either be compatible with Iterator::rewind(): void, or the #[\\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.php on line 61", "PHP Fatal error:  Type of xml_format_exception::$line must be int (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line 42"], "stdout": "\nDeprecated: Return type of moodle_recordset::rewind() should either be compatible with Iterator::rewind(): void, or the #[\\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.php on line 61\n\nFatal error: Type of xml_format_exception::$line must be int (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line 42", "stdout_lines": ["", "Deprecated: Return type of moodle_recordset::rewind() should either be compatible with Iterator::rewind(): void, or the #[\\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice in /opt/iiab/moodle/lib/dml/moodle_recordset.php on line 61", "", "Fatal error: Type of xml_format_exception::$line must be int (as in class Exception) in /opt/iiab/moodle/lib/xmlize.php on line 42"]}

   I set `moodle_install: False` and `moodle_enabled: False` in /etc/iiab/local_vars.yml to continue.

4) The 4th error can be worked around, by setting `mongodb_64bit_version: 6.0` in `roles/mongodb/defaults/main.yml` and forcing the install of bullseye's libssl1.1 with ` when: is_debian_12` much like PR #3397: (and admitting that pre-AVX CPU's from 2010 are less important now, that being the reason we haung onto Mongo 4.4 until now)

   > TASK [mongodb : Install packages: mongodb-org, mongodb-org-server] *****************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'mongodb-org' is available"}

   I used:

   ```
   apt_repository:
     repo: deb http://security.debian.org/debian-security bullseye-security main
   ```

   But the following likely also works:

   ```
   apt_repository:
     repo: deb https://deb.debian.org/debian-security bullseye-security main
   ```

5) The 5th error was like the 2nd:

   > TASK [matomo : Start Collecting Matomo Data] ***************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failed to find required executable \"crontab\" in paths: /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"}

   I set `matomo_install: False` and `matomo_enabled: False` in /etc/iiab/local_vars.yml to continue.

6) Naturally networkd-dispatcher diagnostic hook creation failed:

   > TASK [network : Create networkd-dispatcher diagnostic hook for recording network events] *******************************************************************************
failed: [127.0.0.1] (item={'src': 'hostapd/00-iiab-debug', 'dest': '/etc/networkd-dispatcher/carrier.d/00-iiab-debug'}) => {"ansible_loop_var": "item", "changed": false, "checksum": "969e5f59a1fac94638d39846faa08afcf4f1b536", "item": {"dest": "/etc/networkd-dispatcher/carrier.d/00-iiab-debug", "src": "hostapd/00-iiab-debug"}, "msg": "Destination directory /etc/networkd-dispatcher/carrier.d does not exist"}
failed: [127.0.0.1] (item={'src': 'hostapd/00-iiab-debug', 'dest': '/etc/networkd-dispatcher/degraded.d/00-iiab-debug'}) => {"ansible_loop_var": "item", "changed": false, "checksum": "969e5f59a1fac94638d39846faa08afcf4f1b536", "item": {"dest": "/etc/networkd-dispatcher/degraded.d/00-iiab-debug", "src": "hostapd/00-iiab-debug"}, "msg": "Destination directory /etc/networkd-dispatcher/degraded.d does not exist"}
failed: [127.0.0.1] (item={'src': 'hostapd/00-iiab-debug', 'dest': '/etc/networkd-dispatcher/dormant.d/00-iiab-debug'}) => {"ansible_loop_var": "item", "changed": false, "checksum": "969e5f59a1fac94638d39846faa08afcf4f1b536", "item": {"dest": "/etc/networkd-dispatcher/dormant.d/00-iiab-debug", "src": "hostapd/00-iiab-debug"}, "msg": "Destination directory /etc/networkd-dispatcher/dormant.d does not exist"}
failed: [127.0.0.1] (item={'src': 'hostapd/00-iiab-debug', 'dest': '/etc/networkd-dispatcher/no-carrier.d/00-iiab-debug'}) => {"ansible_loop_var": "item", "changed": false, "checksum": "969e5f59a1fac94638d39846faa08afcf4f1b536", "item": {"dest": "/etc/networkd-dispatcher/no-carrier.d/00-iiab-debug", "src": "hostapd/00-iiab-debug"}, "msg": "Destination directory /etc/networkd-dispatcher/no-carrier.d does not exist"}
failed: [127.0.0.1] (item={'src': 'hostapd/00-iiab-debug', 'dest': '/etc/networkd-dispatcher/off.d/00-iiab-debug'}) => {"ansible_loop_var": "item", "changed": false, "checksum": "969e5f59a1fac94638d39846faa08afcf4f1b536", "item": {"dest": "/etc/networkd-dispatcher/off.d/00-iiab-debug", "src": "hostapd/00-iiab-debug"}, "msg": "Destinatio [... cut ...] f4f1b536", "item": {"dest": "/etc/networkd-dispatcher/routable.d/00-iiab-debug", "src": "hostapd/00-iiab-debug"}, "msg": "Destination directory /etc/networkd-dispatcher/routable.d does not exist"}

   I commented out [Lines 81-83 of roles/network/tasks/main.yml](https://github.com/iiab/iiab/blob/e46e34679ec98eeae0a3ed95b85b1f63846469c4/roles/network/tasks/main.yml#L81-L83) just for now.

7) Admin Console installed and logging into it works!

8) iiab-diagnostics are: http://sprunge.us/aoLf3m?bash

jvonau, 2022-10-13:
https://tracker.debian.org/pkg/networkd-dispatcher
https://bugs.debian.org/cgi-bin/pkgreport.cgi?repeatmerged=no&src=networkd-dispatcher
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1010303
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1014615

holta, 2022-10-13:
Thanks @jvonau for those Debian community's `networkd-dispatcher` links posted above.

Debian 12 Bookworm's failure to support `/etc/crontab` and the `crontab` executable out-of-the-box are highly surprising.  Possibly this is a transient/evolving issue that will resolve itself in coming months?  Either way, whether this is a permanent decision on their part or not &mdash; it seems quite likely that we can solve this quickly, with the equivalent of:

```
apt install cron
```

As indicated by the lastest Debian 12 Bookworm daily build showing...

```
root@box:~# apt list cron
Listing... Done
cron/testing 3.0pl1-150 amd64
```

```
root@box:~# apt show cron
Package: cron
Version: 3.0pl1-150
Priority: important
Section: admin
Maintainer: Javier Fernández-Sanguino Peña <jfs@debian.org>
Installed-Size: 249 kB
Provides: cron-daemon
Pre-Depends: init-system-helpers (>= 1.54~), cron-daemon-common
Depends: libc6 (>= 2.34), libpam0g (>= 0.99.7.1), libselinux1 (>= 3.1~), sensible-utils, libpam-runtime
Recommends: default-mta | mail-transport-agent
Suggests: anacron, logrotate, checksecurity
Conflicts: bcron, cronie, systemd-cron
Replaces: bcron, cronie, systemd-cron
Homepage: https://ftp.isc.org/isc/cron/
Tag: admin::automation, implemented-in::c, interface::daemon, role::program,
 scope::utility
Download-Size: 100 kB
APT-Sources: mirror+file:/etc/apt/mirrors/debian.list bookworm/main amd64 Packages
Description: process scheduling daemon
 The cron daemon is a background process that runs particular programs at
 particular times (for example, every minute, day, week, or month), as
 specified in a crontab. By default, users may also create crontabs of
 their own so that processes are run on their behalf.
 .
 Output from the commands is usually mailed to the system administrator
 (or to the user in question); you should probably install a mail system
 as well so that you can receive these messages.
 .
 This cron package does not provide any system maintenance tasks. Basic
 periodic maintenance tasks are provided by other packages, such
 as checksecurity.
```

holta, 2022-10-19:
Summary of today's test install of IIAB onto Debian 12 (Bookworm's) latest faily build;

- networkd-dispatcher remains missing from Debian 12 (Bookworm)
- Everything else worked, even including Moodle 4.1 LTS's latest pre-release (#3381)
  - Kiwix and Matomo now work with Debian 12, thanks to: PR #3400

iiab-diagnostics: http://sprunge.us/kL49dr?bash

holta, 2022-10-19:
Oops, CLARIF: MongoDB still requires forcible install of libssl1.1 on Debian 12, for those wanting Sugarizer.

So the PR below will be further refined as time goes on &mdash; if indeed MongoDB continues to refuse to supported OS's that have made the transition from libssl1.1 to libssl3:

- PR #3397

holta, 2022-10-19:
> MongoDB still requires forcible install of libssl1.1 on Debian 12, for those wanting Sugarizer.

_Above is no longer true!_

IIAB's code has now been changed, so MongoDB 6.0 is installed on all 64-bit OS's (including Debian 12) going forward:

- PR #3407

holta, 2022-11-05:
Vital Multipass workaround here:

- https://github.com/canonical/multipass/issues/2492#issuecomment-1304590744

(As Multipass is not yet mature enough to reliably delete its own VM's / instances.)

jvonau, 2022-11-05:
What is the relation between Debian 12 and mulitpass?

holta, 2022-11-05:
> What is the relation between Debian 12 and [Multipass] ?

See the top of this ticket:
https://github.com/iiab/iiab/issues/3399#issue-1407189520

jvonau, 2022-11-05:
> See the top of this ticket: [#3399 (comment)](https://github.com/iiab/iiab/issues/3399#issue-1407189520)

So nothing other than it's the guest OS, using the semi-supported lxc backend within multipass, gotcha.

holta, 2022-11-05:
This tkt was created to help ppl test Debian 12 on Multipass.

That is no longer possible in recent days.

Worse, it will completely block your host PC from rebooting, unless you pull the power, e.g. with LXD which is what allows  real testing from other 192.168.0.x devices nearby.

So the responsible thing to do was to warn people.

Hopefully the above problems will...at least be contained...in coming days or so.

holta, 2022-11-06:
Just FYI the Multipass freeze-up arises with all three Debian 12 daily builds over the past 3 days (Nov 4-6) from:

https://cloud.debian.org/images/cloud/bookworm/daily/

(e.g. [debian-12-generic-amd64-daily-20221106-1191.qcow2](https://cloud.debian.org/images/cloud/bookworm/daily/20221106-1191/debian-12-generic-amd64-daily-20221106-1191.qcow2) fails to allow login, in the same way as the prior 2 days, with Multipass "Connection refused" and "Timeout" errors.)

jvonau, 2022-11-08:
> @holta, @jvonau FWIW, this use case is entirely out of Multipass's scope. As I mentioned in [canonical/multipass#2492](https://github.com/canonical/multipass/issues/2492), Multipass is designed to launch Ubuntu VMs, not Debian.

Yup I agree with @ricab, one should limit the use of multipass to what is supported with `multipass find` otherwise just use the workarounds presented within the posted issue when you start walking on the wild side going outside those boundaries.

holta, 2022-11-08:
Let's not overly exaggerate please.

[`multipass launch <image url>`](https://multipass.run/docs/launch-command) works perfectly well.

So long as `<image url>` is a working .qcow2 image, no matter the OS.

The real issue is that Multipass doesn't yet garbage-collect cleanly &mdash; as @ricab has documented thoughtfully here:

- canonical/multipass#2492

And hopefully Multipass 1.12 will solve this in due course, sometime in coming months.

jvonau, 2022-11-08:
`multipass launch --help`
```
Usage: multipass launch [options] [[<remote:>]<image> | <url>]
snip
Arguments:
  image                                 Optional image to launch. If omitted,
                                        then the default Ubuntu LTS will be
                                        used.
                                        <remote> can be either ‘release’ or
                                        ‘daily‘. If <remote> is omitted,
                                        ‘release’ will be used.
                                        <image> can be a partial image hash or
                                        an Ubuntu release version, codename or
                                        alias.
                                        <url> is a custom image URL that is in
                                        http://, https://, or file:// format.
```
Well --help does explicitly mention **_Ubuntu_**, whatever becomes of it @holta keep my name/handle out of this complaint.

holta, 2022-11-08:
> --help does explicitly mention Ubuntu

Again, let's not overly exaggerate:

1) Multipass docs do _**NOT**_ say that Ubuntu instances are mandatory.
2) In fact many posts from Multipass developers in recent years have provided tips to help make this possible (running non-Ubuntu OS's as Multipass instances).
3) I do not expect Canonical to suddenly now begin blocking non-Ubuntu OS's &mdash; inviting the scrutiny of antitrust lawyers.

holta, 2022-11-11:
[Debian 12's daily build cloud installer](https://cloud.debian.org/images/cloud/bookworm/daily/) is still broken 2 weeks later, just FYI.

holta, 2022-11-15:
> [Debian 12 Bookworm's daily build cloud installer](https://cloud.debian.org/images/cloud/bookworm/daily/) is still broken 2 weeks later, just FYI.

Ditto, 5 days later (-:

holta, 2022-11-16:
Happier news:

minetest-server 5.6.1 is already testable today, as part of Debian 12 Bookworm:
https://packages.debian.org/bookworm/minetest-server

(FYI minetest-server >= 5.6 is necessary for those who want https://github.com/mt-mods/dreambuilder_game that @tim-moody recommends.)

holta, 2022-11-28:
Great news Debian 12 Bookworm images once again work.
(Sometime in the past 13 days, they fixed the issue that cropped up exactly a month ago, on Oct 28th.)

So testing of Debian 12 VM's is once again incredibly fast & easy, using commands like:

```
multipass launch -n deb12 -c 2 -d 25G -m 2G https://cloud.debian.org/images/cloud/bookworm/daily/latest/debian-12-generic-amd64-daily.qcow2
```

holta, 2022-11-28:
Just FYI Debian 12 is [still missing](https://github.com/iiab/iiab/issues/3399#issuecomment-1277785160) package [networkd-dispatcher](https://tracker.debian.org/pkg/networkd-dispatcher).

(As in early October, commenting out Line 29 of roles/network/tasks/install.yml works as a basic workaround.)

Hopefully they'll work on resolving this properly before [Debian 12's freeze dates](https://release.debian.org/bookworm/freeze_policy.html#summary) in early 2023.

holta, 2022-11-28:
FYI a change from early October &mdash; kalite can not longer be installed on Debian 12 Bookworm:

```
TASK [kalite : Install packages: python2, python-setuptools, virtualenv (for Python 2)] ********************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'python-setuptools' is available"}
```

Related:

- https://packages.debian.org/bookworm/python-setuptools "Package not available in this suite."
- https://packages.debian.org/bookworm/python3-setuptools "65.5.0-1"

holta, 2022-11-28:
LARGE-sized IIAB install onto latest daily Debian 12 Bookworm is complete &mdash; looks great overall:

- http://sprunge.us/Cp8PIq?bash

Except for the 2 issues mentioned above:
1. Missing package [networkd-dispatcher](https://tracker.debian.org/pkg/networkd-dispatcher)
2. KA Lite needing Python 2 version of missing package [python-setuptools](https://packages.debian.org/bookworm/python-setuptools)

jvonau, 2022-11-29:
The dependencies are being removed first before the main package goes away. It's going to progress to other python2 [packages](https://packages.debian.org/bookworm/all/python/) try to open https://packages.debian.org/bookworm/all/python/python2.7. It not so much the deb files being removed from the repo, the reference to the deb package gets removed from the Release [file](https://deb.debian.org/debian/dists/bookworm/) that contains what is available for the release.

holta, 2022-12-29:
Ubuntu 23.04 issues are extremely similar to Debian 12 issues: (as both OS's are converging quickly in coming months!)

- #3434 
- #3450
  - #3289

holta, 2023-01-21:
Just FYI Debian 12 "Bookworm" has jumped forward to:

- Linux kernel 6.1 LTS
- PHP 8.2

Whereas Ubuntu 23.04 "Lunar Lobster" currently uses:

- Linux kernel 5.15 LTS _(will later use 6.2, if we believe https://youtu.be/JL_NHwNQi48 !)_
- PHP 8.1 _(will later use 8.2, if we believe https://youtu.be/JL_NHwNQi48 !)_

holta, 2023-01-31:
Just FYI Nextcloud 25 is not compatible with PHP 8.2

So errors like the following should hopefully go away after Nextcloud 26 is released on [2022-03-21](https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule#nextcloud-26):

> TASK [nextcloud : Allow Nextcloud access from all hosts and IP addresses -- SEE ALSO THE OLD WAY: https://github.com/iiab/iiab/blob/master/roles/nextcloud/templates/nextcloud.conf.j2.unused] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["php", "/library/www/nextcloud/occ", "config:system:set", "trusted_domains", "1", "--value=*"], "delta": "0:00:00.041838", "end": "2023-01-31 11:59:24.383956", "msg": "non-zero return code", "rc": 1, "start": "2023-01-31 11:59:24.342118", "stderr": "", "stderr_lines": [], "stdout": "This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.1.", "stdout_lines": ["This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.1."]}

holta, 2023-01-31:
> So errors like the following should hopefully go away after Nextcloud 26 is released on [2022-03-21](https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule#nextcloud-26):

FYI Nextcloud 26 pre-releases can be attempted during the interim 7+ weeks if anybody wants to test that, using a line like the following in your [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it%3F) :

```
nextcloud_dl_url: https://download.nextcloud.com/server/prereleases/nextcloud-26.0.0beta1.tar.bz2
```

holta, 2023-01-31:
> FYI Nextcloud 26 pre-releases can be attempted during the interim 7+ weeks

Or try installing this daily build of their very latest:

```
nextcloud_dl_url: https://download.nextcloud.com/server/daily/latest-master.tar.bz2
```

holta, 2023-02-17:
"bookworm in soft freeze"
https://lists.debian.org/debian-devel-announce/2023/02/msg00003.html

holta, 2023-03-07:
Ansible will likely need to be installed in a somewhat different way going forward — as this Debian 12 / Bookworm (pre-release) error indicates:

```

'pip3 install --upgrade ansible-core' will now run:

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python
installation or OS distribution provider. You can override this, at
the risk of breaking your Python installation or OS, by passing --
break-system-packages.
```

Current Debian 12 pre-release shows:

```
root@deb12:~# apt list python3
Listing... Done
python3/testing,now 3.11.2-1 amd64 [installed,automatic]

root@deb12:~# apt list python3-full
Listing... Done
python3-full/testing 3.11.2-1 amd64

root@deb12:~# apt list pipx                                        
Listing... Done
pipx/testing 1.1.0-1 all
```

holta, 2023-03-07:
Of course `apt install ansible-core` would be yet another workaround: (in addition to those mentioned above)

```
root@deb12:~# apt list ansible-core
Listing... Done
ansible-core/testing 2.14.2-1 all
```

holta, 2023-03-17:
FYI Ubuntu 23.04 also now contains Python 3.11, so the fix below (possibly an interim fix, but certainly good enough for now) was put into place, restoring the ability to install IIAB on both Ubuntu 23.04 and Debian 12 pre-releases:

- PR #3491

jvonau, 2023-03-20:
https://github.com/iiab/iiab/issues/3289#issuecomment-1476378975

holta, 2023-05-01:
LARGE-sized IIAB install test on Debian 12 RC2+ ran essentially flawlessly yesterday:

iiab-diagnostics: http://sprunge.us/IiIYRG?en

holta, 2023-05-05:
Interestingly (FYI!) Debian 11 networking is a real mess when a VM (instance) is attempted using Multipass, as documented here:

- https://github.com/iiab/iiab/issues/3568

_In contrast, Debian 12 VM's (instances) continue to work very well within Multipass._

RECAP / CONTEXT: Testing was done using the very latest version of Multipass, from its "edge" release channel, as installed by snap.

holta, 2023-05-27:
Can somebody test a LARGE-sized IIAB install on the brand new Debian 12 Bookworm Release Candidate 4?

holta, 2023-06-10:
- "Debian 12 bookworm released" according to https://debian.org/News/2023/20230610
- Final builds still being published according to https://micronews.debian.org
- A few interesting "hygiene" changes introduced by Debian 12: https://wiki.debian.org/NewInBookworm
- "Chapter 2. What's new in Debian 12" @ https://debian.org/releases/bookworm/amd64/release-notes/ch-whats-new.en.html

holta, 2023-06-10:
FWIW Debian 12 release notes clarify:

> 5.2.4. systemd-resolved has been split into a separate package
>
> The new systemd-resolved package will not be installed automatically on upgrades. If you were using the systemd-resolved system service, please install the new package manually after the upgrade, and note that until it has been installed, DNS resolution might no longer work since the service will not be present on the system. Installing this package will automatically give systemd-resolved control of /etc/resolv.conf. For more information about systemd-resolved, consult the official [documentation](https://www.freedesktop.org/software/systemd/man/systemd-resolved.service.html). Note that systemd-resolved was not, and still is not, the default DNS resolver in Debian. If you have not configured your machine to use systemd-resolved as the DNS resolver, no action is required.

https://www.debian.org/releases/bookworm/amd64/release-notes/ch-information.en.html?ref=news.itsfoss.com#systemd-resolved

holta, 2023-06-11:
LARGE-sized IIAB test install onto Debian 12:

http://sprunge.us/uwBt5I?en

closed 2023-06-11 by holta