iiab/iiab #3404: Mint 21 pop-up error / warning "loading /etc/profile" "sudo: a password is required"

holta, 2022-10-16:
Apparently this occurs when booting all recent IIAB installs on Mint 21:

[image]

iiab-diagnostics: http://sprunge.us/FMrBYY?bash

`sudo -n` might possibly related here:

- #3368

avni, 2022-10-16:
Despite the warning, IIAB works as expected on these new installs.

/cc @pickypet

holta, 2022-10-16:
As https://github.com/iiab/iiab/pull/3368#issuecomment-1253196311 mentions, `sudo -n` is explained in sudo's man page here:

> -n, --non-interactive
Avoid prompting the user for input of any kind. If a password is required for the command to run, sudo will display an error message and exit.

In short this pop-up is somewhat intentional, obviously annoying, and essentially harmless.

However if `/etc/profile*` can be refined in a future-proof way, without dragging IIAB into endless/costly per-distro customizations, this should definitely be considered.

jvonau, 2022-10-18:
#3300

holta, 2022-10-18:
Fixed by:

- PR #3405

closed 2022-10-18 by holta