iiab/iiab #3409: Investigate usbmount 0.0.25 as a possible improvement over 0.0.22 ?

holta, 2022-10-21:
Thanks to Andrew Davison in the UK:
https://github.com/birdslikewires/usbmount/releases

Regrettably Covid killed the prior maintainer of usbmount (Rogério Theodoro de Brito) last year:
https://www.debian.org/News/2021/20210812

Code in question is roles/usb_lib installed by roles/4-server-options:
- https://github.com/iiab/iiab/tree/master/roles/usb_lib#readme
- https://github.com/iiab/iiab/blob/master/roles/usb_lib/tasks/install.yml

Related:
- https://github.com/moodlebox/moodlebox/issues/269

holta, 2022-10-21:
usbmount 0.0.24 is also apparently an option: (if 0.0.25 is not satisfactory?)

- https://github.com/rbrito/usbmount/blob/master/debian/changelog
- https://forums.raspberrypi.com/viewtopic.php?p=1280234&sid=786f365735cb3d7309f87110821b85f8#p1280234

martignoni, 2022-10-21:
Version 0.0.25 is buggy. Use version 0.0.26 (see https://github.com/birdslikewires/usbmount/pull/1).

holta, 2022-10-21:
> Version 0.0.25 is buggy. Use version 0.0.26 (see [birdslikewires/usbmount#1](https://github.com/birdslikewires/usbmount/pull/1)).

Great news if usbmount can be actively maintained for Debian-like and Ubuntu-like Linux OS's!

Next month I and others should have more time to test/confirm — any code paths / use cases you recommend?

martignoni, 2022-10-21:
> Next month I and others should have more time to test/confirm — any code paths / use cases you recommend?

Nothing other than USB sticks with various file systems, with and without label.

holta, 2022-10-21:
> Nothing other than USB sticks with various file systems, with and without label.

Ok!  Which filesystems likely need testing?

ext4, FAT32 / VFAT, and exFAT?

martignoni, 2022-10-22:
Add NTFS and HFS+ which should work too.

holta, 2022-10-22:
We should try to compare 0.0.22 (appears very reliable over recent years) to the proposed 0.0.26.
Unfortunately I don't know the functional differences.
Using these 2 docs should help understand:

- https://github.com/moodlebox/usbmount/blob/martignoni-patch-1/README.md
- https://github.com/moodlebox/usbmount/blob/martignoni-patch-1/debian/changelog

_Many thanks to @martignoni._

holta, 2022-11-08:
>> Which filesystems likely need testing?
>>
>> ext4, FAT32 / VFAT, and exFAT?

> Add NTFS and HFS+ which should work too.

ASIDE- Android 14 is apparently bringing NTFS back life, likewise for external drives:

https://www.androidpolice.com/android-ntfs-dev-timeline/

holta, 2023-07-17:
Interesting alternative, thanks to @martignoni:

- Evaluate `devmon` as a more sustainable solution for USB auto-mounting ?
  moodlebox/moodlebox#299

Earlier work:

- PR #2607

ASIDE: Some also want to disable USB mounting entirely, for security reasons, which currently can be done manually by disabling the `usbmount@dev-sdb` service.

holta, 2023-07-20:
> Evaluate `devmon` as a more sustainable solution for USB auto-mounting ?

FYI this is now implemented by MoodleBox using apt package `udevil` and its `/etc/udevil/udevil.conf` : 

- moodlebox/moodlebox#300