iiab/iiab #3424: Question: is it possible to "archive" some sites into IIAB and access them later?

vlnn, 2022-11-16:
### Expected functionality:
During periods of fast internet I would like to be able to "archive" some sites so they will become available locally (i.e. from IIAB) during periods of no internet. 

### Steps to resolve the question myself:

I've tried to search IIAB's wiki and Issues page for this workflow pattern, but haven't got much luck. I guess there's a way to run zimit to generate zim files and add them manually to IIAB but this seems to be quite a manual task. Also perhaps there is some integration with wayback machine? This could help if I could just "download" some archives from there to my IIAB.

holta, 2022-11-16:
> Keep in touch!

Feel free also to write directly, if you prefer:

holt@unleashkids.org

vlnn, 2022-11-16:
I'm OK with having the proposal discussion open, so community could step in. Let me describe couple of user workflows that I'm interested in -- maybe some of them are already available and I just couldn't find how to use them?

### Archive of bookmarked web pages
Simplest workflow -- I read a page (e.g. https://vim.rtorr.com/) and want to share it with local community using my IIAB. I copy the address of a page into some form at IIAB (or, preferred, just click on some browser extension button) and in some time it is available as is for access either via index at http://box.local/home/ or via search (either at http://box.local:4244/ or http://box.local/kiwix or new one).
One page should be fully copied and represented as is when accessed locally. It's OK to support only static pages while representation is similar to the one in the web -- i.e. I don't expect to save youtube page and have access to the video.

### Archive the whole site
Similar to the previous, but the archive should contain representation of the at least one level of every linked page while it's situated on same domain. E.g. if we use same https://vim.rtorr.com/ example, I store locally the main page and all the localized pages linked under "Additional Resources" pane.
One level is OK, but it would be great to have possibility to either control the level of downloads or get the whole tree (similar to zim files we have pre-made for some sites). Similarly, every page should be searchable as in previous workflow.

It looks to me that this should be possible with, for instance, integration of https://github.com/openzim/zimit both into the IIAB image and admin panel so we can omit spam or non-wanted content.

vlnn, 2022-11-16:
Another way of resolving pretty much the same problem could be integration with the wayback machine -- e.g. if when I find some site I should be able to choose the snapshot (https://web.archive.org/web/20221114180355/https://vim.rtorr.com/ for instance) and get it to my IIAB instance.

tim-moody, 2022-11-16:
@vlnn have a look at the internetarchive role in iiab. It is an attempt to allow pulling content from the Internet Archive into iiab for offline use.

zimit used to use httrack and allow you to scrape a page or site into a zim. The problem is that levels easily get away from you and anything interactive fails. So they have moved to an offline Internet Archive tool. The problem for your purposes is that you  must do this in advance and copy as a zim. The problem for iiab is that it requires an https connection and we don't like the user experience of requiring the user to accept a local ssl certificate.

I do a fair amount of scraping to add modules to iiab and have been slowly building up https://github.com/tim-moody/scraper-tools. I have thought some of integrating this into Admin Console, but there are lots of obstacles.

tim-moody, 2022-11-16:
Let's start over. 

4. I got on the wrong track by not recognizing 4244 as the port for the internetarchive service in iiab, which I then suggested you try even when you had already tried it. That service is supposed to pull material from the internet archive into iiab and then render it. If that does not work, it would be great if you could open a ticket and detail your experience. The expectation is that any site you want is already in the internet archive, rather than requiring you to put it there. If that is not the case or pulling content in via the internetarchive service doesn't work, then documenting that would be a great contribution.

1 -2. The old zimit doesn't exist any more, and the new one is actively maintained, but requires iiab to run on https which we don't do. It doesn't matter if the zimmed site is https or not.

3. yes, if someone decided to implement that.

vlnn, 2022-11-16:
Great, thank you a ton -- and the issue could be closed if noone objects. I'll create a new one regarding non rendered pages from internet archive.

closed 2022-11-16 by vlnn

holta, 2022-11-16:
@vlnn you're raised some important questions here, shared by many.

Educators may in fact be badly underpaid & overworked (in almost all countries) but each is trying to solve a very limited version of the DIY scraping-on-demand problem that's you've above outlined.

(Definitely please link to this ticket for wider context, if/as you open a new one!)