iiab/iiab #3425: Internet archive webcontent should be accessible from IIAB

vlnn, 2022-11-16:
related to the discussion https://github.com/iiab/iiab/issues/3424

### Expected behavior and actual behavior:
I expect IIAB to be able to open web media archived via "Save Page Now" functionality of Wayback Machine so that I can share the snapshots of web sites at IIAB. This will provide admins an easy way to control the resources their IIAB provides.

I can't find a way to store created collection with web sites into the IIAB right now.

### Steps to reproduce the problem:

If I try to open already existing collections of mediatype web using IIAB (link http://box.local:4244/details/spn2-20200601134320) just a message **Unsupported mediatype:web** is shown:
[image]

If I try to open my newly created collection (made with https://web.archive.org/save/https://vim.rtorr.com/) https://web.archive.org/web/20221116180231/https://vim.rtorr.com/ using IIAB, I expect to find it by the URL http://box.local:4244/web/20221116180231/https://vim.rtorr.com/, I get an error `Cannot GET /web/20221116180231/https://vim.rtorr.com/`

[image]

### IIAB version, OS version, hardware details:

Latest image of IIAB for RPi

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

http://sprunge.us/wSpjga

holta, 2022-11-16:
@mitra42,

Are you possibly familiar with the `Unsupported mediatype:web` error in the screenshot above?

Any immediate suggestions to help us help educators & students here?!

mitra42, 2022-11-16:
The Offline Archive only supports the Internet Archive's collections. Most URLs follow the pattern shown, so its a good guess, but the Wayback Machine content has never been supported in the offline Internet Archive, which is why its not in the UI. The project ceased active development before I had a chance to build support for it (there are significant technical challenges doing so).

jvonau, 2022-11-17:
@vlnn Ласкаво просимо, я маю коріння з Малована

By no means am I any sort of js expert just observations from a far, I'll note within view-source:https://vim.rtorr.com/
``` 
    <script>
    window.CACHE_NAME = "3.1.1";

    if ('serviceWorker' in navigator) {
      window.addEventListener('load', function () {
        navigator.serviceWorker.register('/javascripts/sw.js?version=3.1.1');
      });
    }
  </script>
```
>Nov 16 15:28:03 box internetarchive[1172]: 2022-11-16T20:28:03.551Z dweb-transports:httptools Fetch of https://dweb.me/info opened
Nov 16 15:28:32 box internetarchive[1172]: GET /mds/v1/get_related/all/spn2-20200601134320 - - - - ms
Nov 16 15:31:45 box internetarchive[1172]: 2022-11-16T20:31:45.905Z dweb-mirror:mirrorHttp STARTING: /web/20221116180231/https://vim.rtorr.com/
Nov 16 15:31:45 box internetarchive[1172]: 2022-11-16T20:31:45.910Z dweb-mirror:mirrorHttp No file in: /opt/iiab/internetarchive/node_modules/@internetarchive/dweb-archive-dist/web/20221116180231/https:/vim.rtorr.com/ ENOENT: no such file or directory, stat '/opt/iiab/internetarchive/node_modules/@internetarchive/dweb-archive-dist/web/20221116180231/https:/vim.rtorr.com/index.html'
Nov 16 15:31:45 box internetarchive[1172]: 2022-11-16T20:31:45.911Z dweb-mirror:mirrorHttp FAILING: /web/20221116180231/https://vim.rtorr.com/
Nov 16 15:31:45 box internetarchive[1172]: GET /web/20221116180231/https://vim.rtorr.com/ - 404 180 9.426 ms

20221116180231/https:/vim.rtorr.com
Is the string missing a / ^ here? causing the ENOENT by any chance?

mitra42, 2022-11-17:
That's not the problem - as I said above, there is no support for `archive.org/web` since it works completely different than the parts that are implemented. I would not expect what you did with creating that URL to work, and it didn't work - i.e. code behaved as expected.

vlnn, 2022-11-17:
> 20221116180231/https:/vim.rtorr.com Is the string missing a / ^ here? causing the ENOENT by any chance?

Good catch! This was me mistyping when reproducing, but as @mitra42 said, alas, correct url isn't working as well due to unsupported yet format.

tim-moody, 2022-11-17:
@mitra42 thanks for the clarification; can you suggest an item that is expected to work? @vlnn while we're diving into this topic can you verify that a url that is designed to work does?

holta, 2022-11-17:
Earlier references, just fyi:

- internetarchive/dweb-mirror#364
- internetarchive/dweb-mirror#365

mitra42, 2022-11-17:
Thanks @vlnn  - yes - please post that error as a new Issue on [internetarchive/dweb-mirror](https://github.com/internetarchive/dweb-mirror/issues) and I'll take a look when I get a  chance.

vlnn, 2022-11-17:
Done: https://github.com/internetarchive/dweb-mirror/issues/366 has been created. But this issue here I guess is independent, also on really different level, much more general and more on enhancement side.

mitra42, 2022-11-18:
Agreed - I'm taking a look at a number of bugs that you have reported, thanks for that. I've not been looking at the package for a while, and haven't seen anyone else doing so, so have had to recreate my dev environment - wasted too much time today dealing with the bitrot in github (authentication changes) and some other packages. So will have to find another time to actually track own some bugs.

aamoussa, 2022-11-26:
Hello , I have the issue on my new iiab install , ubuntu server  22.04.1 LTS . Here is the return message of **service internetarchive status** : 
`● internetarchive.service - Internet Archive Universal Library service
     Loaded: loaded (/etc/systemd/system/internetarchive.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-11-26 22:32:15 UTC; 17min ago
   Main PID: 69208 (node)
      Tasks: 11 (limit: 4439)
     Memory: 110.0M
        CPU: 1min 10.242s
     CGroup: /system.slice/internetarchive.service
             └─69208 /usr/bin/node ./internetarchive -sc

nov. 26 22:49:43 box internetarchive[69208]: GET /advancedsearch.php?output=json&q=identifier%3A(opensource_audio%20OR%20community)&rows=2&page=1&sort[]=identifier&fl=identifier%2Ctitle%2Ccollection%2Cmediatype%2Cdownloads%2Ccreator%2Cnu>
nov. 26 22:49:43 box internetarchive[69208]: 2022-11-26T22:49:43.747Z dweb-transports:httptools Fetch of https://www-dweb-cors.dev.archive.org/cors/ENGLISHQuestions_201801/Carefree.png opened
nov. 26 22:49:43 box internetarchive[69208]: 2022-11-26T22:49:43.749Z dweb-mirror:MirrorFS ERROR: ArchiveFilePatched.cacheAndOrStream failed TypeError: readableStream.on is not a function at new ReadableStream (/opt/iiab/internetarchive/>
nov. 26 22:49:43 box internetarchive[69208]: 2022-11-26T22:49:43.749Z dweb-mirror:ArchiveFile Unable to cacheOrStream ENGLISHQuestions_201801/Carefree.png
nov. 26 22:49:43 box internetarchive[69208]: 2022-11-26T22:49:43.749Z dweb-mirror:mirrorHttp ERROR: streamArchiveFile failed for ENGLISHQuestions_201801/Carefree.png: readableStream.on is not a function
nov. 26 22:49:43 box internetarchive[69208]: GET /download/ENGLISHQuestions_201801/Carefree.png - 404 35 215.098 ms
nov. 26 22:49:44 box internetarchive[69208]: 2022-11-26T22:49:44.494Z dweb-transports:httptools Fetch of https://be-api.us.archive.org/mds/v1/get_related/all/ENGLISHQuestions_201801 opened
nov. 26 22:49:44 box internetarchive[69208]: 2022-11-26T22:49:44.494Z dweb-mirror:MirrorFS ERROR: ArchiveFilePatched.cacheAndOrStream failed TypeError: readableStream.on is not a function at new ReadableStream (/opt/iiab/internetarchive/>
nov. 26 22:49:47 box internetarchive[69208]: 2022-11-26T22:49:47.075Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
nov. 26 22:49:47 box internetarchive[69208]: 2022-11-26T22:49:47.557Z dweb-transports:httptools Fetch of https://dweb.me/info opened
`

holta, 2022-11-26:
Thank you @aamoussa for the report!

:microscope:

vlnn, 2022-12-24:
> @tim-moody I've tried to open some archived images at http://box.local:4244/details/ia-25-wf (stored in the internet arhive here https://archive.org/details/ia-25-wf) and it opens a viewer for the pictures [image] Alas no picture itself and some errors in internetarchive logs, but this is something I'll address here [internetarchive/dweb-mirror#365](https://github.com/internetarchive/dweb-mirror/issues/365) or on next issues there. But at least I can see a proper web page, metadata under the picture placeholder, list of files, things like that.

This is resolved for me now -- thanks again to @mitra42

mitra42, 2022-12-24:
I'm going to close this issue as its got a lot of points all of which are sorted I believe except support for WayBack which isn't going to happen (its a massive job that requires entirely different techniques)

mitra42, 2022-12-24:
@holta  - do you want to close this - I believe all the point above are now fixed except Wayback support which isn't going to happen as its a large job requiring entirely different techniques (there is for example no list of files to pull down).

vlnn, 2022-12-24:
I agree. It's still a pity there's no simple way to create local copy of some site in iiab context.

holta, 2022-12-25:
> I agree. It's still a pity there's no simple way to create local copy of some site in iiab context.

1000%

Profound thanks @mitra42 for helping everyone "kick the can down the road" — keeping the flame alive allowing us & all to ponder the Larger Question !!

closed 2022-12-25 by holta