iiab/iiab #3427: Installing tmux breaks... ansible?

johnklos, 2022-11-17:
### Expected behavior and actual behavior:

I expect that additional packages should not cause ansible, or other IIAB required packages, to break.

### Steps to reproduce the problem:

After installing IIAB, run apt -y install tmux

Or, install tmux, then try installing IIAB.

### IIAB version, OS version, hardware details:

IIAB version is not reported in /usr/sbin/iiab beyond "8.0". OS is "Linux raspberrypi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux".

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

The "iiab-diagnostics" command exited with tons of "No such file or directory" errors.

When trying to install tmux, this is what I get:

Unpacking tmux (2.8-3) ...
Setting up ansible-core (2.12.10-1ppa~focal) ...
  File "/usr/lib/python3/dist-packages/ansible/playbook/play_context.py", line 202
    if (attr_val := getattr(task, attr, None)) is not None:
                 ^
SyntaxError: invalid syntax

dpkg: error processing package ansible-core (--configure):
 installed ansible-core package post-installation script subprocess returned error exit status 1
Setting up libutempter0:armhf (1.1.6-3) ...
Setting up tmux (2.8-3) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10+rpt2+rpi1+deb10u1) ...
Errors were encountered while processing:
 ansible-core
E: Sub-process /usr/bin/dpkg returned an error code (1)

With iiab not installed, I installed tmux, then tried to install iiab, and got this:

Setting up python3-paramiko (2.4.2-0.1+deb10u1) ...
Setting up ansible-core (2.12.10-1ppa~focal) ...
  File "/usr/lib/python3/dist-packages/ansible/playbook/play_context.py", line 202
    if (attr_val := getattr(task, attr, None)) is not None:
                 ^
SyntaxError: invalid syntax

dpkg: error processing package ansible-core (--configure):
 installed ansible-core package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.28-10+rpt2+rpi1+deb10u1) ...
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 ansible-core
E: Sub-process /usr/bin/dpkg returned an error code (1)

Thanks!

holta, 2022-11-17:
@johnklos what's the result when you try _the very lastest_ **64-bit** Raspberry Pi OS from https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit ?

Please run `sudo iiab-diagnostics` prior to (and after attempting) install tmux, Thanks!

johnklos, 2022-11-17:
This is the very latest, downloaded from just a few hours ago, with a full, "apt -y update ; apt -y upgrade ; apt -y dist-upgrade" run after installation.

After manually creating /etc/iiab/diag and re-running /opt/iiab/iiab/scripts/iiab-diagnostics, I was able to get this output:

https://www.klos.com/~john/__2022-11-17_23:29:13_GMT_iiab.zia.io

This is with tmux installed but with "iiab" itself reporting that ansible error ("if (attr_val := getattr(task, attr, None)) is not None") when trying to run iiab.

Let me know if you'd like me to remove tmux, then rerun iiab and let it do its thing, then re-run /opt/iiab/iiab/scripts/iiab-diagnostics.

Thanks!

holta, 2022-11-17:
FYI this is almost definitely a result of the fact that 32-bit Raspberry Pi OS currently contains many bugs due to its pip wheels / cryptography (and Rust) being missing:

- PR #3421
- piwheels/piwheels#322
- pyca/cryptography#7651

(Use the 64-bit version of Raspberry Pi OS to avoid all these nightmares if you can.)

johnklos, 2022-11-17:
Oh - I installed "Raspberry Pi OS (Legacy)", not "Raspberry Pi OS (64-bit)", because I intend to run this on an original earmv6 system. Is this no longer supported?

holta, 2022-11-17:
> /etc/debian-version: 10.13

Please avoid Debian 10, if possible!

IIAB supports Debian 11, as well as pre-releases of Debian 12.

> I intend to run this on an original earmv6

If absolutely necessary, IIAB still generally runs on 32-bit Raspberry Pi OS for the very old "Zero W" Raspberry Pi's.

(But most everyone else should run the latest 64-bit Raspberry Pi OS, which is documented to be about 48% faster, and much most compitable with Linux packages across the board.)

johnklos, 2022-11-17:
I realize that I installed "Raspberry Pi OS (Legacy)", not "Raspberry Pi OS (64-bit)", because I intend to run this on an earmv6 system.

I don't think the OS itself is buggy - it's support for the OS that's buggy. Obviously the version of python cryptography should not have been bumped for places where Rust can't easily be built and/or run, but I don't know how the politics of Linux work with regards to simply not breaking what already works.

One improvement until someone reverts the C version of python cryptography is to have Ansible emit something less silly than "SyntaxError: invalid syntax" for something it can't find.

holta, 2022-11-18:
> I don't think the OS itself is buggy - it's support for the OS that's buggy.

Personally I'd disagree, when it comes to relying on modern cryptography.

But in any case, IIAB has not supported Debian 10 Buster in a long time.

So this is definitely not an IIAB issue.

johnklos, 2022-11-18:
I don't agree that "relying on modern cryptography" means that the OS is buggy when it's actually the support that's lacking. I only brought it up because, "32-bit Raspberry Pi OS currently contains many bugs" isn't necessarily true - it'd be accurate to say that 32 bit Raspberry Pi OS brings to light many bugs, but the OS itself isn't inherently buggy because of its 32 bitness.

But that's not really the issue. I want to use IIAB to support environments with limited equipment options. If IIAB now requires Rust, and, by extension, requires tons of memory and/or 64 bit processors, then the list of suitable hardware is much smaller than it once was.

If the intention moving forward is to fix IIAB to use the C version of python cryptography, then good. If not, then 32 bit support should be documented as unsupported, unless / until proper Rust support is available for earmv6.

Considering the goals of the IIAB project, this clarification is important.

Thank you for all your hard work!

holta, 2022-11-18:
You are misrepresenting the truth.

I will not judge whether you are doing so maliciously or out of naivete.

The truth remains:

- 32-bit Bullseye Raspberry Pi OS is well-documented to contain serious bugs since August.
- IIAB runs fine on the above OS, even despite.

closed 2022-11-18 by holta

tim-moody, 2022-11-18:
>The latest 32 bit Raspberry Pi OS (Legacy) that's available for download right now for original Pis and Pi Zeros is Debian 10 (buster).

I have the latest raspios running on a W (https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-09-26/2022-09-22-raspios-bullseye-armhf-lite.img.xz). Is there a tacit assumption that to run on legacy hardware you need a legacy OS?

holta, 2022-11-18:
1) What matters most is that Raspberry Pi "Zero W" hardware works with IIAB using the latest [32-bit Raspberry Pi (Bullseye) OS](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit).

   A lot of hard work was invested since August to make this possible &mdash; that's all too easy to take for granted  &mdash; e.g.:

   - PR #3379
   - PR #3421

2) Even despite their (dominant) 32-bit Raspberry Pi OS having a few serious flaws since August (as documented in places like PR #3421) volunteers like Ben Nuttall are working hard on these wheels / cryptography / Rust issues, such that legacy hardware will hopefully still have many uses, for years to come.

   Please support people like Ben if you want to make a legit, meaningful contribution here.

3. A reminder that we're not talking about "Zero 2 W" hardware, which also works fine with the latest 64-bit Raspberry Pi OS (preferably Lite!)

4. Finally, just a reminder that Raspberry Pi OS (Legacy, Buster) is not supported by IIAB, as this [vivid warning](https://github.com/iiab/iiab/blob/master/scripts/local_facts.fact#L76-L89) makes clear:

   [image]

jvonau, 2022-11-18:
Try using the latest 'Raspberry Pi OS Lite' -> https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-09-26/2022-09-22-raspios-bullseye-armhf-lite.img.xz not the 'Raspberry Pi OS (Legacy)' one https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2022-09-26/2022-09-22-raspios-buster-armhf-lite.img.xz that is based on buster. Yup, rust is going to be a thorny issue going forward.

johnklos, 2022-11-18:
> Is there a tacit assumption that to run on legacy hardware you need a legacy OS?

I'm helping people with more than half a dozen Pis already in use from earlier this year before the move from Debian 10 to Debian 11. Since I want to set up a test system that mirrors what they have, I used (legacy).

Since in-place upgrade of Debian from 10 to 11 doesn't work on Pis, and since Debian 10 no longer works, I'll have to figure out how to help them upgrade without breaking and without starting from scratch.

The clarifications are helpful. Thank you!

jvonau, 2022-11-18:
> > I don't think the OS itself is buggy - it's support for the OS that's buggy.
> 
> Personally I'd disagree, when it comes to relying on modern cryptography.
> 

Yet #3380 finds it way into the code base

tim-moody, 2022-11-18:
you could clone one of the legacy machines for a mirror

jvonau, 2022-11-18:
> I'm helping people with more than half a dozen Pis already in use from earlier this year before the move from Debian 10 to Debian 11. Since I want to set up a test system that mirrors what they [...]

https://github.com/iiab/iiab/issues/2978 Notes from when I went down that path

>I'll have to figure out how to help them upgrade without breaking and without starting from scratch.

Install the legacy OS and manually create /opt/iiab and git clone the repos, in each repo
`git checkout -b rollback` for a new working branch
figure out the date/git hash in use on the other machines
`git reset --hard` "githash" where the githash is what is in use on the other machines
and follow the manual directions from the FAQ starting with scripts/ansible
Contact me directly if you like some other pointers.