iiab/iiab #3428: static ip regression

tim-moody, 2022-11-22:
### Expected behavior and actual behavior:
In the past I have set a static ip address in Admin Console, saved, and ICO and got a static IP. Today this resulted in a dynamic ip.

The details are in http://sprunge.us/kdrMiI

I also tried running iiab-networking

The details are in http://sprunge.us/Y2Rdcy

holta, 2022-11-23:
@jvonau clarified that IIAB networking should today work on Ubuntu 22.04 in much same way as on 20.04.

The reason is that Netplan become "authoritative" during the 2 years between Ubuntu 18.04 (almost 5 years ago now) and Ubuntu 20.04 (31 months ago) &mdash; i.e. Netplan has not changed much since.

Still, in contrast with the pre-Netplan era, it sounds like there are a growing number of situations where customizing networking on Ubuntu benefits substantially from (or might possibly require ?) getting to know Netplan, before installing IIAB.

In any case, let's all do our best to document common customizations that are attainable and realistic, in places like:

- https://github.com/iiab/iiab/wiki/IIAB-Networking#common-customizations
- https://github.com/iiab/iiab/wiki/IIAB-Networking#10-netplan

Tangentially related:

- PR #1634
- PR #2380
- PR #3429

jvonau, 2022-11-24:
> Tim, 2:14 PM
I set the following:
gui_desired_network_role: Appliance
gui_static_wan: True
gui_static_wan_ip: 192.168.3.170
gui_static_wan_netmask: 255.255.255.0
gui_static_wan_gateway: 192.168.3.1
gui_static_wan_nameserver: 192.168.3.1
did ICO but still got dhcp ip address

>Jerry 4:59 PM
IIAB--------------------------------------------------------------------------
-rw-r--r-- 1 root root 118 Nov 20 18:34 /etc/netplan/00-installer-config.yaml
                        ...ITS LAST 100 LINES FOLLOW...
\# This is the network config written by 'subiquity'
network:
  ethernets:
    enp88s0:
      dhcp4: true
  version: 2

> Jerry 5:06 PM
> hostapd_install: False but it was installed
There never was a hostapd_install option, If you want to omit hostapd one would have to skip installing network altogether with 'network_install: False' of course that would force the use of 'network_enabled: False' also.

>Tim, 5:08 PM
>There never was a hostapd_install option
OK

>Tim, 7:04 AM
on the current demo machine:
root@iiab-me:\~# cat /etc/netplan/00-installer-config.yaml
\# This is the network config written by 'subiquity'
network:
  ethernets:
    enp0s25:
      dhcp4: true
  version: 2
root@iiab-me:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: enp0s25: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether b8:ae:ed:78:9e:dc brd ff:ff:ff:ff:ff:ff
    inet 192.168.5.170/24 brd 192.168.5.255 scope global dynamic enp0s25
       valid_lft 71448sec preferred_lft 71448sec
    inet6 fe80::baae:edff:fe78:9edc/64 scope link
       valid_lft forever preferred_lft forever

> Jerry 7:09 AM
What are you suggesting?

>Canoe, 7:11 AM
IIAB's Netplan tips are a bit more organized than a few weeks ago:
https://github.com/iiab/iiab/wiki/IIAB-Networking > Common Customizations -> 10. Netplan
IIAB Networking · iiab/iiab Wiki
https://github.com

>Tim, 7:17 AM
I'm suggesting that I made no changes to netplan on either machine and on the old machine configuring for static ip worked even though it looks like it did not in the netplan file. I'm saying that on the new machine I did not get the expected network config from iiab.

>Jerry 7:19 AM
What did the 'old' machine start out as? 22.04 from the start or was it upgraded from an earlier version?

>Tim, 7:20 AM
'old machine' is and was ub 20.04
both are bare metal installs

>Jerry 7:21 AM
[... 93 lines cut ...]
because I have only learned this from the current conversation. when I ran the config I expected the iiab would configure whatever was the network controller for a supported os
I now understand that for ub22 this is not the case

>Jerry 9:28 AM
We could, but that is not what was wanted 3-4 years ago..

>Tim, 9:29 AM
3 - 4 years ago iiab was not responsible for configuring the network?

>Jerry 9:33 AM
yes, but much of the extended functionality has been neutered to support the notion that a user can plug/unplug the WAN at will without changing the intended use of the wired device as per Mr Holt desire
netplan still is buggy and requires an interactive console session
to use 'try' or 'apply'
or that might if been fixed in the last 4 years since 18.04 was released..

>Canoe, 9:38 AM
A week ago I wrote up what Jerry told me applies to Raspberry Pi OS only:
IIAB intentionally does not support attaching Wi-Fi Access Points or Ethernet Routers to Raspberry Pi OS. The reason is that field implementers prefer hassle-free Internet using ad hoc Ethernet cables (e.g. from the Raspberry Pi IIAB to any nearby/live cablemodem). If you must attach a Wi-Fi device to your Raspberry Pi IIAB using an Ethernet cable, consider installing 64-bit Ubuntu Server instead of Raspberry Pi OS from the beginning, before you install IIAB.
https://github.com/iiab/iiab/wiki/IIAB-Networking > Common Customizations > 9. Optionally, Attach a Wi-Fi Access Point or Ethernet Router (LAN Side)
Today is the 1st time I've heard that this applies to other OS's than Raspberry Pi OS.
IIAB Networking · iiab/iiab Wiki
https://github.com

> Tim, 9:43 AM
what you just wrote is completely irrelevant to my problem

jvonau, 2022-11-24:
>Today is the 1st time I've heard that this applies to other OS's than Raspberry Pi OS.

I have to call BS on this one, https://github.com/iiab/iiab/issues/3330#issuecomment-1206793172

tim-moody, 2023-03-20:
Don't think there's anything more to add to this

closed 2023-03-20 by tim-moody