iiab/iiab #3434: munin-node.service doesn't start on current Ubuntu 23.04 daily pre-release

holta, 2022-12-13:
Hardly surprising, given Ubuntu 23.04 is in rapid flux 4 months prior to release (i.e. this worked last month, and might well fix itself in coming months or weeks).

Still, here's the current error just for the record:

> TASK [munin : Enable & Start 'munin-node' systemd service] *************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service munin-node: Job for munin-node.service failed because the control process exited with error code.\nSee \"systemctl status munin-node.service\" and \"journalctl -xeu munin-node.service\" for details.\n"}

jvonau, 2022-12-13:
> See "systemctl status munin-node.service" and "journalctl -xeu munin-node.service" for details.

And those details are what?

holta, 2022-12-29:
Still an issue 27 days later, just FYI.

jvonau, 2023-01-04:
Seems to install today

holta, 2023-01-07:
Not the end of the world:

But just FYI `systemctl restart munin-node` does not work on a fresh install of IIAB on the latest Ubuntu 23.04 daily.  As of today.

holta, 2023-01-13:
Just FYI this same problem seems to happen on Debian 12 Bookworm too.

(Though `systemctl restart munin` seems to work more reliably on Debian 12 than Ubuntu 23.04)

- #3399

holta, 2023-01-13:
Correction/Clarification: `systemctl restart munin-node` does not work on Debian 12 as of today.  Trying it repeatedly does not help.  In any case, here's the result with Node.js 19.3.0 every time:

```# systemctl status munin-node.service

× munin-node.service - Munin Node
     Loaded: loaded (/lib/systemd/system/munin-node.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-01-13 10:41:45 EST; 5s ago
       Docs: man:munin-node(1)
             http://guide.munin-monitoring.org/en/stable-2.0/reference/munin-node.html
    Process: 71948 ExecStartPre=/usr/bin/install -o munin -g munin -d /run/munin (code=exited, status=0/SUCCESS)
    Process: 71949 ExecStart=/usr/sbin/munin-node --foreground $DAEMON_ARGS (code=exited, status=1/FAILURE)
   Main PID: 71949 (code=exited, status=1/FAILURE)
        CPU: 77ms

Jan 13 10:41:45 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 5.
Jan 13 10:41:45 box systemd[1]: Stopped Munin Node.
Jan 13 10:41:45 box systemd[1]: munin-node.service: Start request repeated too quickly.
Jan 13 10:41:45 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Jan 13 10:41:45 box systemd[1]: Failed to start Munin Node.```

holta, 2023-01-13:
Munin has not released on GitHub for almost 4 years now fyi:
https://github.com/munin-monitoring/munin/releases

The writing may be on the wall.  Certainly Munin appears to be far more trouble than it's worth (a decade later, monitoring happens via systemctl and other more modern methods).

jvonau, 2023-01-13:
> Munin has not released on GitHub for almost 4 years now fyi: https://github.com/munin-monitoring/munin/releases

That seems a bit odd given there is [recent activity ](https://github.com/munin-monitoring/munin/commits/master) perhaps you should be looking at the [tags](https://github.com/munin-monitoring/munin/tags) to get the released versions.
```
ubuntu@test-2304-220112:~$ apt list *munin*
Listing... Done
groonga-munin-plugins/lunar 12.1.1+dfsg-1 all
munin-async/lunar 2.0.71-2ubuntu1 all
munin-common/lunar 2.0.71-2ubuntu1 all
munin-doc/lunar 2.0.71-2ubuntu1 all
munin-libvirt-plugins/lunar 0.0.7-1 all
munin-node-c/lunar 0.0.13-1 amd64
munin-node/lunar 2.0.71-2ubuntu1 all
munin-plugins-btrfs/lunar 12-2 all
munin-plugins-c/lunar 0.0.13-1 amd64
munin-plugins-core/lunar 2.0.71-2ubuntu1 all
munin-plugins-extra/lunar 2.0.71-2ubuntu1 all
munin-plugins-java/lunar 2.0.71-2ubuntu1 all
munin/lunar 2.0.71-2ubuntu1 all
pmuninstall/lunar 0.33-1 all
```
It appears that Ubuntu-2304 is carrying the latest offered by upstream.

holta, 2023-01-17:
> It appears that Ubuntu-2304 is carrying the latest offered by upstream.

Indeed.  In any case, the failure continues, when testing the very latest Ubuntu 23.04 daily build today, based on:

```
# apt list munin munin-node
Listing... Done
munin-node/lunar,now 2.0.71-2ubuntu1 all [installed]
munin/lunar,now 2.0.71-2ubuntu1 all [installed]
```

Recap: (1) Service `munin` starts whereas, (2) Service `munin-node` does not start.

It's unclear this really matters.  Especially if other tools (`systemctl status <XYZ>`, etc) have effectively replaced Munin years ago. 
 But FYI.

jvonau, 2023-01-18:
Fresh multipass 2304 20230110 instance
http://sprunge.us/51rZqP
```
ubuntu@portal:/opt/iiab/iiab$ sudo systemctl status munin
● munin.service - LSB: Create munin master directories on boot
     Loaded: loaded (/etc/init.d/munin; generated)
     Active: active (exited) since Wed 2023-01-18 00:48:48 CST; 4min 17s ago
       Docs: man:systemd-sysv-generator(8)
        CPU: 6ms

Jan 18 00:48:48 portal systemd[1]: Starting LSB: Create munin master directories on boot...
Jan 18 00:48:48 portal munin[4342]:  * Starting Preparing munin directories munin
Jan 18 00:48:48 portal systemd[1]: Started LSB: Create munin master directories on boot.
ubuntu@portal:/opt/iiab/iiab$ sudo systemctl status munin-node.service 
● munin-node.service - Munin Node
     Loaded: loaded (/lib/systemd/system/munin-node.service; enabled; preset: enabled)
     Active: active (running) since Wed 2023-01-18 00:48:48 CST; 4min 28s ago
       Docs: man:munin-node(1)
             http://guide.munin-monitoring.org/en/stable-2.0/reference/munin-node.html
   Main PID: 4117 (munin-node)
      Tasks: 1 (limit: 9491)
     Memory: 15.4M
        CPU: 890ms
     CGroup: /system.slice/munin-node.service
             └─4117 /usr/bin/perl -wT /usr/sbin/munin-node --foreground

Jan 18 00:48:47 portal systemd[1]: Starting Munin Node...
Jan 18 00:48:48 portal systemd[1]: Started Munin Node.
```
@holta I have to ask what is the source location of the 'very latest Ubuntu 23.04 daily build today' that you are employing? Would be helpful if something more useful other than 'it doesn't work' like the output like I posted above could be shared.

holta, 2023-01-18:
The higher level question is who truly needs Munin with IIAB.  So far after about a decade the answer appears to nobody.  As such I'm personally not willing to invest further time in tracing & debugging something that people broadly do not need.

Still, if the Munin community cares for other purposes than IIAB: FYI the munin-node systemd service consistently fails on some hardware in recent months (Multipass Ubuntu 23.04 VM's if not others, the latest every time, on a reasonably fast PC in my case) but not on other hardware (an extremely fast CPU, in @jvonau's case).

holta, 2023-01-18:
Again, I'm personally not interested in prioritizing or working something that consistently appears unwanted and unneeded.

But for those that care about Munin for other purposes:

- munin-node.service _consistently fails to start_ when installing IIAB on a fast Ubuntu 23.04 Multipass VM (and possibly other VM's too).
- munin-node.service typically _does start_ consistently after reboot of the VM
- In the most recent case, it was a LARGE-sized IIAB install, with KA Lite and Munin subsequently removed from /etc/iiab/local_vars.yml in order to allow the IIAB install to complete: http://sprunge.us/CwyFbD

jvonau, 2023-01-19:
Perhaps placing the Munin role into the same install category as phpmyadmin, monit, azuracast, calibre, pbx are in (ie install False) is in order as it is clear you have no intention in supporting Munin going forward.

holta, 2023-01-19:
Possibly in early spring when Ubuntu 23.04 is released, and Debian 12's full freeze takes effect.  Whatever decision makes most sense around that time.

holta, 2023-01-31:
> munin-node.service typically does start consistently after reboot of the VM.

Reconfirmed, on the latest Debian 12.  (Not that it's worth fixing, but FYI.)

holta, 2023-04-20:
This is still a problem more than 4 months later, on the released version of Ubuntu 23.04, e.g. with Ubuntu Server 23.04 on Raspberry Pi 😦

_Thank you to @EMG70 who just confirmed this!_

holta, 2023-04-24:
In coming weeks (or days) Munin and vnStat will be considered for removal from LARGE-sized IIAB installs.

The reason is that both are rarely used, and poorly maintained.

Conversely: if there's a desire to patch up and work around, either can be preserved if truly necessary?

holta, 2023-04-30:
Related?

- "munin-node fails to start on traditional sysvinit system"
https://groups.google.com/g/linux.debian.bugs.dist/c/-1z6jgYonOM

holta, 2023-04-30:
This 8-line pattern (that repeats throughout `/var/log/munin/munin-node.log`) might hint at root cause?

```
2023/04/30-15:27:26 Munin::Node::Server (type Net::Server::Fork) starting! pid(24398)
Resolved [*]:4949 to [::]:4949, IPv6
Not including resolved host [0.0.0.0] IPv4 because it will be handled by [::] IPv6
Binding to TCP port 4949 on host :: with IPv6
2023/04/30-15:27:26 Can't connect to TCP port 4949 on :: [Invalid argument]
  at line 66 in file /usr/share/perl5/Net/Server/Proto/TCP.pm
2023/04/30-15:27:26 Server closing!
shutdown() on unopened socket GEN0 at /usr/lib/x86_64-linux-gnu/perl-base/IO/Socket.pm line 325.
```

(On a fresh IIAB install attempt, on Ubuntu Server 23.04 VM, with no IIAB Apps/Services installed.)

While `journalctl -u munin-node | head -40` shows this 6-line-pattern that repeats:

```
Apr 30 15:27:26 box systemd[1]: Starting munin-node.service - Munin Node...
Apr 30 15:27:26 box systemd[1]: munin-node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 15:27:26 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:26 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:26 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 1.
Apr 30 15:27:26 box systemd[1]: Stopped munin-node.service - Munin Node.
Apr 30 15:27:26 box systemd[1]: Starting munin-node.service - Munin Node...
Apr 30 15:27:26 box systemd[1]: munin-node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 15:27:26 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:26 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:26 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 2.
Apr 30 15:27:26 box systemd[1]: Stopped munin-node.service - Munin Node.
Apr 30 15:27:26 box systemd[1]: Starting munin-node.service - Munin Node...
Apr 30 15:27:26 box systemd[1]: munin-node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 15:27:26 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:26 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:27 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 3.
Apr 30 15:27:27 box systemd[1]: Stopped munin-node.service - Munin Node.
Apr 30 15:27:27 box systemd[1]: Starting munin-node.service - Munin Node...
Apr 30 15:27:27 box systemd[1]: munin-node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 15:27:27 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:27 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:27 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 4.
Apr 30 15:27:27 box systemd[1]: Stopped munin-node.service - Munin Node.
Apr 30 15:27:27 box systemd[1]: Starting munin-node.service - Munin Node...
Apr 30 15:27:27 box systemd[1]: munin-node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 15:27:27 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:27 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:27 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 5.
Apr 30 15:27:27 box systemd[1]: Stopped munin-node.service - Munin Node.
Apr 30 15:27:27 box systemd[1]: munin-node.service: Start request repeated too quickly.
Apr 30 15:27:27 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:27 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:33 box systemd[1]: Starting munin-node.service - Munin Node...
Apr 30 15:27:33 box systemd[1]: munin-node.service: Main process exited, code=exited, status=1/FAILURE
Apr 30 15:27:33 box systemd[1]: munin-node.service: Failed with result 'exit-code'.
Apr 30 15:27:33 box systemd[1]: Failed to start munin-node.service - Munin Node.
Apr 30 15:27:33 box systemd[1]: munin-node.service: Scheduled restart job, restart counter is at 1.
Apr 30 15:27:33 box systemd[1]: Stopped munin-node.service - Munin Node.
Apr 30 15:27:33 box systemd[1]: Starting munin-node.service - Munin Node...
```

holta, 2023-04-30:
FYI this line[*] triggers the failure of `munin-node.service` to restart, if IIAB _HASN'T_ been rebooted:_

https://github.com/iiab/iiab/blob/d768757f7affeab24184714628bc19686c886404/roles/network/tasks/install.yml#L62

WHEREAS: `munin-node.service` starts w/o problem if IIAB _HAS_ been rebooted.

[*] Invoked in Stage 1 here:

https://github.com/iiab/iiab/blob/d768757f7affeab24184714628bc19686c886404/roles/1-prep/tasks/main.yml#L38-L40

Perhaps not beautiful, but one easy & effective workaround would be to revert to `net.ipv6.conf.all.disable_ipv6=0` temporarily (e.g. edit `/etc/sysctl.conf` and enact it by running: `sysctl -p`) both before-and-after these 2 places that (re)start `munin-node.service` :

1. https://github.com/iiab/iiab/blob/d768757f7affeab24184714628bc19686c886404/roles/munin/tasks/install.yml#L1-L10
2. https://github.com/iiab/iiab/blob/d768757f7affeab24184714628bc19686c886404/roles/munin/tasks/enable-or-disable.yml#L1-L7

holta, 2023-04-30:
> Perhaps not beautiful, but one easy & effective workaround would be to revert to `net.ipv6.conf.all.disable_ipv6=0` temporarily (e.g. edit `/etc/sysctl.conf` and enact it by running: `sysctl -p`) both before-and-after these 2 places that (re)start `munin-node.service` [...]

Done.  This issue is resolved by:

- PR #3560

holta, 2023-06-02:
Good enough!

jvonau, 2023-06-02:
IPv6 should not be disabled across the board on all interfaces just on br0 as disabling IPv6 has the potential to break internet access on IPv6 only networks that are becoming common place around the globe.

BETLOG, 2023-09-27:
If I understand correctly; this should work..right?

```
$ sudo sysctl -p
net.ipv4.ping_group_range = 1000 1000
net.ipv6.conf.all.disable_ipv6 = 0

$ sudo systemctl stop munin-node

$ sudo systemctl disable munin-node
Synchronizing state of munin-node.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable munin-node
Removed "/etc/systemd/system/multi-user.target.wants/munin-node.service".

$ sudo systemctl enable munin-node
Synchronizing state of munin-node.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable munin-node
Created symlink /etc/systemd/system/multi-user.target.wants/munin-node.service → /lib/systemd/system/munin-node.service.

$ sudo systemctl start munin-node
Job for munin-node.service failed because the control process exited with error code.
See "systemctl status munin-node.service" and "journalctl -xeu munin-node.service" for details.
```

I'm seeing multiple LU and KUbuntu machines are logging normally, and yet munin-node on at least one has been down since 2023-09-05 despite multiple reboots since then.

```
$ cat /var/log/munin/munin-node.log
023/09/27-09:14:05 Munin::Node::Server (type Net::Server::Fork) starting! pid(41724)
Resolved [*]:4949 to [::]:4949, IPv6
Not including resolved host [0.0.0.0] IPv4 because it will be handled by [::] IPv6
Binding to TCP port 4949 on host :: with IPv6
2023/09/27-09:14:05 Can't connect to TCP port 4949 on :: [Invalid argument]
  at line 66 in file /usr/share/perl5/Net/Server/Proto/TCP.pm
2023/09/27-09:14:05 Server closing!
shutdown() on unopened socket GEN0 at /usr/lib/x86_64-linux-gnu/perl-base/IO/Socket.pm line 325.
```

holta, 2023-09-27:
> If I understand correctly; this should work..right?

@BETLOG are you having luck with that?

So far Internet-in-a-Box's Munin hack (#3560) appears to work with Ubuntu 23.04 — but regrettably not with Raspberry Pi OS Bookworm pre-releases:

- #3603

BETLOG, 2023-09-27:
@holta I'm not having any luck with the sequence i posted [above](https://github.com/iiab/iiab/issues/3434#issuecomment-1736462983). But that's why I'm wondering what I'm missing.
I already have an install that is now failing, so I figured disabling net.ipv6.conf.all.disable_ipv6 and then enabling munin-node is what [#3560](https://github.com/iiab/iiab/pull/3560) essentially describes. Or did I misunderstand that? I'm assuming reinstalling isn't relevant here, and only enabling ip6 prior to the enable/start. Or is there some once-off call that only happens at install time that I am missing..and ?cannot? just invoke prior to enabling munin-node?

It may be relevant that I had ip6 disabled in network-manager, since i dont really recall when, but I checked and ENabled it prior to the commands [above](https://github.com/iiab/iiab/issues/3434#issuecomment-1736462983).
...or maybe has the patch not made it to the kubunto repos yet? (i'm not sure how to check that)

holta, 2023-09-27:
@BETLOG your questions are right on target but I do not know the answers!

Can you post your exact same questions to Munin's own repo at https://github.com/munin-monitoring/munin — and link back to this ticket if possible?

BETLOG, 2023-09-27:
> Can you post your exact same questions to Munin's own repo at https://github.com/munin-monitoring/munin — and link back to this ticket if possible?

oh...lol, sorry, my bad, I should have looked at the page header before assuming i had munin git-actual.

https://github.com/munin-monitoring/munin/issues/1579