iiab/iiab #3441: BUG: matomo install is broken

jvonau, 2022-12-23:
### Expected behavior and actual behavior:
See some sort of meaningful webpage when accessing /matomo

[image]

### Steps to reproduce the problem:
navigate to /matomo within a web browser

holta, 2022-12-23:
Coincidentally this week @cwivagg is right now looking into it.

There appears to be different Matomo problems on different OS's {Debian 12, Ubuntu 23.04, ETC}.

Hopefully this will all be understood and/or resolved within a week or so.

jvonau, 2022-12-23:
Should just make Matomo install optional, like we need another php service that could be abused as a backdoor.

jvonau, 2022-12-23:
On a side note @tim-moody iiab-admin-console does not offer the ability to enable/disable Matomo. I would of expected Matomo to appear within the 'Configuration' -> 'Services Enabled' -> 'For Monitoring and Administration' section.

[image]

holta, 2022-12-23:
The ability to disable Matomo after it's installed with IIAB is not implemented (to my knowledge).
And is probably not needed, to be honest.
Still I will ask @cwivagg about such things when I speak to him, hopefully in coming days.

jvonau, 2022-12-23:
> The ability to (programmatically) disable Matomo after it's installed with IIAB is not implemented (to my knowledge). And is probably not needed, to be honest. Still I will ask @cwivagg about such things when I speak to him, hopefully in coming days.

The service must work out of the box or be dropped from the mandatory default install.

holta, 2022-12-23:
@cwivagg just pinged me and he's working on it :heavy_check_mark:

jvonau, 2022-12-27:
What URI is matomo expected to respond to? http://box.lan/matomo, http://WANIP/matomo, and/or http://hostname.domain/matemo?

jvonau, 2022-12-27:
Accessing the website with just http://WANIP/matomo results in a 404 error but if I append index.php I'm presented with:
[image]

Might want to loosen the rules a bit to have matomo respond to the ip address of the machine also. On a side note what is the significance of the /library/www/matomo/offline-service-worker.js file and how is that file used?

shanti-bhardwa, 2022-12-27:
Here is the index.php text

<?php
/**
 * Matomo - free/libre analytics platform
 *
 * @link https://matomo.org
 * @license http://www.gnu.org/licenses/gpl-3.0.html GPL v3 or later
 */

if (!defined('PIWIK_DOCUMENT_ROOT')) {
    define('PIWIK_DOCUMENT_ROOT', dirname(__FILE__) == '/' ? '' : dirname(__FIL>
}
if (file_exists(PIWIK_DOCUMENT_ROOT . '/bootstrap.php')) {
    require_once PIWIK_DOCUMENT_ROOT . '/bootstrap.php';
}
if (!defined('PIWIK_INCLUDE_PATH')) {
    define('PIWIK_INCLUDE_PATH', PIWIK_DOCUMENT_ROOT);
}

require_once PIWIK_INCLUDE_PATH . '/core/bootstrap.php';

jvonau, 2022-12-27:
from chat:
```
shanti, 12:33 AM
yes I get the screen with this test Warning: You are now accessing Matomo from http://192.168.0.158/matomo/index.php, but Matomo has been configured to run at this address: http://box.ftp/matomo/index.php.

jvonau, 12:34 AM
yea, using the link won't work, give me a sec..

@shanti reload the webpage
should get the login page now... 

shanti, 12:37 AM
now i get sign in page

jvonau 12:38 AM
added 'enable_trusted_host_check = 0' to /library/www/matomo/config/config.ini.php
```

jvonau, 2022-12-27:
On a side note [cron](https://github.com/iiab/iiab/blob/master/roles/matomo/tasks/install.yml#L183-L198) should be in enable-or-disable.yml with the ability to disable the cron jobs when matomo is disabled, just like the rest of the roles are laid out. This would augment the ability to disable matomo that already exists within nginx.yml. The other question is around [timezone](https://github.com/iiab/iiab/blob/master/roles/matomo/tasks/install.yml#L154), it would be nice if the times shown were referenced to the local setting of the timezone based on the location deployed.

holta, 2022-12-27:
I'll try to speak with @cwivagg today if he's available.

He's working on it.

jvonau, 2022-12-29:
https://github.com/iiab/iiab/pull/3448#issuecomment-1367481272

jvonau, 2022-12-29:
https://github.com/iiab/iiab/pull/3448#issuecomment-1367510544

jvonau, 2022-12-30:
#3452

jvonau, 2022-12-30:
This piece of software is the most privacy invasive pile of code I have ever came across bar none.

jvonau, 2023-01-04:
Diagnostic --> System Check
```
### Mandatory checks

#### PHP version >= 7.2.5:
 ✔ 8.1.12-1ubuntu1

#### PDO extension:
 ✔ 

#### PDO\MYSQL extension:
 ✔ 

#### MYSQLI extension:
 ✔ 

#### Other required extensions:
 ✔ zlib ✔ json ✔ filter ✔ hash ✔ session

#### Required functions:
 ✔ debug_backtrace ✔ eval ✔ hash ✔ gzcompress ✔ gzuncompress ✔ pack

#### Required PHP configuration (php.ini):
 ✔ session.auto_start = 0 ✔ max_execution_time = 0 OR = -1 OR >= 30

#### Directories with write access:
 ✔ $DOC_ROOT/tmp ✔ $DOC_ROOT/tmp/assets ✔ $DOC_ROOT/tmp/cache ✔ $DOC_ROOT/tmp/climulti ✔ $DOC_ROOT/tmp/latest ✔ $DOC_ROOT/tmp/logs ✔ $DOC_ROOT/tmp/sessions ✔ $DOC_ROOT/tmp/tcpdf ✔ $DOC_ROOT/tmp/templates_c

### Optional checks

#### Required Private Directories:
⚠ Error: $MATOMO_URLconfig/config.ini.php⚠ Error: $MATOMO_URLtmp/cache/tracker/matomocache_general.php
We found that the above URLs are accessible via the browser, but they should NOT be. Allowing them to be accessed can pose a potential security risk since the contents can provide information about your server and potentially your users. Please restrict access to them.<br/><br/>We also found that Matomo's config directory is publicly accessible. While attackers can't read the config now, if your webserver stops executing PHP files for some reason, your MySQL credentials and other information will be available to anyone. Please check your webserver config and deny access to this directory.<br/><br/>Read this to learn more.
#### Recommended Private Directories:
⚠ Error: $MATOMO_URLtmp/⚠ Error: $MATOMO_URLtmp/empty⚠ Error: $MATOMO_URLlang/en.json We found that the above URLs are accessible via the browser, but we recommend they should not be. If possible, please restrict access to them. Read this to learn more.

#### File integrity:
 ✔ 

#### 64-bit PHP Binary:
 ✔ 

#### Tracker status:
 ✔ 

#### Memory limit:
 ✔ 128M

#### Time zone:
 ✔ 

#### Open URL:
 ✔ curl

#### PageSpeed is turned off:
 ✔ 

#### GD > 2.x + FreeType (graphics):
 ✔ 

#### Other extensions:
 ✔ json ✔ libxml ✔ dom ✔ SimpleXML ✔ openssl

#### Other functions:
 ✔ shell_exec ✔ set_time_limit ✔ mail ✔ parse_ini_file ✔ glob ✔ gzopen ✔ md5_file

#### Filesystem:
 ✔ 

#### Set up Cron (faster report-loading):
[... 165 lines cut ...]

#### Earliest invalidation ts_invalidated:
 2023-01-04 12:03:53

#### Latest invalidation ts_invalidated:
 2023-01-04 12:03:53

#### Number of segment invalidations:
 0

#### Number of plugin invalidations:
 0

#### List of plugins being invalidated:
 

#### Anonymize Referrer:
 

#### Do Not Track enabled:
 1
```
https://matomo.org/faq/troubleshooting/how-do-i-fix-the-error-private-directories-are-accessible/
https://github.com/matomo-org/matomo-nginx#readme
https://github.com/matomo-org/matomo-nginx/blob/master/sites-available/matomo.conf

jvonau, 2025-04-06:
Don't care as I disable tracking software.

closed 2025-04-06 by jvonau