iiab/iiab #3442: Please explain the use of ncat within openvpn

jvonau, 2022-12-24:
What is the intended function of the scripts within /etc/openvpn/scripts/?
https://github.com/iiab/iiab/blob/master/roles/openvpn/templates/announce
https://github.com/iiab/iiab/blob/master/roles/openvpn/templates/announcer
https://github.com/iiab/iiab/blob/master/roles/openvpn/templates/silence

holta, 2022-12-24:
@georgejhunt might know?

georgejhunt, 2022-12-24:
Each openvpn client maintains a list of clients it knows about. There is
one client,  which I'll call the server, which is running on an amazon
cloud instance at xscenet.net.  The list of clients known to xscenet.net
openvpn server only has the local addresses on the openvpn subnet
(generated during the exchange of keys when the vpn is set up.

The ncat program (announce), running a bash script, responds to a openvpn
request with the iiab handle and client local ip address. The ncat program
is essentially a simple one function service, running on each iiab client,
which responds to an openvpn request to "identify yourself" from the amazon
cloud instance.

The "silence" script brings down the ncat server when the openvpn tunnel is
being destroyed.

The code that is running at the amazon cloud instance is at
github.com/georgejhunt/xscenet

jvonau, 2022-12-27:
Then can I have a guarantee that no other commands other than '/bin/echo HANDLE = $HANDLE|UUID = $UUID' could ever be run remotely with ncat bound to every ip address that is active?

closed 2025-04-06 by jvonau