iiab/iiab #3444: IIAB 8.0 fails to install on Pi 4 running  Desktop version 2022-09-22-raspios-bullseye-arm64.img on PI 4

shanti-bhardwa, 2022-12-25:
Get this error during install over WIFI only on PI4 with 2022-09-22-raspios-bullseye-arm64.img O/S.

TASK [iiab-admin : If so, install from template: /etc/xdg/lxsession/LXDE-pi/sshpwd-lxde-iiab.sh] *********************************************************************************************************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ansible.errors.AnsibleUndefinedVariable: 'sshd_service' is undefined. 'sshd_service' is undefined
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'sshd_service' is undefined. 'sshd_service' is undefined"}

PLAY RECAP

Cannot do iiab-diagnostics as the install did not complete. Here is the copy of the output.

TASK [iiab-admin : Configure user iiab-admin / password and its group(s), if iiab_admin_user_install] ****************************************************************************************************************************************
included: /opt/iiab/iiab/roles/iiab-admin/tasks/admin-user.yml for 127.0.0.1

TASK [iiab-admin : Does user 'iiab-admin' (iiab_admin_user) exist?] **************************************************************************************************************************************************************************
changed: [127.0.0.1]

TASK [iiab-admin : Establish Linux group 'iiab-admin' group, for login to Admin Console] *****************************************************************************************************************************************************
ok: [127.0.0.1]

TASK [iiab-admin : Configure user 'iiab-admin' with group 'iiab-admin' for login to IIAB's Admin Console (http://box.lan/admin) AND for IIAB community support commands (/usr/bin/iiab-*) at the command-line] *******************************
ok: [127.0.0.1]

TASK [iiab-admin : If user didn't exist, set password to 'g0adm1n'] **************************************************************************************************************************************************************************
skipping: [127.0.0.1]

TASK [iiab-admin : Add user iiab-admin to group sudo, for IIAB community support commands in /usr/bin like {iiab-diagnostics, iiab-hotspot-on, iiab-check-firmware}, if iiab_admin_can_sudo] *************************************************
ok: [127.0.0.1]

TASK [iiab-admin : Remove user iiab-admin from group sudo, if not iiab_admin_can_sudo] *******************************************************************************************************************************************************
skipping: [127.0.0.1]

TASK [iiab-admin : Install password warning(s)] **********************************************************************************************************************************************************************************************
included: /opt/iiab/iiab/roles/iiab-admin/tasks/pwd-warnings.yml for 127.0.0.1

TASK [iiab-admin : Install /etc/profile.d/sshpwd-profile-iiab.sh from template, to issue warnings (during shell/ssh logins) if iiab-admin password is the default] ***************************************************************************
ok: [127.0.0.1]

TASK [iiab-admin : Is /etc/xdg/lxsession/LXDE-pi a directory?] *******************************************************************************************************************************************************************************
ok: [127.0.0.1]

TASK [iiab-admin : If so, install from template: /etc/xdg/lxsession/LXDE-pi/sshpwd-lxde-iiab.sh] *********************************************************************************************************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ansible.errors.AnsibleUndefinedVariable: 'sshd_service' is undefined. 'sshd_service' is undefined
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'sshd_service' is undefined. 'sshd_service' is undefined"}

PLAY RECAP ***********************************************************************************************************************************************************************************************************************************
127.0.0.1                  : ok=57   changed=10   unreachable=0    failed=1    skipped=17   rescued=0    ignored=1 

sb01@raspberrypi:~ $ uname -r
5.15.76-v8+
sb01@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
sb01@raspberrypi:~ $

holta, 2022-12-25:
I'll investigate, thanks @shanti-bhardwa

closed 2022-12-25 by holta

jvonau, 2022-12-26:
There was no error with {{ sshd_service }} until #3440 landed, which moved {{ sshd_service }} to be within roles/sshd/defaults/main.yml as opposed to using vars/default_vars.yml like all the other services touched by that PR.

holta, 2022-12-26:
That's almost true:

To complete the story, another Ansible var `systemctl_program: /bin/systemctl` moved to roles/kiwix/defaults/main.yml as part of PR #3440's reconciliation.

jvonau, 2022-12-26:
> That's almost true:
> 
> To complete the story, another Ansible var `systemctl_program: /bin/systemctl` moved to roles/kiwix/defaults/main.yml as part of PR #3440's reconciliation.

That would make the variable apply only to the kiwix role and would not be a global variable that could be used by any other role.

holta, 2022-12-26:
Exactly.

jvonau, 2022-12-26:
That is what the root cause is of the {{ sshd_service }} issue you just 'fixed', the template is part of the 'iiab-admin' role and the variable is now local to the sshd role only by virtue of there the variable is defined. If you wanted to use {{ sshd_service }} in iiab-admin it would need to be global, which it was when defined in the OS.yml files. The correct place to define that variable would be within vars/default_vars.yml.

holta, 2022-12-26:
> The correct place to define that variable would be within vars/default_vars.yml.

Humbly disagree.  It is a local variable intentionally kept within the non-inflationary (local) context of roles/kiwix for now.

(Too many global variables carries a very high cost.)

jvonau, 2022-12-26:
> > The correct place to define that variable would be within vars/default_vars.yml.
> 
> Humbly disagree. It is a local variable intentionally kept within the non-inflationary (local) context of roles/kiwix for now.
> 
> (Too many global variables carries a very high cost.)

I'm not referring to the red herring of kiwix that you tossed into this conservation, kiwix's role is the only role that uses systemctl_program, this discussion was about sshd_service that was referenced by more than one role.

holta, 2022-12-26:
Same logic applies to both, as I wrote here: https://github.com/iiab/iiab/issues/3444#issuecomment-1365278367

This is discussion isn't leading anywhere.  I'm withdrawing.

jvonau, 2022-12-26:
> Same logic applies to both, as I wrote here: [#3444 (comment)](https://github.com/iiab/iiab/issues/3444#issuecomment-1365278367)
> 
> This is discussion isn't leading anywhere. I'm withdrawing.

I disagree, just wondering which logic is the prevailing thought, having role related variables within role/defaults/main.yml or vars/default_vars.yml given the placement of mysql_service, proxy, proxy_user, smb_service, nmb_service, apache_service and apache_user that were part of OS.yml.

tim-moody, 2022-12-26:
I think the problem is the variable name sounds global, but the use in only in a specific role. probably kiwix_systemctl_prog would have been clearer. Or just hard code in the single place it is used and keep life simple.

holta, 2022-12-26:
> probably kiwix_systemctl_prog would have been clearer

Definitely.  I was tempted to do that last week, but didn't want to impinge on the original author's wishes.

> Or just hard code in the single place it is used and keep life simple.

:heavy_check_mark:

jvonau, 2022-12-26:
> > probably kiwix_systemctl_prog would have been clearer
> 
> Definitely. I was tempted to do that last week, but didn't want to impinge on the original author's wishes.

For the original authors one would have to look way back:
https://github.com/XSCE/xsce/blame/release-6.2/roles/kiwix/tasks/kiwix_install.yml
https://github.com/XSCE/xsce/blame/release-6.2/vars/Debian.yml
https://github.com/XSCE/xsce/blame/release-6.2/vars/Fedora.yml 

> > Or just hard code in the single place it is used and keep life simple.
> 
> heavy_check_mark

Now by the looks of it there was some confusion with the path within the [crontab file](https://github.com/XSCE/xsce/blame/release-6.2/roles/kiwix/tasks/kiwix_install.yml#L75-L85) using is_redhat and is_debuntu as flags for the wanted path and the already defined as of Nov 21, 2016 systemctl_program for [debian](https://github.com/XSCE/xsce/blame/release-6.2/vars/Debian.yml#L15) and [fedora](https://github.com/XSCE/xsce/blame/release-6.2/vars/Fedora.yml#L15). Could of had a single crontab stanza and the different path could of been handled by systemctl_program.

shanti-bhardwa, 2022-12-27:
Ran again after the fix and works OK. See here

http://sprunge.us/lgC3XU

Also tested 2022-09-22-raspios-bullseye-arm64-lite.img and 8.0 installs Ok