iiab/iiab #3450: Install fails on pre-release Ubuntu 23.04

jvonau, 2022-12-29:
### Expected behavior and actual behavior:
install runs to completion
### Steps to reproduce the problem:
curl install routine with option 1 selected
### IIAB version, OS version, hardware details:
as per iiab-diagnostics
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/WVQwvN

holta, 2022-12-29:
http://sprunge.us/WVQwvN?en#n-1280 pasted in here:

> 2022-12-29 01:26:58,457 p=4730 u=root n=ansible | TASK [kalite : Install packages: python2, python-setuptools, virtualenv (for Python 2)] ***
2022-12-29 01:26:58,798 p=4730 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "No package matching 'python-setuptools' is available"}

holta, 2022-12-29:
For now these pre-release OS issue are being documented via:

- https://github.com/iiab/iiab/wiki/IIAB-Platforms > Section 2. ("may require extensive babysitting to get right") > Ubuntu 23.04 > IMPORTANT INSTALL TIPS > #3399

jvonau, 2023-03-20:
https://github.com/iiab/iiab/issues/3289#issuecomment-1476378975

closed 2023-04-09 by jvonau