iiab/iiab #3455: Translate FAQ.IIAB.IO & Admin Console UX to French and Spanish ?

holta, 2023-01-01:
Thanks @nzola for suggesting this idea:

It's probably long overdue :smile: & needs careful thought :heavy_check_mark:

nzola, 2023-01-01:
I came up with this idea because I am deploying IIAB/Lokole box in the Congo DRC and it is a French speaking country. Admins there have a hard time to understand the admin procedures.
I have tried to do superficial manual translation in French on one particular install. But it will be nice and if we had an automation buttons which one can hit to change the languages of the console.

See the screenshot my manual translation in French:
[image]

nzola, 2023-01-01:
[image]

tim-moody, 2023-01-01:
From: @nzola 1/1/2023
Yes you are right. It is actually home page. I wrongly call it Admin. 
Yes home page is mainly what I need.