iiab/iiab #3467: SECURITY: Audit nginx-fpm for potienal exploitablity for CVE-2019-11043

jvonau, 2023-01-16:
https://www.tenable.com/blog/cve-2019-11043-vulnerability-in-php-fpm-could-lead-to-remote-code-execution-on-nginx with an exploit available at https://github.com/neex/phuip-fpizdam. Quick look over points to weakness in the code base.

holta, 2023-01-16:
If one takes "Support for PHP 7 has been **discontinued** [red emphasis] since **03 Nov 2022**" seriously (at https://www.php.net/releases/index.php etc) then OS's like Debian 11 and its PHP 7.4 should be avoided.

Their EOL page makes clear that PHP 7.x communities should assume they are exposing themselves to security vulnerabilities and bugs: https://www.php.net/eol.php

tim-moody, 2023-01-16:
actually it says _using older versions may expose you to security vulnerabilities and bugs that have been fixed in more recent versions of PHP_. In any event 7.4 has the fix for the subject of this ticket.

>OS's like Debian 11 Bullseye and its PHP 7.4 should be avoided.

that would mean avoiding Debian, as 12 will not be released for some time yet. what CVE are you concerned about?

holta, 2023-01-16:
Again just for the record PHP 7 is no longer supported, and communities that embrace it are asking for trouble:

[image]

https://www.php.net/supported-versions

jvonau, 2023-01-16:
Never assume when it come to security but the exploit was [corrected with](https://github.com/php/php-src/commit/ab061f95ca966731b1c84cf5b7b20155c0a1c06a) with change being carried forward into 8+ That doesn't preclude to using better code in the configuration files that would guard against other undiscovered exploits that may or may not be present. I'll point out the use of explicit blocking of sub-paths within [upstream sample configs](https://github.com/matomo-org/matomo-nginx/blob/master/sites-available/matomo.conf). I can point out other examples but I think I made my point.

[image]

The above redacted response points to the nagging question of is RasPiOS the best choice of an OS for web-serving.

holta, 2023-01-16:
> The above redacted response points to the nagging question of is RasPiOS the best choice of an OS for web-serving.

I don't know what was redacted, but yes I agree it's important to remind "enterprise" users that RasPi OS is sometimes fragile.

ASIDE: "there will be no bug or security fixes made to any of the PHP 7 versions" according to https://php.watch/news/2021/11/php7-eol

Then they slightly contradict themselves further down on the page, suggesting that big bucks "enterprise" communities may be able to pay Red Hat and/or Ubuntu to salvage PHP 7.4 in coming years (maybe!)