iiab/iiab #3468: roles/osm-vector-maps installs fail repeatedly when GitHub throttling or ISP peering causes 1 of its ~40 file downloads (at random) to stall

holta, 2023-01-17:
Problem occurred repeatedly all morning and again this afternoon, despite other downloads working perfectly.  On the 4th try (keep running `sudo iiab` many times ;) it finally worked!

Summary: IIAB's OSM role can fails (consistently, in a different place every time) when GitHub file downloads arrives about 1000X more slowly (and more erratically) than on other days.

Likely the problem will go away tomorrow and/or with another ISP's?

In any case the root cause is that IIAB's OSM role needs to download about ~40 different files, and so when downloads become painfully slow and somewhat erratic, a frustrating situation arises where [roles/osm-vector-maps](https://github.com/iiab/iiab/tree/master/roles/osm-vector-maps) gets close to installing every time &mdash; but then one of those ~40 file downloads (at random) causes the entire Ansible job to fail.

(So be it.  Something to monitor if it recurs regularly.)

holta, 2023-01-18:
Just a reminder that IIAB sets a download timeout for good reason, so that such ISP & mirroring glitches are properly brought to the attention of the person installing IIAB:

https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L49-L50

(At the same time everyone is free to raise this number in /etc/iiab/local_vars yml from 100 seconds to a higher number — e.g. if they use an old/unreliable/rural Internet connection — or face download/mirroring/ISP/peering issues further upstream as in this case!)

jvonau, 2023-01-19:
One could clone the repo with a depth of 1 and then use 'copy' in place of get_url but that results in reduced space available within the system due to duplicate files being present.

holta, 2023-01-31:
ISP/peering issue has not recurred in 2 weeks thankfully; good enough.

closed 2023-01-31 by holta