iiab/iiab #3472: nodesource 18.x LTS installer appears to block all IIAB installs (?) due to expired https cert

holta, 2023-01-31:
Likely they'll fix this very shortly in coming hours hopefully?  As countless thousands of people are being affected, unable to install Node.js in the usual/mainline way.

Just for example on Debian 12, running `apt update` shows:

```
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 184.25.58.144 443]
...
W: Failed to fetch https://deb.nodesource.com/node_18.x/dists/bookworm/InRelease  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 184.25.58.144 443]
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

FYITthe usual interim workaround probably doesn't work in this case unfortunately, i.e. adding the following line in /etc/iiab/local_vars.yml probably doesn't change things in this case:

```
nodejs_version: 19.x
```

Upstream issue:

- nodesource/distributions#1507

MarianoRD, 2023-01-31:
Duplicate of https://github.com/nodesource/distributions/issues/1507

holta, 2023-01-31:
Nodesource.com fixed their cert ~30min ago.

closed 2023-01-31 by holta

tim-moody, 2023-01-31:
many prefer nvm to install node as supports multiple versions.