iiab/iiab #3475: BUG: Find a better place for pid files

jvonau, 2023-02-08:
### Expected behavior and actual behavior:
Not see pid files from running apps in the root of the filesystem.
```
jerryv@box:/opt/iiab/iiab $ ls -la /jupyterhub-proxy.pid 
-rw-r--r-- 1 root root 3 Feb  4 08:17 /jupyterhub-proxy.pid
```
### Steps to reproduce the problem:
install jupyterhub
### IIAB version, OS version, hardware details:
current master
Should be using /run as the location for the pid files as the distros do.
```
jerryv@box:/opt/iiab/iiab $ ls -la /run/*.pid
-rw-r--r-- 1 root root 4 Feb  4 08:17 /run/acpid.pid
-rw-r--r-- 1 root root 4 Feb  4 08:17 /run/crond.pid
-rw-r--r-- 1 root root 6 Feb  8 05:54 /run/dhcpcd.pid
-rw-r--r-- 1 root root 6 Feb  8 05:54 /run/hostapd.pid
-rw-r--r-- 1 root root 4 Feb  4 08:17 /run/iiab-cmdsrv.pid
-rw-r--r-- 1 root root 5 Feb  8 05:48 /run/nginx.pid
-rw-r--r-- 1 root root 4 Feb  4 08:17 /run/rngd.pid
-rw-r--r-- 1 root root 4 Feb  4 08:17 /run/sshd.pid
```
Perhaps a configuration [variable](https://github.com/iiab/iiab/blob/master/roles/jupyterhub/templates/jupyterhub_config.py.j2#L470-L472) could be set?

holta, 2023-02-08:
Presumably `jupyterhub_config.py.j2` needs other modernizations as well, as the years ago by:

- jupyterhub/jupyterhub#3004

Above recommendation:

```python
c.ConfigurableHTTPProxy.pid_file = "/var/run/jupyterhub-proxy.pid"
```

jvonau, 2023-02-08:
Note that /var/run/ is a symlink to /run on modern OS, I would just use /run. On a side note perhaps [upstream jupyterhub_config.py](https://github.com/jupyterhub/jupyterhub/blob/main/testing/jupyterhub_config.py) could be utilized and then customized with only the lines that are of interest being changed with lineinfile or ini_file in place using a severely outdated template that is currently in this repo.

holta, 2023-02-08:
> perhaps upstream jupyterhub_config.py could be utilized and then customized

That would be best yes, if it doesn't introduce brittleness of other kinds

jvonau, 2023-02-08:
How you fix this issue is up to you, just correct this oversight. On a side note, more brittle than not being able to install on [RasPiOS-32](https://github.com/iiab/iiab/pull/3421#issuecomment-1309743628)?

holta, 2023-02-10:
Resolved by:

- PR #3476

closed 2023-02-10 by holta