iiab/iiab #3489: Asterisk did not install on 64-bit RasPiOS 2023-02-21

holta, 2023-03-08:
On #3487 @jvonau noted:

> there are apt packages with 'armhf' during 'install_prereq install' of asterisk, that is surely wrong as the arch is wrong, should be arm64

http://sprunge.us/4HdW7J?en#n-3460 (pasted in below) should be investigated to track this down (if there's time, given the reality that Asterisk & FreePBX don't really belong on Raspberry Pi).  Still, they did install successfully in the past, so this might turn out to be very easily fixable.

Just FYI Asterisk & FreePBX were installed successfully on Debian 11 as part of IIAB on x86_64 just last week (i.e. on a regular PC/laptop/VM).

> 3460 /opt/iiab/iiab/iiab-debug.log-2023-02-23 23:07:58,593 p=36116 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "export DEBIAN_FRONTEND=noninteractive && ./contrib/scripts/install_prereq install", "delta": "0:02:43.449113", "end": "2023-02-23 23:07:58.368214", "msg": "non-zero return code", "rc": 1, "start": "2023-02-23 23:05:14.919101", "stderr": "", "stderr_lines": [], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following package was automatically installed and is no longer required:\n  libfuse2\nUse 'apt autoremove' to remove it.\nThe following additional packages will be installed:\n  aptitude-common libcwidget4 libsigc++-2.0-0v5 libxapian30\nSuggested packages:\n  apt-xapian-index aptitude-doc-en | aptitude-doc debtags libcwidget-dev\n  xapian-tools\nThe following NEW packages will be installed:\n  aptitude aptitude-common libcwidget4 libsigc++-2.0-0v5 libxapian30\n0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.\nNeed to get 4,275 kB of archives.\nAfter this operation, 18.3 MB of additional disk space will be used.\nGet:1 http://deb.debian.org/debian bullseye/main arm64 aptitude-common all 0.8.13-3 [1,776 kB]\nGet:2 http://deb.debian.org/debian bullseye/main arm64 libsigc++-2.0-0v5 arm64 2.10.4-2 [30.6 kB]\nGet:3 http://deb.debian.org/debian bullseye/main arm64 libcwidget4 arm64 0.5.18-5 [280 kB]\nGet:4 http://deb.debian.org/debian bullseye/main arm64 libxapian30 arm64 1.4.18-3 [1,013 kB]\nGet:5 http://deb.debian.org/debian bullseye/main arm64 aptitude arm64 0.8.13-3 [1,175 kB]\nFetched 4,275 kB in 0s (11.9 MB/s)\nSelecting previously unselected package aptitude-common.\r\n(Reading database ... \r(Reading database ... 5%\r(Reading database ... 10%\r(Reading database ... 15%\r(Reading database ... 20%\r(Reading database ... 25%\r(Reading database ... 30%\r(Reading database ... 35%\r(Reading database ... 40%\r(Reading database ... 45%\r(Reading database ... 50%\r(Reading database ... 55%\r(Reading database ... 60%\r(Reading database ... 65%\r(Reading database ... 70%\r(Reading database ... 75%\r(Reading database ... 80%\r(Reading database ... 85%\r(Reading database ... 90%\r(Reading database ... 95%\r(Reading database ... 100%\r(Reading database ... 125503 files and directories currently installed.)\r\nPreparing to unpack .../aptitude-common_0.8.13-3_all.deb ...\r\nUnpacking aptitude-common (0.8.13-3) ...\r\nSelecting previously unselected package libsigc++-2.0-0v5:arm64.\r\nPreparing to unpack .../libsigc++-2.0-0v5_2.10.4-2_arm64.deb ...\r\nUnpacking libsigc++-2.0-0v5:arm64 (2.10.4-2) ...\r\nSelecting previously unselected package libcwidget4:arm64.\r\nPreparing to unpack .../libcwidget4_0.5.18-5_arm64.deb ...\r\nUnpacking libcwidget4:arm64 (0.5.18-5) ...\r\nSelecting previously unselected package libxapian30:arm64.\r\nPreparing to unpack .../libxapian30_1.4.18-3_arm64.deb ...\r\nUnpacking libxapian30:arm64 (1.4.18-3) ...\r\nSelecting previously unselected package aptitude.\r\nPreparing to unpack .../aptitude_0.8.13-3_arm64.deb ...\r\nUnpacking aptitude (0.8.13-3) ...\r\nSetting up libxapian30:arm64 (1.4.18-3) ...\r\nSetting up libsigc++-2.0-0v5:arm64 (2.10.4-2) ...\r\nSetting up aptitude-common (0.8.13-3) ...\r\nSetting up libcwidget4:arm64 (0.5.18-5) ...\r\nSetting up aptitude (0.8.13-3) ...\r\nupdate-alternatives: using /usr/bin/aptitude-curses to provide /usr/bin/aptitude (aptitude) in auto mode\r\nProcessing triggers for man-db (2.9.4-2) ...\r\nProcessing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...\r\nHit http://security.debian.org/debian-security bullseye-security InRelease\nHit http://deb.debian.org/debian bullseye InRelease\nHit http://deb.debian.org/debian bullseye-updates InRelease\nHit http://archive.raspberrypi.org/debian bullseye InRelease\nHit http://ppa.launchpad.net/learningequality/kolibri/ubuntu focal InRelease\nHit https://deb.nodesource.com/node_18.x bullseye InRelease\nReading package lists...\nBuilding dependency tree...\nReading state information...\nInitializing package states...\nWriting extended state information...\nBuilding tag database...\nReading package lists...\nBuilding dependency tree...\nReading state information...\nReading extended state information...\nInitializing package states...\nWriting extended state information...\nBuilding tag database...\nThe following NEW packages will be installed:\n  autoconf{a} automake{a} autotools-dev{a} bash:armhf bcc{a} bin86{a} binutils-common:armhf{a} binutils-dev:armhf{b} bison:armhf build-essential:armhf{b} bzip2:armhf comerr-dev:armhf{a} doxygen:armhf elks-libc{a} flex:armhf fonts-liberation{a} freetds-common{a} freetds-dev:armhf gcc-10-base:armhf{a} gir1.2-glib-2.0:armhf{a} gir1.2-gmime-3.0{a} gir1.2-gmime-3.0:armhf{a} gir1.2-ical-3.0{a} gir1.2-ical-3.0:armhf{a} graphviz:armhf icu-devtools{a} krb5-locales{a} krb5-multidev:armhf{a} libapr1:armhf{a} libaprutil1:armhf{a} libasound2:armhf{a} libasound2-dev libasound2-dev:armhf libassuan0:armhf{a} libaudit1:armhf{a} libbinutils:armhf{a} libblkid-dev{a} libblkid-dev:armhf{a} libblkid1:armhf{a} libbluetooth-dev libbluetooth-dev:armhf libbluetooth3:armhf{a} libbrotli1:armhf{a} libbsd-dev{a} libbsd-dev:armhf{a} libbsd0:armhf{a} libbz2-1.0:armhf{a} libc-client2007e:armhf{a} libc-client2007e-dev:armhf libc6:armhf{a} libc6-dev:armhf{a} libcairo2:armhf{a} libcap-dev libcap-dev:armhf libcap-ng0:armhf{a} libcap2:armhf{a} libcdt5:armhf{a} libcfg-dev libcfg-dev:armhf libcfg7{a} libcfg7:armhf{a} libcgraph6:armhf{a} libclang-cpp11:armhf{a} libclang1-11:armhf{a} libcodec2-0.9:armhf{a} libcodec2-dev:armhf libcom-err2:armhf{a} libcorosync-common-dev libcorosync-common-dev:armhf libcorosync-common4{a} libcorosync-common4:armhf{a} libcpg-dev libcpg-dev:armhf libcpg4{a} libcpg4:armhf{a} libcrypt-dev:armhf{a} libcrypt1:armhf{a} libct4:armhf{a} libctf-nobfd0:armhf{a} libctf0:armhf{a} libcurl4:armhf{a} libcurl4-openssl-dev libcurl4-openssl-dev:armhf libdatrie1:armhf{a} libdb-dev{ab} libdb-dev:armhf{a} libdb5.3:armhf{a} libdb5.3-dev:armhf{a} libdbus-1-3:armhf{a} libdbus-1-dev:armhf{a} libdeflate-dev{a} libdeflate-dev:armhf{a} libdeflate0:armhf{a} libedit-dev libedit-dev:armhf libedit2:armhf{a} libevent-2.1-7:armhf{a} libexpat1:armhf{a} libffi-dev{a} libffi-dev:armhf{a} libffi7:armhf{a} libfftw3-bin{a} libfftw3-dev libfftw3-dev:armhf libfftw3-double3:armhf{a} libfftw3-long3{a} libfftw3-single3:armhf{a} libfl-dev:armhf{a} libfl2:armhf{a} libflac-dev{a} libflac-dev:armhf{a} libflac8:armhf{a} libfontconfig1:armhf{a} libfreetype6:armhf{a} libfribidi0:armhf{a} libgcc-s1:armhf{a} libgcrypt20:armhf{a} libgd3:armhf{a} libgdbm-compat4:armhf{a} libgdbm6:armhf{a} libgirepository-1.0-1:armhf{a} libglib2.0-0:armhf{a} libglib2.0-dev{a} libglib2.0-dev:armhf{a} libglib2.0-dev-bin{a} libgmime-3.0-0{a} libgmime-3.0-0:armhf{a} libgmime-3.0-dev libgmime-3.0-dev:armhf libgmp10:armhf{a} libgnutls30:armhf{a} libgomp1:armhf{a} libgpg-error-l10n{a} libgpg-error0:armhf{a} libgpgme11:armhf{a} libgpm2:armhf{a} libgraphite2-3:armhf{a} libgsm1:armhf{a} libgsm1-dev libgsm1-dev:armhf libgssapi-krb5-2:armhf{a} libgssrpc4:armhf{a} libgts-0.7-5{a} libgts-0.7-5:armhf{a} libgts-bin{a} libgvc6:armhf{a} libgvpr2:armhf{a} libharfbuzz0b:armhf{a} libhogweed6:armhf{a} libical-dev libical-dev:armhf libical3:armhf{a} libice6:armhf{a} libicu-dev{a} libicu-dev:armhf{a} libicu67:armhf{a} libidn2-0:armhf{a} libiksemel-dev libiksemel-dev:armhf libiksemel3{a} libiksemel3:armhf{a} libjack-jackd2-0:armhf{a} libjack-jackd2-dev libjack-jackd2-dev:armhf libjansson-dev libjansson-dev:armhf libjansson4:armhf{a} libjbig-dev{a} libjbig-dev:armhf{a} libjbig0:armhf{a} libjpeg-dev{a} libjpeg-dev:armhf{a} libjpeg62-turbo:armhf{a} libjpeg62-turbo-dev{a} libjpeg62-turbo-dev:armhf{a} libk5crypto3:armhf{a} libkadm5clnt-mit12:armhf{a} libkadm5srv-mit12:armhf{a} libkdb5-10:armhf{a} libkeyutils1:armh [... cut ...] ut will NOT be installed:", "  libtool-bin ", "0 packages upgraded, 392 newly installed, 10 to remove and 0 not upgraded.", "Need to get 176 MB of archives. After unpacking 574 MB will be used.", "The following ESSENTIAL packages will be REMOVED!", "  bash ", "", "WARNING: Performing this action will probably cause your system to break!", "         Do NOT continue unless you know EXACTLY what you are doing!", "To continue, type the phrase \"I am aware that this is a very bad idea\":", "Abort."]}

jvonau, 2023-03-13:
Noting that RasPiOS is 32bit which equates to 'armhf' has no mention of 'foreign-architectures' within iiab-summary's Arch2 below while 64-bit RasPiOS does mention ['foreign-architectures'](http://sprunge.us/4HdW7J?en#n-21) at Arch2
```
Arch1: armhf   Arch2: 
```
Refs: https://github.com/iiab/iiab/issues/2873#issuecomment-884353815 299adb17da3d8bb3e97f7cb87276c23b5f90984c

Knowing both 32bit and 64bit images are built with the same [tool](https://github.com/RPi-Distro/pi-gen/) abet different branches, a quick [diff](https://github.com/RPi-Distro/pi-gen/compare/master...arm64) shows that https://github.com/RPi-Distro/pi-gen/commit/99399d30fc7c80553cf50899388ff8ecb278867d introduced `dpkg --add-architecture armhf` into the 64bit images on July 30, 2020 and has been that way ever since. Don't know why this was done, the QA department of this project could ask upstream the reasons for the change but suspect that is the root cause of this issue here.

Two workarounds come to mind, first just undo the change with `dpkg --remove-architecture armhf` just before the 'install_prereq install' stanze. The second would be to install what `install_prereq install` does (below) without using `install_prereq`
```
# Basic build system:
PACKAGES_DEBIAN="build-essential pkg-config"
# Asterisk: basic requirements:
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libedit-dev libjansson-dev libsqlite3-dev uuid-dev libxml2-dev"
# Asterisk: for addons:
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libspeex-dev libspeexdsp-dev libogg-dev libvorbis-dev libasound2-dev portaudio19-dev libcurl4-openssl-dev xmlstarlet bison flex"
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libpq-dev unixodbc-dev libneon27-dev libgmime-2.6-dev libgmime-3.0-dev liblua5.2-dev liburiparser-dev libxslt1-dev libssl-dev"
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libmysqlclient-dev libbluetooth-dev libradcli-dev freetds-dev libosptk-dev libjack-jackd2-dev bash libcap-dev"
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libsnmp-dev libiksemel-dev libcorosync-common-dev libcpg-dev libcfg-dev libnewt-dev libpopt-dev libical-dev libspandsp-dev"
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libresample1-dev libc-client2007e-dev binutils-dev libsrtp0-dev libsrtp2-dev libgsm1-dev doxygen graphviz zlib1g-dev libldap2-dev"
PACKAGES_DEBIAN="$PACKAGES_DEBIAN libcodec2-dev libfftw3-dev libsndfile1-dev libunbound-dev"
# Asterisk: for the unpackaged below:
PACKAGES_DEBIAN="$PACKAGES_DEBIAN wget subversion"
# Asterisk: for ./configure --with-pjproject-bundled:
PACKAGES_DEBIAN="$PACKAGES_DEBIAN bzip2 patch"
```
Personally I would leave out the installation of 'aptitude' that `install_prereq` forces upon the enduser in a different part of the script and on a hunch omit 'bash' from the above list as the error message mentions removing bash to install the armhf version.

jvonau, 2023-03-13:
One data point of interest on a pristine 64bit image would be `apt install bash` and see if 'armhf' would get dragged in to the mix (one does not have to go through with the install just note what is going to be installed) if the 'armhf' version is not being installed proceed to install aptitude with `apt install aptitude` and repeat the `apt install bash` test.

holta, 2023-03-13:
Great Tips👍

I'm optimistic this will be solved soon in coming months (ideally weeks, when I/anybody hopefully find time to work on this).

jvonau, 2023-03-17:
Until then take steps to prevent users from shooting themselves when the combination is known not to install.

holta, 2023-03-17:
I prefer that we (volunteers who use Asterisk) get more involved in coming weeks/months, to help confirm and understand the situation.

I'm optimistic this will happen in the spring, ideally rather soon, as 64-bit RasPiOS gains more traction.  Possibly even serving a real purpose ;)

jvonau, 2023-03-19:
You do realize that this arch mismatch issue has been faced before on X64_64 when upstream was using i686 libraries in the past, that is where the [asterisk_dependencies.yml](https://github.com/holta/iiab/blob/master/roles/pbx/tasks/asterisk.yml#L5-L7) file came into play in order to have install_prereq not puke in the same way on that arch in the past. You have been presented with 2 potential workarounds, I can only point to the well, you must carry your own water to put the fire out.

EMG70, 2023-03-31:
iiab with Astersiks 19 & 20 diagnostics
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
em@box:~ $

http://sprunge.us/L9zu3v
http://sprunge.us/J6L0Ym

EMG70, 2023-03-31:
[image]

EMG70, 2023-03-31:
I hope I posted on the right platform

EMG70, 2023-03-31:
I got same error with Asterisk version 19 and 20 on 64 bit RasPios

holta, 2023-03-31:
Thank you @EMG70 for posting.

I don't guarantee IIAB can fix this.

Bbut hopefully your iiab-diagnostics links will shed proper light — as to why it used to work — and obviously no longer does 🤔

jvonau, 2023-04-01:
> I don't guarantee IIAB can fix this.
 
I can but will not.
> But hopefully your iiab-diagnostics links will shed proper light — as to why it used to work — and obviously no longer does 

Don't think it used to work on 64-bit RasPiOS ever but works on 32bit. Also works on Ubuntu-arm64 use the MATE-22.04-desktop or U-server but if you have one where RaspPiOS-64-bit worked in the past post the iiab-diagnostics.

jvonau, 2023-04-01:
>pbx_use_apache: True    # 2021-08-17: Set either to 'False' if nec -- please
pbx_use_nginx: False     # read github.com/iiab/iiab/issues/2914 & #2916, THX!

So much for putting the effort into nginx's config and then not promoting the nginx configuration, waste of my time.

holta, 2023-04-01:
@EMG70 tried disabling NGINX just to see if that would make any difference.

Whether or not he makes progress in the end, it sounds like he volunteered many hours today just in case this helps others in his home country (Zimbabwe).

jvonau, 2023-04-01:
Pending Gerrit Account setup for pull request.

jvonau, 2023-04-01:
The second option in https://github.com/iiab/iiab/issues/3489#issuecomment-1466057006 is viable as it was in the past or patch install_prereq as noted in above PR prior to installing.

holta, 2023-04-01:
> The second option in [#3489 (comment)](https://github.com/iiab/iiab/issues/3489#issuecomment-1466057006) is viable as it was in the past or [patch](https://patch-diff.githubusercontent.com/raw/asterisk/asterisk/pull/32.diff) install_prereq as noted in above PR prior to installing.

Promising.

@EMG70 please take a look In April if possible.

Regardless if resolution (one way or another) isn't easy in 64-bit RasPiOS.

ASIDE: @EMG70 tried essentially every different version of RasPiOS yesterday, if I'm understanding his many attempts, and had no luck with any of them 😵‍💫

holta, 2023-04-01:
CLARIF: @EMG70 is trying to become more adept with the Linux Command-Line.

So possibly this will help over the course of April.

(If "surgery" is indeed required...)

EMG70, 2023-04-01:
I will give it a try in April.Thank you very much.

jvonau, 2023-04-01:
Remember this role was originally intended for ubuntu only in the past, but in the course of testing 32-bit RasPiOS the install just worked as there is no arm64 deb packages within the apt repos to be confused over. 
Random package from PACKAGES_DEBIAN above:
```
jerryv@raspi-lite-64bit:~ $ apt list libsnmp-dev
Listing... Done
libsnmp-dev/stable,stable-security,now 5.9+dfsg-4+deb11u1 arm64 [installed]
libsnmp-dev/stable,stable-security 5.9+dfsg-4+deb11u1 armhf
```
That is the crux of the issue, aptitude was trying install both arches without the patch as the 64-bit image is using debian repos where there are both 32&64bit deb packages within the apt repos while 32-bit raspbian only offers 'armhf'

https://docs.ansible.com/ansible/latest/collections/ansible/posix/patch_module.html should someone take the lead and produce a PR while I **might** get the patch accepted upstream.

holta, 2023-04-01:
> ubuntu only

Just an Aside: @EMG70 has Asterisk & FreePBX working well on Debian 11.

(Whereas Ubuntu 22.04+ and its cousin OS Mint 21 are not yet appropriate, as FreePBX 16 does not work with PHP 8.)

Hypothetically the company [Sangoma] will resolve this dilemma, by releasing FreePBX 17 in the coming year-or-so.  Certainly their corporate customers _cannot be happy_ that PHP 7 was [End-of-Life'd](https://www.php.net/eol.php) in [November 2022](https://www.php.net/releases/), exposing all FreePBX installations to an increasing number of security vulnerabilities going forward.

jvonau, 2023-04-01:
> > ubuntu only
> 
> Just an Aside: @EMG70 has Asterisk & FreePBX working well on Debian 11.
>

I'll guess on X86_64 where the stock install_prereq deals correctly with 'i386' in the mixed debian apt repo.
 
> (Whereas Ubuntu 22.04+ and its cousin OS Mint 21 are not yet appropriate, as FreePBX 16 does not "yet" work with PHP 8.)
>

Odd thought that did work or is this a case of trying the bleeding edge? I'll check if I find time..

holta, 2023-04-01:
Asterisk 20 and FreePBX 16 install on x86_64 Debian 11, yes.

Whereas FreePBX 16 refuses to install on any PHP 8 OS (I did not try to force it to proceed).

jvonau, 2023-04-02:
U-22.04 VM installs but fails at the database provisioning
ansible
```
TASK [pbx : FreePBX - 2-step install - won't run if /var/www/html/freepbx already exists - FAST W/ GITHUB (OR freepbx-16.0-latest.tgz CAN TAKE 3-12 MIN OR LONGER!)] ***
changed: [127.0.0.1] => (item=./start_asterisk start)
failed: [127.0.0.1] (item=./install -n --webroot /var/www/html/freepbx --dbuser asterisk --dbpass asterisk) => {"ansible_loop_var": "item", "changed": true, "cmd": ["./install", "-n", "--webroot", "/var/www/html/freepbx", "--dbuser", "asterisk", "--dbpass", "asterisk"], "delta": "0:00:00.041930", "end": "2023-04-02 00:36:39.426387", "item": "./install -n --webroot /var/www/html/freepbx --dbuser asterisk --dbpass asterisk", "msg": "non-zero return code", "rc": 255, "start": "2023-04-02 00:36:39.384457", "stderr": "PHP Fatal error:  Declaration of FreePBX\\Database::query() must be compatible with PDO::query(string $query, ?int $fetchMode = null, mixed ...$fetchModeArgs): PDOStatement|false in /opt/iiab/freepbx/amp_conf/htdocs/admin/libraries/BMO/Database.class.php on line 240", "stderr_lines": ["PHP Fatal error:  Declaration of FreePBX\\Database::query() must be compatible with PDO::query(string $query, ?int $fetchMode = null, mixed ...$fetchModeArgs): PDOStatement|false in /opt/iiab/freepbx/amp_conf/htdocs/admin/libraries/BMO/Database.class.php on line 240"], "stdout": "Checking if SELinux is enabled...Its not (good)!\nReading /etc/asterisk/asterisk.conf...Done\nChecking if Asterisk is running and we can talk to it as the 'asterisk' user...Yes. Determined Asterisk version to be: 20.2.0\nChecking if NodeJS is installed and we can get a version from it...Yes. Determined NodeJS version to be: 18.15.0\nPreliminary checks done. Starting FreePBX Installation\nChecking if this is a new install...Yes (No /etc/freepbx.conf file detected)\nDatabase installation checking credentials and permissions..Connected!", "stdout_lines": ["Checking if SELinux is enabled...Its not (good)!", "Reading /etc/asterisk/asterisk.conf...Done", "Checking if Asterisk is running and we can talk to it as the 'asterisk' user...Yes. Determined Asterisk version to be: 20.2.0", "Checking if NodeJS is installed and we can get a version from it...Yes. Determined NodeJS version to be: 18.15.0", "Preliminary checks done. Starting FreePBX Installation", "Checking if this is a new install...Yes (No /etc/freepbx.conf file detected)", "Database installation checking credentials and permissions..Connected!"]}
```
Cmdline
```
ubuntu@box:/opt/iiab/freepbx$ sudo ./install -n --webroot /var/www/html/freepbx --dbuser asterisk --dbpass asterisk
Checking if SELinux is enabled...Its not (good)!
Reading /etc/asterisk/asterisk.conf...Done
Checking if Asterisk is running and we can talk to it as the 'asterisk' user...Yes. Determined Asterisk version to be: 20.2.0
Checking if NodeJS is installed and we can get a version from it...Yes. Determined NodeJS version to be: 18.15.0
Preliminary checks done. Starting FreePBX Installation
Checking if this is a new install...Yes (No /etc/freepbx.conf file detected)
Database installation checking credentials and permissions..Connected!
PHP Fatal error:  Declaration of FreePBX\Database::query() must be compatible with PDO::query(string $query, ?int $fetchMode = null, mixed ...$fetchModeArgs): PDOStatement|false in /opt/iiab/freepbx/amp_conf/htdocs/admin/libraries/BMO/Database.class.php on line 240
```
noted warning from cmdline running of install_prereq
```
apt install "$PACKAGES_DEBIAN"
Reading package lists...
Done Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libmysqlclient-dev is not available, but is referred to by another package. This may mean that the package is
missing, has been obsoleted, or is only available from another source However the following packages replace it:
  libmariadb-dev-compat:armhf libmariadb-dev:armhf libmariadb-dev-compat libmariadb-dev
```

holta, 2023-04-02:
Just FYI @EMG70 successfully installed Asterisk & FreePBX on 32-bit Debian, simply as a test (on x86_64).

jvonau, 2023-04-02:
> Just FYI @EMG70 successfully installed Asterisk & FreePBX on 32-bit Debian, simply as a test (on x86_64).

Process used and logs or it didn't happen

jvonau, 2023-04-02:
out of the box RasPiOS64bit
```
sudo apt update
Get:1 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Hit:2 http://deb.debian.org/debian bullseye InRelease                                                                   
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]                                                 
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]                          
Get:5 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [232 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main armhf Packages [232 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main Translation-en [155 kB]
Get:8 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [12.0 kB]
Get:9 http://deb.debian.org/debian bullseye-updates/main armhf Packages [12.0 kB]
Get:10 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Get:11 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [315 kB]
Fetched 1,381 kB in 2s (610 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
13 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
Now test a theory 
```
sudo dpkg --remove-architecture armhf

sudo apt update
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease                                                                           
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease                                                                   
Hit:4 https://deb.nodesource.com/node_18.x bullseye InRelease       
Hit:5 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

apt list libsnmp-dev
Listing... Done
libsnmp-dev/stable,stable-security,now 5.9+dfsg-4+deb11u1 arm64 [installed]
```

holta, 2023-04-03:
Good Enough!

Thanks @jvonau and @EMG70 for making this happen:

- PR #3523

closed 2023-04-03 by holta

jvonau, 2023-04-29:
https://github.com/asterisk/asterisk/issues/37