iiab/iiab #3495: IIAB's WiFi client (student) devices fail to access Internet despite 'iiab_gateway_enabled: True'

holta, 2023-03-17:
Thanks to TK for reporting both iiab-diagnostics and screenshots below:

http://sprunge.us/whcvih?en

[image]

holta, 2023-03-17:
I'm assuming this is a bug (TK knows what he's doing after many years!)

But hopefully someone else can confirm over the wkd or in coming days — as this is indeed functionality that (some!) communities really want — and deserves fixing if indeed confirmed as bug👍

jvonau, 2023-03-18:
> I'm assuming this is a bug (TK knows what he's doing after many years!)

That is a bad assumption, things can subtly change under the hood and the power user may not be aware of the change. Now lets analyze the information presented. The issue is the iptables rules are lacking the needed lines to allow the traffic
```
2034 COMMAND: /usr/bin/sudo iptables-save    # Firewall rules
2035 
2036 # Generated by iptables-save v1.8.7 on Fri Mar 17 13:40:53 2023
2037 *filter
2038 :INPUT ACCEPT [0:0]
2039 :FORWARD ACCEPT [0:0]
2040 :OUTPUT ACCEPT [540092:185855773]
2041 -A INPUT -i lo -j ACCEPT
2042 -A INPUT -p tcp -m tcp --dport 111 -j DROP
2043 -A INPUT -p udp -m udp --dport 111 -j DROP
2044 -A INPUT -p tcp -m tcp --dport 3306 -j DROP
2045 -A INPUT -p udp -m udp --dport 3306 -j DROP
2046 -A INPUT -p tcp -m tcp --dport 5432 -j DROP
2047 -A INPUT -p udp -m udp --dport 5432 -j DROP
2048 -A INPUT -p tcp -m tcp --dport 5984 -j DROP
2049 -A INPUT -p udp -m udp --dport 5984 -j DROP
2050 -A INPUT -m state --state RELATED,ESTABLISHED -j ACCEPT
2051 -A INPUT -i br0 -m state --state NEW -j ACCEPT
2052 -A INPUT -p udp -m udp --dport 5353 -j ACCEPT
2053 -A INPUT -i eth0 -p tcp -m tcp --dport 22 -m state --state NEW -j ACCEPT
2054 -A INPUT -i eth0 -p tcp -m tcp --dport 80 -m state --state NEW -j ACCEPT
2055 -A INPUT -i eth0 -p tcp -m tcp --dport 10000:10100 -m state --state NEW -j ACCEPT
2056 -A INPUT -i eth0 -p tcp -m tcp --dport 12080 -m state --state NEW -j ACCEPT
2057 -A INPUT -i eth0 -p tcp -m tcp --dport 12443 -m state --state NEW -j ACCEPT
2058 -A INPUT -i eth0 -p tcp -m tcp --dport 8080 -m state --state NEW -j ACCEPT
2059 -A INPUT -i eth0 -p tcp -m tcp --dport 8083 -m state --state NEW -j ACCEPT
2060 -A INPUT -i eth0 -p tcp -m tcp --dport 631 -m state --state NEW -j ACCEPT
2061 -A INPUT -i eth0 -p tcp -m tcp --dport 4244 -m state --state NEW -j ACCEPT
2062 -A INPUT -i eth0 -p tcp -m tcp --dport 8000 -m state --state NEW -j ACCEPT
2063 -A INPUT -i eth0 -p tcp -m tcp --dport 8008 -m state --state NEW -j ACCEPT
2064 -A INPUT -i eth0 -p tcp -m tcp --dport 3000 -m state --state NEW -j ACCEPT
2065 -A INPUT -i eth0 -p tcp -m tcp --dport 8009 -m state --state NEW -j ACCEPT
2066 -A INPUT -i eth0 -p udp -m udp --dport 30000 -m state --state NEW -j ACCEPT
2067 -A INPUT -i eth0 -p tcp -m tcp --dport 1883 -m state --state NEW -j ACCEPT
2068 -A INPUT -i eth0 -p tcp -m tcp --dport 1880 -m state --state NEW -j ACCEPT
2069 -A INPUT -i eth0 -p tcp -m tcp --dport 8089 -m state --state NEW -j ACCEPT
2070 -A INPUT -i eth0 -p tcp -m tcp --dport 9091 -m state --state NEW -j ACCEPT
2071 -A INPUT -i eth0 -p tcp -m tcp --dport 51413 -m state --state NEW -j ACCEPT
2072 -A INPUT -i eth0 -j DROP
2073 -A FORWARD -i eth0 -o br0 -m state --state RELATED,ESTABLISHED -j ACCEPT
2074 -A FORWARD -i br0 -o eth0 -j ACCEPT
2075 -A FORWARD -i eth0 -o br0 -j DROP
2076 COMMIT
2077 # Completed on Fri Mar 17 13:40:53 2023
```
What is missing are the MASQ rules
```
*nat
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
-A POSTROUTING -o eth0 -j MASQUERADE
COMMIT
```  
The inclusion of the above rules are dependent upon the setting of IIAB_GATEWAY_ENABLED in /etc/iiab/iiab.env and are acted upon when iiab-gen-iptables is run.

> 63 -rw-r--r-- 1 root root 400 Mar 17 08:11 /etc/iiab/iiab.env
71 IIAB_GATEWAY_ENABLED=True

The modified time of /etc/iiab/iiab.env is after the noted end of the install of
```
2512 2023-03-17 07:24:23,096 p=59945 u=root n=ansible | TASK [js-menu : Create Feedback Database] **********************************************************************************************************************
2513 2023-03-17 07:24:23,941 p=59945 u=root n=ansible | changed: [127.0.0.1]
2514 2023-03-17 07:24:24,282 p=59945 u=root n=ansible | PLAY RECAP *****************************************************************************************************************************************************
2515 2023-03-17 07:24:24,284 p=59945 u=root n=ansible | 127.0.0.1                  : ok=71   changed=63   unreachable=0    failed=0    skipped=4    rescued=0    ignored=0
[... 8 lines cut ...]
Next iiab-configure was run to install 
> 807 postgresql_installed: True
 808 moodle_installed: True
 809 minetest_installed: True 
```
2297 2023-03-17 08:09:41,920 p=2182 u=root n=ansible | TASK [9-local-addons : Recording STAGE 9 HAS COMPLETED ====================] ***
2298 2023-03-17 08:09:42,551 p=2182 u=root n=ansible | changed: [127.0.0.1]
2299 2023-03-17 08:09:42,819 p=2182 u=root n=ansible | PLAY RECAP *********************************************************************
2300 2023-03-17 08:09:42,819 p=2182 u=root n=ansible | 127.0.0.1                  : ok=276  changed=90   unreachable=0    failed=0    skipped=97   rescued=0    ig
2408 2023-03-17 10:27:00,395 p=1964 u=root n=ansible | TASK [network : Restart wpa_supplicant service] ********************************
2409 2023-03-17 10:27:01,566 p=1964 u=root n=ansible | changed: [127.0.0.1] => (item=wpa_supplicant)
2410 2023-03-17 10:27:01,629 p=1964 u=root n=ansible | TASK [network : Create /etc/hostapd/hostapd.conf and backup .iiab from template if needed] ***
```
Now it gets a bit weird, the logging is not complete and at that point iiab-gen-iptables would not of ran yet.
```
2407 2023-03-17 10:27:00,246 p=1964 u=root n=ansible | included: /opt/iiab/iiab/roles/network/tasks/restart.yml for 127.0.0.1
[... cut ...]
The tick box would suggest that is the setting from local_vars.yml or has been user selected in the GUI but 'save' has not been applied as there are no lines in local_vars.yml from admin-console and 'ICO' has not been run as there are no lines in iiab-debug.log

Comes back to what looks like iiab-network not completing due to a powering off situation but I can only go by what is presented.
If everything else is working I would suggest `sudo iiab-internet-on` otherwise run 'sudo iiab-network' and let it complete. Could run ICO in the admin-console that runs the network play also.

holta, 2023-03-18:
Extremely helpful 👍

I'll ask TK to see how he makes out.

holta, 2023-03-18:
TK confirms `iiab-internet-on` worked, PRICELESS!

(Are any further diagnostics needed from him, to try to better understand how this arose?)

jvonau, 2023-03-18:
Sorry no magic crystal ball over here. All I can say is iiab-network.log was truncated/not complete. Why iiab-network stopped logging would of been the result of something interrupting it, what did it I have no clue but it looks like sudden loss of power to me. I'm not going to spend any more time debugging reports of network issues when it is very clear from the logging that the routine did not complete, as there should be a summary printed at the end of a successful run. Don't see the summary in the log file *something* interrupted the routine, try it again could be the next course of action as there is no clear failure point within the logging pointing to anything within the code base.

jvonau, 2023-03-18:
'iiab-internet-on' is not a workaround rather a documented feature to quickly (dis|en)able access to the internet without needing to rerun the full network role. https://github.com/iiab/iiab/pull/2728

holta, 2023-03-18:
While TK doesn't mention any power interruptions, as you say anything's possible.

Ok if that's as much analysis as is realistic here, good enough 👍

closed 2023-03-18 by holta