iiab/iiab #3504: kolibri manage runs on wrong directory

tim-moody, 2023-03-22:
unlike kalite, the kolibri script does not set the HOME directory so content goes into the .kolibri directory of the user (root)

```
root@iiab-me-new:~# cat /usr/bin/kalite
#!/bin/bash
KALITE_HOME=/library/ka-lite
export KALITE_HOME
/usr/local/kalite/venv/bin/kalite $@

```

```
root@iiab-kolibri:~# cat /usr/bin/kolibri
#!/usr/bin/python3
# EASY-INSTALL-ENTRY-SCRIPT: 'kolibri==0.15.12','console_scripts','kolibri'
__requires__ = 'kolibri==0.15.12'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('kolibri==0.15.12', 'console_scripts', 'kolibri')()
    )

```

The kolibri service sets:

```
Environment=KOLIBRI_USER=kolibri
Environment=KOLIBRI_HOME=/library/kolibri
Environment=KOLIBRI_HTTP_PORT=8009
Environment=KOLIBRI_URL_PATH_PREFIX=/kolibri
User=kolibri

```

tim-moody, 2023-03-22:
if I wrap /usr/bin/kolibri with a script to set env variables I get 

`ERROR: Loading plugin configuration failed with error 'The parent of your KOLIBRI_HOME does not exist: /home/kolibri'`

note that /etc/passwd has that as kolibri's home dir with /bin/false as shell

holta, 2023-03-22:
Is it confirmed that setting a KOLIBRI_HOME environment vat is really necessary?

FYI Kolibri reads such settings from several different places, one of which is /etc/kolibri/daemon.conf

Best practices remain ambiguous (last I checked anyway?!) as to where Kolibri prefers we set these kinds of vars, e.g. there's probably 3-or-4 different places that KOLIBRI_USER can be set, including /etc/kolibri/username

See _"Where should KOLIBRI_HOME and KOLIBRI_USER properly be set & stored — among all the many choices — to install a relatively future-proof + resilient/robust Kolibri?"_ below:

- learningequality/kolibri-installer-debian#115

holta, 2023-03-22:
> KOLIBRI_HOME environment var

> `/etc/kolibri/daemon.conf`

> `/etc/kolibri/username`

It appears there are ~5 different places where such Kolibri settings can be established (including the 3 above) if the following are (maybe?) also possible:

- /etc/kolibri/conf.d/*
- $KOLIBRI_HOME/options.ini

Simplifying would be great, if Kolibri has possibly updated their "official" recommendations over the past 6 months?  That would be awesome if so, and @jredrejo might know!

PS @benjaoming was a huge help getting this organized ~5 years ago, but he's no longer working with/on Kolibri I don't think.

PPS @radinamatic is extremely helpful on all topics (no joke!) so she might also know.

tim-moody, 2023-03-22:
This works

```
cmd = '/usr/bin/kolibri manage importchannel network c1f2b7e6ac9f56a2bb44fa7a48b66dce'
args = shlex.split(cmd)
kol_env = {}
kol_env['KOLIBRI_HOME'] = '/library/kolibri'

p = subprocess.Popen(args, env=kol_env, group='www-data', user='kolibri')
```

jvonau, 2023-03-23:
> note that /etc/passwd has that as kolibri's home dir with /bin/false as shell

Yea that would be wrong, the user's home directory should be listed as /library/kolibri, this can be traced to the lack of using 'home:  /library/kolibri' and using 'create_home: no' within ansible's [user module](https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/install.yml#L7-L23) when the user is created, see https://docs.ansible.com/ansible/latest/collections/ansible/builtin/user_module.html

tim-moody, 2023-03-23:
>Create '/usr/local/bin/kolibri' because /usr/local/bin is first on the $PATH with the export statements needed then call the original '/usr/bin/kolibri'

@jvonau what you did for kalite works really well. I just wasn't able to get an equivalent for kolibri. Most of the problems have to do with needing the script to run as the kolibri user, who has no home, no shell, and no password. The environment variables I got after a few iterations.

The kolibri script that retrieves the environment makes assumptions that the service does not.

tim-moody, 2023-03-23:
>the user's home directory should be listed as /library/kolibri,

I think you'll end up with home being set to /library/kolibri/.kolibri

benjaoming, 2023-03-23:
> @jvonau what you did for kalite works really well. I just wasn't able to get an equivalent for kolibri. Most of the problems have to do with needing the script to run as the kolibri user, who has no home, no shell, and no password. The environment variables I got after a few iterations.

You can definitely let the Kolibri user have a home and a shell and even a password. The fact that they don't exist is just so that the Debian package isn't creating user accounts on someone else's system with shells or logins :)

Good luck :+1:

jvonau, 2023-03-23:
> > the user's home directory should be listed as /library/kolibri,
> 
> I think you'll end up with home being set to /library/kolibri/.kolibri

Not as the user is created, the deb install might alter that after the fact but lets get the foundation correct.
> You can definitely let the Kolibri user have a home and a shell and even a password. The fact that they don't exist is just so that the Debian package isn't creating user accounts on someone else's [...]

Most well behaved daemons don't have a home directory set to within /home but pointing to somewhere else like /var/lib nor a password set, that would be the model I'm following.

benjaoming, 2023-03-23:
Yup, that's how it should be, hence it's the defaults of the Kolibri .deb package :+1:

jvonau, 2023-03-23:
> @jvonau what you did for kalite works really well. I just wasn't able to get an equivalent for kolibri. Most of the problems have to do with needing the script to run as the kolibri user, who has no [...]

If it is just the user that is the issue I would suggest 'su -u' or 'runuser' in /usr/local/bin/kolibri as /usr/bin/kolibri is python and doesn't need a login shell just like kalite Ref: https://github.com/learningequality/kolibri-installer-debian/blob/master/debian/startup/kolibri.init

tim-moody, 2023-03-23:
btw, the python code above runs with the stock install of no home, no shell, and no password and sets the user and group properly for the imported .sqlite3

jvonau, 2023-03-23:
From /etc/kolibri/README
`sudo su -l `cat /etc/kolibri/username` -c "source /etc/default/kolibri && kolibri manage shell -c 'import os; print(\"KOLIBRI_HOME is: {}\".format(os.environ[\"KOLIBRI_HOME\"]))'"`
```
sudo su -l `cat /etc/kolibri/username` -c "source /etc/default/kolibri && kolibri manage shell -c 'import os; print(\"KOLIBRI_HOME is: {}\".format(os.environ[\"KOLIBRI_HOME\"]))'"
sudo: unable to resolve host box.wp.shawcable.net: Name or service not known
su: warning: cannot change directory to /home/kolibri: No such file or directory
```
The home directory within /etc/passwd needs to be corrected and that is the problem I pointed out with the ansible user module above.

holta, 2023-03-23:
> @benjaoming kolibri manage outputs INFO and requires pressing enter to acknowledge. Any way to suppress that? I tried -v 0, but still had to press enter.

Of course in bash there are established ways to force a simulated pressing of [Enter]:

https://stackoverflow.com/questions/6264596/simulating-enter-keypress-in-bash-script/

(But hopefully we can avoid that in this case!)

jvonau, 2023-03-23:
22.04 VM
```
TASK [kolibri : Add signed Kolibri PPA 'jammy' (if Ubuntu 22.04+ or Mint 21 or Debian 12)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: W:An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://archive.ubuntu.com/ubuntu jammy InRelease: Couldn't create temporary file /tmp/apt.conf.Kar4Xq for passing config to apt-key, W:An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://security.ubuntu.com/ubuntu jammy-security InRelease: Couldn't create temporary file /tmp/apt.conf.rASlb8 for passing config to apt-key, W:GPG error: http://ppa.launchpad.net/learningequality/kolibri/ubuntu jammy InRelease: Couldn't create temporary file /tmp/apt.conf.WE318u for passing config to apt-key, E:The repository 'http://ppa.launchpad.net/learningequality/kolibri/ubuntu jammy InRelease' is not signed., W:Updating from such a repository can't be done securely, and is therefore disabled by default., W:See apt-secure(8) manpage for repository creation and user configuration details., W:An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://archive.ubuntu.com/ubuntu jammy-updates InRelease: Couldn't create temporary file /tmp/apt.conf.hjgJN2 for passing config to apt-key, W:An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://archive.ubuntu.com/ubuntu jammy-backports InRelease: Couldn't create temporary file /tmp/apt.conf.fsdDC4 for passing config to apt-key"}
```

jvonau, 2023-03-23:
Pending PR.

jvonau, 2023-03-23:
> This works
> 
> ```
> cmd = '/usr/bin/kolibri manage importchannel network c1f2b7e6ac9f56a2bb44fa7a48b66dce'
> args = shlex.split(cmd)
> kol_env = {}
> kol_env['KOLIBRI_HOME'] = '/library/kolibri'
> 
> p = subprocess.Popen(args, env=kol_env, group='www-data', user='kolibri')
> ```

Isn't this functionality already build into kolibri's GUI natively?

jvonau, 2023-03-23:
> > Isn't this functionality already build into kolibri's GUI natively?
> 
> Yes. What's needed is facilitating provisioning of "learning bouquets" in a more automated way. Allowing many more people to benefit.

Seems like a while loop with a bunch of channel imports. Where does the list of channels originate from? Feels more like a addition that would be welcome upstream, why not work directly with upstream?

holta, 2023-03-23:
> Where does the list of channels originate from?

At the moment 131 are available at https://catalog learningequality.org = https://studio.learningequality.org/en/channels/#/public

Possibly there are other "public library" places too?

Ideally many others will appear.

> why not work directly with upstream?

Indeed everybody is doing their best.  Half the people on this ticket are current or former Kolibri employees.

benjaoming, 2023-03-24:
Just to clarify the previous comment: I didn't look in the Debian installer's codebase, I just wondered out loud that if the system service's user doesn't have a shell by default, then that can be a good decision. However, I see that it does and there's probably a reason for that: https://github.com/learningequality/kolibri-installer-debian/blob/master/debian/kolibri.scripts-common#L251

(I'm not actively developing on this project anymore and I can't remember why the user is constructed the way it is)

holta, 2023-03-24:
> [If] system service's user doesn't have a shell by default, then that can be a good decision.

That was my intuition as well (as Kolibri's run very well with Internet-in-a-Box for many years in this fashion!)

> However, I see that it does and there's probably a reason for that: https://github.com/learningequality/kolibri-installer-debian/blob/master/debian/kolibri.scripts-common#L251

Thanks much @benjaoming for highlighting.  Ideally Kolibri can make an official recommendation in future 👍

tim-moody, 2023-03-24:
>Where does the list of channels originate from?

the catalog is at kolibri_api = 'https://studio.learningequality.org/api/public/v1/channels'

channel ids for public channels are expected to remain constant

for kalite the preset topics are in content.json such as

 "kalite": {
    "lang_code": "en",
    "topics": ["khan/"]

for kolibri an equivalent would be needed with channel id and an optional list of node ids

the only downside is that the ids are uuids so not human intelligible, but they are constants