iiab/iiab #3506: kolibri global vars

tim-moody, 2023-03-23:
kolibri_home and kolibri_user should be in default_vars so they can be used outside of the kolibri role

holta, 2023-03-23:
> advertising usage to people who might want that

i.e. so far this hasn't been an issue, but quite possibly it's now time to promote awareness of these 2 vars more widely going forward...

tim-moody, 2023-03-23:
has nothing to do with awareness. if they are local to the kolibri role they are not visible anywhere else. at the moment I have hard coded the values. if they are put in defaults then they can be read anywhere is ansible

holta, 2023-03-23:
Ok ✅ 

I guess that's awareness of another kind (e.g. "machine awareness").

With most all implementers still not expected to be needing these 2 vars, if I'm understanding you correctly.

holta, 2023-03-24:
Done:

- PR #3507

closed 2023-03-24 by holta