iiab/iiab #3512: Linphone sip extensions cannot be reached when iphone or desktop is locked.It seems this is a known issue.Online solutions advise to enabled push notifications but still doesn’t work.Apparently a battery saving trick from Apple. I am trying to source android phones to test in meantime. I have tried Zoiper but push notification is a paid for feature.All solutions welcome.

EMG70, 2023-03-27:
### Expected behavior and actual behavior:

### Steps to reproduce the problem:

### IIAB version, OS version, hardware details:

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

EMG70, 2023-03-27:
> Thank you @EMG70 for reporting ✅
> 
> If possible let's link to the most pragmatic Android workaround (or more real solution) if you can suggest one for IIAB's Guide below, in the coming week or so?
> 
> https://github.com/iiab/iiab/tree/master/roles/pbx#readme
> 
> cc: @lemueldsouza

OK I will look into Android solutions in the coming week.

holta, 2023-04-01:
ASIDE: @EMG70 worked hard today trying a few different things to get Asterisk 19 or 20 working on 64-bit Raspberry Pi OS:

- #3489

Sorry it appears not to have worked out 🕊️

holta, 2023-04-15:
Workaround / solution posted to pbx/README.adoc thanks to @EMG70 !

- PR #3533

closed 2023-04-15 by holta