iiab/iiab #3516: Strange switch to 64bit kernel on 32bit RasPiOS

jvonau, 2023-03-29:
This is just a heads up that something changed in the behavior on 32bit RasPiOS that might cause some weird errors going forward.
http://sprunge.us/xLDTSv?en

Highlights below
  18 Raspberry Pi reference 2022-09-22
  19 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage2
  20 /etc/debian-version: 11.6
  21 display-manager? inactive   Arch1: armhf   Arch2: 
  22 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 unknown
  23 Cortex-A72   CPU(s): 4   Mem: 3794 Swap: 99
  24 Raspberry Pi 4 Model B Rev 1.1
  98 install_date = 2023-01-03T18:10:51Z
109 runtime_date = 2023-03-17T12:17:00Z
110 ansible_version = 2.14.3
111 kernel = 5.15.84-v7l+

Note line '22 6.1.19-v8+' that is the 64bit kernel now running but as of 2023-03-17 the machine was running '111 kernel = 5.15.84-v7l+' There is no 'arm_64bit=1' present in config.txt as the docs suggest that is needed to boot the 64bit kernel.
```
jerryv@box:~ $ cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
```
/var/log/apt/history.log
```
Start-Date: 2023-03-10  06:51:20
Commandline: /usr/bin/apt-get -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold install unixodbc=2.3.6-0.1+b1 sox=14.4.2+git20190427-2+deb11u1 php7.4-bcmath=7.4.33-1+deb11u1 php7.4-cgi=7.4.33-1+deb11u1 php7.4-curl=7.4.33-1+deb11u1 php7.4-gd=7.4.33-1+deb11u1 php7.4-imap=7.4.33-1+deb11u1 php7.4-mbstring=7.4.33-1+deb11u1 php-pear=1:1.10.12+submodules+notgz+20210212-1 php7.4-snmp=7.4.33-1+deb11u1 php7.4-xml=7.4.33-1+deb11u1 php7.4-zip=7.4.33-1+deb11u1 cmake=3.18.4-2+rpt1+rpi1+deb11u1
Requested-By: jerryv (1000)
Install: php7.4-snmp:armhf (7.4.33-1+deb11u1), php7.4-mbstring:armhf (7.4.33-1+deb11u1), libonig5:armhf (6.9.6-1.1, automatic), php7.4-gd:armhf (7.4.33-1+deb11u1), php7.4-curl:armhf (7.4.33-1+deb11u1), libwavpack1:armhf (5.4.0-1, automatic), php7.4-imap:armhf (7.4.33-1+deb11u1), php7.4-bcmath:armhf (7.4.33-1+deb11u1), libarchive13:armhf (3.4.3-2+deb11u1, automatic), sox:armhf (14.4.2+git20190427-2+deb11u1), libopencore-amrnb0:armhf (0.1.5-1, automatic), libopencore-amrwb0:armhf (0.1.5-1, automatic), php7.4-cgi:armhf (7.4.33-1+deb11u1), php7.4-xml:armhf (7.4.33-1+deb11u1), php7.4-zip:armhf (7.4.33-1+deb11u1), libsox3:armhf (14.4.2+git20190427-2+deb11u1, automatic), php-pear:armhf (1:1.10.12+submodules+notgz+20210212-1), libsox-fmt-alsa:armhf (14.4.2+git20190427-2+deb11u1, automatic), librhash0:armhf (1.4.1-2, automatic), snmp:armhf (5.9+dfsg-4+deb11u1, automatic), libzip4:armhf (1.7.3-1, automatic), cmake-data:armhf (3.18.4-2+rpt1+rpi1+deb11u1, automatic), libsox-fmt-base:armhf (14.4.2+git20190427-2+deb11u1, automatic), unixodbc:armhf (2.3.6-0.1+b1), cmake:armhf (3.18.4-2+rpt1+rpi1+deb11u1)
End-Date: 2023-03-10  06:53:39

Start-Date: 2023-03-17  06:30:16
Commandline: apt upgrade
Requested-By: jerryv (1000)
Upgrade: raspberrypi-bootloader:armhf (1:1.20230106-1, 1:1.20230306-1), libcurl3-gnutls:armhf (7.74.0-1.3+deb11u5, 7.74.0-1.3+deb11u7), raspi-config:armhf (20221214, 20230214), python3-picamera2:armhf (0.3.8-1, 0.3.9-1), nodejs:armhf (18.14.2-deb-1nodesource1, 18.15.0-deb-1nodesource1), libcamera-apps-lite:armhf (0~git20230125+9f08463-1, 0~git20230301+54a781d-1), kolibri:armhf (0.15.11-0ubuntu1, 0.15.12-0ubuntu1), vcdbg:armhf (1:1.20230106-1, 1:1.20230306-1), libcamera0:armhf (0~git20230124+9b860a66-1, 0~git20230302+923f5d70-1), raspberrypi-kernel:armhf (1:1.20230106-1, 1:1.20230306-1), python3-libcamera:armhf (0~git20230124+9b860a66-1, 0~git20230302+923f5d70-1), python3-v4l2:armhf (0.3.1-1, 0.3.2-1), linux-libc-dev:armhf (1:1.20230106-1, 1:1.20230306-1)
End-Date: 2023-03-17  06:39:52

Start-Date: 2023-03-22  16:52:36
Commandline: apt install apt-rdepends
Requested-By: jerryv (1000)
Install: apt-rdepends:armhf (1.3.0-7), libapt-pkg-perl:armhf (0.1.39, automatic)
End-Date: 2023-03-22  16:52:40

Start-Date: 2023-03-28  07:36:25
Commandline: apt upgrade
Requested-By: jerryv (1000)
Upgrade: raspberrypi-bootloader:armhf (1:1.20230306-1, 1:1.20230317-1), sox:armhf (14.4.2+git20190427-2+deb11u1, 14.4.2+git20190427-2+deb11u2), libsox3:armhf (14.4.2+git20190427-2+deb11u1, 14.4.2+git20190427-2+deb11u2), libsox-fmt-alsa:armhf (14.4.2+git20190427-2+deb11u1, 14.4.2+git20190427-2+deb11u2), vcdbg:armhf (1:1.20230306-1, 1:1.20230317-1), libsox-fmt-base:armhf (14.4.2+git20190427-2+deb11u1, 14.4.2+git20190427-2+deb11u2), raspberrypi-kernel:armhf (1:1.20230306-1, 1:1.20230317-1), linux-libc-dev:armhf (1:1.20230306-1, 1:1.20230317-1)
End-Date: 2023-03-28  07:43:23
```
'runtime_date = 2023-03-17T12:17:00Z' is later than 'End-Date: 2023-03-17  06:39:52' so that kernel was installed and the machine rebooted prior to the last ansible run that recorded the latest '[runtime]' entries.

In no way did I expect this installation to switch from using 'v7l+' to 'v8+' without having 'arm_64bit=1' present in config.txt. Running the 64bit kernel and having 32bit userspace might confuse some third-party installers when it comes to which binaries to install based on the running kernel.
```
jerryv@box:~ $ uname -a
Linux box 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
```

jvonau, 2023-03-29:
Looks like the change is intentional https://github.com/raspberrypi/firmware/issues/1795

jvonau, 2023-03-29:
informational

closed 2023-03-29 by jvonau

tim-moody, 2023-03-29:
so is the 8.1 image that says 32 bit 32 or 64? it's being used on rpi w.

holta, 2023-03-29:
> so is the 8.1 image that says 32 bit 32 or 64? it's being used on rpi w.

It sounds like this new 64-bit behavior of the 32-bit RasPiOS began around 2 weeks ago when they pushed Linux kernel 6.1:

https://forums.raspberrypi.com/viewtopic.php?p=2088935

If that's confirmed, then the 32-bit IIAB 8.1 Preview 1 image from 2023-02-25 is indeed 32-bits here:

https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images-~-Summary

(Of course apt update / upgrading might cause problems on Zero W, as the Raspberry Pi Foundation didn't exactly warn people before enacting this change ;)

jvonau, 2023-03-29:
This issue is with the latest round of apt updates for RasPiOS-32bit, the boot-loader will boot the 64-bit kernel on 64bit capable hardware without being told to do so, to boot the 32bit kernel one needs to alter /boot/config.txt as above to retain the older behavior. Pure 32bit hardware should not see a change in behavior.

holta, 2023-03-30:
While Sugarizer is rarely needed on Zero W (32-bit) in practice, I agree that it definitely wouldn't hurt to know more of the facts here.

(And if so, to help others understand, what's realistic and what isn't.)

jvonau, 2023-03-31:
```
apt list mongo*
Listing... Done
mongocli/focal 1.29.0 arm64
mongodb-atlas-cli/focal 1.6.0 arm64
mongodb-atlas/focal 1.6.0 arm64
mongodb-database-tools/focal 100.7.0 arm64
mongodb-mongosh-shared-openssl11/focal 1.8.0 arm64
mongodb-mongosh/focal 1.8.0 arm64
mongodb-org-database-tools-extra/focal 5.0.15 arm64
mongodb-org-database/focal 5.0.15 arm64
mongodb-org-mongos/focal 5.0.15 arm64
mongodb-org-server/focal 5.0.15 arm64
mongodb-org-shell/focal 5.0.15 arm64
mongodb-org-tools/focal 5.0.15 arm64
mongodb-org/focal 5.0.15 arm64
mongoose/stable 1:5.8.1+dfsg-2 armhf

sudo apt install mongodb-org
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mongodb-database-tools:arm64 : Depends: libc6:arm64 but it is not installable
                                Depends: libgssapi-krb5-2:arm64 but it is not installable
                                Depends: libkrb5-3:arm64 but it is not installable
                                Depends: libk5crypto3:arm64 but it is not installable
                                Depends: libcomerr2:arm64 but it is not installable
                                Depends: libkrb5support0:arm64 but it is not installable
                                Depends: libkeyutils1:arm64 but it is not installable
 mongodb-org:arm64 : Depends: mongodb-mongosh:arm64
 mongodb-org-mongos:arm64 : Depends: libc6:arm64 (>= 2.29) but it is not installable
                            Depends: libcurl4:arm64 (>= 7.16.2) but it is not installable
                            Depends: libgcc-s1:arm64 (>= 4.5) but it is not installable
                            Depends: liblzma5:arm64 (>= 5.1.1alpha+20110809) but it is not installable
                            Depends: libssl1.1:arm64 (>= 1.1.1) but it is not installable
 mongodb-org-server:arm64 : Depends: libc6:arm64 (>= 2.29) but it is not installable
                            Depends: libcurl4:arm64 (>= 7.16.2) but it is not installable
                            Depends: libgcc-s1:arm64 (>= 4.5) but it is not installable
                            Depends: liblzma5:arm64 (>= 5.1.1alpha+20110809) but it is not installable
                            Depends: libssl1.1:arm64 (>= 1.1.1) but it is not installable
 mongodb-org-shell:arm64 : Depends: libc6:arm64 (>= 2.29) but it is not installable
                           Depends: libcurl4:arm64 (>= 7.16.2) but it is not installable
                           Depends: libgcc-s1:arm64 (>= 4.5) but it is not installable
                           Depends: liblzma5:arm64 (>= 5.1.1alpha+20110809) but it is not installable
                           Depends: libssl1.1:arm64 (>= 1.1.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

holta, 2023-05-03:
> Looks like the change is intentional [raspberrypi/firmware#1795](https://github.com/raspberrypi/firmware/issues/1795)

Ramifications with the latest Linux 6.1 LTS kernels (as used by RasPiOS) are still somewhat evolving...

- raspberrypi/linux#5408

Related:

- https://github.com/iiab/iiab/pull/3422#issuecomment-1533441463