iiab/iiab #3526: How to pre-test RasPiOS 12

jvonau, 2023-04-04:
Install the latest arm64 image of RasPiOS and boot
```
sudo apt update
sudo apt -y upgrade
sudo apt -y autoremove
sudo reboot 
```
```
sudo sed -i 's/bullseye/bookworm/g' /etc/apt/sources.list

sudo apt update
sudo apt -y dist-upgrade
# Don't go too far away there a couple of confirmations to confirm just press enter when prompted.
# Any changed files should be a result of the image creation process I would presume. 
# I hate suggesting that method but you can stop and read the actions and decide for yourself.

sudo apt -y autoremove
sudo reboot
```
sudo raspi-config and set the country code for the WiFi and other settings. Might want to try the NetworkManager option at this time.
Make the required changes in local_facts.fact and provide a OS.yml file #3525
Install IIAB as per normal.

jvonau, 2023-04-04:
Just a how to note, not for general consumption but I thought I would share.

holta, 2023-04-04:
> Install IIAB as per normal.

Did an IIAB install largely complete?

(Either way, instructions like this are indeed extremely useful for newcomers.)

jvonau, 2023-04-04:
I went with the 'none' version of local_vars as a starting point, installed clean.

jvonau, 2023-04-05:
FYI

jerryv@box:~ $ apt list python2*
Listing... Done

jvonau, 2023-04-05:
https://github.com/iiab/iiab/issues/3289#issuecomment-1497728119
> perhaps using the last deb package(s) released from an older install.

http://ftp.debian.org/debian/pool/main/p/python2.7/?C=M;O=D
```
wget http://ftp.debian.org/debian/pool/main/p/python2.7/libpython2.7-minimal_2.7.18-8_arm64.deb
sudo apt install ./libpython2.7-minimal_2.7.18-8_arm64.deb
wget http://ftp.debian.org/debian/pool/main/p/python2.7/python2.7-minimal_2.7.18-8_arm64.deb
sudo apt install ./python2.7-minimal_2.7.18-8_arm64.deb
```
The order is important and are 'dpkg_arch' dependent, that was the reasoning behind my original #3518 having dpkg_arch available for other roles.

holta, 2023-04-11:
@jvonau mentionsat https://github.com/iiab/iiab/pull/3535#issuecomment-1503771152 :

> http://raspbian.raspberrypi.org/ has a new file [snapshotindex.txt](http://raspbian.raspberrypi.org/snapshotindex.txt) suggesting there will be a new bookworm release sooner or later when you look at the bottom of the file

jvonau, 2023-04-11:
I don't have time to try this but should work:
Install the latest armhf image of RasPiOS and boot
```
sudo apt update
sudo apt -y upgrade
sudo apt -y autoremove

apt info apt info libpython2.7-stdlib > ~/libpython.txt and attach or pastebin the file to PR3535 please.

sudo reboot 

sudo sed -i 's/bullseye/bookworm/g' /etc/apt/sources.list

sudo apt update
sudo apt -y dist-upgrade
# Don't go too far away there a couple of confirmations to confirm just press enter when prompted.
# Any changed files should be a result of the image creation process I would presume. 
# I hate suggesting that method but you can stop and read the actions and decide for yourself.

sudo apt -y autoremove
sudo reboot
```
Might not all be in place yet but post how far you get to.

holta, 2023-04-11:
@EMG70 would you have time in the coming week, to try the instructions immediately above?

EMG70, 2023-04-11:
That’s OK, I will try it this week and let you know.

jvonau, 2023-04-11:
Stepping through an upgrade of an older install
```
jerryv@box:~ $ sudo apt update
Hit:1 http://ppa.launchpad.net/learningequality/kolibri/ubuntu focal InRelease
Hit:2 https://deb.nodesource.com/node_18.x bullseye InRelease                                          
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease                                         
Get:4 http://raspbian.raspberrypi.org/raspbian bookworm InRelease [15.0 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian bookworm/main armhf Packages [14.5 MB]
Get:6 http://raspbian.raspberrypi.org/raspbian bookworm/contrib armhf Packages [62.3 kB]
Get:7 http://raspbian.raspberrypi.org/raspbian bookworm/non-free armhf Packages [118 kB]
Get:8 http://raspbian.raspberrypi.org/raspbian bookworm/rpi armhf Packages [1,360 B]
Fetched 14.7 MB in 9s (1,689 kB/s)                                                                                                      
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
795 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

jvonau, 2023-04-11:
```
sudo apt -y dist-upgrade
The following packages will be REMOVED:
  fuse hardlink libnfsidmap2 libnginx-mod-http-lua libsemanage1 libsepol1-dev mariadb-client-10.5 mariadb-client-core-10.5
  mariadb-server-10.5 mariadb-server-core-10.5 nginx-extras php-pear php7.4-bcmath php7.4-cgi php7.4-cli php7.4-common php7.4-curl
  php7.4-fpm php7.4-gd php7.4-imap php7.4-json php7.4-mbstring php7.4-mysql php7.4-opcache php7.4-readline php7.4-snmp php7.4-xml
  php7.4-zip python-pip-whl python3-kms++ python3-libcamera python3-picamera2 python3-pidng python3-simplejpeg python3-spidev
  python3.9-venv
```
most of these will not be installed on a fresh image.

jvonau, 2023-04-11:
This test box never had kalite installed so no python2 was ever installed http://sprunge.us/M2I5sQ
```
jerryv@box:~ $ apt list *python2*
Listing... Done
idle-python2.7/testing 2.7.18-13.2 all
libpython2-dbg/testing 2.7.18-3 armhf
libpython2-dev/testing 2.7.18-3 armhf
libpython2-stdlib/testing 2.7.18-3 armhf
libpython2.7-dbg/testing 2.7.18-13.2 armhf
libpython2.7-dev/testing 2.7.18-13.2 armhf
libpython2.7-minimal/testing 2.7.18-13.2 armhf
libpython2.7-stdlib/testing 2.7.18-13.2 armhf
libpython2.7-testsuite/testing 2.7.18-13.2 all
libpython2.7/testing 2.7.18-13.2 armhf
python2-dbg/testing 2.7.18-3 armhf
python2-dev/testing 2.7.18-3 armhf
python2-doc/testing 2.7.18-3 all
python2-minimal/testing 2.7.18-3 armhf
python2-setuptools-whl/testing 44.1.1-1.2 all
python2.7-dbg/testing 2.7.18-13.2 armhf
python2.7-dev/testing 2.7.18-13.2 armhf
python2.7-doc/testing 2.7.18-13.2 all
python2.7-examples/testing 2.7.18-13.2 all
python2.7-minimal/testing 2.7.18-13.2 armhf
python2.7/testing 2.7.18-13.2 armhf
python2/testing 2.7.18-3 armhf
```
Looks like they are going to ship python2

jvonau, 2023-04-11:
And it installs too
```
jerryv@box:~ $ sudo apt install python2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib python2-minimal python2.7 python2.7-minimal
Suggested packages:
  python2-doc python-tk python2.7-doc binfmt-support
The following NEW packages will be installed:
  libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib python2 python2-minimal python2.7 python2.7-minimal
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,631 kB of archives.
After this operation, 16.6 MB of additional disk space will be used.
Get:1 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf libpython2.7-minimal armhf 2.7.18-13.2 [397 kB]
Get:2 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf python2.7-minimal armhf 2.7.18-13.2 [1,046 kB]
Get:3 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf python2-minimal armhf 2.7.18-3 [34.2 kB]
Get:4 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf libpython2.7-stdlib armhf 2.7.18-13.2 [1,797 kB]
Get:5 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf python2.7 armhf 2.7.18-13.2 [311 kB]
Get:6 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf libpython2-stdlib armhf 2.7.18-3 [21.2 kB]
Get:7 https://mirrors.switch.ca/raspbian/raspbian bookworm/main armhf python2 armhf 2.7.18-3 [24.5 kB]
Fetched 3,631 kB in 3s (1,109 kB/s) 
Selecting previously unselected package libpython2.7-minimal:armhf.
(Reading database ... 82616 files and directories currently installed.)
Preparing to unpack .../0-libpython2.7-minimal_2.7.18-13.2_armhf.deb ...
Unpacking libpython2.7-minimal:armhf (2.7.18-13.2) ...
Selecting previously unselected package python2.7-minimal.
Preparing to unpack .../1-python2.7-minimal_2.7.18-13.2_armhf.deb ...
Unpacking python2.7-minimal (2.7.18-13.2) ...
Selecting previously unselected package python2-minimal.
Preparing to unpack .../2-python2-minimal_2.7.18-3_armhf.deb ...
Unpacking python2-minimal (2.7.18-3) ...
Selecting previously unselected package libpython2.7-stdlib:armhf.
Preparing to unpack .../3-libpython2.7-stdlib_2.7.18-13.2_armhf.deb ...
Unpacking libpython2.7-stdlib:armhf (2.7.18-13.2) ...
Selecting previously unselected package python2.7.
Preparing to unpack .../4-python2.7_2.7.18-13.2_armhf.deb ...
Unpacking python2.7 (2.7.18-13.2) ...
Selecting previously unselected package libpython2-stdlib:armhf.
Preparing to unpack .../5-libpython2-stdlib_2.7.18-3_armhf.deb ...
Unpacking libpython2-stdlib:armhf (2.7.18-3) ...
Setting up libpython2.7-minimal:armhf (2.7.18-13.2) ...
Setting up python2.7-minimal (2.7.18-13.2) ...
Linking and byte-compiling packages for runtime python2.7...
Setting up python2-minimal (2.7.18-3) ...
Selecting previously unselected package python2.
(Reading database ... 83363 files and directories currently installed.)
Preparing to unpack .../python2_2.7.18-3_armhf.deb ...
Unpacking python2 (2.7.18-3) ...
Setting up libpython2.7-stdlib:armhf (2.7.18-13.2) ...
Setting up python2.7 (2.7.18-13.2) ...
Setting up libpython2-stdlib:armhf (2.7.18-3) ...
Setting up python2 (2.7.18-3) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for mailcap (3.70+nmu1) ...
```

jvonau, 2023-04-11:
installing ansible might be a pain
```
Create virtual environment for Ansible
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible-core
  Using cached https://www.piwheels.org/simple/ansible-core/ansible_core-2.14.4-py3-none-any.whl (2.2 MB)
Collecting jinja2>=3.0.0
  Using cached https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting PyYAML>=5.1
  Downloading PyYAML-6.0.tar.gz (124 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 125.0/125.0 kB 1.2 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting cryptography
  Downloading cryptography-40.0.1.tar.gz (624 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 625.0/625.0 kB 3.1 MB/s eta 0:00:00
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [77 lines of output]
      Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
      Collecting setuptools!=60.9.0,>=40.6.0
        Using cached https://www.piwheels.org/simple/setuptools/setuptools-67.6.1-py3-none-any.whl (1.1 MB)
      Collecting wheel
        Using cached https://www.piwheels.org/simple/wheel/wheel-0.40.0-py3-none-any.whl (64 kB)
      Collecting cffi>=1.12
        Downloading cffi-1.15.1.tar.gz (508 kB)
           ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 508.5/508.5 kB 3.1 MB/s eta 0:00:00
        Preparing metadata (setup.py): started
        Preparing metadata (setup.py): finished with status 'done'
      Collecting setuptools-rust>=0.11.4
        Downloading https://www.piwheels.org/simple/setuptools-rust/setuptools_rust-1.5.2-py3-none-any.whl (23 kB)
      Collecting pycparser
        Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
      Collecting semantic-version<3,>=2.8.2
        Downloading https://www.piwheels.org/simple/semantic-version/semantic_version-2.10.0-py2.py3-none-any.whl (19 kB)
      Collecting typing-extensions>=3.7.4.3
        Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.5.0-py3-none-any.whl (27 kB)
      Installing collected packages: wheel, typing-extensions, setuptools, semantic-version, pycparser, setuptools-rust, cffi
        DEPRECATION: cffi is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
        Running setup.py install for cffi: started
        Running setup.py install for cffi: finished with status 'error'
        error: subprocess-exited-with-error
      
        × Running setup.py install for cffi did not run successfully.
        │ exit code: 1
        ╰─> [40 lines of output]
            /usr/local/ansible/lib/python3.11/site-packages/setuptools/config/setupcfg.py:515: SetuptoolsDeprecationWarning: The license_file parameter is deprecated, use license_files instead.
              warnings.warn(msg, warning_class)
            running install
            /usr/local/ansible/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
              warnings.warn(
            running build
            running build_py
            creating build
            creating build/lib.linux-aarch64-cpython-311
            creating build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/lock.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/cparser.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/verifier.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/vengine_gen.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/backend_ctypes.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/vengine_cpy.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/setuptools_ext.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/error.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/cffi_opcode.py -> build/lib.linux-aarch64-cpython-311/cffi
            copying cffi/__init__.py -> build/lib.linux-aarch64-cpython-311/cffi
[... 22 lines cut ...]
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: legacy-install-failure
      
      × Encountered error while trying to install package.
      ╰─> cffi
      
      note: This is an issue with the package mentioned above, not pip.
      hint: See above for output from the failure.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```
Note the -aarch64- in the above suggests that python's pip is using `uname` while the userspace is armhf
```
jerryv@box:/opt/iiab/iiab $ dpkg --print-architecture 
armhf
```
Lets see where is goes in a few days and see if a wheel is created for cffi on piwheels

jvonau, 2023-04-11:
Don't try until there is a new wheel with cp311 within the name in the above [url.](https://www.piwheels.org/simple/cffi/)

EMG70, 2023-04-12:
> Don't try until there is a new wheel with cp311 within the name in the above [url.](https://www.piwheels.org/simple/cffi/)

Noted Jerry,Thanks

holta, 2023-04-25:
> > wait a few days before [testing 32-bit RasPiOS](https://github.com/iiab/iiab/issues/3526#issuecomment-1503806882) ?

> Don't try until there is a new wheel with cp311 within the name in the above [url.](https://www.piwheels.org/simple/cffi/)

Looks like it might be a few weeks?   (Or worst case months!)

holta, 2023-04-27:
FYI Debian 12 Bookworm is planning an early release on 2023-06-10, so RasPiOS 12 should be expected in a similar timeframe, e.g. perhaps June 2023 or thereabouts:

https://lists.debian.org/debian-devel-announce/2023/04/msg00007.html

holta, 2023-05-15:
> Don't try until there is a new wheel with cp311 within the name in the above [url.](https://www.piwheels.org/simple/cffi/)

It might be many weeks possibly, or even months worst case.

(Certainly `cp311` does not yet appear in the above listing!)

holta, 2023-05-25:
> Don't try until there is a new wheel with cp311 within the name in the above [url.](https://www.piwheels.org/simple/cffi/)

Just FYI 6 weeks later there's still no progress beyond wheels 40.0.2 in the colorful chart here:

- https://www.piwheels.org/project/cryptography/

@jvonau @EMG70, with evidence growing that RasPiOS 12 will be released rather soon (presumably roughly around the same time as Debian 12 itself on 2023-06-10 ??) let's assess in early June... what testing makes most sense 🏗️

holta, 2023-05-28:
Details/Preview:

http://raspbian.raspberrypi.org/raspbian/dists/bookworm-staging/

holta, 2023-06-11:
Some ancient-but-still-interesting Raspberry Pi tips pasted in here from https://wiki.debian.org/RaspberryPi :

> Usage Tips
> - Many problems are/were caused by using a poor quality power supply. For stable and reliable operations you need a good and reliable adapter with a good cable. The shorter and thicker the better.
> - You can safely attach simple (ie low-power) USB devices to a RPi. But if you attach many and/or USB devices which consume a 'lot' of power, like SSD/HDD, consider using a powered hub for those.
> - The quality of SD cards provided in 'RPi kits' is often of poor quality. Investing in higher quality and higher capacity cards can improve the endurance significantly. It's good to have much free space on SD cards.
> - Putting/Adding swap space on a SD card is generally a (very) bad idea. Swap is no substitute for RAM and it can significantly reduce the lifespan of your SD card.
> - If you're running Debian Buster with a 4.19 kernel, consider upgrading to the current Debian Stable version (Bullseye) with a 5.10 kernel. Many improvements for RPi devices have landed in the 5.10 kernel. And running the current Debian Stable release is generally a good advice.

holta, 2023-06-12:
Just FYI, even without any surgery to its apt configuration, 64-bit RasPiOS Lite looks to be starting to set itself up to introduce Debian 12 Bookworm changes — based on this `apt update` output anyway:

```
N: Repository 'http://security.debian.org/debian-security bullseye-security InRelease' changed its 'Suite' value from 'stable-security' to 'oldstable-security'
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://deb.debian.org/debian bullseye-updates InRelease' changed its 'Suite' value from 'stable-updates' to 'oldstable-updates'
```

holta, 2023-06-14:
@EMG70 will work on testing pre-release Raspberry Pi OS 12 later this week 💯

Likely focusing on 64-bit Raspberry Pi OS Lite.  With LARGE-sized IIAB install attempt(s).

So I'm re-opening this ticket.  Or he can open a new ticket if he prefers.

holta, 2023-06-15:
Not official, but there's gossip that `/var/log/{syslog,messages}` might be going away:

https://forums.raspberrypi.com/viewtopic.php?t=352477#p2112927

As Debian 12 Bookworm no longer installs `rsyslog` by default:

https://www.debian.org/releases/stable/amd64/release-notes/ch-information.en.html#changes-to-system-logging

holta, 2023-06-16:
@EMG70's generally successful test (except for Munin!) of a LARGE-sized install of IIAB onto pre-release "RasPiOS 12" is explained/summarized here:

- #3603

closed 2023-06-16 by jvonau

holta, 2023-07-29:
Huge thanks to @EMG70 who just installed LARGE-sized IIAB onto RasPiOS 12 "with desktop" on a Raspberry Pi 400, as documented by iiab-diagnostics:

https://paste2.org/dd099UYt

CLARIF: Munin did not start until the RPi 400 machine was rebooted.  He then ran `sudo iiab` to continue/complete the IIAB software install.

Related:

- #3434
  - PR #3560
- #3612

holta, 2023-07-30:
Related:

- moodlebox/moodlebox#287
- moodlebox/moodlebox#305

holta, 2023-09-07:
> Don't try until there is a new wheel with cp311 within the name in the above [url.](https://www.piwheels.org/simple/cffi/)

Done, 5 months later! 💯 

Where CFFI =  Foreign Function Interface

https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp311-cp311-linux_armv7l.whl#sha256=b7d4d03d3abebb8b6623c14c20e2f9defec95aabde9f5e8999af483eeab7ae18

https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp311-cp311-linux_armv6l.whl#sha256=b7d4d03d3abebb8b6623c14c20e2f9defec95aabde9f5e8999af483eeab7ae18

@EMG70 let's do a LARGE-sized install onto "Raspberry Pi OS 12" (desktop version ideally? or Lite version?) if you have time in the coming 10 days!

holta, 2023-09-07:
> sudo sed -i 's/bullseye/bookworm/g' /etc/apt/sources.list

1) @EMG70 in addition to the above, at that time I recommend you *also* run:

   ```sed -i 's/bullseye/bookworm/g' /etc/apt/sources.list.d/raspi.list```

2) After your OS upgrade from RasPiOS 11 Lite to RasPiOS 12 Lite is completed, try to install **SMALL-sized IIAB** with PR #3632 as follows:

   ```curl iiab.io/install.txt | bash -s 3632```

3) If your IIAB install on "32-bit" RasPiOS 12 Lite works, please also test your microSD card in a truly 32-bit Raspberry Pi Zero W if you have one (to check the Wi-Fi hotspot works well!)

holta, 2023-09-08:
CFFI issues persist for @EMG70 on today's new test on "32-bit" (sort of!) RasPiOS 12 Lite:

[image]

(He should post iiab-diagnostics shortly to give a bit more context!)

holta, 2023-09-08:
Same issue seems to occur with `cryptography 40.0.1` too :/

And @EMG70 also tried booting his RPi 400 in 32-bit mode using:

```echo 'arm_64bit=0' >> /boot/config.txt
reboot```

In short, he tried both kinds of boots, 32-bit armv7l kernel and 64-bit aarch64 kernel, each running on RPi 400, and no luck in either case.  FYI he saved the "32-bit" RasPiOS 12 Lite test [scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible) using techniques just like the following:

```
mv /etc/iiab /etc/iiab.old
mv /opt/iiab/iiab /opt/iiab/iiab.old
mv /usr/local/ansible /usr/local/ansible.old
sudo iiab
```

A bit more detail in this screenshot below, showing what appears almost immediately as/after it tries to install cryptography 40.0.1 to the new `/usr/local/ansible` venv:

![c79fb206-42b6-41d8-a178-0b9c9f9d13f9](https://github.com/iiab/iiab/assets/2458907/54ac3022-b264-458e-9838-c1163c8fd388)

holta, 2023-09-09:
FYI @EMG70 and I intend to give "32-bit" RasPiOS 12 (Bookworm) another crack on Friday September 22:

Whether or not **cffi** wheel for Python 3.11 shows improvement here... ?

- https://www.piwheels.org/project/cffi/
  - "1.15.1 from 2022-06-30"
- https://www.piwheels.org/simple/cffi/
  - [epeg_cffi-0.2-cp311-cp311-linux_armv7l.whl](https://www.piwheels.org/simple/epeg-cffi/epeg_cffi-0.2-cp311-cp311-linux_armv7l.whl#sha256=a72c45130726b6266749ca904f55a23d79ca1a5024d94f06d4cdb48ea8e6b51b)
  - [epeg_cffi-0.2-cp311-cp311-linux_armv6l.whl](https://www.piwheels.org/simple/epeg-cffi/epeg_cffi-0.2-cp311-cp311-linux_armv6l.whl#sha256=a72c45130726b6266749ca904f55a23d79ca1a5024d94f06d4cdb48ea8e6b51b)

holta, 2023-09-09:
Full paste of the latest cffi error on a fresh "32-bit" RasPiOS 12 Lite below — running with a 64-bit kernel in this case — in the hope this helps us & all move the ball forward:

<details>
<summary>OS/Architecture/Prep Details</summary>

```
root@raspberrypi:~# cat /etc/issue
Raspbian GNU/Linux 12 \n \l

root@raspberrypi:~# cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

root@raspberrypi:~# uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

root@raspberrypi:~# tail -1 /proc/cpuinfo
Model           : Raspberry Pi 4 Model B Rev 1.1

root@raspberrypi:~# apt install python3-venv
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  python3-distutils python3-lib2to3 python3-pip-whl python3-setuptools-whl python3.11-venv
The following NEW packages will be installed:
  python3-distutils python3-lib2to3 python3-pip-whl python3-setuptools-whl python3-venv python3.11-venv
0 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,043 kB of archives.
After this operation, 4,197 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://raspbian.mirror.constant.com/raspbian bookworm/main armhf python3-lib2to3 all 3.11.2-3 [76.3 kB]
Get:2 http://raspbian.mirror.constant.com/raspbian bookworm/main armhf python3-distutils all 3.11.2-3 [131 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bookworm/main armhf python3-pip-whl all 23.0.1+dfsg-1 [1,717 kB]
Get:4 http://raspbian.mirror.constant.com/raspbian bookworm/main armhf python3-setuptools-whl all 66.1.1-1 [1,111 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian bookworm/main armhf python3-venv armhf 3.11.2-1 [1,188 B]
Get:5 http://raspbian.mirror.constant.com/raspbian bookworm/main armhf python3.11-venv armhf 3.11.2-6 [5,892 B]
Fetched 3,043 kB in 2s (1,409 kB/s)
Selecting previously unselected package python3-lib2to3.
(Reading database ... 44873 files and directories currently installed.)
Preparing to unpack .../0-python3-lib2to3_3.11.2-3_all.deb ...
Unpacking python3-lib2to3 (3.11.2-3) ...
Selecting previously unselected package python3-distutils.
Preparing to unpack .../1-python3-distutils_3.11.2-3_all.deb ...
Unpacking python3-distutils (3.11.2-3) ...
Selecting previously unselected package python3-pip-whl.
Preparing to unpack .../2-python3-pip-whl_23.0.1+dfsg-1_all.deb ...
Unpacking python3-pip-whl (23.0.1+dfsg-1) ...
Selecting previously unselected package python3-setuptools-whl.
Preparing to unpack .../3-python3-setuptools-whl_66.1.1-1_all.deb ...
Unpacking python3-setuptools-whl (66.1.1-1) ...
Selecting previously unselected package python3.11-venv.
Preparing to unpack .../4-python3.11-venv_3.11.2-6_armhf.deb ...
Unpacking python3.11-venv (3.11.2-6) ...
Selecting previously unselected package python3-venv.
Preparing to unpack .../5-python3-venv_3.11.2-1_armhf.deb ...
Unpacking python3-venv (3.11.2-1) ...
Setting up python3-setuptools-whl (66.1.1-1) ...
Setting up python3-pip-whl (23.0.1+dfsg-1) ...
Setting up python3-lib2to3 (3.11.2-3) ...
Setting up python3-distutils (3.11.2-3) ...
Setting up python3.11-venv (3.11.2-6) ...
Setting up python3-venv (3.11.2-1) ...

root@raspberrypi:~# python3 -m venv /usr/local/ansible
```
</details>

```
root@raspberrypi:~# /usr/local/ansible/bin/python3 -m pip install --upgrade ansible-core
[... 104 lines cut ...]
                2 | #include <Python.h>
                  |          ^~~~~~~~~~
            compilation terminated.
            error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
            [end of output]

        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: legacy-install-failure

      × Encountered error while trying to install package.
      ╰─> cffi

      note: This is an issue with the package mentioned above, not pip.
      hint: See above for output from the failure.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

holta, 2023-09-10:
@EMG70 I've posted upstream to make sure we stay on track with RasPiOS12 — as it evolves towards final release / general availability — presumably very quickly now in coming weeks:

- piwheels/packages#390

holta, 2023-09-11:
1) @EMG70: Great news, the piwheels folk helped clear up the main problem, so this PR clears up several nagging issues: ✅

   - PR #3632

2) Main remaining issue (on "32-bit" RasPiOS 12) appears to be KA Lite not finding Python 2:

   - #3635

holta, 2023-09-26:
@EMG70 just FYI the Calibre-Web install issue on RasPiOS Bookworm (#3642) appears under control now, but might still be improved/converging further here:

- piwheels/packages#391

So let's test everything we can carefully on/around Friday 2023-10-06 if you have time to help then, as RasPiOS 12 is making big strides behind the scenes and presumably is now on final approach very soon!

EMG70, 2023-09-26:
> @EMG70 just FYI the Calibre-Web install issue on RasPiOS Bookworm (#3642) appears under control now, but might still be further improved as it converges here:
> 
> * [Missing package: lxml piwheels/packages#391](https://github.com/piwheels/packages/issues/391)
> 
> So let's test everything we can carefully on/around Friday 2023-10-06 if you have time to help then, as RasPiOS 12 is making big strides behind the scenes, and presumably is now on final approach very soon!

No problem. I will get in touch a few days before 06/10/23.

holta, 2023-09-26:
> No problem. I will get in touch a few days before 06/10/23.

Awesome.  ASIDE / lots of grassroots folk are now tested Raspberry Pi OS Bookworm, so hopefully that means it'll be quite solid when it's released in coming-weeks-or-so, many months after Debian 12 Bookworm itself was released 2023-06-10:

- https://raspberrytips.com/update-raspberry-pi-latest-version/
- https://gist.github.com/jauderho/6b7d42030e264a135450ecc0ba521bd8
- https://blog.fernvenue.com/archives/upgrade-raspberrypi-to-debian-12-bookworm/
- https://raspberrypi.stackexchange.com/questions/143341/is-it-possible-to-install-bookworm-testing

holta, 2023-09-26:
> @EMG70 just FYI the Calibre-Web install issue on RasPiOS Bookworm (#3642) appears under control now, but might still be further improved as it converges here:
> 
> * [Missing package: lxml piwheels/packages#391](https://github.com/piwheels/packages/issues/391)

Awesome News: piwheels.org just fixed it (lxml 4.9.3 for Python 3.11) today after I pinged them about this failed build!

holta, 2023-10-05:
Two generally very successful RasPiOS 12/Bookworm pre-release tests — huge thanks to @EMG70 — in both cases installing LARGE-sized IIAB using an RPi 400:

Fully 32-bit using `arm_64bit=0` in `/boot/config.txt` :
http://sprunge.us/H36mDb?en

32-bit with 64-bit kernel:
http://sprunge.us/oGYxI6?en

There were a few errors like **ERROR: NetworkManager is not running.** — in both cases above — when running **sudo iiab-network** as this fully 32-bit screenshot shows:

[image]

holta, 2023-10-11:
Now that [Raspberry Pi OS Bookworm](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/) is released ([64-bit release notes](https://downloads.raspberrypi.com/raspios_arm64/release_notes.txt)), follow-up is on this new ticket:

- #3657