iiab/iiab #3536: April 2023 Testing update

holta, 2023-04-12:
1. **Huge thanks to @EMG70** who justed successfully tested a LARGE-sized IIAB install + Asterisk and FreePBX on 64-bit RasPiOS with desktop on RPi 400:

   http://sprunge.us/dE9Rf7?en

2. 32-bit testing for Raspberry Pi Zero W can hopefully happen in the coming 10 days:

- https://github.com/iiab/iiab/issues/3526#issuecomment-1504042835

3. 64-bit Ubuntu Server 23.04 testing should follow, either after it's released 2023-04-20, or possibly beforehand:

- https://github.com/iiab/iiab/pull/3535#issuecomment-1505860697

jvonau, 2023-04-12:
The other option to test would be RasPiOS11 -> 12 on arm64 from https://github.com/iiab/iiab/issues/3526#issue-1654621428

holta, 2023-04-12:
And later in May, preliminary poking at Ubuntu 23.10 daily builds can begin:

- #3528

holta, 2023-04-17:
Testing is extremely hard work.  That goes without saying but it still needs to be said!

In short, yet another Ubuntu-focused testing task is here:

- PR #3537

holta, 2023-05-15:
4 of the 6 above checkboxes are basically done!

Thanks to hard work from @EMG70 and @jvonau especially.

When Debian 12 (Bookworm) is released on Saturday 2023-06-10, we should take another closer look at those final 2 tasks — i.e. Raspberry Pi OS 12 pre-releases if not final releases — evaluating both 32-bit OS/operations for Zero W and 64-bit for other Raspberry Pi's!