iiab/iiab #3538: Calibre-Web requirements fail to install: "netifaces.c:1:10: fatal error: Python.h: No such file or directory" (64-bit RasPiOS)

holta, 2023-04-16:
Is it possible @ozzieisaacs or @jvonau might recognize this issue?

This error is fully reproducible, on a clean/fresh 64-bit Raspberry Pi OS Lite.

It seems to have arisen suddenly since ~2 days ago — is it possible something about Calibre-Web's requirements changed since April 13 or 14 ?

> TASK [calibre-web : Download Calibre-Web dependencies from 'requirements.txt' into python3 virtual environment /usr/local/calibre-web-py3] ***

> fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/local/calibre-web-py3/bin/pip3", "install", "-r", "/usr/local/calibre-web-py3/requirements.txt"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nCollecting APScheduler<3.11.0,>=3.6.3\n  Downloading https://www.piwheels.org/simple/apscheduler/APScheduler-3.10.1-py3-none-any.whl (59 kB)\nCollecting werkzeug<2.1.0\n  Downloading https://www.piwheels.org/simple/werkzeug/Werkzeug-2.0.3-py3-none-any.whl (289 kB)\nCollecting Babel<3.0,>=1.3\n  Downloading https://www.piwheels.org/simple/babel/Babel-2.12.1-py3-none-any.whl (10.1 MB)\nCollecting Flask-Babel<3.1.0,>=0.11.1\n  Downloading https://www.piwheels.org/simple/flask-babel/flask_babel-3.0.1-py3-none-any.whl (11 kB)\nCollecting Flask-Login<0.6.3,>=0.3.2\n  Downloading Flask_Login-0.6.2-py3-none-any.whl (17 kB)\nCollecting Flask-Principal<0.5.1,>=0.3.2\n  Downloading https://www.piwheels.org/simple/flask-principal/Flask_Principal-0.4.0-py3-none-any.whl (6.7 kB)\nCollecting Flask<2.3.0,>=1.0.2\n  Downloading https://www.piwheels.org/simple/flask/Flask-2.2.3-py3-none-any.whl (101 kB)\nCollecting iso-639<0.5.0,>=0.4.5\n  Downloading https://www.piwheels.org/simple/iso-639/iso_639-0.4.5-py3-none-any.whl (171 kB)\nCollecting PyPDF<3.8.0,>=3.0.0\n  Downloading https://www.piwheels.org/simple/pypdf/pypdf-3.7.1-py3-none-any.whl (248 kB)\nCollecting pytz>=2016.10\n  Downloading https://www.piwheels.org/simple/pytz/pytz-2023.3-py3-none-any.whl (502 kB)\nCollecting requests<2.29.0,>=2.11.1\n  Downloading https://www.piwheels.org/simple/requests/requests-2.28.2-py3-none-any.whl (62 kB)\nCollecting SQLAlchemy<2.0.0,>=1.3.0\n  Downloading SQLAlchemy-1.4.47-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (1.6 MB)\nCollecting tornado<6.3,>=4.1\n  Downloading tornado-6.2-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (424 kB)\nCollecting Wand<0.7.0,>=0.4.4\n  Downloading https://www.piwheels.org/simple/wand/Wand-0.6.11-py3-none-any.whl (143 kB)\nCollecting unidecode<1.4.0,>=0.04.19\n  Downloading https://www.piwheels.org/simple/unidecode/Unidecode-1.3.6-py3-none-any.whl (235 kB)\nCollecting lxml<5.0.0,>=3.8.0\n  Downloading lxml-4.9.2-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.manylinux_2_24_aarch64.whl (6.8 MB)\nCollecting flask-wtf<1.2.0,>=0.14.2\n  Downloading https://www.piwheels.org/simple/flask-wtf/Flask_WTF-1.1.1-py3-none-any.whl (12 kB)\nCollecting chardet<4.1.0,>=3.0.0\n  Downloading https://www.piwheels.org/simple/chardet/chardet-4.0.0-py2.py3-none-any.whl (178 kB)\nCollecting advocate<1.1.0,>=1.0.0\n  Downloading https://www.piwheels.org/simple/advocate/advocate-1.0.0-py2.py3-none-any.whl (38 kB)\nCollecting Flask-Limiter<3.4.0,>=2.3.0\n  Downloading https://www.piwheels.org/simple/flask-limiter/Flask_Limiter-3.3.0-py3-none-any.whl (27 kB)\nCollecting netifaces>=0.10.5\n  Downloading netifaces-0.11.0.tar.gz (30 kB)\nCollecting urllib3<2.0,>=1.22\n  Downloading https://www.piwheels.org/simple/urllib3/urllib3-1.26.15-py2.py3-none-any.whl (140 kB)\nCollecting pyasn1\n  Downloading https://www.piwheels.org/simple/pyasn1/pyasn1-0.4.8-py2.py3-none-any.whl (77 kB)\nCollecting ndg-httpsclient\n  Downloading https://www.piwheels.org/simple/ndg-httpsclient/ndg_httpsclient-0.5.1-py3-none-any.whl (34 kB)\nCollecting pyopenssl\n  Downloading https://www.piwheels.org/simple/pyopenssl/pyOpenSSL-23.1.1-py3-none-any.whl (57 kB)\nCollecting six\n  Downloading https://www.piwheels.org/simple/six/six-1.16.0-py2.py3-none-any.whl (11 kB)\nCollecting tzlocal!=3.*,>=2.0\n  Downloading https://www.piwheels.org/simple/tzlocal/tzlocal-4.3-py3-none-any.whl (20 kB)\nRequirement already satisfied: setuptools>=0.7 in /usr/local/calibre-web-py3/lib/python3.9/site-packages (from APScheduler<3.11.0,>=3.6.3->-r /usr/local/calibre-web-py3/requirements.txt (line 1)) (44.1.1)\nCollecting itsdangerous>=2.0\n  Downloading https://www.piwheels.org/simple/itsdangerous/itsdangerous-2.1.2-py3-none-any.whl (15 kB)\nCollecting importlib-metadata>=3.6.0\n  Downloading https://www.piwheels.org/simple/importlib-metadata/importlib_metadata-6.4.1-py3-none-any.whl (22 kB)\nCollecting Flask<2.3.0,>=1.0.2\n  Downloading https://www.piwheels.org/simple/flask/Flask-2.2.2-py3-none-any.whl (101 kB)\n  Downloading https://www.piwheels.org/simple/flask/Flask-2.2.1-py3-none-any.whl (101 kB)\nCollecting Jinja2>=3.0\n  Using cached https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)\nCollecting Flask<2.3.0,>=1.0.2\n  Downloading https://www.piwheels.org/simple/flask/Flask-2.2.0-py3-none-any.whl (101 kB)\n  Downloading https://www.piwheels.org/simple/flask/Flask-2.1.3-py3-none-any.whl (95 kB)\nCollecting click>=8.0\n  Downloading https://www.piwheels.org/simple/click/click-8.1.3-py3-none-any.whl (96 kB)\nCollecting pytz>=2016.10\n  Downloading https://www.piwheels.org/simple/pytz/pytz-2022.7.1-py3-none-any.whl (499 kB)\nCollecting ordered-set<5,>4\n  Downloading https://www.piwheels.org/simple/ordered-set/ordered_set-4.1.0-py3-none-any.whl (7.6 kB)\nCollecting typing-extensions>=4\n  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.5.0-py3-none-any.whl (27 kB)\nCollecting rich<14,>=12\n  Downloading https://www.piwheels.org/simple/rich/rich-13.3.4-py3-none-any.whl (238 kB)\nCollecting limits>=2.8\n  Downloading https://www.piwheels.org/simple/limits/limits-3.3.1-py3-none-any.whl (43 kB)\nCollecting blinker\n  Downloading https://www.piwheels.org/simple/blinker/blinker-1.6.2-py3-none-any.whl (13 kB)\nCollecting WTForms\n  Downloading https://www.piwheels.org/simple/wtforms/WTForms-3.0.1-py3-none-any.whl (136 kB)\nCollecting idna<4,>=2.5\n  Downloading https://www.piwheels.org/simple/idna/idna-3.4-py3-none-any.whl (61 kB)\nCollecting charset-normalizer<4,>=2\n  Downloading charset_normalizer-3.1.0-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (196 kB)\nCollecting certifi>=2017.4.17\n  Downloading https://www.piwheels.org/simple/certifi/certifi-2022.12.7-py3-none-any.whl (155 kB)\nCollecting greenlet!=0.4.17\n  Downloading greenlet-2.0.2-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (607 kB)\nCollecting zipp>=0.5\n  Downloading https://www.piwheels.org/simple/zipp/zipp-3.15.0-py3-none-any.whl (6.8 kB)\nCollecting MarkupSafe>=2.0\n  Using cached MarkupSafe-2.1.2-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (26 kB)\nCollecting deprecated>=1.2\n  Downloading https://www.piwheels.org/simple/deprecated/Deprecated-1.2.13-py2.py3-none-any.whl (9.5 kB)\nCollecting packaging<24,>=21\n  Using cached https://www.piwheels.org/simple/packaging/packaging-23.1-py3-none-any.whl (48 kB)\nCollecting importlib-resources>=1.3\n  Downloading https://www.piwheels.org/simple/importlib-resources/importlib_resources-5.12.0-py3-none-any.whl (36 kB)\nCollecting wrapt<2,>=1.10\n  Downloading wrapt-1.15.0-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (78 kB)\nCollecting pygments<3.0.0,>=2.13.0\n  Downloading https://www.piwheels.org/simple/pygments/Pygments-2.15.0-py3-none-any.whl (1.1 MB)\nCollecting markdown-it-py<3.0.0,>=2.2.0\n  Downloading markdown_it_py-2.2.0-py3-none-any.whl (84 kB)\nCollecting mdurl~=0.1\n  Downloading https://www.piwheels.org/simple/mdurl/mdurl-0.1.2-py3-none-any.whl (10.0 kB)\nCollecting pytz-deprecation-shim\n  Downloading https://www.piwheels.org/simple/pytz-deprecation-shim/pytz_deprecation_shim-0.1.0.post0-py2.py3-none-any.whl (15 kB)\nCollecting cryptography<41,>=38.0.0\n  Using cached cryptography-40.0.2-cp36-abi3-manylinux_2_28_aarch64.whl (3.5 MB)\nCollecting cffi>=1.12\n  Using cached cffi-1.15.1-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (448 kB)\nCollecting pycparser\n  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)\nCollecting tzdata\n  Downloading https://www.piwheels.org/simple/tzdata/tzdata-2023.3-py2.py3-none-any.whl (341 kB)\nUsing legacy 'setup.py install' for netifaces, since package 'wheel' is  [... cut ...]
display-manager? inactive   Arch1: arm64   Arch2: armhf
6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 unknown
Cortex-A72   CPU(s): 4   Mem: 3794 Swap: 1023
Raspberry Pi 4 Model B Rev 1.2
Temperature(s): 50147

Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G  3.8G  111G   4% /      ZIMs: 0 OER2Go: 0 Apps
2B: 1

lo eth0 wlan0 br0 tun0
openvpn_enabled: True
openvpn_handle: 184-raspios64-srv-unittest
192.168.0.184 10.10.10.10 10.8.0.46
```

Possibly related:

- PR #3503

jvonau, 2023-04-16:
>fatal error: Python.h: No such file or directory\n

check if python3-dev is installed.

jvonau, 2023-04-16:
https://pypi.org/project/netifaces/#files the underling issue the wheel is missing causing the need to compile (netifaces-0.11.0-cp39-cp39-manylinux_2_5_arm64.manylinux1_arm64.whl), much like the cp310 or cp311 for other wheels. Note on armhf the dependency is met with [piwheels.](https://www.piwheels.org/project/netifaces/) no compiler needed.

jvonau, 2023-04-16:
Might be a good time to test 3503

holta, 2023-04-16:
> > fatal error: Python.h: No such file or directory\n
> 
> check if python3-dev is installed. [#3503 (comment)](https://github.com/iiab/iiab/pull/3503#issuecomment-1480193792)

`python3-dev` not installed FYI:

```
root@box:~# apt list python3-dev
Listing... Done
python3-dev/stable 3.9.2-3 arm64
python3-dev/stable 3.9.2-3 armhf
```

holta, 2023-04-16:
> Might be a good time to test 3503

Looks potentially perfect:

Is testing on 64-bit RasPiOS likely sufficient?

- PR #3503

(Or are there other OS's potentially important to test?)

jvonau, 2023-04-16:
The only os that has a netifaces wheel available is RasPiOS-11-armhf the rest need to use the apt supplied version or compile it. If you are going to try 3503 on this test box please remove /usr/local/calibre-web-py3 before applying the PR.

holta, 2023-04-16:
> The only os that has a netifaces wheel available is RasPiOS-11-armhf the rest need to use the apt supplied version or compile it. If you are going to try 3503 on this test box please remove /usr/local/calibre-web-py3 before applying the PR.

Oops I tested right before seeing the above message:
- https://github.com/iiab/iiab/pull/3503#issuecomment-1510362731

So additional test results (with automated removal of `/usr/local/calibre-web-py3`) are here:
- https://github.com/iiab/iiab/pull/3503#issuecomment-1510385616

holta, 2023-04-16:
Should now be solved by the revised:

- PR #3503

closed 2023-04-16 by holta