iiab/iiab #3539: Vnstat error when building iiab

EMG70, 2023-04-17:
### Expected behavior and actual behavior:

### Steps to reproduce the problem:
Tried running iiab many times and been failing in past few days at same stage.
### IIAB version, OS version, hardware details:
Iiab 8.1 ,Debian 11

### Please run the `iiab-diagnostics` 
command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

http://sprunge.us/wEnNC5

holta, 2023-04-17:
Thanks @EMG70 for reporting!
Highlighting the error message here:

>  
1910 2023-04-17 15:09:31,923 p=3937 u=root n=ansible | TASK [vnstat : Create database for WAN to collect vnStat data] *****************
1911 2023-04-17 15:09:32,159 p=3937 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/usr/bin/vnstat -i wlp2s0", "delta": "0:00:00.004179", "end": "2023-04-17 15:09:32.135351", "msg": "non-zero return code", "rc": 1, "start": "2023-04-17 15:09:32.131172", "stderr": "", "stderr_lines": [], "stdout": "Error: Failed to open database \"/var/lib/vnstat/vnstat.db\" in read-only mode.\nThe vnStat daemon should have created the database when started.\nCheck that it is configured and running. See also \"man vnstatd\".", "stdout_lines": ["Error: Failed to open database \"/var/lib/vnstat/vnstat.db\" in read-only mode.", "The vnStat daemon should have created the database when started.", "Check that it is configured and running. See also \"man vnstatd\"."]}

holta, 2023-04-17:
FYI vnStat ("a network traffic monitor for Linux and BSD") may not be commonly used with IIAB, but at least it's actively maintained:

https://github.com/vergoh/vnstat/releases

holta, 2023-04-17:
ASIDE: IIAB currently uses this `/etc/vnstat.conf` template derived from 2011-06-01's vnStat 1.11

https://github.com/iiab/iiab/blob/master/roles/vnstat/templates/vnstat.conf.j2

If vnStat is genuinely useful to certain IIAB implementers/communities, this should probably be updated to vnStat 2.10 (released 2022-10-22) or similar?

holta, 2023-04-17:
vnStat's changelog, covering more than 20 years:

https://humdi.net/vnstat/CHANGES

holta, 2023-04-18:
Should we assume this problem applies to Debian 11 only?

(And perhaps also RasPiOS 11?)

holta, 2023-04-22:
vnStat version 2.6.3 according to Debian 11:

https://packages.debian.org/bullseye/vnstat

holta, 2023-04-24:
In coming weeks (or days) vnStat and Munin will be considered for removal from LARGE-sized IIAB installs.

The reason is that both are rarely used, and poorly maintained.

Conversely: if there's a desire to patch up and work around, either can be preserved if truly necessary?

Related:

- #3434

tim-moody, 2023-04-24:
on ub 22.04.1 apt install vnstat had no complaints

holta, 2023-04-24:
> on ub 22.04.1 apt install vnstat had no complaints

CLARIF: Please see the error at the top of this ticket.

(In short: the vnstat apt package installed fine.  But IIAB's roles/vnstat/tasks/install.yml -> "Create database for WAN to collect vnStat data" failed every single time.)

Recap: If vnStat is genuinely needed + used (?) then the above error is likely fixable.

jvonau, 2023-04-24:
Stop pushing Debian flavors, problem solved.

jvonau, 2023-04-24:
30 seconds with google turned up: https://github.com/vergoh/vnstat/issues/134

holta, 2023-05-01:
> Tried running iiab many times and been failing in past few days at same stage.

> http://sprunge.us/wEnNC5?en

@EMG70 this should now be fixed, I hope!  As a result of:

- PR #3562 

Can you confirm in the coming week if possible — with a similar install of IIAB on Debian 11 on your Dell?

holta, 2023-05-01:
> OK,I will try and let you know this week.

Awesome.

(You're the only one who's been able to reproduce this failure, so your verdict is essential here!)

holta, 2023-06-01:
vnStat issues all now appear to be solved by:

- PR #3562

closed 2023-06-01 by holta