iiab/iiab #3541: Changing RPi's WiFi channel required these 4 steps: (1) Change 'host_channel: 6' in local_vars.yml (2) iiab-hotspot-off (3) Reboot (4) ./iiab-network

holta, 2023-04-18:
FYI the Admin Console approach also did not work, if one needs to change a Raspberry Pi IIAB's internal WiFi hotspot's to another 2.4 GHz channel.

Not a serious bug but FYI, something to look into.

Thankfully this recipe below did work, every time: _(the key seems to be rebooting in Step 3)_

1. Set IIAB's new (desired) 2.4 GHz WiFi channel, by changing `host_channel: 6` in `/etc/iiab/local_vars.yml`
2. Run: `sudo iiab-hotspot-off`
3. Reboot the Raspberry Pi
4. Run:
   `cd /opt/iiab/iiab`
   `./iiab-network`

FWIW 'wifi_up_down: False' was set in /etc/iiab/local_vars.yml on the particular Raspberry Pi 4 in question.  Which probably doesn't matter but just FYI.  And the IIAB in question had been installed a few weeks back (test machine now gone).

RECAP: The above recipe...is the only way I found to change WiFi channel, after trying/testing many other dead-ends.  So this ticket serves to help others, documenting a solid workaround (better than manually hacking `/etc/hostapd/hostapd.conf` !)

jvonau, 2023-04-20:
> FYI the Admin Console approach also did not work, if one needs to change a Raspberry Pi IIAB's internal WiFi hotspot's to another 2.4 GHz channel.
>

Seems like stale code, or perhaps you/Tim are misunderstanding something, which OS?
Should I stop maintaining the network role if admin-console wants to run the show?
> Not a serious bug but FYI, something to look into.
>

Please expand on the bug you allude to, is upstream wifi in use?
> Thankfully this recipe below did work, every time: _(the key seems to be rebooting in Step 3)_
> 
>     1. Set IIAB's new (desired) 2.4 GHz WiFi channel, by changing `host_channel: 6` in `/etc/iiab/local_vars.yml`
>

Would depend on if there is upstream wifi in use 'host_channel' needs to match what is in use on wlan0 otherwise this 'host_channel' value is respected. Might be that iiab-test-wifi is misbehaving somewhat when upstream wifi is not in use but you have not established that as fact yet.
>     2. Run: `sudo iiab-hotspot-off`
>

Good move you followed directions
>     3. Reboot the Raspberry Pi
>

Don't think that is needed but whatever 
>     4. Run: `sudo iiab-network`
> 
> 
> FWIW 'wifi_up_down: False' was set in /etc/iiab/local_vars.yml on the particular Raspberry Pi 4 in question. Which probably doesn't matter but just FYI. And the IIAB in question had been installed a few weeks back (test machine now gone).

Actually **toggling** 'wifi_up_down: **False'** **requires** a pass though iiab-network to adjust the required files and services but you don't mention when you edited the file so your recipe might be off.

You forgot to mention reboot after iiab-network should be preformed.
> 
> RECAP: The above recipe...is the only way I found to change WiFi channel, after trying/testing many other dead-ends. So this ticket serves to help others, documenting a solid workaround (better than manually hacking `/etc/hostapd/hostapd.conf` !)

Again you need to better describe the situation where the channel in hostapd.conf is not what is expected. Running iiab-test-wifi by hand might point to the issue you might be having.

jvonau, 2023-04-20:
For the record my recommend method still is:

1. iiab-hotspot-off
2. edit local_vars.yml for channel and/or wifi_up_down, also applies to setting up a new/changing an upstream wifi connection.
3. iiab-network
4. reboot

jvonau, 2023-04-20:
The client_wifi_channel recorded in iiab.ini would be of interest if the value is in the 2.4GHz(1-14) range. The assumption is if there is a value present use it, works fine when the upstream wifi is actually being used. Lets see if I can tighten that up to just test when the wifi is providing access to the internet.

holta, 2023-04-20:
> Seems like stale code

It don't think it was stale code, as it was earlier this month sometime, not so long ago.

> which OS?

64-bit Raspberry Pi OS Lite.  Running on a Raspberry Pi 4.

> upstream wifi in use?

No.  Just a clarification that the IIAB hotspot was working throughout (both before and after I shifted it from one 2.4GHz channel to another).

> 3. Reboot the Raspberry Pi
> 4. Run: `sudo iiab-network`

CLARIF: I was changing `host_channel: 6` to `host_channel: 1` or `host_channel: 11` in /etc/iiab/local_vars.yml and it was having no effect in /etc/hostapd/hostapd.conf

Until I discovered the clincher — which was to reboot just prior to running `sudo iiab-network` — for whatever reason that was the breakthrough, and confirmed repeatedly.

Otherwise the WiFi channel was not being updated within /etc/hostapd/hostapd.conf

> **toggling** 'wifi_up_down: **False' requires** a pass though iiab-network to adjust the required files and services

Good to know.

Though in this situation, it was not being toggled — `wifi_up_down: False` was set during IIAB install, and never subsequently changed. 

> reboot after iiab-network should be [performed]

I know that's officially recommended.  But in this particular situation, an additional reboot after `sudo iiab-network` neither helped nor hindered.

i.e. `sudo iiab-network` either worked to change the hotspot to another channel or it didn't — and certainly any additional reboots did not help fix `/etc/hostapd/hostapd.conf`

> Running iiab-test-wifi by hand might point to the issue

_Thanks for mentioning that._  I don't yet understand this tool.  Quick test on a similar machine (also with `wifi_up_down: False`) shows:

```
root@box:~# iiab-test-wifi
ssid is
Couldn't find ssid  to use exiting
```

Just FWIW there are no spaces in that working hotspot's SSID.  In any case spaces are legal in SSID's.

> client_wifi_channel recorded in iiab.ini would be of interest

Ok!  It's definitely repeatable thankfully, as it was a vanilla SMALL-sized install with `wifi_up_down: False` and not much anything else changed.  I'll get back to this in the coming week.  With iiab-diagnostics etc.

jvonau, 2023-04-20:
> CLARIF: I was changing host_channel: 6 to host_channel: 1 or host_channel: 11 in /etc/iiab/local_vars.yml and it was having no effect in /etc/hostapd/hostapd.conf
> Until I discovered the clincher — [...]

Think always running the client channel detection might be the root cause, picking some arbitrary channel from the surrounding wifi environment or the previous setting. The channel setting is intended for when upstream wifi access is needed along with hotspot functionality See https://github.com/iiab/iiab/pull/3542/commits/4fe56e787efbaa120aa1c1be2b9f2447ab72773f

tim-moody, 2023-04-20:
>Seems like stale code

When all this settles down I could use some help updating the network section of Adm Cons. Or could just strip it out.

jvonau, 2023-04-20:
> > Seems like stale code
> 
> When all this settles down I could use some help updating the network section of Adm Cons. Or could just strip it out.

Other than what Adam alluded to in setting channels, static addresses might work as is, haven't tried in along time, what issues need resolving? Is 'Internal Wi-Fi Access Point' using local_vars or are you editing hostapd.conf directly?

tim-moody, 2023-04-20:
most everything runs through iiab-network, but there were a couple of 'shortcuts' like trying to turn on blue tooth and other stuff I don't fully remember, which were rpi only. But I'd also like to review all the settable network parameters to make sure they are on a form somewhere. nothing has been done to all this in a long time.

jvonau, 2023-04-20:
wpa_credentials looks sane but the hardcoded country_code is less than ideal. Agreed, I haven't looked too close in a long time and except for some fiddling around iiab-hotspot-on|off the code is the same for the last 2-3 years.

holta, 2023-04-21:
> Thankfully this recipe below did work, every time: _(the key seems to be rebooting in Step 3)_
> 
> 1. Set IIAB's new (desired) 2.4 GHz WiFi channel, by changing `host_channel: 6` in `/etc/iiab/local_vars.yml`
> 2. Run: `sudo iiab-hotspot-off`
> 3. Reboot the Raspberry Pi
> 4. Run: `sudo iiab-network`

@EMG70 can you try the above recipe WITHOUT Step. 3?

(Please try to reproduce the failure, by SKIPPING OVER Step 3.)

And then post an [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) link, if indeed the WiFi Channel does NOT change within your /etc/hostapd/hostapd.conf ?

ASIDE: I don't think the OS matters, e.g. use Ubuntu Server 23.04 or any OS you want.  So long as the IIAB contains an internal WiFi hotspot.

EMG70, 2023-04-21:
I will give it a try from Sunday.

holta, 2023-04-21:
Thanks @EMG70:

▶️ My own tests had _wifi_up_down: **False**_ around [Line 94](https://github.com/iiab/iiab/blob/671343b9e9cf03b5c3fc9d0227a514b8ded91f9f/vars/local_vars_small.yml#L94) in [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?) (as mentioned above).

▶️ If Necessary: try toggling that variable (wifi_up_down) to see if that has any effect?

_Any & all Precise Failure Pattern(s) that you can reproduce will be incredibly valuable!_

holta, 2023-04-22:
@EMG70

To "kill 2 birds with one stone" can you please install IIAB with PR #3542 as one of your WiFi channel-changing tests?

- Please use Ubuntu Server 23.04 as the OS.
- Install it on a Raspberry Pi or laptop with an internal WiFi hotspot (access point) capability.
- Initiate the install of IIAB with PR #3542 by running: `curl iiab.txt/install.txt | bash -s 3542`
- Choose Zero (0) IIAB apps during the IIAB install — rather than SMALL-sized (1), MEDIUM-sized (2), or LARGE-size (3).
- Prior to running `sudo iiab` (which installs IIAB) please turn on Admin Console around [Lines 33-34](https://github.com/iiab/iiab/blob/671343b9e9cf03b5c3fc9d0227a514b8ded91f9f/vars/local_vars_small.yml#L33-L34) of your [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?):

  ```
  admin_console_install: True
  admin_console_enabled: True
  ```
- Run `sudo iiab` as many times as necessary, until the IIAB install hopefully completes, as indicated by this message on your screen:

  ```
  ┌───────────────────────────────────────────────────────────┐
  │                                                           │
  │   INTERNET-IN-A-BOX (IIAB) SOFTWARE INSTALL IS COMPLETE   │
  │                                                           │
  └───────────────────────────────────────────────────────────┘
  ```

jvonau, 2023-04-22:
If one is testing 3542 do not configure upstream wifi, that masks this channel changing issue, just use the wired connection to install and test with.

jvonau, 2023-04-22:
> Thanks @EMG70:
> 
> arrow_forward My own tests had _wifi_up_down: **False**_ around [Line 94](https://github.com/iiab/iiab/blob/671343b9e9cf03b5c3fc9d0227a514b8ded91f9f/vars/local_vars_small.yml#L94) in [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?) (as mentioned above).
> 
> arrow_forward If Necessary: try toggling that variable (wifi_up_down) to see if that has any effect?
> 
> _Any & all Precise Failure Pattern(s) that you can reproduce will be incredibly valuable!_

Remember to run `sudo iiab-hotspot-off` prior to any edits to local_vars for network variables then run `sudo iiab-network` to apply the changes.

holta, 2023-04-22:
> If one is testing 3542 do not configure upstream wifi, that masks this channel changing issue, just use the wired connection to install and test with.

@EMG70 please start by setting _wifi_up_down: **False**_ around [Line 94](https://github.com/iiab/iiab/blob/671343b9e9cf03b5c3fc9d0227a514b8ded91f9f/vars/local_vars_small.yml#L94) of your /etc/iiab/local_vars.yml before running `sudo iiab`

Thanks.  Keep it simple.  Other scenarios can be tested later.

holta, 2023-04-22:
> run sudo iiab-hotspot-off prior to any edits to local_vars for network variables post-install

What's the reasoning above?  Personally I'm not understanding, given [iiab-hotspot-off](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hotspot-off) does not modify `/etc/iiab/local_vars.yml`

FYI iiab-hotspot-off appears to be extremely simple in this situation (on RasPiOS anyway!) and possibly could be made even simpler if we remove a bit of harmless duplication:

```
sed -i -e "s/^HOSTAPD_ENABLED.*/HOSTAPD_ENABLED=False/" /etc/iiab/iiab.env
systemctl disable hostapd    # DUPLICATE BELOW
systemctl stop hostapd    # DUPLICATE BELOW
# hotspot-off before ap0_updown
sed -i -e "s/^denyinterfaces/#denyinterfaces/" /etc/dhcpcd.conf
systemctl disable hostapd    # DUPLICATE ABOVE
systemctl stop hostapd    # DUPLICATE ABOVE
systemctl daemon-reload
systemctl restart dhcpcd
```

jvonau, 2023-04-22:
/usr/bin/iiab-hotspot-on contents can differ depending on the situation, look at the [template](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hotspot-on)

jvonau, 2023-04-22:
If you think you can improve the code go for it, the contents differ for [iiab-hotspot-off](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hotspot-off) depending on the situation.

holta, 2023-04-22:
> contents differ for [iiab-hotspot-off](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hotspot-off) depending on the situation.

Definitely.

Side observation was that... there's a bit of duplicate code in iiab-hotspot-off... that is completely harmless... we can tighten up later.

(In any case, not relevant to the situation at hand!)

jvonau, 2023-04-22:
> Definitely.
>
> Side observation was that... there's a bit of duplicate code in iiab-hotspot-off... that is completely harmless... we can tighten up later.

Please explain how there is duplicate code when what you are observing is based on conditions present. The code is written for readability and clarity, showing the different steps for different situations encountered. Helps with newbies understanding what might be involved before questions are made. Now what don't you understand? Perhaps I might clarify to help bring you up to speed on the subject matter.

EMG70, 2023-04-22:
> @EMG70
> 
> To "kill 2 birds with one stone" can you please install IIAB with PR #3542 as one of your WiFi channel-changing tests?
> 
> * Please use Ubuntu Server 23.04 as the OS.
> * Install it on a Raspberry Pi or laptop with an internal WiFi hotspot (access point) capability.
> * Initiate the install of IIAB with PR [Reg dom2 #3542](https://github.com/iiab/iiab/pull/3542) by running: `curl iiab.txt/install.txt | bash -s 3542`
> * Choose Zero (0) IIAB apps (for unit-testing) during the IIAB install — rather than SMALL-sized (1), MEDIUM-sized (2), or LARGE-size (3).
> * Prior to running `sudo iiab` (which installs IIAB) please turn on Admin Console around [Lines 33-34](https://github.com/iiab/iiab/blob/671343b9e9cf03b5c3fc9d0227a514b8ded91f9f/vars/local_vars_small.yml#L33-L34) of your [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?):
>   ```
>   admin_console_install: True
>   admin_console_enabled: True
>   ```
> * Run `sudo iiab` as many times as necessary, until the IIAB install hopefully completes, as indicated by this message on your screen:
>   ```
>   ┌───────────────────────────────────────────────────────────┐
>   │                                                           │
>   │   INTERNET-IN-A-BOX (IIAB) SOFTWARE INSTALL IS COMPLETE   │
>   │                                                           │
>   └───────────────────────────────────────────────────────────┘
>   ```

Please confirm you have all agreed that I follow these steps .

jvonau, 2023-04-22:
\ > Please confirm you have all agreed that I follow these steps .

Should be fine, just remember to stop to edit local_vars.yml just after the file is retrieved when asked during 'iiab'.
Change wifi_up_down and the channel you want to use in addition to changing admin_console_* flags, continue with 'iiab'.

holta, 2023-04-22:
> Please explain how there is duplicate code

I don't mean to distract from the WiFi channel-changing issues at hand.

But here's a "straw man" PR with some tiny/possible clarification & de-dup'ing perhaps:

- PR #3546

holta, 2023-04-22:
> Change wifi_up_down and the channel you want to use in addition to changing admin_console_* flags, continue with 'iiab'.

CLARIF: There's no need to change the WiFi Channel initially, when installing IIAB.  Either way, that works fine.

(The problem arises when attempting to change the WiFi Channel _later on..._)

holta, 2023-04-22:
> code is written for readability and clarity, showing the different steps for different situations encountered. Helps with newbies understanding what might be involved before questions are made.

Yes.  Extremely helpful 👍

(For people wanting to see what [iiab-hotspot-off](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hotspot-off) and [iiab-hotspot-on](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hotspot-on) are doing!)

holta, 2023-04-22:
FYI @EMG70 confirmed that Ubuntu Server 23.04 on RPi 400 fails to change IIAB's WiFi channel in the exact the same way as 64-bit RasPiOS Lite on RPi 4.

RECAP: If you skip Step 3. below, regrettably it will fail to update the WiFi channel...

2. Run: sudo iiab-hotspot-off
3. Reboot the Raspberry Pi
4. Run: sudo iiab-network

His initial iiab-diagnostics, before he tried changing the channel many, many, many times: http://sprunge.us/AJuFov?en

(ASIDE: He did not have time to do the exact same experiments with `wifi_up_down: True`, but might have time to try those separate experiments tomorrow.)

jvonau, 2023-04-23:
> FYI @EMG70 confirmed that Ubuntu Server 23.04 on RPi 400 fails to change IIAB's WiFi hotspot channel, in the exact the same way as 64-bit RasPiOS Lite on RPi 4.
> 
> RECAP: If you skip Step 3. below, regrettably it will fail to use the WiFi channel set in /etc/iiab/local_vars.yml :
> 
>     2. Run: `sudo iiab-hotspot-off`
> 
>     3. Reboot the Raspberry Pi
> 
>     4. Run: `sudo iiab-network`
> 
> 
> Here is his initial iiab-diagnostics, before he tried changing the channel many, many, many times: http://sprunge.us/AJuFov?en
> 
> CLARIF / SUMMARY: He ran `sudo grep channel /etc/hostapd/hostapd.conf` almost constantly (to monitor when it successfully changed... and conversely when it unfortunately failed to change).
> 
> ASIDE: He did not have time to do the exact same experiments with `wifi_up_down: True`, but might have time to try those separate experiments tomorrow.

Testing method is flawed, upstream wifi is in use see lines 800-819 so the channel detection was run as the is a wifi gateway. There is no hope of using hostapd while wlan0 is active and wifi_up_down is False so I don't know that this test proves other than being a waste of time.

holta, 2023-04-23:
> > http://sprunge.us/AJuFov?en

> Testing method is flawed, upstream wifi is in use see lines 800-819, 999-1008

@EMG70 IIAB install instructions very strongly emphasize using Ethernet (cable) _instead of_ WiFi as your connection to the Internet, during IIAB installation especially.

If you have time Sunday or in coming days: can you redo the experimentation using the mainline/required setup (i.e. verify that your IIAB is connected to the Internet with an Ethernet cable _NOT_ with WiFi, Thanks!)

Reminder: If in the end you choose Ubuntu Server 23.04 as your OS, remember that [you might possibly need to customize /etc/netplan/02-iiab-config.yaml to get Ethernet working](https://github.com/iiab/iiab/pull/3537#issuecomment-1513909783).  

Conversely: Start with any other OS (e.g. Raspberry Pi OS, Mint 21, Debian 11, Ubuntu 22.04, whatever you're most comfortable with) if you find that easier to get your Ethernet-cable-to-the-Internet instantly working, on a machine with internal WiFi AP capability.

jvonau, 2023-04-23:
> @EMG70 IIAB install instructions very strongly emphasize using Ethernet (cable) _instead of_ WiFi as your connection to the Internet, during IIAB installation especially.
>
> If you have time Sunday or [...]

Or alter network-config on the first partition pre-install after writing the image with the wifi info embedded, adding in the missing ethernets section, just be mindful of the indents:
```
network:
  version: 2
  ethernets:
    eth0:
      dhcp4: true
      optional: true
  wifis:
  ---
```
as noted in https://github.com/iiab/iiab/pull/3537#issuecomment-1513909783, 
> Conversely: Start with any other OS (e.g. Raspberry Pi OS, Mint 21, Debian 11, Ubuntu 22.04, whatever you're most comfortable with) if you find that easier to get your Ethernet-cable-to-the-Internet instantly working, on a machine with internal WiFi AP capability.

Or don't embed any wifi info then ethernet will get used by default. Think that is a bit of a bug in 'imager' should really add the ethernets: section anyway given that 'optional' works correctly.

jvonau, 2023-04-23:
Wifi only installs with 'wifi_up_down: False' can work but you have more hoops to jump through. On Ubuntu server the /etc/netplan/* stanza for the wifi needs to be disabled and iiab-hotspot-on needs to be run before rebooting.

holta, 2023-04-23:
> Or don't embed any wifi info then ethernet will get used by default. Think that is a bit of a bug in 'imager' should really add the ethernets: section anyway given that 'optional' works correctly.

💯

jvonau, 2023-04-23:
> Wifi only installs with 'wifi_up_down: False' can work but you have more hoops to jump through. On Ubuntu server the /etc/netplan/* stanza for the wifi needs to be disabled and iiab-hotspot-on needs to be run before rebooting.

Won't work on RasPiOS as `ip link set wlan0 down` and `systemctl start hostapd` in 'hotspot-on' will result in some not too nice interactions when run across wifi only installs. That could be altered to omit those commands and add the message to reboot, seems like the emphasis was for the command to work across wired connections.

jvonau, 2023-04-24:
Just keep using the workaround, just turns into a circus every time. I don't jump through hoops just to improve what I gave birth too. Happy Hacking!!

holta, 2023-04-27:
Appears solved by:

- PR #3549

Which should be merged soon, after many days of hard work thanks to @jvonau.

Thanks also to @EMG70's assistance heavily testing this day after day, on 3-or-more different OS's and HW combinations.

holta, 2023-04-27:
Fixed!

_Huge thanks to @jvonau and @EMG70_

🏗️

closed 2023-04-27 by holta