iiab/iiab #3543: Can I assume devices are being connected via this VPN tunnel?

jvonau, 2023-04-20:
@jvonau ,Not sure this issue is related to this PR.
I have accidentally found out that if I connect my server to an ethernet cable and VPN is enabled post install, I can access iiab home page from any device connected to my Sky home router without hooking up via IIAB hotspot.
This happens across OS systems ,just tried on Debian 11 desktop,23.04 server and desktop ,same happens.
Can I assume devices are being connected via this VPN tunnel?

_Originally posted by @EMG70 in https://github.com/iiab/iiab/issues/3537#issuecomment-1516922628_

EMG70, 2023-04-20:
Apologies,this has nothing to do with VPN indeed. I have disabled VPN but devices can still connect to IIAB via home router without connecting to iiab hotspot.
[image]
[image]

jvonau, 2023-04-20:
```
6: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 500
    link/none 
    inet 10.8.0.46 peer 10.8.0.45/32 scope global tun0
       valid_lft forever preferred_lft forever
```

jvonau, 2023-04-20:
https://github.com/iiab/iiab/pull/3537#issuecomment-1516985248

holta, 2023-04-20:
@EMG70 will re-test causality when he has time tomorrow:

- His home router sometimes (helpfully!) resolves `box` to an IIAB at 192.168.0.x — while unexpected this is convenient when it happens — as every other device in his house can then browse to http://box (suddenly without having to type in http://192.168.0.33 or http://192.168.0.38 !)

- Tomorrow he will try to verify under what exact conditions this happens (and under what exact conditions this does not happen!)

- _@EMG70 please also verify that all other devices in the house... that happen to be named "box" ...are turned off ?_

EMG70, 2023-04-21:
It was DHCP enabled in my router . I disabled it and couldn’t access box/ via home router as before.
I think it’s a good feature though, people can still access their mobile data and have access to IIAB learning at same time.Apologies for the wild goose chase.

EMG70, 2023-04-21:
[image]

IP 192.168.0.33 is the server

EMG70, 2023-04-21:
[image]

EMG70, 2023-04-21:
Not sure about security side if iiab (open network) is accessible from an online router.

holta, 2023-04-21:
> It was DHCP enabled in my router . I disabled it and couldn’t access box/ via home router as before.

When DHCP is enabled in your router, does toggling VPN on/off on the IIAB have any effect... on *http://box* and *box/* URL's working / not working on devices throughout the house?

(Or no effect at all?)

> I think it’s a good feature though, people can still access their mobile data and have access to IIAB learning at same time.

That depends on the phone?

As mentioned below, "Mobile Data" needs to be turned off in many Samsung and Huawei phones:

http://FAQ.IIAB.IO -> "12 My Android device says "Connected, no Internet" and won't browse http://box"

EMG70, 2023-04-21:
I can still access IIAB via my home router with VPN set to True or False.All gadgets can still access IIAB via home router.

closed 2023-04-21 by holta